jblance/mpp-solar #2: extra status label

JeroenSchwab, 2019-07-06:
I have an MPPsolar PIP5048MS and I have it connected via the serial port to an raspberry pi3 and i am running your script. 
if i run the script like:
mpp-solar -d /dev/ttyUSB0 -b 2400 -c QPIGS

I get this:
ac_input_frequency            	49.9            Hz
ac_input_voltage              	233.6           V
ac_output_active_power        	0386            W
ac_output_apparent_power      	0437            VA
ac_output_frequency           	50.0            Hz
ac_output_load                	008             %
ac_output_voltage             	230.1           V
battery_capacity              	100             %
battery_charging_current      	000             A
battery_discharge_current     	00007           A
battery_voltage               	53.60           V
battery_voltage_from_scc      	53.61           V
bus_voltage                   	374             V
device_status                 	
	- Add SBU priority version - No
	- Configuration unchanged
	- SCC firmware version updated
	- Load on
	- Battery voltage not to steady while charging
	- Charging on
	- SCC charging on
	- AC charging off 
inverter_heat_sink_temperature	0035            Deg_C
pv_input_current_for_battery  	0000            A
pv_input_voltage              	070.7           V
unknown_value_in_response     	010             

I think the "unknown_value_in_response" is the Wattage of the PV. Becouse this is the only value that is different then when i look at the menu on the inverter it self.
I looked at the code to find out how I could add this to the influxdb but I can not figure it out. 
Can you help me to add this?

jblance, 2019-07-06:
Hi

Your inverter is returning more info than mine for the qpigs command. When it does this it will display the result but put it as unknown.

To update what the responses mean there is a qpigs.json in the commands folder. In there you can see a 'response' field. You will need to add in the details (watch the commas and where its added)

You'll need to reinstall it after changing it i think

If you can't get it to work let me know and I'll send a file for you when I get back to a computer

JeroenSchwab, 2019-07-06:
Hi,
tanx for the help I will try it and let you know if it worked.

JeroenSchwab, 2019-07-07:
Hi,
I added some things and I think I get even more values back. I added the extra fields have way and then all the values are not right. if i add them to the end I get this. (pv_input_ power, 1 and 2)
 
mpp-solar -d /dev/ttyUSB0 -b 2400 -c QPIGS

ac_input_frequency            	49.9            Hz
ac_input_voltage              	226.0           V
ac_output_active_power        	2085            W
ac_output_apparent_power      	2101            VA
ac_output_frequency           	49.9            Hz
ac_output_load                	042             %
ac_output_voltage             	226.0           V
battery_capacity              	100             %
battery_charging_current      	000             A
battery_discharge_current     	00000           A
battery_voltage               	55.50           V
battery_voltage_from_scc      	55.56           V
bus_voltage                   	393             V
device_status                 	
	- Add SBU priority version - No
	- Configuration changed
	- SCC firmware version updated
	- Load on
	- Battery voltage not to steady while charging
	- Charging on
	- SCC charging on
	- AC charging off 
inverter_heat_sink_temperature	0034            Deg_C
pv_input_current_for_battery  	0000            A
pv_input_power                	110             W
pv_input_power1               	00              W
pv_input_power2               	00038           W
pv_input_voltage              	136.0           V

Is there a way to see what the inverter is sending out so i can find out what each value is?
like this line in de qpigs.json file
test_responses": [
            "(000.0 00.0 230.0 49.9 0161 0119 003 460 57.50 012 100 0069 0014 103.8 57.45 00000 00110110 00 00 00856 010$\u008c\r"
        ],

jblance, 2019-07-07:
-R will show the raw output

JeroenSchwab, 2019-07-07:
I tryed that but it gives me:
mpp-solar -d /dev/ttyUSB0 -b 2400 -R
(92931809100909��

jblance, 2019-07-07:
You'll need to include the command too, ie add '-c QPIGS'

JeroenSchwab, 2019-07-08:
I can hit my self for the head... it works.
It now gives me:
mpp-solar -d /dev/ttyUSB0 -b 2400 -c QPIGS -R
(223.1 49.9 223.1 49.9 1003 0872 020 393 55.60 000 100 0035 0000 132.1 55.56 00000 01110110 00 00 00038 1104�

jblance, 2019-07-31:
Great to hear. I'll close this as resolved for you now.

closed 2019-07-31 by jblance