jblance/mpp-solar #3: Support for HV LV 2424?

ghettodev, 2019-07-29:
Hi, 

I have the 2424 MPP Solar Hybrid inverter and I was trying to run your script against it. It seems to hang on execution though with no output with either DEBUG or the RAW flag. Is the device not supported? 

If not, is there anything I can do to help get it working? I'd love to gather some metrics from this thing.

jblance, 2019-07-29:
I have no idea if the commands are the same or even similar between the 4048 and the 2424. Do have have any details of the serial commands that your unit supports?

ghettodev, 2019-07-29:
I'd be love to help debug and collect any information I can. Is there a resource you can point me to? I've never tried to monitor a serial port before.

I'd also be happy to talk about offering bounty for the work.

jblance, 2019-07-30:
First can I check how you are setup.
What computer is connected to the inverter?
How is it connected? 
What do you use to communicate at the moment?
Does the above work?

ghettodev, 2019-07-30:
> What computer is connected to the inverter?

I've connected on win10 and Ubuntu 19.04

> How is it connected?

With a pl2303 serial to usb converter connected to the rj45 to serial thing that came with the inverter. 

> What do you use to communicate at the moment?

The solarpower java app that came on the cdrom.

> Does the above work?

The app works. When I connect with the serial adapter, the script hangs with no output. 

> What is the specs and full model of the inverter and size of the battery pack

https://www.mppsolar.com/v3/split-phase-lv-series/

Inverter:

### PV INPUT / SOLAR CHARGING

| PV INPUT / SOLAR CHARGING |                           |
| ------------------------- | ------------------------- |
| Max PV Input Power        | 2000w                     |
| Max PV Input Volt         | 145VDC (open circuit Voc) |
| MPPT Range                | 30 - 115VDC               |
| Number of PV Input        | 1                         |
| Max Charging Current      | **80A**                   |

### BATTERY OUTPUT MODE

| BATTERY OUTPUT MODE          |                               |
| ---------------------------- | ----------------------------- |
| Output Waveform              | Pure Sine Wave                |
| Nominal Output Volt          | 101/110/120 VAC, Single Phase |
| Frequency                    | 50 / 60Hz, auto-sensing       |
| AC Input Transfer Relay      | 40A                           |
| Max Efficiency               | >90%                          |
| Max Utility Charging         | 60A                           |
| Max Utility + Solar Charging | **140A**                      |

### ENVIRONMENTAL / MECHANICAL SPECIFICATIONS

|                    |                           |
| ------------------ | ------------------------- |
| Communication Port | USB / Dry Contact         |
| Operating Temp     | .0 -50°C                  |
| Operating Humidity | 0 -90% RH (No Condensing) |
| Dimension          | 468*295*120               |
| Net Weight         | 11Kg                      |

It's a 24 volt model hooked up to two battleborns in series.

jblance, 2019-07-31:
Ok, when connected via Ubuntu, what dev does it show as?
If you send a simple command like QID via the mpp-solar script does it respond?
If not we might need to add some extra debug to diag
Sniffing the serial would help but can be a pain
Dmesg and syslog of physical connect and disconnect of the adapter might give us some hints

ghettodev, 2019-08-09:
Sorry for the delay. 

The serial port is mounted as `/dev/ttyUSB0`. 

The issue with the scripts hanging in my original post seems to be related to having the direct USB connection from the device plugged in at the same time as the serial adapter. If I unplug the USB then I can get some output from your scripts.

```bash 
username@username-wp:~$ sudo mpp-solar -c QID
serial_number                 	0   
```
Here's some output from some other commands, I'll only post the ones that produce unique output:

```bash
username@username-wp:~$ sudo mpp-solar -s
ERROR:root:Response invalid as insufficient number of elements in response. Got 1, expected as least 17
ERROR:root:Response invalid as insufficient number of elements in response. Got 1, expected as least 17
ERROR:root:Response invalid as insufficient number of elements in response. Got 1, expected as least 17
ERROR:root:Response invalid as insufficient number of elements in response. Got 1, expected as least 17
CRITICAL:root:Command execution failed
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.1.3', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/__init__.py", line 37, in main
    fullStatus = mp.getFullStatus()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mpputils.py", line 55, in getFullStatus
    data = self.mp.execute("Q1").response_dict
AttributeError: 'NoneType' object has no attribute 'response_dict'

username@username-wp:~$ sudo mpp-solar -ll DEBUG -b 2400 -c QBOOT
dsp_has_bootstrap             	No    

username@username-wp:~$ sudo mpp-solar -ll DEBUG -b 2400 -c QDI
ERROR:root:Response invalid as insufficient number of elements in response. Got 1, expected as least 25
ERROR:root:Response invalid as insufficient number of elements in response. Got 1, expected as least 25
ERROR:root:Response invalid as insufficient number of elements in response. Got 1, expected as least 25
ERROR:root:Response invalid as insufficient number of elements in response. Got 1, expected as least 25
CRITICAL:root:Command execution failed
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.1.3', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/__init__.py", line 56, in main
    results = mp.getResponseDict(args.command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mpputils.py", line 38, in getResponseDict
    return self.mp.execute(cmd).response_dict
AttributeError: 'NoneType' object has no attribute 'response_dict'

username@username-wp:~$ sudo mpp-solar -ll DEBUG -b 2400 -c QFLAG
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.1.3', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/__init__.py", line 56, in main
    results = mp.getResponseDict(args.command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mpputils.py", line 38, in getResponseDict
    return self.mp.execute(cmd).response_dict
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppcommands.py", line 155, in execute
    return self.doSerialCommand(command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppcommands.py", line 138, in doSerialCommand
    command.set_response(response_line)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppcommand.py", line 105, in set_response
    self.response_dict = self.get_response_dict()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppcommand.py", line 221, in get_response_dict
    msgs[resp_format[2][item]['name']] = [status, '']
KeyError: '0'

username@username-wp:~$ sudo mpp-solar -ll DEBUG -b 2400 -c QMCHGCR
response                      	0 

username@username-wp:~$ sudo mpp-solar -ll DEBUG -b 2400 -c QOPM
output_mode                   	single machine output

username@username-wp:~$ sudo mpp-solar -ll DEBUG -b 2400 -c QPIWS
warning_status          

username@username-wp:~$ sudo mpp-solar -ll DEBUG -b 2400 -c QPIWS2
CRITICAL:root:Command not found
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.1.3', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/__init__.py", line 56, in main
    results = mp.getResponseDict(args.command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mpputils.py", line 38, in getResponseDict
    return self.mp.execute(cmd).response_dict
AttributeError: 'NoneType' object has no attribute 'response_dict'
```

I'm highly motivated to get this working. I've realized half the fun is the monitoring so let me know if I can do anything to help. Also, if you can point me to a good tutorial about serial port sniffing I could probably dig something up. 

Are you familiar with gitcoin? I'd be happy to set up a bounty for you.

jblance, 2019-08-10:
It looks like the comms are not working at all. Can you make sure that nothing else is using the device. Maybe try a `sudo fuser /dev/ttyUSB0` (when the adapter is connected) 

Also have a look at dmesg just after plugging the serial adapter in

ghettodev, 2019-08-10:
```bash
[57379.239747] pl2303 ttyUSB0: pl2303 converter now disconnected from ttyUSB0
[57379.239816] pl2303 1-4.2:1.0: device disconnected
[57383.300383] usb 1-4.2: new full-speed USB device number 9 using xhci_hcd
[57383.401701] usb 1-4.2: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 3.00                                                          
[57383.401720] usb 1-4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[57383.401730] usb 1-4.2: Product: USB-Serial Controller
[57383.401741] usb 1-4.2: Manufacturer: Prolific Technology Inc.
[57383.411613] pl2303 1-4.2:1.0: pl2303 converter detected
[57383.414131] usb 1-4.2: pl2303 converter now attached to ttyUSB0
```
If I `watch` fuser it shows a pid only when the mpp-solar script accessing the device so I'm reasonably sure it's talking to the device.

jblance, 2019-08-11:
Does the java app give any indication of the connection speed?

We are either not connecting correctly or not sending the right info as far as I can tell so far. 

Next steps are either new code to debug the connection or sniff the data while the java app is working

ghettodev, 2019-08-11:
Based on this [doc](http://forums.aeva.asn.au/uploads/293/HS_MS_MSX_RS232_Protocol_20140822_after_current_upgrade.pdf) I'm assuming it's probably running at the same connection settings as the others so I'm using 2400. 

I installed interceptty and ran this `sudo mpp-solar -t -s  -b 2400 -d /dev/ttyDUMMY -ll DEBUG -R` against it:

```bash
 sudo interceptty -s 'ispeed 2400 ospeed 2400' -l /dev/ttyUSB0 /dev/ttyDUMMY | interceptty-nicedump

>  5e 30 1b e3 0d                | ^0[ESC][CR]
<  51 31 1b fc 0d                | Q1[ESC][CR]
>  5e 30 1b e3 0d                | ^0[ESC][CR]
<  51 31 1b fc 0d                | Q1[ESC][CR]
>  5e 30 1b e3 0d                | ^0[ESC][CR]
<  51 31 1b fc 0d                | Q1[ESC][CR]
>  5e 30 1b e3 0d                | ^0[ESC][CR]
<  51 31 1b fc 0d                | Q1[ESC][CR]
```

You are right, it doesn't look like it's talking to the device.

When I run the java app it sends this to the serial port:

```bash
>  5e 30 1b e3 0d 5e 30 1b e3 0d | ^0[ESC][CR]^0[ESC][CR]
>  5e 30 e3 5e 30 1b e3 0d 30 e3 | ^0^0[ESC][CR]0
>  0d 5e 30 e3 5e 1b e3 5e 1b e3 | [CR]^0^[ESC]^[ESC]
>  0d 5e 30 1b e3 30 5e 30 1b e3 | [CR]^0[ESC]0^0[ESC]
>  5e 30 1b 30 1b e3 0d 5e 30 1b | ^0[ESC]0[ESC][CR]^0[ESC]
>  e3 5e 30 35 38 03 5e 44 30 30 | ^058[ETX]^D00
>  35 31 3b 03 5e 44 5e 30 30 31 | 51;[ETX]^D^001
>  38 3b 03 0d 5e 30 35 38 03 5e | 8;[ETX][CR]^058[ETX]^
>  44 30 38 03 0d 38 5e 44 30 30 | D08[ETX][CR]8^D00
>  31 38 3b 03 0d 39 30 30 30 30 | 18;[ETX][CR]90000
>  30 32 35 34 31 39 36 30 30 30 | 0254196000
>  38 35 30 30 30 30 ab b7 0d 44 | 850000[CR]D
>  32 34 36 31 30 31 30 38 35 30 | 2461010850
>  30 30 30 ab b7 0d 5e 30 35 31 | 000[CR]^051
>  39 36 31 31 30 31 30 38 35 0d | 961101085[CR]
>  39 30 31 30 30 30 38 30 ab 0d | 90100080[CR]
>  5e 31 34 36 31 31 30 38 35 30 | ^146110850
>  30 ab 5e 30 39 36 31 39 30 31 | 0^0961901
>  30 38 35 30 30 5e 44 35 39 31 | 08500^D591
>  39 36 31 30 30 30 38 30 30 30 | 9610008000
>  30 30 ab b7 0d 5e 30 32 35 31 | 00[CR]^0251
>  34 39 36 31 31 31 39 30 36 31 | 4961119061
>  30 30 30 38 35 30 30 30 30 30 | 0008500000
>  30 5e 44 30 30 31 3b 0d 44 30 | 0^D001;[CR]D0
>  30 35 31 38 3b 03 0d 44 30 35 | 0518;[ETX][CR]D05
```

Which it repeats every few minutes or so.

I can't figure out how to tell the java app to listen to ttyDUMMY yet.

jblance, 2019-08-11:
That's the doc that the code is based on, though the Q1 command is undocumented :-/

It looks like your inverter uses a different set of commands. In the first instance you can see Q1 being sent and a response of essentially 'huh?'. You could try again with QID or another query command from the doc to see if the response is different.

The commands sent from the java app do not look like any of the commands from the doc though. 

Is there any password required to connect via the java app?
Is the java app downloadable from anywhere? (Or have you disassembled it?)
Is there any manuals etc with the java app or device that might provide hints?

jblance, 2019-08-20:
Looking at the serial capture I'm not sure that we are getting clean data
It looks like `^` is the start character and if we break on `0d` (carriage return) then we kinda get:
```
5e 30 1b e3 0d 			| ^0 ? ? [CR]
5e 30 1b e3 0d 			| ^0 ? ? [CR]
5e 30 e3 			|
5e 30 1b e3 0d 			| ^0 ? ? [CR]
30 e3 0d 			| 
5e 30 e3 
5e 1b e3 
5e 1b e3 0d 			| 
5e 30 1b e3 30 
5e 30 1b e3 			| 
5e 30 1b 30 1b e3 0d 		| ^00 ã
5e 30 1b e3			| 
5e 30 35 38 03 
5e 44 30 30 35 31 3b 03 	| 
5e 44 
5e 30 30 31 38 3b 03 0d		| ^0018;
5e 30 35 38 03 
5e 44 30 38 03 0d 		| ^D08
38 
5e 44 30 30 31 38 3b 03 0d	| ^D0018 ;
39 30 30 30 30 30 32 35 34 31 39 36 30 30 30 38 35 30 30 30 30 ab b7 0d	| 900000254196000850000 «·
44 32 34 36 31 30 31 30 38 35 30 30 30 30 ab b7 0d 			| D2461010850000 «·
5e 30 35 31 39 36 31 31 30 31 30 38 35 0d 				| ^0519611010 85
39 30 31 30 30 30 38 30 ab 0d 						| 9010008 0«
5e 31 34 36 31 31 30 38 35 30 30 ab 					| ^146110850 0«
5e 30 39 36 31 39 30 31 30 38 35 30 30 					| ^096190108500
5e 44 35 39 31 39 36 31 30 30 30 38 30 30 30 30 30 ab b7 0d		| ^D591961000800000 «·
5e 30 32 35 31 34 39 36 31 31 31 39 30 36 31 30 30 30 38 35 30 30 30 30 30 30  | ^0251496111906100085000000
5e 44 30 30 31 3b 0d 							| ^D00 1;
44 30 30 35 31 38 3b 03 0d						| D00518 ;
```

Now we see some `5e 30 1b e3 0d` like when the script is sending a `Q1` command above
The CRC check for `5e 30` gives `1b e3` so that all works
I cant get any of the remaining `5e .... 0d` sequences to calculate the correct crc though

Maybe turning off (or extending to maximum time) the updates in the java app might allow a less noisy capture?

Sorry I cant be any real help on this

closed 2019-09-18 by ghettodev

jblance, 2020-06-10:
FYI This looks like the PI17 or PI18 protocol - I have been sent these protocol definitions now