jblance/mpp-solar #4: 4048 MSD

abc00113, 2019-09-05:
hi got 2 PIP 4048 MSD they come with usb only connection
connected via us as /dev/usb/hiddev0+1
please help me get it fixed
this is what i got on my pi and the same on my laptop

pi@raspberrypi:~/Downloads/mpp-solar $ sudo mpp-solar -d /dev/usb/hiddev1 -s
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.1.3', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/__init__.py", line 37, in main
    fullStatus = mp.getFullStatus()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mpputils.py", line 55, in getFullStatus
    data = self.mp.execute("Q1").response_dict
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppcommands.py", line 155, in execute
    return self.doSerialCommand(command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppcommands.py", line 125, in doSerialCommand
    with serial.serial_for_url(self._serial_device, self._baud_rate) as s:
  File "/usr/lib/python2.7/dist-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 326, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (22, 'Invalid argument')

abc00113, 2019-09-05:
maybe lsusb or dmesg helps while i think the device is maybe wrong

pi@raspberrypi:~/Downloads/mpp-solar $ lsusb
Bus 001 Device 006: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 001 Device 005: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

[164802.950377] usb 1-1.5: new low-speed USB device number 6 using dwc_otg
[164803.119405] usb 1-1.5: New USB device found, idVendor=0665, idProduct=5161, bcdDevice= 0.02
[164803.119432] usb 1-1.5: New USB device strings: Mfr=3, Product=1, SerialNumber=0
[164803.143009] hid-generic 0003:0665:5161.0003: hiddev97,hidraw1: USB HID v1.11 Device [HID 0665:5161] on usb-20980000.usb-1.5/input0

jblance, 2019-09-08:
The code uses a simple serial connection approach.
I dont think that the normal USB connection works in this way.
Likely it needs something like http://libusb.sourceforge.net/api-1.0/group__dev.html (from https://stackoverflow.com/questions/12542799/communication-with-the-usb-device-in-python)

I think my old PIPs have direct USB as well as the serial - when I get a chance I'll have a look and see if I can figure out how to make this work.

abc00113, 2019-09-09:
ok thx for the answer.
I will close the issue.
if u need some infos about my MSD just let me know (and it's also 3-4years old now i think)
will try with usb to serial adapters now just thougth this should work without the adapters while there is a build in adapter like shown in lsusb

jblance, 2019-09-11:
have checked in some real rough code that seems to work (sometimes)...
[stolen mostly from cabellop5]
still needs more error checking especially for failure to connect to device

jblance, 2019-09-11:
I've improved the code a bit - it now (should) deal with connection errors
Debuging should now work....

Give it a try and let me know how it works for you (I've done some tests using direct USB and /dev/hidraw0)

closed 2019-09-17 by jblance

abc00113, 2019-09-18:
hi sorry i wanted to answer earlier but....
at the moment i run emoncms and solpiplog from njfaria and iḿ pretty happy with that for now
had no chance to test your new code untill now
just tell me if i should test it quick for feedback about my inverters