jblance/mpp-solar #5: Error in test_mppinverter.py

hotear, 2019-10-30:
In:
     def test_execute_qid_usb_cmd(self):
         """ test execute of QID command (Direct USB connection)"""
-        inverter = mppinverter.mppInverter('/dev/hidraw1')
+        inverter = mppinverter.mppInverter('/dev/hidraw0')

jblance, 2019-10-30:
Can you provide more information about the error?
Can you provide some detail on your setup?

hotear, 2019-10-30:
Sorry, was in a hurry when I wrote that.  I've got my 5048MS connected to a
Pi 4 via USB (direct) and it appears (as you say in your readme) as
hidraw0.  When I ran setup.py test, that test didn't seem to be working and
I noticed it was trying to talk to hidraw1 rather than hidraw0 so I changed
it locally and that fixed it for me.  Just thought it was a typo.

jblance, 2019-11-07:
ok, those tests are mainly for travis code testing anyway - I'll look at amending them a little to work if the device doesnt open

closed 2019-11-07 by jblance