jblance/mpp-solar #6: Support python3

hotear, 2019-10-30:
Would be great if this worked under python3.

jblance, 2019-10-30:
What doesnt work?
The travis tests should test 3.6 as well as 2.7?

hotear, 2019-10-30:
I was getting errors when trying to run it under python3.  One error as related to 208 in mppinverter.py:
os.write(usb0, send)
python3 did not like that send is a string and said it should be bytes like.
When I tweaked that line it then was complaining about line 219:
log.debug('USB read error', e.strerror)
python3 did not seem to think that the exception e had a strerror element.
When I tweaked that line it seemed to run but didn't seem to get any messages to/from the pip.

This is using the direct USB connection hidraw0 on a raspberry pi 4.

jblance, 2019-11-07:
how did you alter the `os.write` line?
can you run the script in debug mode and include the output please

hotear, 2019-11-07:
Sorry, this may have been my mistake.  I was running the mpp-solar command line but had altered the first line to make it use python3 instead of python2.  I've been using the actual python modules now under python3 without issue.  I think this can probably be closed.  Nice work btw.

jblance, 2019-11-07:
ok, thanks for the update

closed 2019-11-07 by jblance