jblance/mpp-solar #7: UnicodeDecodeError

hotear, 2019-12-12:
I'm using the command:
**sudo mpp-solar -s -d /dev/hidraw0** but I keep getting a failure and I can't figure out why.  I've tried it on another pi and it works fine there so there is nothing wrong with the physical cable or the inverter etc, it must be something to do with the setup on my non working pi.  The working pi has nothing else installed on it and the non working pi does (not sure if that is related to the issue or not).  I've run the command above with the -D flag and this is the output:

2019-12-12 12:16:05,316 - MPP-Solar - __init__.py - DEBUG - command QID
2019-12-12 12:16:05,317 - MPP-Solar - __init__.py - DEBUG - Serial device used: /dev/hidraw0, baud rate: 2400
2019-12-12 12:16:05,317 - MPP-Solar - mppinverter.py - DEBUG - Device matches hidraw regex
2019-12-12 12:16:05,318 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/invalid.json
2019-12-12 12:16:05,319 - MPP-Solar - mppinverter.py - DEBUG - Error processing JSON in /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/invalid.json
2019-12-12 12:16:05,319 - MPP-Solar - mppinverter.py - DEBUG - Expecting property name: line 3 column 3 (char 25)
2019-12-12 12:16:05,320 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/mchgc.json
2019-12-12 12:16:05,320 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for MCHGC
2019-12-12 12:16:05,321 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for MCHGC
2019-12-12 12:16:05,321 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 9d a2 9da2
2019-12-12 12:16:05,321 - MPP-Solar - mppcommand.py - DEBUG - Full command: MCHGC��
2019-12-12 12:16:05,322 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/muchgc.json
2019-12-12 12:16:05,322 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for MUCHGC
2019-12-12 12:16:05,323 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for MUCHGC
2019-12-12 12:16:05,323 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC bf 86 bf86
2019-12-12 12:16:05,323 - MPP-Solar - mppcommand.py - DEBUG - Full command: MUCHGC��
2019-12-12 12:16:05,324 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pbt.json
2019-12-12 12:16:05,324 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PBT
2019-12-12 12:16:05,325 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for PBT
2019-12-12 12:16:05,325 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 2f 11 2f11
2019-12-12 12:16:05,326 - MPP-Solar - mppcommand.py - DEBUG - Full command: PBT/
2019-12-12 12:16:05,326 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pcp.json
2019-12-12 12:16:05,327 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PCP
2019-12-12 12:16:05,327 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for PCP
2019-12-12 12:16:05,328 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 5c a4 5ca4
2019-12-12 12:16:05,328 - MPP-Solar - mppcommand.py - DEBUG - Full command: PCP\�
2019-12-12 12:16:05,328 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pcvv.json
2019-12-12 12:16:05,329 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PCVV
2019-12-12 12:16:05,329 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for PCVV
2019-12-12 12:16:05,329 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC af ec afec
2019-12-12 12:16:05,330 - MPP-Solar - mppcommand.py - DEBUG - Full command: PCVV��
2019-12-12 12:16:05,330 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pop.json
2019-12-12 12:16:05,331 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for POP
2019-12-12 12:16:05,331 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for POP
2019-12-12 12:16:05,331 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 19 c9 19c9
2019-12-12 12:16:05,332 - MPP-Solar - mppcommand.py - DEBUG - Full command: POP�
2019-12-12 12:16:05,332 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/psdv.json
2019-12-12 12:16:05,333 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PSDV
2019-12-12 12:16:05,333 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for PSDV
2019-12-12 12:16:05,333 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 89 9e 899e
2019-12-12 12:16:05,334 - MPP-Solar - mppcommand.py - DEBUG - Full command: PSDV��
2019-12-12 12:16:05,334 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/q1.json
2019-12-12 12:16:05,334 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for Q1
2019-12-12 12:16:05,335 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for Q1
2019-12-12 12:16:05,335 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 1b fc 1bfc
2019-12-12 12:16:05,335 - MPP-Solar - mppcommand.py - DEBUG - Full command: Q1
2019-12-12 12:16:05,336 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/qboot.json
2019-12-12 12:16:05,336 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for QBOOT
2019-12-12 12:16:05,336 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for QBOOT
2019-12-12 12:16:05,337 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC b 88 b88
2019-12-12 12:16:05,337 - MPP-Solar - mppcommand.py - DEBUG - Full command: QBOOT
 �
2019-12-12 12:16:05,337 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/qdi.json
2019-12-12 12:16:05,337 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for QDI
2019-12-12 12:16:05,337 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for QDI
2019-12-12 12:16:05,338 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 71 1b 711b
2019-12-12 12:16:05,338 - MPP-Solar - mppcommand.py - DEBUG - Full command: QDIq
019-12-12 12:16:05,338 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/qflag.json
2019-12-12 12:16:05,338 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for QFLAG
2019-12-12 12:16:05,338 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for QFLAG
2019-12-12 12:16:05,339 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 98 74 9874
2019-12-12 12:16:05,339 - MPP-Solar - mppcommand.py - DEBUG - Full command: QFLAG�t
2019-12-12 12:16:05,339 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/qid.json
2019-12-12 12:16:05,339 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for QID
2019-12-12 12:16:05,339 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for QID
2019-12-12 12:16:05,339 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC d6 ea d6ea
2019-12-12 12:16:05,339 - MPP-Solar - mppcommand.py - DEBUG - Full command: QID��
[... 54 lines cut ...]
2019-12-12 12:16:05,350 - MPP-Solar - mppcommand.py - DEBUG - Full command: QVFW2��
2019-12-12 12:16:05,350 - MPP-Solar - mppinverter.py - DEBUG - Searching for cmd 'Q1'
2019-12-12 12:16:05,350 - MPP-Solar - mppinverter.py - DEBUG - DIRECT USB connection: executing Q1
Q1 query

Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.1.3', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/__init__.py", line 44, in main
    fullStatus = mp.getFullStatus()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mpputils.py", line 51, in getFullStatus
    data = self.getResponseDict("Q1")
[... cut ...]
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppinverter.py", line 242, in execute
    return self._doDirectUsbCommand(command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppinverter.py", line 208, in _doDirectUsbCommand
    os.write(usb0, send.encode('utf-8'))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfc in position 3: ordinal not in range(128)

It seems to be an issue with a character in the message it is trying to send to the mpp?

Thanks

Nick

hotear, 2019-12-12:
Update:
I commented out the encode('utf-8') part and then it seems to work.

jblance, 2020-03-28:
Glad you got it working
Have removed Q1 from the status data now as well

closed 2020-03-28 by jblance