jblance/mpp-solar #8: No handlers could be found for logger "MPP-Solar"

tjfayaz120, 2020-03-06:
Hi this code works great on my pip 4048GE however I do get this annoying warning. Any idea how I can clear this error up? It happens every time I do a query although it does actually work. Thanks

jblance, 2020-03-07:
This is the logger (or lack of it) complaining

Can you try running with debug on ( -D) and paste the results (including the full command you are using)

Regards

tjfayaz120, 2020-03-07:
Hi here is the output

`root@raspberrypi:~# mpp-solar -D -d /dev/hidraw0 -s
2020-03-07 10:21:46,037 - MPP-Solar - __init__.py - DEBUG - command QID
2020-03-07 10:21:46,038 - MPP-Solar - __init__.py - DEBUG - Serial device used: /dev/hidraw0, baud rate: 2400
2020-03-07 10:21:46,039 - MPP-Solar - mppinverter.py - DEBUG - Device matches hidraw regex
2020-03-07 10:21:46,041 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/invalid.json
2020-03-07 10:21:46,041 - MPP-Solar - mppinverter.py - DEBUG - Error processing JSON in /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/invalid.json
2020-03-07 10:21:46,042 - MPP-Solar - mppinverter.py - DEBUG - Expecting property name: line 3 column 3 (char 25)
2020-03-07 10:21:46,042 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/mchgc.json
2020-03-07 10:21:46,043 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for MCHGC
2020-03-07 10:21:46,044 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for MCHGC
2020-03-07 10:21:46,045 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 9d a2 9da2
2020-03-07 10:21:46,045 - MPP-Solar - mppcommand.py - DEBUG - Full command: MCHGC��
2020-03-07 10:21:46,045 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/muchgc.json
2020-03-07 10:21:46,047 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for MUCHGC
2020-03-07 10:21:46,047 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for MUCHGC
2020-03-07 10:21:46,048 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC bf 86 bf86
2020-03-07 10:21:46,048 - MPP-Solar - mppcommand.py - DEBUG - Full command: MUCHGC��
2020-03-07 10:21:46,048 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pbt.json
2020-03-07 10:21:46,049 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PBT
2020-03-07 10:21:46,050 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for PBT
2020-03-07 10:21:46,050 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 2f 11 2f11
2020-03-07 10:21:46,051 - MPP-Solar - mppcommand.py - DEBUG - Full command: PBT/
2020-03-07 10:21:46,051 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pcp.json
2020-03-07 10:21:46,052 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PCP
2020-03-07 10:21:46,052 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for PCP
2020-03-07 10:21:46,053 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 5c a4 5ca4
2020-03-07 10:21:46,053 - MPP-Solar - mppcommand.py - DEBUG - Full command: PCP\�
2020-03-07 10:21:46,054 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pcvv.json
2020-03-07 10:21:46,055 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PCVV
2020-03-07 10:21:46,055 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for PCVV
2020-03-07 10:21:46,056 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC af ec afec
2020-03-07 10:21:46,056 - MPP-Solar - mppcommand.py - DEBUG - Full command: PCVV��
2020-03-07 10:21:46,056 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pop.json
2020-03-07 10:21:46,057 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for POP
2020-03-07 10:21:46,058 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for POP
2020-03-07 10:21:46,058 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 19 c9 19c9
2020-03-07 10:21:46,059 - MPP-Solar - mppcommand.py - DEBUG - Full command: POP�
2020-03-07 10:21:46,059 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/psdv.json
2020-03-07 10:21:46,060 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PSDV
2020-03-07 10:21:46,061 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for PSDV
2020-03-07 10:21:46,061 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 89 9e 899e
2020-03-07 10:21:46,061 - MPP-Solar - mppcommand.py - DEBUG - Full command: PSDV��
2020-03-07 10:21:46,062 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/q1.json
2020-03-07 10:21:46,062 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for Q1
2020-03-07 10:21:46,063 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for Q1
2020-03-07 10:21:46,063 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 1b fc 1bfc
2020-03-07 10:21:46,063 - MPP-Solar - mppcommand.py - DEBUG - Full command: Q1
2020-03-07 10:21:46,064 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/qboot.json
2020-03-07 10:21:46,064 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for QBOOT
[... cut ...]
is_charging_on                	0               True - 1/False - 0
is_configuration_changed      	0               True - 1/False - 0
is_load_on                    	1               True - 1/False - 0
is_sbu_priority_version_added 	0               True - 1/False - 0
is_scc_charging_on            	0               True - 1/False - 0
is_scc_firmware_updated       	0               True - 1/False - 0
pv_input_current_for_battery  	00.0            A
pv_input_voltage              	000.0           V
unknown_value_in_response     	011`

jblance, 2020-03-10:
Does it still fail without debug?
Did you update (as i made a few changes to the code)

tjfayaz120, 2020-03-14:
> Does it still fail without debug?
> Did you update (as i made a few changes to the code)

Hi I just did an update to the latest code and still the warning appears :(

`No handlers could be found for logger "MPP-Solar"`

jblance, 2020-03-19:
Thats weird.

I have added a specific `set level to WARNING` for the logger if not in debug, hopefully that might resolve the error

jblance, 2020-03-28:
I have also reworking the logging code so it should now have removed that error
(as at version 0.2.6+

closed 2020-04-02 by jblance

tjfayaz120, 2020-04-04:
> I have also reworking the logging code so it should now have removed that error
> (as at version 0.2.6+

thanks it works now :)