jblance/mpp-solar #9: Noob here. Please be gentle.

DMIINC, 2020-03-19:
I do have mpp solar LV 5048 inverters. I am trying to incorporate what you doing here with my system. I can not get past this problem. Any insight would be helpful.

pi@raspberrypi:~ $ mpp-solar -l
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.1.3', 'console_scripts', 'mpp-solar')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
ImportError: No module named mppsolar

jblance, 2020-03-19:
Hi

How did you install the package?
if you used `python ./setup.py install` maybe try as root i.e. `sudo python ./setup.py install`
and ensure that you are trying to run it away from where you cloned the repo

DMIINC, 2020-03-19:
I did as sudo python. Second part has me intrigued. I did all of it from directory where it was downloaded and extracted
Thanks,

DMIINC, 2020-03-19:
I got past this issue. If i do mpp-solar -s all I get is this:

pi@raspberrypi:~ $ mpp-solar -s
================ Status ==================
Parameter                     	Value           Unit
pi@raspberrypi:~ $ 

Nothing else. I am using USB connection. Do I have to do anything to activate it?
Sorry if this is bothersome.

Thanks you

jblance, 2020-03-19:
Hi

If you are using the USB connection (not USB to serial) you will need to tell it what device to use, try
`mpp-solar -s -d /dev/hidraw9`
`-D` is useful for extra debug information, so if the above doesnt work, try 
`mpp-solar -s -d /dev/hidraw9 -D` and post the output (it will be a lot)

DMIINC, 2020-03-19:
First command did not work. Second one, however:

pi@raspberrypi:~ $ mpp-solar -s -d /dev/hidraw0 -D
2020-03-19 19:02:33,983 - MPP-Solar - __init__.py - DEBUG - command QID
2020-03-19 19:02:33,983 - MPP-Solar - __init__.py - DEBUG - Serial device used: /dev/hidraw0, baud rate: 2400
2020-03-19 19:02:33,984 - MPP-Solar - mppinverter.py - DEBUG - Device matches hidraw regex
2020-03-19 19:02:33,985 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/f.json
2020-03-19 19:02:33,986 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for F
2020-03-19 19:02:33,986 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for F
2020-03-19 19:02:33,986 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 29 2 2902
2020-03-19 19:02:33,987 - MPP-Solar - mppcommand.py - DEBUG - Full command: F)
2020-03-19 19:02:33,987 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/invalid.json
2020-03-19 19:02:33,987 - MPP-Solar - mppinverter.py - DEBUG - Error processing JSON in /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/invalid.json
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppinverter.py", line 60, in getCommandsFromJson
    data = json.load(f)
  File "/usr/lib/python2.7/json/__init__.py", line 291, in load
    **kw)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 380, in raw_decode
    obj, end = self.scan_once(s, idx)
ValueError: Expecting property name: line 3 column 3 (char 25)
2020-03-19 19:02:33,989 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/mchgc.json
2020-03-19 19:02:33,990 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for MCHGC
2020-03-19 19:02:33,990 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for MCHGC
2020-03-19 19:02:33,990 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 9d a2 9da2
2020-03-19 19:02:33,990 - MPP-Solar - mppcommand.py - DEBUG - Full command: MCHGC��
2020-03-19 19:02:33,991 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/mnchgc.json
2020-03-19 19:02:33,992 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for MNCHGC
2020-03-19 19:02:33,992 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for MNCHGC
2020-03-19 19:02:33,992 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 57 23 5723
2020-03-19 19:02:33,993 - MPP-Solar - mppcommand.py - DEBUG - Full command: MNCHGCW#
2020-03-19 19:02:33,993 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/muchgc.json
2020-03-19 19:02:33,993 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for MUCHGC
2020-03-19 19:02:33,994 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for MUCHGC
2020-03-19 19:02:33,994 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC bf 86 bf86
2020-03-19 19:02:33,994 - MPP-Solar - mppcommand.py - DEBUG - Full command: MUCHGC��
2020-03-19 19:02:33,995 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pbcv.json
2020-03-19 19:02:33,995 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PBCV
2020-03-19 19:02:33,996 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for PBCV
2020-03-19 19:02:33,996 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC 64 5a 645a
2020-03-19 19:02:33,997 - MPP-Solar - mppcommand.py - DEBUG - Full command: PBCVdZ
2020-03-19 19:02:33,997 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pbdv.json
2020-03-19 19:02:34,000 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PBDV
2020-03-19 19:02:34,000 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for PBDV
2020-03-19 19:02:34,000 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC fd cd fdcd
2020-03-19 19:02:34,001 - MPP-Solar - mppcommand.py - DEBUG - Full command: PBDV��
2020-03-19 19:02:34,001 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pbft.json
2020-03-19 19:02:34,002 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PBFT
2020-03-19 19:02:34,002 - MPP-Solar - mppcommand.py - INFO - Calculating CRC for PBFT
2020-03-19 19:02:34,002 - MPP-Solar - mppcommand.py - DEBUG - Generated CRC bb ed bbed
2020-03-19 19:02:34,003 - MPP-Solar - mppcommand.py - DEBUG - Full command: PBFT��
2020-03-19 19:02:34,003 - MPP-Solar - mppinverter.py - DEBUG - Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/commands/pbt.json
2020-03-19 19:02:34,003 - MPP-Solar - mppcommand.py - DEBUG - Generate full command for PBT
[... cut ...]
  File "/usr/lib/python2.7/logging/__init__.py", line 465, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 329, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Logged from file mppinverter.py, line 209
2020-03-19 19:02:34,050 - MPP-Solar - mppcommand.py - INFO - No response
================ Status ==================
Parameter                     	Value           Unit
pi@raspberrypi:~ $ 

Thank you

jblance, 2020-03-19:
OK, so its not getting a response from the inverter
try a `dmesg` and see if you can see what the connection to the inverter is

DMIINC, 2020-03-19:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.97-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1294 SMP Thu Jan 30 13:21:14 GMT 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s36928 r8192 d24512 u69632
[    0.000000] pcpu-alloc: s36928 r8192 d24512 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=64M cma=256M video=HDMI-A-1:1920x1080M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0 smsc95xx.macaddr=DC:A6:32:23:78:EB vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=97709164-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 3735564K/4050944K available (8192K kernel code, 687K rwdata, 2408K rodata, 2048K init, 850K bss, 53236K reserved, 262144K cma-reserved, 3264512K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 688 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 851 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28692 entries in 85 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000018] Switching to timer-based delay loop, resolution 18ns
[    0.000238] Console: colour dummy device 80x30
[    0.000262] console [tty1] enabled
[    0.000310] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000326] pid_max: default: 32768 minimum: 301
[    0.000601] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000618] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001375] CPU: Testing write buffer coherency: ok
[    0.001774] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002415] Setting up static identity map for 0x200000 - 0x20003c
[    0.002573] rcu: Hierarchical SRCU implementation.
[    0.003424] smp: Bringing up secondary CPUs ...
[    0.004211] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005138] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006020] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006156] smp: Brought up 1 node, 4 CPUs
[    0.006169] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.006178] CPU: All CPU(s) started in HYP mode.
[    0.006187] CPU: Virtualization extensions available.
[    0.006971] devtmpfs: initialized
[    0.017433] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.017641] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017661] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.025129] pinctrl core: initialized pinctrl subsystem
[    0.025977] NET: Registered protocol family 16
[    0.028758] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.030259] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.030269] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.030456] Serial: AMBA PL011 UART driver
[... 311 lines cut ...]
[  644.988919] w1_master_driver w1_bus_master1: Attaching one wire slave 00.300000000000 crc be
[  644.997308] w1_master_driver w1_bus_master1: Family 0 for 00.300000000000.be is not registered.
[  694.379966] w1_master_driver w1_bus_master1: Attaching one wire slave 00.b00000000000 crc 32
[... cut ...]
[ 1130.071852] w1_master_driver w1_bus_master1: Family 0 for 00.a80000000000.6d is not registered.
[ 1193.585575] w1_master_driver w1_bus_master1: Attaching one wire slave 00.680000000000 crc a7
[ 1193.594971] w1_master_driver w1_bus_master1: Family 0 for 00.680000000000.a7 is not registered.
[ 1233.986829] w1_master_driver w1_bus_master1: Attaching one wire slave 00.e80000000000 crc 2b
[ 1233.994766] w1_master_driver w1_bus_master1: Family 0 for 00.e80000000000.2b is not registered.
pi@raspberrypi:~ $

jblance, 2020-03-19:
seems like multiple devices connected - so if you are close to the pi/inverter try removing the USB cable and reconnecting it, then run `dmesg` and `tail /var/log/messages`

PS what OS / version are you using?

jblance, 2020-03-19:
though on further search it looks like `hidraw2`
so try `mpp-solar -s -d /dev/hidraw2`

DMIINC, 2020-03-19:
yes I do have two LV5048 inverters connected together.

DMIINC, 2020-03-19:
Raspbian Buster with desktop and recommended software
Image with desktop and recommended software based on Debian Buster
Version:February 2020
Release date:2020-02-13
Kernel version:4.19

jblance, 2020-03-19:
sorry, I meant multiple devices on the Pi USB
I run a parallel setup as well, so that shouldn't be a problem

DMIINC, 2020-03-19:
pi@raspberrypi:~ $ tail /var/log/messages
Mar 19 19:20:29 raspberrypi kernel: [ 1809.881740] w1_master_driver w1_bus_master1: Attaching one wire slave 00.440000000000 crc 27
Mar 19 19:20:29 raspberrypi kernel: [ 1809.888978] w1_master_driver w1_bus_master1: Family 0 for 00.440000000000.27 is not registered.
Mar 19 19:21:20 raspberrypi kernel: [ 1860.632439] w1_master_driver w1_bus_master1: Attaching one wire slave 00.c40000000000 crc ab
Mar 19 19:21:20 raspberrypi kernel: [ 1860.643078] w1_master_driver w1_bus_master1: Family 0 for 00.c40000000000.ab is not registered.
Mar 19 19:21:45 raspberrypi kernel: [ 1885.562542] w1_master_driver w1_bus_master1: Attaching one wire slave 00.240000000000 crc 42
Mar 19 19:21:45 raspberrypi kernel: [ 1885.573806] w1_master_driver w1_bus_master1: Family 0 for 00.240000000000.42 is not registered.
Mar 19 19:22:48 raspberrypi kernel: [ 1949.153106] w1_master_driver w1_bus_master1: Attaching one wire slave 00.a40000000000 crc ce
Mar 19 19:22:48 raspberrypi kernel: [ 1949.160838] w1_master_driver w1_bus_master1: Family 0 for 00.a40000000000.ce is not registered.
Mar 19 19:23:16 raspberrypi kernel: [ 1976.633157] w1_master_driver w1_bus_master1: Attaching one wire slave 00.640000000000 crc 04
Mar 19 19:23:16 raspberrypi kernel: [ 1976.638177] w1_master_driver w1_bus_master1: Family 0 for 00.640000000000.04 is not registered.
pi@raspberrypi:~ $

DMIINC, 2020-03-19:
I do logitech usb dongle for wireless mouse and keyboard

jblance, 2020-03-19:
try `mpp-solar -s -d /dev/hidraw2`

DMIINC, 2020-03-19:
pi@raspberrypi:~ $ mpp-solar -s -d /dev/hidraw2
================ Status ==================
Parameter                     	Value           Unit
pi@raspberrypi:~ $

jblance, 2020-03-19:
can you try disconnecting the inverter USB and reconnecting the running `dmesg`

DMIINC, 2020-03-19:
Will do in 10 minutes.

Dalibor Majksner
Hera Realty
2444 Duluth Highway STE 203
Duluth GA 30097
404-710-4876

DMIINC, 2020-03-19:
Thank you btw

Dalibor Majksner
Hera Realty
2444 Duluth Highway STE 203
Duluth GA 30097
404-710-4876

DMIINC, 2020-03-19:
I did disconnect and reconnected

pi@raspberrypi:~ $ mpp-solar -s -d /dev/hidraw2
================ Status ==================
Parameter                     	Value           Unit
pi@raspberrypi:~ $

DMIINC, 2020-03-19:
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.97-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1294 SMP Thu Jan 30 13:21:14 GMT 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s36928 r8192 d24512 u69632
[    0.000000] pcpu-alloc: s36928 r8192 d24512 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=64M cma=256M video=HDMI-A-1:1920x1080M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0 smsc95xx.macaddr=DC:A6:32:23:78:EB vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=97709164-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 3735564K/4050944K available (8192K kernel code, 687K rwdata, 2408K rodata, 2048K init, 850K bss, 53236K reserved, 262144K cma-reserved, 3264512K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 688 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 851 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28692 entries in 85 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000018] Switching to timer-based delay loop, resolution 18ns
[    0.000238] Console: colour dummy device 80x30
[    0.000262] console [tty1] enabled
[    0.000310] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000326] pid_max: default: 32768 minimum: 301
[    0.000601] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000618] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001375] CPU: Testing write buffer coherency: ok
[    0.001774] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002415] Setting up static identity map for 0x200000 - 0x20003c
[    0.002573] rcu: Hierarchical SRCU implementation.
[    0.003424] smp: Bringing up secondary CPUs ...
[    0.004211] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005138] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006020] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006156] smp: Brought up 1 node, 4 CPUs
[    0.006169] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.006178] CPU: All CPU(s) started in HYP mode.
[    0.006187] CPU: Virtualization extensions available.
[    0.006971] devtmpfs: initialized
[    0.017433] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.017641] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017661] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.025129] pinctrl core: initialized pinctrl subsystem
[    0.025977] NET: Registered protocol family 16
[    0.028758] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.030259] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.030269] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 405 lines cut ...]
[ 3024.329072] w1_master_driver w1_bus_master1: Family 0 for 00.5c0000000000.78 is not registered.
[ 3076.362356] w1_master_driver w1_bus_master1: Attaching one wire slave 00.dc0000000000 crc f4
[ 3076.369499] w1_master_driver w1_bus_master1: Family 0 for 00.dc0000000000.f4 is not registered.
[... cut ...]
[ 3383.136283] w1_master_driver w1_bus_master1: Family 0 for 00.820000000000.30 is not registered.
[ 3448.085675] w1_master_driver w1_bus_master1: Attaching one wire slave 00.420000000000 crc fa
[ 3448.093142] w1_master_driver w1_bus_master1: Family 0 for 00.420000000000.fa is not registered.
[ 3500.266305] w1_master_driver w1_bus_master1: Attaching one wire slave 00.c20000000000 crc 76
[ 3500.275459] w1_master_driver w1_bus_master1: Family 0 for 00.c20000000000.76 is not registered.
pi@raspberrypi:~ $

jblance, 2020-03-19:
it definitely looks like hidraw2
maybe `sudo mpp-solar -s -d /dev/hidraw2` ?

jblance, 2020-03-19:
and a `ls -ld /dev/hidraw2`

DMIINC, 2020-03-19:
pi@raspberrypi:~ $ sudo mpp-solar -s -d /dev/hidraw2
2020-03-19 19:57:59,544 - MPP-Solar - mppcommand.py - ERROR - Response invalid as insufficient number of elements in response. Got 1, expected as least 17
================ Status ==================
Parameter                     	Value           Unit
ac_input_frequency            	60.0            Hz
ac_input_voltage              	123.2           V
ac_output_active_power        	0614            W
ac_output_apparent_power      	0640            VA
ac_output_frequency           	60.0            Hz
ac_output_load                	025             %
ac_output_voltage             	123.2           V
battery_capacity              	100             %
battery_charging_current      	003             A
battery_discharge_current     	00000           A
battery_voltage               	56.10           V
battery_voltage_from_scc      	00.00           V
bus_voltage                   	228             V
inverter_heat_sink_temperature	0038            Deg_C
is_battery_voltage_to_steady_while_charging	0               True - 1/False - 0
is_charging_on                	1               True - 1/False - 0
is_configuration_changed      	1               True - 1/False - 0
is_load_on                    	0               True - 1/False - 0
is_sbu_priority_version_added 	0               True - 1/False - 0
is_scc_charging_on            	0               True - 1/False - 0
is_scc_firmware_updated       	0               True - 1/False - 0
pv_input_current_for_battery  	0000            A
pv_input_voltage              	000.0           V
unknown_value_in_response     	110             
pi@raspberrypi:~ $

DMIINC, 2020-03-19:
pi@raspberrypi:~ $ ls -ld /dev/hidraw2
crw------- 1 root root 245, 2 Mar 19 19:45 /dev/hidraw2
pi@raspberrypi:~ $

jblance, 2020-03-20:
ok that looks better
only root has access, so sudo is needed
not sure why the count is out as you are getting the data

DMIINC, 2020-03-20:
LV5048 is 2 inverters combined into one. So, 2 LV5048 = 4 inverters. 
Do not know if that has anything to do with it

Thank you

jblance, 2020-03-20:
it shouldn't the `QPGS<n>` command might be interesting though
i.e. `sudo mpp-solar -d /dev/hidraw2 -c QPGSn` where n = 0,1,2,3 

Glad it works for you

DMIINC, 2020-03-20:
pi@raspberrypi:~ $ sudo mpp-solar -d /dev/hidraw2 -c QPGSn where n = 0,1,2,3
usage: mpp-solar [-h] [-c COMMAND] [-D] [-d DEVICE] [-b BAUD] [-l] [-s] [-t]
                 [-R]
mpp-solar: error: unrecognized arguments: where n = 0,1,2,3

jblance, 2020-03-20:
sorry you have to change n, i.e.
`sudo mpp-solar -d /dev/hidraw2 -c QPGS0`

DMIINC, 2020-03-20:
i see

DMIINC, 2020-03-20:
sudo mpp-solar -d /dev/hidraw2 -c QPGS0
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.1.3', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/__init__.py", line 66, in main
    results = mp.getResponseDict(args.command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mpputils.py", line 37, in getResponseDict
    return self.inverter.execute(cmd).getResponseDict()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppinverter.py", line 251, in execute
    return self._doDirectUsbCommand(command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppinverter.py", line 217, in _doDirectUsbCommand
    os.write(usb0, send)
OSError: [Errno 22] Invalid argument

DMIINC, 2020-03-20:
i@raspberrypi:~ $ sudo mpp-solar -d /dev/hidraw2 -c QPGS1
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.1.3', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/__init__.py", line 66, in main
    results = mp.getResponseDict(args.command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mpputils.py", line 37, in getResponseDict
    return self.inverter.execute(cmd).getResponseDict()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppinverter.py", line 251, in execute
    return self._doDirectUsbCommand(command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppinverter.py", line 217, in _doDirectUsbCommand
    os.write(usb0, send)
OSError: [Errno 22] Invalid argument
pi@raspberrypi:~ $

jblance, 2020-03-20:
interesting, maybe I've broken that command
i'll need to check
can you try 
`sudo mpp-solar -d /dev/hidraw2 -c QPGS0 -R`

DMIINC, 2020-03-20:
pi@raspberrypi:~ $ sudo mpp-solar -d /dev/hidraw2 -c QPGS0 -R
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.1.3', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/__init__.py", line 64, in main
    print(mp.getResponse(args.command))
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mpputils.py", line 40, in getResponse
    return self.inverter.execute(cmd).getResponse()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppinverter.py", line 251, in execute
    return self._doDirectUsbCommand(command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.1.3-py2.7.egg/mppsolar/mppinverter.py", line 217, in _doDirectUsbCommand
    os.write(usb0, send)
OSError: [Errno 22] Invalid argument
pi@raspberrypi:~ $

DMIINC, 2020-03-20:
LV5048 = split phase inverter 240 Volt North American standard. Anyways since provided Watch Power app is awful. I am trying to get something working for grafana.

Thank you so much. I am finally going in right direction.

jblance, 2020-03-20:
No problem - good luck!!

DMIINC, 2020-03-20:
So do you think that I would be able to see all of the inverters?

jblance, 2020-03-20:
should be able to this is what the QPGSx command is for (and used to work for me)
i changed my router recently and now cant connect to my monitoring Pi over the network so my testing is a little limited
my setup isnt split phase though - i just parallel 2 inverters (as I am off-grid and solar powered)

DMIINC, 2020-03-20:
Ok I will ponder on this. I think it will work. However and for whatever reason even this command is not working anymore sudo mpp-solar -s -d /dev/hidraw2. It worked the first time.
Anyway. Thanks for the help and let me know your pay pal or something. I want to buy you a lunch. You provided some help here.

jblance, 2020-03-28:
Closing as seems to be working except the LV5048 parallel commands as per other issue

No need for lunch - just glad it can be of a little use to someone else - let me know if you have any improvements in your monitoring setup!

closed 2020-03-28 by jblance