jblance/mpp-solar #10: Need some help please :)

mwoodage, 2020-03-24:
Hello, firstly thank you for the code, it's just what i've been looking for given that WatchPower is so bad!
I've managed to get the information from my inverter and can use `mpp-solar -s` to get the following:-
pi@raspberrypi:~ $ mpp-solar -s
No handlers could be found for logger "MPP-Solar"
================ Status ==================
Parameter                     	Value           Unit
ac_input_frequency            	00.0            Hz
ac_input_voltage              	001.0           V
ac_output_active_power        	0000            W
ac_output_apparent_power      	0000            VA
ac_output_frequency           	50.0            Hz
ac_output_load                	000             %
ac_output_voltage             	228.0           V
battery_capacity              	051             %
battery_charging_current      	000             A
battery_discharge_current     	00000           A
battery_voltage               	25.05           V
battery_voltage_from_scc      	25.12           V
bus_voltage                   	352             V
inverter_heat_sink_temperature	0636            Deg_C
is_ac_charging_on             	0               True - 1/False - 0
is_battery_voltage_to_steady_while_charging	0               True - 1/False - 0
is_charging_on                	1               True - 1/False - 0
is_configuration_changed      	0               True - 1/False - 0
is_load_on                    	1               True - 1/False - 0
is_sbu_priority_version_added 	1               True - 1/False - 0
is_scc_charging_on            	1               True - 1/False - 0
is_scc_firmware_updated       	0               True - 1/False - 0
pv_input_current_for_battery  	0001            A
pv_input_voltage              	035.3           V
unknown_value_in_response     	000             

However, i would like to send this information to an MQTT broker and send it maybe once every 30secs or so. Do you know how I could do this? 
I'm not great with coding so might need a bit of help, but i'm learning fast!

Thanks in advance,
Martin

jblance, 2020-03-24:
Hi
There is a MQTT helper command (though it is very 'work in progress')
So if you have a MQTT broker running (that doesnt require a username/password)
`mpp-info-pub -d /dev/ttyUSB0 -q 'mqttbroker_name'` should publish a bunch of stuff

mwoodage, 2020-03-24:
Thanks I’ll give that a try tomorrow and see what happens 👍

mwoodage, 2020-03-25:
Hello, not sure if i've got this right, but i get the following error - any ideas?

Thanks

pi@raspberrypi:~ $ mpp-info-pub -d /dev/ttyUSB0 -q 'mqttbroker_mosquitto'
Traceback (most recent call last):
  File "/usr/local/bin/mpp-info-pub", line 11, in <module>
    load_entry_point('mpp-solar==0.2.2', 'console_scripts', 'mpp-info-pub')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/mpp_info_pub.py", line 12, in <module>
    import paho.mqtt.publish as publish
ImportError: No module named paho.mqtt.publish

jblance, 2020-03-25:
The key is the last line. You need the Mqtt module for python. Install it with `sudo pip install paho-mqtt`

If pip is not installed either, then do a `sudo apt-get install python-pip` before the above command

jblance, 2020-03-25:
Also make sure `mqttbroker_mosquitto` resolves to the ip address of your broker and remove the quotes `'` around the name. The -q value must be the hostname or IP address of the mqtt  broker

mwoodage, 2020-03-25:
Cool, thanks. I've managed to get paho-mqtt installed, but i now get the error 'connection refused'.
My MQTT broker is on a server with the ip address of 192.168.1.120, the raspberry pi is local to the inverter and on ip address 192.168.1.188. Both are connected via cables to a local network switch so i'm assuming i don't need to open a port? Also there is no firewall running on the server (mac mini).

Thank you so much for your help on this, i think its really close to being resolved :) 
Martin

pi@raspberrypi:~ $ mpp-info-pub -d /dev/ttyUSB0 -q 192.168.1.120
ERROR:MPP-Solar:Response invalid as insufficient number of elements in response. Got 1, expected as least 17
Traceback (most recent call last):
  File "/usr/local/bin/mpp-info-pub", line 11, in <module>
    load_entry_point('mpp-solar==0.2.2', 'console_scripts', 'mpp-info-pub')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/mpp_info_pub.py", line 57, in main
    publish.multiple(msgs, hostname=args.broker)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/publish.py", line 163, in multiple
    client.connect(hostname, port, keepalive)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 937, in connect
    return self.reconnect()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1071, in reconnect
    sock = self._create_socket_connection()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 3522, in _create_socket_connection
    return socket.create_connection(addr, source_address=source, timeout=self._keepalive)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
socket.error: [Errno 111] Connection refused

jblance, 2020-03-25:
You might want to check the connection to the broker from the pi
First on the broker (assuming you are using mosquitto) you can check pub / sub etc with 2 console windows
In window 1 run `mosquitto_sub -t "#"` (this subscribes to all topics, so if you have other publishers you will get a lot of noise
In windows 2 run ` mosquitto_pub -h localhost -t "test/sub_subtest" -m "this is a test message"`
This should have the test message show up in window 1
Then on the pi connected to the inverter try  `mosquitto_pub -h 192.168.1.120 -t "test/sub_test" -m "test from the pi"` (you may need to install `mosquitto-clients` on the pi)

If this all works, then we are back to checking the python code

mwoodage, 2020-03-26:
Hello, yes i can get that to work between the server and the raspberry pi :) Thank you.
I just need to figure out how it connects with my home control system (Indigo). I'm using a plugin for Indigo [https://github.com/FlyingDiver/Indigo-MQTT](url) which receives the MQTT data. 

However it's this part that i now need to figure out, as i can send the `mosquitto_pub -h localhost -t "test/sub_subtest" -m "this is a test message"` and have that activate a variable within Indigo, but i don't know how to receive the data that is being sent out when we use the `mpp-info-pub -d /dev/ttyUSB0 -q 192.168.1.120`. I think i need to subscribe to the incoming message? I'll keep trying and let you know if i can make it work :)

Thanks for all your help so far, its been amazing
Martin

jblance, 2020-03-26:
Glad things are moving in the right direction
If you used the following command on the broker (`192.168.1.120`?) `mosquitto_sub -t "#" -v` all topics will be subscribed to and the topics will print out, so if on the pi you run the `mpp-info-pub -d /dev/ttyUSB0 -q 192.168.1.120` you should what the topics and messages are. e.g. ``` mosquitto_sub -t "#" -v
9293333010501/status/sync_frequency/value 50.00
9293333010501/status/sync_frequency/unit (null)
9293333010501/status/is_load_on/value 1
9293333010501/status/is_load_on/unit True - 1/False - 0
9293333010501/status/ac_input_frequency/value 00.0
9293333010501/status/ac_input_frequency/unit Hz
9293333010501/status/inverter_charge_status/value float
9293333010501/status/inverter_charge_status/unit (null)
9293333010501/status/pv_input_voltage/value 103.8
9293333010501/status/pv_input_voltage/unit V
...```
in the above the first part is the topic "9293333010501/status/pv_input_voltage/value" and the last item is the payload "103.8". it uses the inverter serial number as the first part of the topic

jblance, 2020-03-28:
think this is resolved so closing - let me know if you need any more help

mwoodage, 2020-03-29:
Hello, i've almost got this working, sorry for not replying yesterday. It's mainly how I receive the data at the other end that i just need to figure out - i've got it working though and thank you for your help with that :) . One last question would be, how can we get `mpp-info-pub -d /dev/ttyUSB0 -q 192.168.1.120` to repeat every 1min? 
Thanks again for your help, its been fantastic, Martin

jblance, 2020-03-29:
I'd use crontab
`crontab -e` to edit it and an entry like
```
* * * * * /usr/local/bin/mpp-info-pub -d /dev/ttyUSB0 -q 192.168.1.120
```

mwoodage, 2020-03-29:
Excellent, thanks i'll give that a go.

mwoodage, 2020-03-30:
Hello, just to let you know I've got all this working now :). Many thanks for all your help - once i get everything working and displaying on a local web page i'll post a screen shot.

Thanks 
Martin

mwoodage, 2021-04-05:
Hello, last year you helped me set up reading data from my inverter and publish it via MQTT and it's worked fantastically. Unfortunately i've needed to replace my raspberry pi and rebuild the system. 
I've got everything thing working again, apart from publishing the data via MQTT. In the past i used `mpp-info-pub -d /dev/ttyUSB0 -q 192.168.1.120 ` but this no longer works.
Are you able to help getting the MQTT data published to 192.168.1.120?

I've tried `mpp-solar -c QPIGS -q 192.168.1.120` but this only publishes the data locally.

Thanks in advance, 
Martin

jblance, 2021-04-05:
You need to add an output option to your above command. Eg add ```-o mqtt``` for the default mqtt type of output. If you use -o it will give a list of available output modules and a brief description, one if the other mqtt output modules may suit your needs better

mwoodage, 2021-04-06:
Hi John, thanks for the speedy reply. i've added `-o mqtt` but now i'm getting connection refused, which i'm working on trying to resolve at the server end, as i'm assuming this is where the problem now lies?

> pi@raspberrypi:~ $ mpp-solar -c QPIGS -q 192.168.1.120 -o mqtt
WARNING:SerialIO:send_and_receive: Serial read error: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
WARNING:mqtt:output: Error publishing MQTT messages to broker '192.168.1.120' on port '1883' with auth 'None'
WARNING:mqtt:[Errno 111] Connection refused
pi@raspberrypi:~ $ 

Really appreciate your help, thank you 
Martin

jblance, 2021-04-06:
You have 2 issues
1. No response on /dev/ttyUSB0, is this the correct port? Is anything else connected?
2. The mqtt broker is denying the connection, does it allow connection from the pi? Does it require authentication? Check using mosquito_pub on the pi (might need in install mosquitto-client package)

mwoodage, 2021-04-06:
Ah ok, i'm not sure about issue 1 - nothing else is connected to the raspberry pi apart from the inverter into one of the USB ports.
Regarding Issue 2, the mqtt broker at the server end has not changed and can see the connection between the raspberry pi and the server (mac mini). No authentication required.

If i type `sudo Isof -nPi :1883` into terminal on the mac mini i get the following :-
> Mac-Mini:~ mwoodage$ sudo lsof -nPi :1883
COMMAND   PID     USER   FD   TYPE             DEVICE SIZE/OFF NODE NAME
IndigoPlu 586 mwoodage   10u  IPv4 0x1274c26f9ee16a4f      0t0  TCP 192.168.1.120:50934->192.168.1.98:1883 (ESTABLISHED)
IndigoPlu 586 mwoodage   11u  IPv4 0x1274c26f981aa77f      0t0  TCP 192.168.1.120:50937->192.168.1.98:1883 (ESTABLISHED)
IndigoPlu 586 mwoodage   15u  IPv4 0x1274c26f9d6cad1f      0t0  TCP 192.168.1.120:50940->192.168.1.98:1883 (ESTABLISHED)
IndigoPlu 586 mwoodage   18u  IPv4 0x1274c26f9ee173b7      0t0  TCP 192.168.1.120:50943->192.168.1.98:1883 (ESTABLISHED)
IndigoPlu 586 mwoodage   21u  IPv4 0x1274c26f9ee5777f      0t0  TCP 192.168.1.120:50946->192.168.1.98:1883 (ESTABLISHED)

Do you think it would be easier for me to use the previous version of mpp-solar ?

Thanks Martin

jblance, 2021-04-06:
That looks to be 120 connecting to 98? I thought 120 was the broker?

mwoodage, 2021-04-06:
Umm, 98 is the raspberry pi and 120 is the server mac mini with the broker on. Maybe i need to check that......
That will need to be tomorrow as it's nearly bed time in the UK!

mwoodage, 2021-04-12:
Hi John,
I've been trying this again over the weekend and still can't get it to work. Personally i think its at the server end that i need to get right, but i need to confirm with you that what i have done at the PI end is correct.
I can use the following `mpp-solar -c QPIGS -q 192.168.1.120` and get the results below, which would confirm that the inverter is connected and mpp-solar is collecting the data. What i can't confirm is it's sending the data to 192.168.1.120 
I've have mosquitto installed on the PI which is connected to the server and can see that its connected, however i'm not getting any results when i run `mpp-solar -c QPIGS -q 192.168.1.120`

Any ideas?

[image]

Really appreciate any help you can give,

Thanks 
Martin

jblance, 2021-04-12:
mpp-solar now has a number of options for output - by default (i.e. if you dont specify anything) it will just display to the screen (as in your screenshot)
if you run ``` mpp-solar -o ``` you will get a list of available output modules
```mpp-solar -c QPIGS -q 192.168.1.120 -o influx_mqtt``` will send the output to the broker on 192.168.1.120 (on the default port) in the format `influx`

mwoodage, 2021-04-14:
Hi John, you'll be please to know that i've managed to get this working again - half my problems were needing to port forward 1883 on my router! Once i'd figured that out and with your help it's all working again. 
Thank you so much for your help,

Martin

mwoodage, 2021-04-14:
Almost done! Just need to get the info to send every minute - before i've used Crontab -e with the following command, but that's not working any more    `1 * * * * /usr/local/bin/mpp-info-pub -d /dev/ttyUSB0 -q 192.168.1.120`
Do you know how this can be done?

Thanks again Martin

jblance, 2021-04-14:
You'll need to change to bit from ```mpp-info-pub``` onwards to the command you got working on the command line (  the reason behind the change was the make the commands the same, so testing on the command line would give the same results

mwoodage, 2021-04-14:
Umm, i've done that but its not working - is this correct?

[image]

jblance, 2021-04-14:
No, the command part of the line must work on the command line (cron just runs a command at a specified interval)

You to fully specify the command as cron has a different path environment compared to your shell, so it should be more like ```/usr/local/bin/mpp-solar -c QPIGS -q 192.168.1.81 -o mqtt```
But whatever you put as the command in cron must work on the command line, so test there first

mwoodage, 2021-04-15:
Ok, i've tried `/usr/local/bin/mpp-solar -c QPIGS -q 192.168.1.81 -o mqtt` in terminal and that works. However once i set it up in Cron nothing happens.

I've goggled some ideas but can't get anything to work. I've tried service cron status and get the following:-
It's showing that cron is active and running - can you see anything in this that indicates a problem? Sorry i know cron is not part of your code and therefore don't need to provide help, but i'm great full for all the help you've given so far, i think we're nearly there :)

> pi@raspberrypi:~ $ service cron status
> ● cron.service - Regular background program processing daemon
>    Loaded: loaded (/lib/systemd/system/cron.service; enabled; vendor preset: enabled)
>    Active: active (running) since Thu 2021-04-15 13:05:20 BST; 1h 21min ago
>      Docs: man:cron(8)
>  Main PID: 341 (cron)
>     Tasks: 1 (limit: 2062)
>    CGroup: /system.slice/cron.service
>            └─341 /usr/sbin/cron -f
> 
> Apr 15 14:26:01 raspberrypi CRON[20745]: pam_unix(cron:session): session opened for user root by (uid=0)
> Apr 15 14:26:01 raspberrypi CRON[20748]: (root) CMD (/usr/local/bin/ mpp-solar -c QPIGS -q 192.168.1.81 -o mqtt)
> Apr 15 14:26:01 raspberrypi CRON[20749]: (root) CMD (yourCommand >/dev/null 2>&1)
> Apr 15 14:26:01 raspberrypi CRON[20745]: pam_unix(cron:session): session closed for user root
> Apr 15 14:26:01 raspberrypi CRON[20744]: pam_unix(cron:session): session closed for user root
> Apr 15 14:27:01 raspberrypi CRON[25037]: pam_unix(cron:session): session opened for user root by (uid=0)
> Apr 15 14:27:01 raspberrypi CRON[25038]: pam_unix(cron:session): session opened for user root by (uid=0)
> Apr 15 14:27:01 raspberrypi CRON[25040]: (root) CMD (/usr/local/bin/ mpp-solar -c QPIGS -q 192.168.1.81 -o mqtt)
> Apr 15 14:27:01 raspberrypi CRON[25037]: pam_unix(cron:session): session closed for user root
> Apr 15 14:27:01 raspberrypi CRON[25038]: pam_unix(cron:session): session closed for user root
> pi@raspberrypi:~ $ 
>

mwoodage, 2021-04-15:
I've just noticed that it's working but every hour, not every min........

mwoodage, 2021-04-15:
All sorted - i was adding 1 * * * * to crontab, which was working every hour. I've since just used * * * * * which is now working every minute - yay, its finally all working.
Thanks so much for all your help,

Martin

jblance, 2021-04-15:
That makes sense, the 1 would mean it is to run at 1 min past the hour.

Good to hear it you've got it working

closed 2021-04-15 by jblance