jblance/mpp-solar #11: Adding commands for LV5048.

DMIINC, 2020-03-26:
[LV5048 5KW protocol-20190222 for customer.pdf](https://github.com/jblance/mpp-solar/files/4386569/LV5048.5KW.protocol-20190222.for.customer.pdf)
Hello,

Thank you for your help last week. I was in touch with MPP Solar and they provided communication protocol for LV5048. See attached. Is there way to integrate some of these commands with what you have written so far?

jblance, 2020-03-26:
Hi

Yep - if you want to do it just for you - the approach is too add/change the json files in mpp-solar/commands folder

For me to do it more generically would take a bit longer as I'd need to add a way to filter commands based on what device you have (though that is probably a good idea for the future)

DMIINC, 2020-03-26:
Which file? Is it lot of typing. I have a daughter at the house with nothing to do due to virus. Is it mostly copy paste type of adding commands?

Thanks,

jblance, 2020-03-26:
Depends what you want to change...
While the mpp-solar can do a summary for you with `-s` etc. You can also just use the raw commands like `-c QPIGS`, some of these are the same (e.g. QPIGS), some have changed a little (like QPGSn) and some are new (e.g. QPIGS2)
e.g. to update QPGSn you'd edit the `mppsolar/commands/qpgs.json` (from the downloaded content)
to create support for a new command like QPIGS2 copy `mppsolar/commands/qpigs.json` to `mppsolar/commands/qpig2.json` and update the content

Then re-install the package with `sudo python ./setup.py install`

Hope this helps - if not let me know your priorities and I'll try to get time to look at implementing something

DMIINC, 2020-03-26:
I really do I appreciate your help. But before we put a lot of work on this, what would be the easiest way to see if these commands can communicate with all 6 inverters. 2 per chassis for 3 chassis. All I can do now is one half of one chassis.

jblance, 2020-03-26:
So what command is working now?
If you are using `-s`, try `-c QPIGS`  then `-c QPGS0`, and `-c QPGS1`, and `-c QPGS2` and let me see what you get

DMIINC, 2020-03-26:
i@raspberrypi:~ $ sudo mpp-solar -c QPIGS -R
(122.5 59.9 122.5 59.9 0539 0486 021 228 56.10 003 100 0036 0000 000.0 00.00 00000 0100010 00 00 00000 110�p
pi@raspberrypi:~ $

DMIINC, 2020-03-26:
pi@raspberrypi:~ $ sudo mpp-solar -c QPGS0 -R
2020-03-26 19:16:27,944 - MPP-Solar - mppcommand.py - ERROR - Response invalid as insufficient number of elements in response. Got 1, expected as least 27
(NAKss
pi@raspberrypi:~ $ sudo mpp-solar -c QPGS0 
2020-03-26 19:16:52,081 - MPP-Solar - mppcommand.py - ERROR - Response invalid as insufficient number of elements in response. Got 1, expected as least 27
pi@raspberrypi:~ $ i@raspberrypi:~ $ sudo mpp-solar -c QPIGS -R
bash: i@raspberrypi:~: command not found

DMIINC, 2020-03-26:
pi@raspberrypi:~ $ sudo mpp-solar -c QPI
protocol_id                   	PI41

jblance, 2020-03-26:
what about `sudo mpp-solar -c QPGS1 -R` ?

DMIINC, 2020-03-26:
pi@raspberrypi:~ $ sudo mpp-solar -c QPGS1 -R
2020-03-26 19:33:02,609 - MPP-Solar - mppcommand.py - ERROR - Response invalid as insufficient number of elements in response. Got 1, expected as least 27
(NAKss

Same thing for additional numbers

DMIINC, 2020-03-26:
Yes usb to serial

jblance, 2020-03-26:
So, `sudo mpp-solar -c QPIGS -R` works?
what about `sudo mpp-solar -c QMOD` and `sudo mpp-solar -c QDI` ?

(coz QPGS0 should work accord to the doc you sent

DMIINC, 2020-03-26:
pi@raspberrypi:~ $ sudo mpp-solar -c QMOD
device_mode                   	Line            
pi@raspberrypi:~ $ sudo mpp-solar -c QDI
ac_output_frequency           	60.0            Hz
ac_output_voltage             	120.0           V
battery_bulk_charge_voltage   	56.4            V
battery_float_charge_voltage  	54.0            V
battery_recharge_voltage      	46.0            V
battery_redischarge_voltage   	54.0            V
battery_type                  	AGM             
battery_under_voltage         	42.0            V
buzzer                        	enabled         
charger_source_priority       	Solar + Utility 
input_voltage_range           	Appliance       
lcd_backlight                 	enabled         
lcd_reset_to_default          	enabled         
max_ac_charging_current       	0030            A
max_charging_current          	60              A
output_mode                   	parallel output 
output_source_priority        	Utility first   
over_temperature_restart      	disabled        
overload_bypass               	disabled        
overload_restart              	disabled        
power_saving                  	disabled        
primary_source_interrupt_alarm	enabled         
pv_ok_condition               	As long as one unit of inverters has connect PV, parallel system will consider PV OK 
pv_power_balance              	PV input max current will be the max charged current 
record_fault_code             	disabled        
unknown_value_in_response     	000  

Yes, on all 3 commands

DMIINC, 2020-03-26:
Unless all of them need to be connected via usb or serial to raspberry pi

jblance, 2020-03-27:
thats not what the doc says (i.e. it implies you can get details from any using QPGSn)
I've added a `QP2GSn` command, can you install the latest from git and try `mpp-solar -c QP2GS0` and maybe `mpp-solar -c QP2GS0 -D` if it doesnt work

DMIINC, 2020-03-27:
pi@raspberrypi:~ $ mpp-solar -c QP2GS0 -M LV5048
ERROR:MPP-Solar:Response invalid as insufficient number of elements in response. Got 1, expected as least 17
pi@raspberrypi:~ $ mpp-solar -c QP2GS0 -M LV5048 -D
INFO:MPP-Solar:command QP2GS0
INFO:MPP-Solar:Serial device used: /dev/ttyUSB0, baud rate: 2400
INFO:MPP-Solar:Loading commands for inverter model: LV5048
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full command for F
DEBUG:MPP-Solar:Calculating CRC for F
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full command: F)
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for MCHGC
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full command: MCHGC��
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for MNCHGC
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full command: MNCHGCW#
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/commands/muchgc.json
DEBUG:MPP-Solar:Command MUCHGC supported all inverters
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full command for MUCHGC
DEBUG:MPP-Solar:Calculating CRC for MUCHGC
DEBUG:MPP-Solar:Generated CRC bf 86 bf86
DEBUG:MPP-Solar:Full command: MUCHGC��
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/commands/pbcv.json
DEBUG:MPP-Solar:Command PBCV supported all inverters
INFO:MPP-Solar:... command PBCV loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full command for PBCV
DEBUG:MPP-Solar:Calculating CRC for PBCV
DEBUG:MPP-Solar:Generated CRC 64 5a 645a
DEBUG:MPP-Solar:Full command: PBCVdZ
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/commands/pbdv.json
DEBUG:MPP-Solar:Command PBDV supported all inverters
INFO:MPP-Solar:... command PBDV loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full command for PBDV
DEBUG:MPP-Solar:Calculating CRC for PBDV
DEBUG:MPP-Solar:Generated CRC fd cd fdcd
DEBUG:MPP-Solar:Full command: PBDV��
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/commands/pbft.json
DEBUG:MPP-Solar:Command PBFT supported all inverters
INFO:MPP-Solar:... command PBFT loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full command for PBFT
DEBUG:MPP-Solar:Calculating CRC for PBFT
DEBUG:MPP-Solar:Generated CRC bb ed bbed
DEBUG:MPP-Solar:Full command: PBFT��
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/commands/pbt.json
DEBUG:MPP-Solar:Command PBT supported all inverters
INFO:MPP-Solar:... command PBT loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full command for PBT
DEBUG:MPP-Solar:Calculating CRC for PBT
DEBUG:MPP-Solar:Generated CRC 2f 11 2f11
DEBUG:MPP-Solar:Full command: PBT/
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/commands/pcp.json
DEBUG:MPP-Solar:Command PCP supported all inverters
INFO:MPP-Solar:... command PCP loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full command for PCP
DEBUG:MPP-Solar:Calculating CRC for PCP
DEBUG:MPP-Solar:Generated CRC 5c a4 5ca4
DEBUG:MPP-Solar:Full command: PCP\�
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/commands/pcvv.json
[... 198 lines cut ...]
DEBUG:MPP-Solar:Calculating CRC for QVFW2
DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
DEBUG:MPP-Solar:Full command: QVFW2��
DEBUG:MPP-Solar:Searching for cmd 'QP2GS0'
DEBUG:MPP-Solar:Matched: QP2GSn Value: 0
DEBUG:MPP-Solar:Generate full command for QP2GSn0
DEBUG:MPP-Solar:Calculating CRC for QP2GSn0
DEBUG:MPP-Solar:Generated CRC 4c 56 4c56
DEBUG:MPP-Solar:Full command: QP2GSn0LV
INFO:MPP-Solar:SERIAL connection: executing QP2GSn
Parallel Information inquiry
 -- example: QP2GS1 queries the values of various metrics from instance 1 of parallel setup Inverters (numbers from 0)

DEBUG:MPP-Solar:port /dev/ttyUSB0, baudrate 2400
DEBUG:MPP-Solar:Command execution attempt 1...
DEBUG:MPP-Solar:serial response was: (NAKss
DEBUG:MPP-Solar:Response length: 7
DEBUG:MPP-Solar:CRC resp	73 73
DEBUG:MPP-Solar:Calculating CRC for (NAK
DEBUG:MPP-Solar:Generated CRC 73 73 7373
DEBUG:MPP-Solar:CRC calc	73 73
DEBUG:MPP-Solar:CRCs match
ERROR:MPP-Solar:Response invalid as insufficient number of elements in response. Got 1, expected as least 17
INFO:MPP-Solar:Invalid response
pi@raspberrypi:~ $

jblance, 2020-03-28:
Weird - your inverter doesn't seem to respond to the new commands in the documentation, or the documentation is incorrrect? Maybe you have a 2 digit ID or the ID isnt 0-3??
Did it come with software?

DMIINC, 2020-03-28:
Yes. Windows side. It is called watch power. https://usa-mpp-solar.com/manuals/Software/WatchPowerV1.13SP3(latest)/  and you can find Linux stuff on that link.

DMIINC, 2020-03-28:
I know just weird. I have been pulling my hair out for last 4 weeks. Just can't figure it out.

jblance, 2020-03-28:
Can you check the inverters to see what ID's are set
Page39 of the user manual has some details
https://usa-mpp-solar.com/manuals/LV5048%20split%20phase%20manual-20191024.pdf

DMIINC, 2020-03-28:
They are set as they requested it. System is performing flawlessly.

jblance, 2020-03-28:
So the dip switches have the IDs set to 0 and 1?

DMIINC, 2020-03-28:
If you look at the diagram all three are set as they want them to be.

DMIINC, 2020-03-28:
Dalibor Majksner
Hera Realty
2444 Duluth Highway STE 203
Duluth GA 30097
404-710-4876

DMIINC, 2020-03-28:
[image]
[image]
[image]

DMIINC, 2020-03-28:
Watch power app does see all three.

jblance, 2020-03-28:
Nice setup
I was meaning the ID number - was wondering if they were set to something other than 0,1,2 - the manual says it could be 0-7 and as long as they are all different everything will work
[image]
[image]

DMIINC, 2020-03-28:
Yes they are different
#1 off off off
#2 on off off     and 
#3 off on off

jblance, 2020-03-28:
Does the command QPIGS2 work?
i.e.  `mpp-solar -c QP2GS0 -M LV5048 -I`

DMIINC, 2020-03-28:
pi@raspberrypi:~ $ sudo mpp-solar -c QPIGS2 -M LV5048 -I
INFO:MPP-Solar:command QPIGS2
INFO:MPP-Solar:Serial device used: /dev/ttyUSB0, baud rate: 2400
INFO:MPP-Solar:Loading commands for inverter model: LV5048
INFO:MPP-Solar:... command F loaded for inverter model: LV5048
INFO:MPP-Solar:... command MCHGC loaded for inverter model: LV5048
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: LV5048
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: LV5048
INFO:MPP-Solar:... command PBCV loaded for inverter model: LV5048
INFO:MPP-Solar:... command PBDV loaded for inverter model: LV5048
INFO:MPP-Solar:... command PBFT loaded for inverter model: LV5048
INFO:MPP-Solar:... command PBT loaded for inverter model: LV5048
INFO:MPP-Solar:... command PCP loaded for inverter model: LV5048
INFO:MPP-Solar:... command PCVV loaded for inverter model: LV5048
INFO:MPP-Solar:... command PEPD loaded for inverter model: LV5048
INFO:MPP-Solar:... command PF loaded for inverter model: LV5048
INFO:MPP-Solar:... command PGR loaded for inverter model: LV5048
INFO:MPP-Solar:... command POP loaded for inverter model: LV5048
INFO:MPP-Solar:... command POPM loaded for inverter model: LV5048
INFO:MPP-Solar:... command PPCP loaded for inverter model: LV5048
INFO:MPP-Solar:... command PPVOKC loaded for inverter model: LV5048
INFO:MPP-Solar:... command PSDV loaded for inverter model: LV5048
INFO:MPP-Solar:... command PSPB loaded for inverter model: LV5048
INFO:MPP-Solar:... command Q1 loaded for inverter model: LV5048
INFO:MPP-Solar:... command QBOOT loaded for inverter model: LV5048
INFO:MPP-Solar:... command QDI loaded for inverter model: LV5048
INFO:MPP-Solar:... command QFLAG loaded for inverter model: LV5048
INFO:MPP-Solar:... command QID loaded for inverter model: LV5048
INFO:MPP-Solar:... command QMCHGCR loaded for inverter model: LV5048
INFO:MPP-Solar:... command QMOD loaded for inverter model: LV5048
INFO:MPP-Solar:... command QMUCHGCR loaded for inverter model: LV5048
INFO:MPP-Solar:... command QOPM loaded for inverter model: LV5048
INFO:MPP-Solar:... command QP2GSn loaded for inverter model: LV5048
INFO:MPP-Solar:... command QPGSn loaded for inverter model: LV5048
INFO:MPP-Solar:... command QPI loaded for inverter model: LV5048
INFO:MPP-Solar:... command QPIGS loaded for inverter model: LV5048
INFO:MPP-Solar:... command QPIGS2 loaded for inverter model: LV5048
INFO:MPP-Solar:... command QPIRI loaded for inverter model: LV5048
INFO:MPP-Solar:... command QPIWS loaded for inverter model: LV5048
INFO:MPP-Solar:... command QVFW loaded for inverter model: LV5048
INFO:MPP-Solar:... command QVFW2 loaded for inverter model: LV5048
INFO:MPP-Solar:SERIAL connection: executing QPIGS2
General Status Parameters inquiry
 -- queries the value of various metrics from the Inverter

is_l2_ac_charging_on          	0               True - 1/False - 0
is_l2_line_not_ok             	0               True - 1/False - 0
is_l2_scc_charging_on         	0               True - 1/False - 0
is_l2_scc_ok                  	0               True - 1/False - 0
is_load_on                    	1               True - 1/False - 0
l2_ac_input_frequency         	60.0            Hz
l2_ac_input_voltage           	123.8           V
l2_ac_output_active_power     	0492            W
l2_ac_output_apparent_power   	0539            VA
l2_ac_output_frequency        	60.0            Hz
l2_ac_output_load             	021             %
l2_ac_output_voltage          	119.8           V
l2_battery_voltage            	52.20           V
pv2_battery_charging_current  	0000            A
pv2_input_voltage             	000.0           V
reserved                      	1               True - 1/False - 0
pi@raspberrypi:~ $

jblance, 2020-03-28:
Interesting - so the only ones that dont seem to work are the parallel commands
can you run `mpp-solar -c QPIRI -M LV5048` for me please
Also, what is the box below the 3 inverters?

DMIINC, 2020-03-28:
pi@raspberrypi:~ $ sudo mpp-solar -c QPIRI -M LV5048
WARNING:MPP-Solar:Serial read error: list index out of range
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.2.2', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/__init__.py", line 74, in main
    results = mp.getResponseDict(args.command)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/mpputils.py", line 37, in getResponseDict
    return self.inverter.getResponseDict(cmd)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/mppinverter.py", line 182, in getResponseDict
    return result.getResponseDict()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.2.2-py2.7.egg/mppsolar/mppcommand.py", line 196, in getResponseDict
    msgs[key] = [resp_format[2][int(result)], '']
IndexError: list index out of range

DMIINC, 2020-03-28:
Box houses DC breakers for each inverter and generic battery monitor.

jblance, 2020-03-28:
Bummer, ok try `sudo mpp-solar -c QPIRI -M LV5048 -R`

DMIINC, 2020-03-28:
pi@raspberrypi:~ $ sudo mpp-solar -c QPIRI -M LV5048 -R
WARNING:MPP-Solar:Serial read error: list index out of range
(120.0 20.8 120.0 60.0 20.8 2500 2500 48.0 47.0 43.0 56.1 56.1 2 40 060 0 0 0 9 01 0 6 55.0 0 1 000��

jblance, 2020-03-28:
OK, so fixed the above error - but I just can't figure out why any command with the parallel ID fails to work - it looks like the command isn't recognized but it is the command as per the documentation.

DMIINC, 2020-03-28:
Thanks . I will  reinstall in the morning and try again. I really appreciate your help.

DMIINC, 2020-03-28:
QPIGS Inverter general status parameters inquiry for Line 1 of one inverter. It works.
QPIGS2 Inverter general status parameters inquiry for Line 2 of one inverter. It works.

With these commands above I see all of the information from one inverter. 
Since L1 and L2 are 180 deg out that gives 240 volts across them. Full phase. 
L1 to N = 120 Volts L2 to N = 120 Volts for split phase.

For this to work, I think each inverter will need it's own connection to Raspberry Pi. USB to Serial. I do not think that we know the code to pick up all of the information that is floating around through parallel cables between inverters.

DMIINC, 2020-03-28:
Bought 2 more USB to serial connectors. Looking better. Shows no errors but no information comes back for QP2GSn QPGSn

pi@raspberrypi:~ $ sudo mpp-solar -c QP2GS0 -M LV5048
pi@raspberrypi:~ $ sudo mpp-solar -c QP2GS1 -M LV5048
pi@raspberrypi:~ $ sudo mpp-solar -c QPGS1 -M LV5048
pi@raspberrypi:~ $ sudo mpp-solar -c QPGS2 -M LV5048
pi@raspberrypi:~ $ sudo mpp-solar -c QPIGS -M LV5048
battery_capacity              	100             %
battery_charging_current      	000             A
battery_discharge_current     	00000           A
battery_voltage               	56.20           V
battery_voltage_from_scc      	00.00           V
bus_voltage                   	225             V
inverter_heat_sink_temperature	0036            Deg_C
is_battery_over               	0               True - 1/False - 0
is_battery_under              	0               True - 1/False - 0
is_l1_ac_charging_on          	0               True - 1/False - 0
is_l1_line_not_ok             	1               True - 1/False - 0
is_l1_scc_charging_on         	0               True - 1/False - 0
is_l1_scc_ok                  	0               True - 1/False - 0
is_load_on                    	0               True - 1/False - 0
l1_ac_input_frequency         	59.9            Hz
l1_ac_input_voltage           	122.8           V
l1_ac_output_active_power     	0392            W
l1_ac_output_apparent_power   	0392            VA
l1_ac_output_frequency        	59.9            Hz
l1_ac_output_load             	015             %
l1_ac_output_voltage          	122.8           V
pv_input_current_for_battery  	0000            A
pv_input_voltage              	000.0           V
unknown_value_in_response     	110             
pi@raspberrypi:~ $ sudo mpp-solar -c QPIGS2 -M LV5048
is_l2_ac_charging_on          	0               True - 1/False - 0
is_l2_line_not_ok             	0               True - 1/False - 0
is_l2_scc_charging_on         	0               True - 1/False - 0
is_l2_scc_ok                  	0               True - 1/False - 0
is_load_on                    	1               True - 1/False - 0
l2_ac_input_frequency         	59.9            Hz
l2_ac_input_voltage           	123.5           V
l2_ac_output_active_power     	0094            W
l2_ac_output_apparent_power   	0111            VA
l2_ac_output_frequency        	59.9            Hz
l2_ac_output_load             	004             %
l2_ac_output_voltage          	123.5           V
l2_battery_voltage            	56.20           V
pv2_battery_charging_current  	0000            A
pv2_input_voltage             	000.0           V
reserved                      	0               True - 1/False - 0

jblance, 2020-03-28:
Great that you have a solution. Just strange that the commands dont work as per the docs. My parallel 4048s work with QPGS0 and 1 showing the data from the 2 inverters while only connected to one of them.

DMIINC, 2020-03-28:
Well. Now when I type pi@raspberrypi:~ $ sudo mpp-solar -c QP2GS0 -M LV5048 it returns as this

pi@raspberrypi:~ $ sudo mpp-solar -c QP2GS0 -M LV5048
pi@raspberrypi:~ $ 

It does not give error or anything else.

jblance, 2020-03-28:
Sorry, yes i was cleaning up some of the log messages
I have added a bit more information to the response if the command fails now - e.g.
`mpp-solar -c QPIRI -d TEST -M LV5048`
```
error                           Invalid response
response                        (NAKss
```

DMIINC, 2020-03-28:
Will test it out. Thank you.

cowpen, 2020-03-28:
Very glad to see you guys working together on a solution to this. I have an LV5048 and have been struggling for a couple of months trying to get a data logged in a useful format with "solar-sis" for the 4048. Your efforts are very much appreciated!

DMIINC, 2020-03-28:
Ok. I see what you mean. Well. It is another wall barrier. I need to put my hard thinking hat on.

jblance, 2020-03-28:
ok, so the java code shows the the is QPGS0 - also have some indication that it tries multiple times...
e.g. the below tries 3 times
```
for (try_i = 0, try_i = 0; try_i < 3; ++try_i) {
                ret = handler.excuteCommand("QPGS0", true);
                if (!ret.equals("(NAK")) {
                    break;
                }
            }
```

Maybe try bruting it?
`for x in 1 2 3 4 5; do echo $x; sudo mpp-solar -c QP2GS0 -M LV5048; done`

DMIINC, 2020-03-28:
With my thinking hat on........ can it be done via USB connection?
Three USB cables = 3 devices
There for mpp-solar -c QPIGS or QPIGS2 -d /dev/hidraw0 or 1 or 2

DMIINC, 2020-03-28:
How do I try that command?
Maybe try bruting it? Can you try the below:
for x in 1 2 3 4 5; do echo $x; sudo mpp-solar -c QP2GS0 -M LV5048; done

DMIINC, 2020-03-28:
pi@raspberrypi:~ $ echo $1; sudo mpp-solar -c QP2GS0 -M LV5048; done
bash: syntax error near unexpected token `done'
pi@raspberrypi:~ $ echo $1; sudo mpp-solar -c QP2GS0 -M LV5048;

error                         	Invalid response 
response                      	(NAKss          
pi@raspberrypi:~ $ echo $1; sudo mpp-solar -c QP2GS0 -M LV5048

error                         	Invalid response 
response                      	(NAKss          
pi@raspberrypi:~ $

jblance, 2020-03-28:
Yes if they are direct USB from the Pi to the Inverter they will appear and 3 different hidraw devices
If they are 3 USB-serial converters they will be 3 different /dev/ttyUSBn devices (where n is a number like 0,1,2,3 etc)
Originally I had 2 USB-serial converters (one to each inverter) and no other USB devices on my pi and the 2 devices were /dev/ttyUSB0 and /dev/ttyUSB1

Note if you connect both the USB and serial ports on the inverter problems occur (at least for me only one worked)

jblance, 2020-03-28:
just copy the entire line on the command line
`$ for x in 1 2 3 4 5; do echo $x; sudo mpp-solar -c QP2GS0 -M LV5048; done`
and hit enter

DMIINC, 2020-03-28:
No I only have USB to Serial cables on Raspberry Pi with wireless keyboard and mouse
Of course. I do understand.

DMIINC, 2020-03-28:
pi@raspberrypi:~ $ for x in 1 2 3 4 5; do echo $x; sudo mpp-solar -c QP2GS0 -M LV5048; done
1
error                         	Invalid response 
response                      	(NAKss          
2
error                         	Invalid response 
response                      	(NAKss          
3
error                         	Invalid response 
response                      	(NAKss          
4
error                         	Invalid response 
response                      	(NAKss          
5
error                         	Invalid response 
response                      	(NAKss          
pi@raspberrypi:~ $ for x in 1 2 3 4 5; do echo $x; sudo mpp-solar -c QPGS0 -M LV5048; done
1
error                         	Invalid response 
response                      	(NAKss          
2
error                         	Invalid response 
response                      	(NAKss          
3
error                         	Invalid response 
response                      	(NAKss          
4
error                         	Invalid response 
response                      	(NAKss          
5
error                         	Invalid response 
response                      	(NAKss          
pi@raspberrypi:~ $

jblance, 2020-03-28:
at least it is consistent :-(
just on the off chance that we have some other ID can you try
` for x in 0 1 2 3 4 5 6 7; do sudo mpp-solar -c QPGS${x} -M LV5048; done`

if all USB-serial adapters are on the same device you should info from the other inverters with
`sudo mpp-solar -c QID -d /dev/ttyUSB0`
`sudo mpp-solar -c QID -d /dev/ttyUSB1`
`sudo mpp-solar -c QID -d /dev/ttyUSB2`

QID should be the serial number

jblance, 2020-03-29:
@cowpen  
> Very glad to see you guys working together on a solution to this. I have an LV5048 and have been struggling for a couple of months trying to get a data logged in a useful format with "solar-sis" for the 4048. Your efforts are very much appreciated!
Is it working for you?

DMIINC, 2020-03-29:
pi@raspberrypi:~ $ for x in 0 1 2 3 4 5 6 7; do sudo mpp-solar -c QPGS${x} -M LV5048; done
error                         	Invalid response 
response                      	(NAKss          
error                         	Invalid response 
response                      	(NAKss          
error                         	Invalid response 
response                      	(NAKss          
error                         	Invalid response 
response                      	(NAKss          
error                         	Invalid response 
response                      	(NAKss          
error                         	Invalid response 
response                      	(NAKss          
error                         	Invalid response 
response                      	(NAKss          
error                         	Invalid response 
response                      	(NAKss          
pi@raspberrypi:~ $

DMIINC, 2020-03-29:
pi@raspberrypi:~ $ sudo mpp-solar -c QID -d /dev/ttyUSB0
serial_number                 	92912001100085  
pi@raspberrypi:~ $ sudo mpp-solar -c QID -d /dev/ttyUSB1
serial_number                 	92911906100045  
pi@raspberrypi:~ $ sudo mpp-solar -c QID -d /dev/ttyUSB2
serial_number                 	92911906100023  
pi@raspberrypi:~ $ 

All of the inverters are there.
Serial number correspond to physical stickers on them.

DMIINC, 2020-03-29:
Check this out

pi@raspberrypi:~ $ sudo mpp-solar -c QPIGS -d /dev/ttyUSB2
ac_input_frequency            	59.9            Hz
ac_input_voltage              	123.5           V
ac_output_active_power        	0426            W
ac_output_apparent_power      	0444            VA
ac_output_frequency           	59.9            Hz
ac_output_load                	017             %
ac_output_voltage             	123.5           V
battery_capacity              	100             %
battery_charging_current      	000             A
battery_discharge_current     	00000           A
battery_voltage               	56.10           V
battery_voltage_from_scc      	00.00           V
bus_voltage                   	226             V
inverter_heat_sink_temperature	0039            Deg_C
is_battery_voltage_to_steady_while_charging	0               True - 1/False - 0
is_charging_on                	1               True - 1/False - 0
is_configuration_changed      	0               True - 1/False - 0
is_load_on                    	0               True - 1/False - 0
is_sbu_priority_version_added 	0               True - 1/False - 0
is_scc_charging_on            	0               True - 1/False - 0
is_scc_firmware_updated       	0               True - 1/False - 0
pv_input_current_for_battery  	0000            A
pv_input_voltage              	000.0           V
unknown_value_in_response     	110             
pi@raspberrypi:~ $ sudo mpp-solar -c QPIGS2 -d /dev/ttyUSB2
CRITICAL:MPP-Solar:Command not found
pi@raspberrypi:~ $ sudo mpp-solar -c QPIGS2 -d /dev/ttyUSB2 -M LV5048
is_l2_ac_charging_on          	0               True - 1/False - 0
is_l2_line_not_ok             	0               True - 1/False - 0
is_l2_scc_charging_on         	0               True - 1/False - 0
is_l2_scc_ok                  	0               True - 1/False - 0
is_load_on                    	1               True - 1/False - 0
l2_ac_input_frequency         	59.9            Hz
l2_ac_input_voltage           	124.2           V
l2_ac_output_active_power     	0242            W
l2_ac_output_apparent_power   	0260            VA
l2_ac_output_frequency        	59.9            Hz
l2_ac_output_load             	010             %
l2_ac_output_voltage          	124.2           V
l2_battery_voltage            	56.20           V
pv2_battery_charging_current  	0000            A
pv2_input_voltage             	000.0           V
reserved                      	0               True - 1/False - 0
pi@raspberrypi:~ $

jblance, 2020-03-29:
So at least you can query all of the inverters - no idea why my parallel commands don't work

There is also an undocumented command that sends info to a MQTT broker if that is useful (it might need tweaking to support the new model parameter

DMIINC, 2020-03-29:
Once again thanks for all the support you have provided so far. I see you closed other thread. Still want to buy you a lunch regardless. HaHa.

More elegant solution, of course would be using -QPGSn or -QP2GSn. Why we can't do that I do not know. Last time I wrote some ""serious" code was in  1990 using BASIC, LOL. So I do understand basic principles of programming and how difficult it can be to debug. So question I can ask would be:
1. Would updating Python to latest and greatest help?
2. Reinstalling OS on Pi to something that you know it works?
3. Shift focus away from trying to get those commands to work to start implementing database and Grafana.

[194 more replies]

closed 2020-04-04 by jblance