jblance/mpp-solar #12: Directory structure has changed?

BarryRGreen, 2020-03-28:
Hi

Great video and thanks very much for providing the code too.

I am trying to do something very similar - I have an MPP Solar PIP 5048 MK attached to my solar panels and home-made battery (10kWh, made from recycled e-Bike batteries). The set-up is working fine but I need something much more than what WatchPower can provide. 

While following your video, I noticed that the download image I used differed somewhat from the one used in the video. In particular, the directory structure was different. I don't mind doing the coding, but was hoping that your code would provide a very quick 'springboard' to my project. 

I can't find the basic scripts that handle the interfacing with the RS232 output of the PIP.

Please can you direct me to where I can find these, or if they are not included in the image currently downloadable, are there some 'starter scripts' that you can provide for the PIP?

I don't have a Batrium (yet!), so the code for the Batrium interface doesn't help.

Many Thanks,

jblance, 2020-03-28:
I think you might be in the wrong place? No batrium code in this repo, just some python code that talks to the inverters

jblance, 2020-03-29:
Closing as not related to this code base

closed 2020-03-29 by jblance

BarryRGreen, 2020-03-29:
Hi 
Thanks for replying - and indeed you were perfectly correct. I posted my comment to the wrong GitHub (was meant to be on "Solar-sis" - similar topic)!
Thanks again........