jblance/mpp-solar #13: everything installed but does not start!

cheomfp, 2020-04-13:
Hello!
I have tried to make everything work but I am not successful.
everything is installed, but I have not seen anything in GRAFANA. I think I am missing some configuration.
the investor responds to the compando, here is an attached photo:
[image]

jblance, 2020-04-13:
So mpp-solar is working

If you are trying to graph data in grafana, you will have to check each step is working 
- is the data being published to MQTT?
- is telegraf loading it into influx
- is there data in influx
- what data does the influx source give you in grafana

cheomfp, 2020-04-13:
Yes, I have everything tested and installed, the only thing is that I cannot see anything in Grafana.
Could you come in and take a look at teamviwer? plss
ID 1402080010
pass: 2616

jblance, 2020-04-13:
I dont wont to start that level of support - also its not wise to publish those details
Please post the test results for the above

cheomfp, 2020-04-13:
I apologize that I am a beginner in this, and have been learning a little from you to find an alternative to the software that the inverter brings (watchpower).

jblance, 2020-04-13:
Thats ok - one reason for you to do this is too learn...
- MQTT test is subscribing to all topics and validating the inverter data is being published (post the results)
- telegraf test is to check the status of the telegraf service (and post results, especially the log items)
- influx - connect to the db and show measurements, then look at the contents of one of them
- grafana, start by defining a query for a table

cheomfp, 2020-04-13:
Can you give me instructions to obtain the information you are requesting?

[image]

cheomfp, 2020-04-13:
Influx
[image]

jblance, 2020-04-13:
so first off, it doesnt look  like you are sending any info on a regular basis
you need to setup cron or the mpp-solar service

jblance, 2020-04-13:
influx is empty - by default telegraf will put computer stats in there. have you configured telegraf?
what does `sudo systemctl status telegraf` give

cheomfp, 2020-04-13:
TELEGRAF
[image]

jblance, 2020-04-13:
you truncated the last few lines (including an error) what do they say

cheomfp, 2020-04-13:
[image]

jblance, 2020-04-13:
have you created the database `telegraf` in influx?

cheomfp, 2020-04-13:
I think I haven't created it yet.
Could you tell me how to do it?

jblance, 2020-04-13:
`influx`
```create database telegraf
use telegraf

create user grafana with password '<passwordhere>' with all privileges
grant all privileges on telegraf to grafana
```

cheomfp, 2020-04-13:
> `influx`
> 
> ```
> use telegraf
> 
> create user grafana with password '<passwordhere>' with all privileges
> grant all privileges on telegraf to grafana
> ```

yes, I have done this before

cheomfp, 2020-04-13:
I think I have done everything that is in your publication, what I would lack is how to make the grafana read from my inverter

cheomfp, 2020-04-13:
What I want to show in grafana would be:
the load of L1, L2
production of solar panels PV1, PV2
Maybe over time, as you learn a little about this, keep adding a few more things: D

jblance, 2020-04-13:
you might have done it, but telegraf is failing to write to influx and you are not regularly sending information to MQTT
So there is no way that grafana will be able to display anything from influx

cheomfp, 2020-04-13:
Sorry if I don't understand very well, since my English is not very good. my language is spanish :P

jblance, 2020-04-13:
can you post your `/etc/telegraf/telegraf.d/influx-output.conf`

cheomfp, 2020-04-13:
> puedes publicar tu `/etc/telegraf/telegraf.d/influx-output.conf`

[image]

jblance, 2020-04-13:
Thats weird, coz the error in the telegraf service mentions database `telegraf`

jblance, 2020-04-13:
did you edit the /etc/telegraf/telegraf.conf file?

cheomfp, 2020-04-13:
not yet!
[image]

jblance, 2020-04-13:
ok - you shouldn't need to
can you post the output of the above (as it displays the file without all the comments and blank lines)
also try restarting telegraf `sudo systemctl restart telegraf`

cheomfp, 2020-04-13:
> did you edit the /etc/telegraf/telegraf.conf file?
> can you post the output of `egrep -v "#|^$" /etc/telegraf/telegraf.conf`

[image]

jblance, 2020-04-13:
ok, you do need to edit it, find the line with `[[outputs.influxdb]]` and put a `#` at the start (i.e. comment out that line)
then restart telegraf as above
then post the output of `sudo systemctl status telegraf`

cheomfp, 2020-04-13:
> ok, you do need to edit it, find the line with `[[outputs.influxdb]]` and put a `#` at the start (i.e. comment out that line)
> then restart telegraf as above
> then post the output of `sudo systemctl status telegraf`

[image]
It seems to me that now it gave no error

jblance, 2020-04-13:
now on a command line
`influx`
`use home`
`show measurements` 
and post that output

cheomfp, 2020-04-13:
> looking much better - telegraf seems to be working
> now on a command line
> `influx`
> `use home`
> `show measurements`
> and post that output

[image]

jblance, 2020-04-13:
ok so now you just need to be running the command regularly - did you follow the Schedule regular queries step? https://github.com/jblance/mpp-solar/blob/master/docs/MQTT_Influx_Grafana.md#schedule-regular-queries-and-mqtt-message-posts
What command are you wanting the output of?

cheomfp, 2020-04-13:
1- total production of solar panels.
2- Current load of lines L1, L2.
3- Voltage of the Batteries.

jblance, 2020-04-13:
you need to pick the commands you want to run, looks like the LV5048 version of `QPIGS`
so you need to use cron to schedule `/usr/local/bin/mpp-info-pub -q localhost -M LV5048 -c QPIGS -i > /home/pi/cron.out 2>&1`
or setup the mpp-solar service

cheomfp, 2020-04-13:
mpp-solar service configuration
[image]

jblance, 2020-04-13:
you need to edit `/etc/mpp-solar/mpp-solar.conf`

jblance, 2020-04-13:
actually you'll be better using cron as you need sudo for access to /dev/hidraw

cheomfp, 2020-04-14:
I'm really lost, I don't know what I have to do. it will take me a little more time to learn how to do all this

jblance, 2020-04-14:
the readme has a section on cron:

Schedule Regular Queries (and MQTT message posts)
Use cron or similar to schedule the command to run on a regular basis e.g. for cron crontab -e to edit and add a line like

`* * * * * /usr/local/bin/mpp-info-pub -q localhost -q localhost -M LV5048 -c QPIGS -i > /home/pi/cron.out 2>&1`

start with that

cheomfp, 2020-04-14:
Something like this?
[image]

jblance, 2020-04-14:
Yes. You'll need to add `sudo ` after the * and before the / though

Then check the mosquito sub window

cheomfp, 2020-04-14:
I am very grateful for your patience and help. If there is a way to make any kind of donation to contribute to the project, please let me know, I am willing to contribute.
I will continue reading a little more to see if I have any progress.😉

DMIINC, 2020-04-14:
Order in which install should be done in my opinion. jblance posted the picture which hints as to what needs to be done and in which order. But in any case:
1. MPP Solar Inverter Python Package and test with sample command.
2. Setup Mosquito MQTT broker and test
3. Install Influxdb. Setup database.
4. Install  Telegraf and setup.
5. Install python-systemd and add mpp-solar as service.
6. Start influxdb and make sure that you are collecting data.

Follow instructions to the last letter.

Only after all of the above is working correctly should you install Grafana and start visualizing you database.

cheomfp, 2020-04-14:
> Order in which install should be done in my opinion. jblance posted the picture which hints as to what needs to be done and in which order. But in any case:
> 
> 1. Install MPP Solar Inverter Python Package and test with sample commands.
> 2. Setup Mosquito MQTT broker and test
> 3. Install Influxdb. Setup database.
> 4. Install  Telegraf and setup.
> 5. Install python-systemd and add mpp-solar as service.
> 6. Start influxdb and make sure that you are collecting data.
> 
> Follow instructions to the last letter. DO NOT USE CRONTAB.
> 
> Only after all of the above is working correctly should you install Grafana and start visualizing you database.

Hello!
Thanks for the suggestion, I did it the same way!
Here I attach a screenshot where you see the command running every minute.
[image]

DMIINC, 2020-04-14:
Awesome. Did you start using grafana?

cheomfp, 2020-04-14:
> Awesome. Did you start using grafana?

I am using it but there are no graphics to display.
[image]

DMIINC, 2020-04-14:
It looks like you need to figure out how to use Grafana.

jblance, 2020-04-14:
what is in influx?

cheomfp, 2020-04-14:
I think I am progressing very well, thanks to the help of all of you!
[image]

jblance, 2020-04-14:
great!

cheomfp, 2020-04-14:
This is what I have at the moment, just one phase (L1) and I am looking for the other phase of the inverter (L2). I have been testing various commands but still without success:
* * * * * /usr/local/bin/mpp-info-pub -q localhost -c QPIGS,QPIGS2 -i > /home/pi/cron.out 2>&1 -d /dev/hidraw0
[image]
[image]

jblance, 2020-04-15:
you need to include your model in the command to be able to access LV5048
only commands (of which QPIGS2 is one)
also ensure you are either using cron OR the service not both

cheomfp, 2020-04-15:
I launched this command and got no results.
* * * * * /usr/local/bin/mpp-info-pub -q localhost -M LV5048 -c QPIGS2 -i > /home/pi/cron.out 2>&1 -d /dev/hidraw0

cheomfp, 2020-04-15:
with this command if you can succeed!
* * * * * /usr/local/bin/mpp-info-pub -q localhost -M LV5048 -c QPIGS -i > /home/pi/cron.out 2>&1 -d /dev/hidraw0

jblance, 2020-04-15:
try on command line first
you'll need sudo unless you have allowed your current user access to
/dev/hidraw0

cheomfp, 2020-04-15:
> try on command line first you'll need sudo unless you have allowed your current user access to /dev/hidraw0
> […](#)

* * * * *sudo /usr/local/bin/mpp-info-pub -q localhost -M LV5048 -c QPIGS2 -i > /home/pi/cron.out 2>&1 -d /dev/hidraw0

If you mean this, I have already tried it and it is waiting for an answer.

jblance, 2020-04-15:
you didnt include `-d /dev/hidraw0` in the command (it needs to be before the `>`
I'm confused how `/usr/local/bin/mpp-info-pub -q localhost -M LV5048 -c QPIGS -i > /home/pi/cron.out 2>&1 -d /dev/hidraw0` could have worked for the same reason
these lines are for entering into crontab - not command line
the `* * * * *` is tell cron to run the following command every minute
the `> /home/pi/cron.out 2>&1 ` piece redirects the output and errors of the command to a file
the `/usr/local/bin/mpp-info-pub -q localhost -M LV5048 -c QPIGS -i` is the command
the `-d /dev/hidraw0` on the end should just be ignored

for testing running the commands in a command window is best
once you have the command working as expected you can then add it to crontab

cheomfp, 2020-04-15:
with the QPIGS command it works perfectly for the LV 5048 model, but as for the other QPIGS2 it does not recognize it

jblance, 2020-04-15:
if you are meaning the example above that you said worked, there are no values in the response

comment out any entries in crontab (add a `#` at the start of the line)
restart the mosquitto_sub process (so no history is there)

on the command line run `sudo /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -M LV5048 -c QPIGS2 -i `
if you get no response in the mosquitto_sub window, on the command line run `sudo mpp-solar -d /dev/hidraw0 -M LV5048 -c QPIGS2`
if still no response (in the command window) run `sudo mpp-solar -d /dev/hidraw0 -M LV5048 -c QPIGS2 -D`

cheomfp, 2020-04-15:
[image]

[76 more replies]

closed 2020-04-15 by jblance