jblance/mpp-solar #16: Failed to start MPP Solar Service

dricote, 2020-04-26:
Hello, I cannot start the service, I use a raspi2 and Raspbian Buster, when I enter this I get the following error.

```
pi@PIP5048MG:~ $ systemctl --user status mpp-solar
● mpp-solar.service - MPP Solar Service
   Loaded: loaded (/etc/systemd/user/mpp-solar.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-04-26 06:10:55 CEST; 2min 23s ago
  Process: 789 ExecStart=/usr/bin/python /usr/local/bin/mpp-solar-service -c /etc/mpp-solar/m
 Main PID: 789 (code=exited, status=1/FAILURE)

abr 26 06:10:55 PIP5048MG systemd[542]: Failed to start MPP Solar Service.
abr 26 06:10:55 PIP5048MG systemd[542]: mpp-solar.service: Service RestartSec=100ms expired,
abr 26 06:10:55 PIP5048MG systemd[542]: mpp-solar.service: Scheduled restart job, restart cou
abr 26 06:10:55 PIP5048MG systemd[542]: Stopped MPP Solar Service.
abr 26 06:10:55 PIP5048MG systemd[542]: mpp-solar.service: Start request repeated too quickly
abr 26 06:10:55 PIP5048MG systemd[542]: mpp-solar.service: Failed with result 'exit-code'.
abr 26 06:10:55 PIP5048MG systemd[542]: Failed to start MPP Solar Service.
```
Thanks

dricote, 2020-04-26:
This is my mpp-solar.conf

```
[SETUP]
# Number of seconds to pause between command execution loop
# 0 is no pause, greater than 60 will cause service restarts
pause=5
mqtt_broker=localhost

# This example would work on 2x PIP4048 installed in parallel
[Inverter_1]
model=standard
port=/dev/hidraw0
baud=2400
command=QPGS0
tag=QPGS0
format=influx2
```

jblance, 2020-04-26:
Hi
try running the service command on the command line 
`/usr/bin/python /usr/local/bin/mpp-solar-service -c /etc/mpp-solar/mpp-solar.conf`
next try the command line version:
`mpp-info-pub -q localhost -d /dev/hidraw0 -c QPGS0 -I`
finally try the standard mpp-solar command (which allows extra info)
`mpp-solar -d /dev/hidraw0 -c QPGS0 -I` (change `-I` to `-D` for even more info on this command)
post any output

Cheers

dricote, 2020-04-26:
When trying to lift the service it tells me the following
/usr/bin/python /usr/local/bin/mpp-solar-service -c /etc/mpp-solar/mpp-solar.conf

```
pi@PIP5048MG:~ $ /usr/bin/python /usr/local/bin/mpp-solar-service -c /etc/mpp-solar/mpp-solar.conf
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar-service", line 11, in <module>
    load_entry_point('mpp-solar==0.3.1', 'console_scripts', 'mpp-solar-service')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.3.1-py2.7.egg/mppsolar/mpp_solar_service.py", line 11, in <module>
    import paho.mqtt.publish as publish
ImportError: No module named paho.mqtt.publish
```

dricote, 2020-04-26:
Seen the error it gave, I have looked at another issue and installed paho-mqtt with `sudo pip install paho-mqtt` and the result has been the following:

```
pi@PIP5048MG:~ $ /usr/bin/python /usr/local/bin/mpp-solar-service -c /etc/mpp-solar/mpp-solar.conf
MPP-Solar-Service: Initializing ...
MPP-Solar-Service: Config file: /etc/mpp-solar/mpp-solar.conf
MPP-Solar-Service: Config setting - pause: 5
MPP-Solar-Service: Config setting - mqtt_broker: localhost
MPP-Solar-Service: Config setting - command sections found: 1
MPP-Solar-Service: item {'tag': u'QPGS0', 'command': u'QPGS0', 'mp': <mppsolar.mpputils.mppUtils instance at 0x75dd45f8>, 'format': u'influx2'}
```

Then I have activated the service normally with `systemctl --user start mpp-solar`

```
pi@PIP5048MG:~ $ systemctl --user start mpp-solar
pi@PIP5048MG:~ $ systemctl --user status mpp-solar
● mpp-solar.service - MPP Solar Service
   Loaded: loaded (/etc/systemd/user/mpp-solar.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-04-26 18:27:44 CEST; 9s ago
 Main PID: 1532 (python)
   CGroup: /user.slice/user-1000.slice/user@1000.service/mpp-solar.service
           └─1532 /usr/bin/python /usr/local/bin/mpp-solar-service -c /etc/mpp-solar/mpp-solar.

abr 26 18:27:42 PIP5048MG systemd[413]: Starting MPP Solar Service...
abr 26 18:27:44 PIP5048MG python[1532]: MPP-Solar-Service: Initializing ...
abr 26 18:27:44 PIP5048MG python[1532]: MPP-Solar-Service: Config file: /etc/mpp-solar/mpp-sola
abr 26 18:27:44 PIP5048MG python[1532]: MPP-Solar-Service: Config setting - pause: 5
abr 26 18:27:44 PIP5048MG python[1532]: MPP-Solar-Service: Config setting - mqtt_broker: localh
abr 26 18:27:44 PIP5048MG python[1532]: MPP-Solar-Service: Config setting - command sections fo
abr 26 18:27:44 PIP5048MG python[1532]: MPP-Solar-Service: item {'tag': u'QPGS0', 'command': u'
abr 26 18:27:44 PIP5048MG systemd[413]: Started MPP Solar Service.
```
I had some packages retained when upgrade that I do not know if they would have to see, they are the following:
`binutils binutils-arm-linux-gnueabihf binutils-common libbinutils`
I have installed them and then I have done the installation of `paho-mqtt`

Then i tried `mpp-info-pub -q localhost -d /dev/hidraw0 -c QPGS0 -I` with `sudo` in front because otherwise it doesn't work for me, and in another console I have received the following:

```
pi@PIP5048MG:~ $ mosquitto_sub -h localhost -v -t "#"
mpp-solar mpp-solar,command=QPGS0 parallel_instance_number="valid"
mpp-solar mpp-solar,command=QPGS0 serial_number=9.29319031031e+13
mpp-solar mpp-solar,command=QPGS0 work_mode="Battery Mode"
mpp-solar mpp-solar,command=QPGS0 fault_code="No fault"
mpp-solar mpp-solar,command=QPGS0 grid_voltage=0.0
mpp-solar mpp-solar,command=QPGS0 grid_frequency=0.0
mpp-solar mpp-solar,command=QPGS0 ac_output_voltage=230.1
mpp-solar mpp-solar,command=QPGS0 ac_output_frequency=50.0
mpp-solar mpp-solar,command=QPGS0 ac_output_apparent_power=115.0
mpp-solar mpp-solar,command=QPGS0 ac_output_active_power=108.0
mpp-solar mpp-solar,command=QPGS0 load_percentage=2.0
mpp-solar mpp-solar,command=QPGS0 battery_voltage=53.3
mpp-solar mpp-solar,command=QPGS0 battery_charging_current=2.0
mpp-solar mpp-solar,command=QPGS0 battery_capacity=75.0
mpp-solar mpp-solar,command=QPGS0 pv_input_voltage=223.2
mpp-solar mpp-solar,command=QPGS0 total_charging_current=2.0
mpp-solar mpp-solar,command=QPGS0 total_ac_output_apparent_power=115.0
mpp-solar mpp-solar,command=QPGS0 total_output_active_power=108.0
mpp-solar mpp-solar,command=QPGS0 total_ac_output_percentage=2.0
mpp-solar mpp-solar,command=QPGS0 is_scc_ok=1
mpp-solar mpp-solar,command=QPGS0 is_ac_charging=0
mpp-solar mpp-solar,command=QPGS0 is_scc_charging=1
mpp-solar mpp-solar,command=QPGS0 is_battery_over_voltage=0
mpp-solar mpp-solar,command=QPGS0 is_battery_under_voltage=0
mpp-solar mpp-solar,command=QPGS0 is_line_lost=1
mpp-solar mpp-solar,command=QPGS0 is_load_on=1
mpp-solar mpp-solar,command=QPGS0 is_configuration_changed=0
mpp-solar mpp-solar,command=QPGS0 output_mode="single machine"
mpp-solar mpp-solar,command=QPGS0 charger_source_priority="Solar + Utility"
[... 418 lines cut ...]
grid_voltage                    000.0           V
is_ac_charging                  0               True - 1/False - 0
is_battery_over_voltage         0               True - 1/False - 0
is_battery_under_voltage        0               True - 1/False - 0
is_configuration_changed        0               True - 1/False - 0
is_line_lost                    1               True - 1/False - 0
is_load_on                      1               True - 1/False - 0
is_scc_charging                 1               True - 1/False - 0
is_scc_ok                       1               True - 1/False - 0
load_percentage                 002             %
max_ac_charger_current          50              A
max_charger_current             080             A
max_charger_range               080             A
output_mode                     single machine
parallel_instance_number        valid
pv_input_current                00              A
pv_input_voltage                222.6           V
serial_number                   92931903103094
total_ac_output_apparent_power  00115           VA
total_ac_output_percentage      001             %
total_charging_current          000             A
total_output_active_power       00099           W
work_mode                       Battery Mode
```
Cheers!

dricote, 2020-04-26:
as a troubleshooting, tell you, when installing grafana this not found:

Start grafana and set to run at boot
```
sudo systemctl unmask grafana
sudo systemctl start grafana
sudo systemctl enable grafana
```
In the grafana installation it indicates the following:

```
### NOT starting on installation, please execute the following statements to configure grafana to start automatically using systemd
 sudo /bin/systemctl daemon-reload
 sudo /bin/systemctl enable grafana-server
### You can start grafana-server by executing
 sudo /bin/systemctl start grafana-server
```

I thought it interesting to tell you

Cheers!

jblance, 2020-04-26:
After installing the missing package it looks to be working. Do you get info in the mosquito_sub window from the service?
If not it is likely that you need to configure you system to allow your user access to /dev/hidraw0

dricote, 2020-04-27:
I have decided to run the daemon as root and with this it already works

`mv /etc/systemd/user/mpp-solar.service /etc/systemd/system/mpp-solar.service`

jblance, 2020-04-27:
ok, thats another way to do it
Sounds like this issue is now resolved

closed 2020-04-27 by jblance