jblance/mpp-solar #18: new project

volkerjaenisch, 2020-05-03:
@all

I suggest that we open a new GH repo for the task.

e.g. 
https://github.com/jblance/solar_power

jblance, 2020-05-03:
happy to create a repo
have we determined the purpose - is it 'framework to abstract complex internal states and provide a simple API that integrates with MQTT / HASS etc'?

volkerjaenisch, 2020-05-03:
Yep that covers it. 
Now lets fish for a good name.
It should contain a gist of
"solar" "management" "current" 

To my astonishment "solar" is free on GH as well as solar-mangement

I opt for short words so let it be "solar"

volkerjaenisch, 2020-05-03:
Probably we will need several repos. One for each framework.

But let us start with a python package name that is worth to remember. What about

solarman

jblance, 2020-05-04:
created repo 'solarman' - we can move discussions there

closed 2020-05-04 by jblance