jblance/mpp-solar #19: mppsolar Hybrid V2

preussal, 2020-05-17:
Hi,
Hi,
I tried to use your software with an MPP Solar Hybrid V2 inverter.
But I don't get any data.

mpp-solar --device /dev/hidraw0 -D

> INFO:MPP-Solar:command QID
> INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
> DEBUG:MPP-Solar:Device matches hidraw regex
> INFO:MPP-Solar:Loading commands for inverter model: standard
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/f.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command F supported all inverters
> INFO:MPP-Solar:... command F loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for F
> DEBUG:MPP-Solar:Calculating CRC for b'F'
> DEBUG:MPP-Solar:Generated CRC 29 2 2902
> DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MCHGC supported all inverters
> INFO:MPP-Solar:... command MCHGC loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for MCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
> DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
> DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mnchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MNCHGC supported all inverters
> INFO:MPP-Solar:... command MNCHGC loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for MNCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
> DEBUG:MPP-Solar:Generated CRC 57 23 5723
> DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/muchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MUCHGC supported all inverters
> INFO:MPP-Solar:... command MUCHGC loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for MUCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
> DEBUG:MPP-Solar:Generated CRC bf 86 bf86
> DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbcv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBCV supported all inverters
> INFO:MPP-Solar:... command PBCV loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for PBCV
> DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
> DEBUG:MPP-Solar:Generated CRC 64 5a 645a
> DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbdv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBDV supported all inverters
> INFO:MPP-Solar:... command PBDV loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for PBDV
> DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
> DEBUG:MPP-Solar:Generated CRC fd cd fdcd
> DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbft.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBFT supported all inverters
> INFO:MPP-Solar:... command PBFT loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for PBFT
> DEBUG:MPP-Solar:Calculating CRC for b'PBFT'
> DEBUG:MPP-Solar:Generated CRC bb ed bbed
> DEBUG:MPP-Solar:Full byte command: b'PBFT\xbb\xed\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbt.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBT supported all inverters
[... 530 lines cut ...]
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command QVFW2 supported all inverters
> INFO:MPP-Solar:... command QVFW2 loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for QVFW2
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
> DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
> DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
> DEBUG:MPP-Solar:Searching for cmd 'QPIGS'
> INFO:MPP-Solar:DIRECT USB connection: executing QPIGS
> General Status Parameters inquiry
>  -- queries the value of various metrics from the Inverter
> 
> DEBUG:MPP-Solar:length of to_send: 8
> DEBUG:MPP-Solar:1 chunk send
> DEBUG:MPP-Solar:usb byte_response was: b'^0\x1b\xe3\r'
> ================ Status ==================
> Parameter                       Value           Unit
> ac_input_voltage                0               V
> 

In Solar Power I turned on the debug log and attached it to you.
It would be really great if this would also work with the Hybrid V2.
Unfortunately there is no ARM version of the SolarPower software.

[2020-05-17.log](https://github.com/jblance/mpp-solar/files/4639628/2020-05-17.log)

jblance, 2020-05-17:
If you have connected via USB directly you might need to run as root, i.e. ```sudo mpp-solar --device /dev/hidraw0 -D -c QPIRI```

The log shows the commands without 'Q' ie 'PIRI' etc but that may just be the way the logging is working

preussal, 2020-05-18:
i try it and same resault

dmesg

> [Mon May 18 07:36:01 2020] usb 1-1.1.3: new low-speed USB device number 9 using dwc_otg
> [Mon May 18 07:36:01 2020] usb 1-1.1.3: New USB device found, idVendor=0665, idProduct=5161, bcdDevice= 0.02
> [Mon May 18 07:36:01 2020] usb 1-1.1.3: New USB device strings: Mfr=3, Product=1, SerialNumber=0
> [Mon May 18 07:36:01 2020] hid-generic 0003:0665:5161.0004: hiddev96,hidraw0: USB HID v1.11 Device [HID 0665:5161] on usb-3f980000.usb-1.1.3/input0

sudo mpp-solar --device /dev/hidraw0 -D -c QPIRI

> INFO:MPP-Solar:command QPIRI
> INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
> DEBUG:MPP-Solar:Device matches hidraw regex
> INFO:MPP-Solar:Loading commands for inverter model: standard
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/f.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command F supported all inverters
> INFO:MPP-Solar:... command F loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for F
> DEBUG:MPP-Solar:Calculating CRC for b'F'
> DEBUG:MPP-Solar:Generated CRC 29 2 2902
> DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MCHGC supported all inverters
> INFO:MPP-Solar:... command MCHGC loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for MCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
> DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
> DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mnchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MNCHGC supported all inverters
> INFO:MPP-Solar:... command MNCHGC loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for MNCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
> DEBUG:MPP-Solar:Generated CRC 57 23 5723
> DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/muchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MUCHGC supported all inverters
> INFO:MPP-Solar:... command MUCHGC loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for MUCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
> DEBUG:MPP-Solar:Generated CRC bf 86 bf86
> DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbcv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBCV supported all inverters
> INFO:MPP-Solar:... command PBCV loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for PBCV
> DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
> DEBUG:MPP-Solar:Generated CRC 64 5a 645a
> DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbdv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBDV supported all inverters
> INFO:MPP-Solar:... command PBDV loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for PBDV
> DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
> DEBUG:MPP-Solar:Generated CRC fd cd fdcd
> DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbft.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBFT supported all inverters
> INFO:MPP-Solar:... command PBFT loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for PBFT
> DEBUG:MPP-Solar:Calculating CRC for b'PBFT'
> DEBUG:MPP-Solar:Generated CRC bb ed bbed
[... 219 lines cut ...]
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/qvfw.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command QVFW supported all inverters
> INFO:MPP-Solar:... command QVFW loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for QVFW
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW'
> DEBUG:MPP-Solar:Generated CRC 62 99 6299
> DEBUG:MPP-Solar:Full byte command: b'QVFWb\x99\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/qvfw2.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command QVFW2 supported all inverters
> INFO:MPP-Solar:... command QVFW2 loaded for inverter model: standard
> DEBUG:MPP-Solar:Generate full byte command for QVFW2
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
> DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
> DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
> DEBUG:MPP-Solar:Searching for cmd 'QPIRI'
> INFO:MPP-Solar:DIRECT USB connection: executing QPIRI
> Current Settings inquiry
>  -- queries the current settings from the Inverter
> 
> DEBUG:MPP-Solar:length of to_send: 8
> DEBUG:MPP-Solar:1 chunk send
> DEBUG:MPP-Solar:usb byte_response was: b'^0\x1b\xe3\r'
> ac_input_voltage                0               V

DMIINC, 2020-05-18:
I am starting to think that grid tie inverters might have different protocol.

jblance, 2020-05-18:
Yeah looks like different commands - though the CRC looks the same
The above response ```b'^0\x1b\xe3\r'``` 0x1b 0xe3 is the 'correct' crc for ```^0```
But the other inverters response starts with a ```(``` and would likely be NAK for an unknown command
Actually looking at the log shows the same response to a QPI

? What is the model number of your inverter?
? can you update to the latest version from github and try
```sudo mpp-solar --device /dev/hidraw0 -D -c ID -M V2 -R```

preussal, 2020-05-19:
Hi,
The Model Name is Hybrid V2 5048
[image]

I also think the protocol is different. here I found a forum post that writes something about it.
https://forums.aeva.asn.au/viewtopic.php?p=76852#p76852

sudo mpp-solar --device /dev/hidraw0 -D -c ID -M V2 -R

> INFO:MPP-Solar:command ID
> INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
> DEBUG:MPP-Solar:Device matches hidraw regex
> INFO:MPP-Solar:Loading commands for inverter model: V2
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/f.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command F supported all inverters
> INFO:MPP-Solar:... command F loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for F
> DEBUG:MPP-Solar:Calculating CRC for b'F'
> DEBUG:MPP-Solar:Generated CRC 29 2 2902
> DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/id.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command ID supported by model V2
> INFO:MPP-Solar:... command ID loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for ID
> DEBUG:MPP-Solar:Calculating CRC for b'ID'
> DEBUG:MPP-Solar:Generated CRC bf 14 bf14
> DEBUG:MPP-Solar:Full byte command: b'ID\xbf\x14\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MCHGC supported all inverters
> INFO:MPP-Solar:... command MCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
> DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
> DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mnchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MNCHGC supported all inverters
> INFO:MPP-Solar:... command MNCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MNCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
> DEBUG:MPP-Solar:Generated CRC 57 23 5723
> DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/muchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MUCHGC supported all inverters
> INFO:MPP-Solar:... command MUCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MUCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
> DEBUG:MPP-Solar:Generated CRC bf 86 bf86
> DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbcv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBCV supported all inverters
> INFO:MPP-Solar:... command PBCV loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBCV
> DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
> DEBUG:MPP-Solar:Generated CRC 64 5a 645a
> DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbdv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBDV supported all inverters
> INFO:MPP-Solar:... command PBDV loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBDV
> DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
> DEBUG:MPP-Solar:Generated CRC fd cd fdcd
> DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbft.json
[... 232 lines cut ...]
> DEBUG:MPP-Solar:Full byte command: b'QVFWb\x99\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/qvfw2.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command QVFW2 supported all inverters
> INFO:MPP-Solar:... command QVFW2 loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for QVFW2
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
> DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
> DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
> DEBUG:MPP-Solar:Searching for cmd 'ID'
> INFO:MPP-Solar:DIRECT USB connection: executing ID
> Device Serial Number inquiry
>  -- queries the device serial number
> 
> DEBUG:MPP-Solar:length of to_send: 5
> DEBUG:MPP-Solar:1 chunk send
> DEBUG:MPP-Solar:usb byte_response was: b'^0\x1b\xe3\r'
> 0

I currently use the one USB gateway software from www.net-usb.com
USBIPdoes not really work with.
But I'm also not willing to pay $ 160 for this USB gateway software.
Especially since I would like the data in my InfluxDB.

thank you for your help

jblance, 2020-05-19:
Most (all) the people that I have seen get this work are using a raspberry pi connected directly to the inverter. I'd suggest connecting directly first (either direct USB or a USB-to-serial adapter) before moving to a more complicated connection approach.

That said it looks like the inverter is responding correctly (as per the log and the link you posted)
I'm waiting to get my account approved so i can download the linked protocol document

preussal, 2020-05-19:
the inverter is connected to the Rasperry Pi via USB.
I only used the Net-USB program so that I can share the USB port from the Rasperie Pie to my laptop. And this is where the SolarPower app runs.
Unfortunately, the SolarPower app is not available for ARM only for 32 and 64 bit Linux.

jblance, 2020-05-19:
OIC - sorry my mistake
I've made a few rough tweaks to the code to see if it will work without a CRC
Can you update to the latest from github and try 
```sudo mpp-solar -d /dev/hidraw0 -D -c ID -M V2 -R``` 
again please

preussal, 2020-05-19:
Sure

sudo mpp-solar -d /dev/hidraw0 -D -c ID -M V2 -R

> INFO:MPP-Solar:command ID
> INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
> DEBUG:MPP-Solar:Device matches hidraw regex
> INFO:MPP-Solar:Loading commands for inverter model: V2
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/f.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command F supported all inverters
> INFO:MPP-Solar:... command F loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for F
> DEBUG:MPP-Solar:Calculating CRC for b'F'
> DEBUG:MPP-Solar:Generated CRC 29 2 2902
> DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/id.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command ID supported by model V2
> INFO:MPP-Solar:... command ID loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for ID
> DEBUG:MPP-Solar:Full byte command: b'ID\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MCHGC supported all inverters
> INFO:MPP-Solar:... command MCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
> DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
> DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mnchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MNCHGC supported all inverters
> INFO:MPP-Solar:... command MNCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MNCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
> DEBUG:MPP-Solar:Generated CRC 57 23 5723
> DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/muchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MUCHGC supported all inverters
> INFO:MPP-Solar:... command MUCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MUCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
> DEBUG:MPP-Solar:Generated CRC bf 86 bf86
> DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbcv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBCV supported all inverters
> INFO:MPP-Solar:... command PBCV loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBCV
> DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
> DEBUG:MPP-Solar:Generated CRC 64 5a 645a
> DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbdv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBDV supported all inverters
> INFO:MPP-Solar:... command PBDV loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBDV
> DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
> DEBUG:MPP-Solar:Generated CRC fd cd fdcd
> DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbft.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBFT supported all inverters
> INFO:MPP-Solar:... command PBFT loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBFT
> DEBUG:MPP-Solar:Calculating CRC for b'PBFT'
> DEBUG:MPP-Solar:Generated CRC bb ed bbed
> DEBUG:MPP-Solar:Full byte command: b'PBFT\xbb\xed\r'
[... 218 lines cut ...]
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/qvfw.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command QVFW supported all inverters
> INFO:MPP-Solar:... command QVFW loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for QVFW
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW'
> DEBUG:MPP-Solar:Generated CRC 62 99 6299
> DEBUG:MPP-Solar:Full byte command: b'QVFWb\x99\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/qvfw2.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command QVFW2 supported all inverters
> INFO:MPP-Solar:... command QVFW2 loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for QVFW2
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
> DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
> DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
> DEBUG:MPP-Solar:Searching for cmd 'ID'
> INFO:MPP-Solar:DIRECT USB connection: executing ID
> Device Serial Number inquiry
>  -- queries the device serial number
> 
> DEBUG:MPP-Solar:length of to_send: 3
> DEBUG:MPP-Solar:1 chunk send
> DEBUG:MPP-Solar:usb byte_response was: b'^0\x1b\xe3\r'
> 0

jblance, 2020-05-19:
ok, got a manual that might work (is apparently for P18 version of the protocol)

One more attempt, from the latest git version try:
```sudo mpp-solar -d /dev/hidraw0 -D -c ID -M V2 -R```

preussal, 2020-05-19:
sudo mpp-solar -d /dev/hidraw0 -D -c ^P003ID -M V2 -R

> INFO:MPP-Solar:command ^P003ID
> INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
> DEBUG:MPP-Solar:Device matches hidraw regex
> INFO:MPP-Solar:Loading commands for inverter model: V2
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/f.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command F supported all inverters
> INFO:MPP-Solar:... command F loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for F
> DEBUG:MPP-Solar:Calculating CRC for b'F'
> DEBUG:MPP-Solar:Generated CRC 29 2 2902
> DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/id.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command ^P003ID supported by model V2
> INFO:MPP-Solar:... command ^P003ID loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for ^P003ID
> DEBUG:MPP-Solar:Full byte command: b'^P003ID\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MCHGC supported all inverters
> INFO:MPP-Solar:... command MCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
> DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
> DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mnchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MNCHGC supported all inverters
> INFO:MPP-Solar:... command MNCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MNCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
> DEBUG:MPP-Solar:Generated CRC 57 23 5723
> DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/muchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MUCHGC supported all inverters
> INFO:MPP-Solar:... command MUCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MUCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
> DEBUG:MPP-Solar:Generated CRC bf 86 bf86
> DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbcv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBCV supported all inverters
> INFO:MPP-Solar:... command PBCV loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBCV
> DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
> DEBUG:MPP-Solar:Generated CRC 64 5a 645a
> DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbdv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBDV supported all inverters
> INFO:MPP-Solar:... command PBDV loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBDV
> DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
> DEBUG:MPP-Solar:Generated CRC fd cd fdcd
> DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbft.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBFT supported all inverters
> INFO:MPP-Solar:... command PBFT loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBFT
> DEBUG:MPP-Solar:Calculating CRC for b'PBFT'
> DEBUG:MPP-Solar:Generated CRC bb ed bbed
> DEBUG:MPP-Solar:Full byte command: b'PBFT\xbb\xed\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbt.json
> DEBUG:MPP-Solar:-----No supports
[... 837 lines cut ...]
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/qvfw.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command QVFW supported all inverters
> INFO:MPP-Solar:... command QVFW loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for QVFW
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW'
> DEBUG:MPP-Solar:Generated CRC 62 99 6299
> DEBUG:MPP-Solar:Full byte command: b'QVFWb\x99\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/qvfw2.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command QVFW2 supported all inverters
> INFO:MPP-Solar:... command QVFW2 loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for QVFW2
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
> DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
> DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
> DEBUG:MPP-Solar:Searching for cmd '^P003ID'
> INFO:MPP-Solar:DIRECT USB connection: executing ^P003ID
> Device Serial Number inquiry
>  -- queries the device serial number
> 
> DEBUG:MPP-Solar:length of to_send: 8
> DEBUG:MPP-Solar:1 chunk send
> DEBUG:MPP-Solar:usb byte_response was: b'^0\x1b\xe3\r'
> 0

jblance, 2020-05-19:
OK, so have access to the SA forum with another protocol document  (turns out the one I found was PI17, not 18)
So have updated the test ID command for PI18
Can you update from github and try
```sudo mpp-solar -d /dev/hidraw0 -D -c ^P005ID -M V2 -R```

preussal, 2020-05-19:
sudo mpp-solar -d /dev/hidraw0 -D -c ^P005ID -M V2 -R

> INFO:MPP-Solar:command ^P005ID
> INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
> DEBUG:MPP-Solar:Device matches hidraw regex
> INFO:MPP-Solar:Loading commands for inverter model: V2
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/f.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command F supported all inverters
> INFO:MPP-Solar:... command F loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for F
> DEBUG:MPP-Solar:Calculating CRC for b'F'
> DEBUG:MPP-Solar:Generated CRC 29 2 2902
> DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/id.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command ^P005ID supported by model V2
> INFO:MPP-Solar:... command ^P005ID loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for ^P005ID
> DEBUG:MPP-Solar:Calculating CRC for b'^P005ID'
> DEBUG:MPP-Solar:Generated CRC 19 cd 19cd
> DEBUG:MPP-Solar:Full byte command: b'^P005ID\x19\xcd\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MCHGC supported all inverters
> INFO:MPP-Solar:... command MCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
> DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
> DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/mnchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MNCHGC supported all inverters
> INFO:MPP-Solar:... command MNCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MNCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
> DEBUG:MPP-Solar:Generated CRC 57 23 5723
> DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/muchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MUCHGC supported all inverters
> INFO:MPP-Solar:... command MUCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MUCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
> DEBUG:MPP-Solar:Generated CRC bf 86 bf86
> DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbcv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBCV supported all inverters
> INFO:MPP-Solar:... command PBCV loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBCV
> DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
> DEBUG:MPP-Solar:Generated CRC 64 5a 645a
> DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbdv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBDV supported all inverters
> INFO:MPP-Solar:... command PBDV loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBDV
> DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
> DEBUG:MPP-Solar:Generated CRC fd cd fdcd
> DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/pbft.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBFT supported all inverters
> INFO:MPP-Solar:... command PBFT loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBFT
> DEBUG:MPP-Solar:Calculating CRC for b'PBFT'
> DEBUG:MPP-Solar:Generated CRC bb ed bbed
> DEBUG:MPP-Solar:Full byte command: b'PBFT\xbb\xed\r'
[... 222 lines cut ...]
> DEBUG:MPP-Solar:Generate full byte command for QVFW
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW'
> DEBUG:MPP-Solar:Generated CRC 62 99 6299
> DEBUG:MPP-Solar:Full byte command: b'QVFWb\x99\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.3.1-py3.7.egg/mppsolar/commands/qvfw2.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command QVFW2 supported all inverters
> INFO:MPP-Solar:... command QVFW2 loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for QVFW2
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
> DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
> DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
> DEBUG:MPP-Solar:Searching for cmd '^P005ID'
> INFO:MPP-Solar:DIRECT USB connection: executing ^P005ID
> Device Serial Number inquiry
>  -- queries the device serial number
> 
> DEBUG:MPP-Solar:length of to_send: 10
> DEBUG:MPP-Solar:2 chunk send
> DEBUG:MPP-Solar:usb byte_response was: b'^0\x1b\xe3\r'
> 0

I think you're trying to get her serial number. I don't get this via SolarPower either.
The marked field is to be entered.
[image]

preussal, 2020-05-19:
[image]

jblance, 2020-05-20:
Its just that the log you sent has a result for this command
```[2020-05-17 11:16:35][ID]1496132004100199000000```

have changed the command to the protocol query , try (on latest github version)
```sudo mpp-solar -d /dev/hidraw0 -D -c ^P005PI -M V2 -R```

preussal, 2020-05-21:
sudo mpp-solar -d /dev/hidraw0 -D -c ^P005PI -M V2 -R

> INFO:MPP-Solar:command ^P005PI
> INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
> DEBUG:MPP-Solar:Device matches hidraw regex
> INFO:MPP-Solar:Loading commands for inverter model: V2
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/f.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command F supported all inverters
> INFO:MPP-Solar:... command F loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for F
> DEBUG:MPP-Solar:Calculating CRC for b'F'
> DEBUG:MPP-Solar:Generated CRC 29 2 2902
> DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/id.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command ^P005PI supported by model V2
> INFO:MPP-Solar:... command ^P005PI loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for ^P005PI
> DEBUG:MPP-Solar:Calculating CRC for b'^P005PI'
> DEBUG:MPP-Solar:Generated CRC 71 8b 718b
> DEBUG:MPP-Solar:Full byte command: b'^P005PIq\x8b\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/mchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MCHGC supported all inverters
> INFO:MPP-Solar:... command MCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
> DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
> DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/mnchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MNCHGC supported all inverters
> INFO:MPP-Solar:... command MNCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MNCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
> DEBUG:MPP-Solar:Generated CRC 57 23 5723
> DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/muchgc.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command MUCHGC supported all inverters
> INFO:MPP-Solar:... command MUCHGC loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for MUCHGC
> DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
> DEBUG:MPP-Solar:Generated CRC bf 86 bf86
> DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbcv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBCV supported all inverters
> INFO:MPP-Solar:... command PBCV loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBCV
> DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
> DEBUG:MPP-Solar:Generated CRC 64 5a 645a
> DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbdv.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBDV supported all inverters
> INFO:MPP-Solar:... command PBDV loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBDV
> DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
> DEBUG:MPP-Solar:Generated CRC fd cd fdcd
> DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbft.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command PBFT supported all inverters
> INFO:MPP-Solar:... command PBFT loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for PBFT
> DEBUG:MPP-Solar:Calculating CRC for b'PBFT'
> DEBUG:MPP-Solar:Generated CRC bb ed bbed
> DEBUG:MPP-Solar:Full byte command: b'PBFT\xbb\xed\r'
[... 218 lines cut ...]
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/qvfw.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command QVFW supported all inverters
> INFO:MPP-Solar:... command QVFW loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for QVFW
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW'
> DEBUG:MPP-Solar:Generated CRC 62 99 6299
> DEBUG:MPP-Solar:Full byte command: b'QVFWb\x99\r'
> DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/qvfw2.json
> DEBUG:MPP-Solar:-----No supports
> DEBUG:MPP-Solar:Command QVFW2 supported all inverters
> INFO:MPP-Solar:... command QVFW2 loaded for inverter model: V2
> DEBUG:MPP-Solar:Generate full byte command for QVFW2
> DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
> DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
> DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
> DEBUG:MPP-Solar:Searching for cmd '^P005PI'
> INFO:MPP-Solar:DIRECT USB connection: executing ^P005PI
> TESTING Device Protocol Version inquiry
>  -- queries the device protocol version
> 
> DEBUG:MPP-Solar:length of to_send: 10
> DEBUG:MPP-Solar:2 chunk send
> DEBUG:MPP-Solar:usb byte_response was: b'^D00518;\x03\r'
> D00518

jblance, 2020-05-21:
ok so that worked
so it is possible to add the commands for your inverter (though they dont follow the same response etc as the other supported inverters)
What information is it that you are after?
What are you trying to do with it after?

(Ideally to support this different protocol I'd change the code to support the quite different format)

preussal, 2020-05-22:
So before I had grid tie inverters on a separate electricity meter.
Two ModBus electricity meters with which I got the house electricity and the solar electricity.

I would like to have this data from the hybrid inverter, that would be great

PV1 input voltage V
Pv1 input power W
Grid voltage V
Grid frequency Hz
Battery voltage V
Battery Capacity%
Battery charging current A
Battery discharge current A
Grid output voltage V
Grid output frequency Hz
AC output apperent power VA
AC output active power W
Output load percent%

The data then ends up in the InfluxDB and I then evaluate it via grafana.

my built power meter
https://www.secretisland.de/raspberry-pi-als-powermeter/
Instructions are in German, of which I now have two in use, and work quite well

jblance, 2020-05-22:
That is doable reasonably quickly
If you have a look in the protocol PDF (Infini-Solar V protocol 20170926-PI18.pdf in the docs folder) and find the query commands you want implemented, I'll try to find time to get them created and a way to get the answer in a better format

preussal, 2020-05-22:
I think these are the queries that bring the most.

^P005ET<CRC><cr>: Query total generated energy

I don't understand the difference. But one of those will be the live dates. I think the P007PIRI
^P007PIRI<CRC><cr>: Query rated information
^P005GS<CRC><cr>: Query general status

I get the information in which mode the inverter is currently operating.
^P006MOD<CRC><cr>: Query working mode

it would also be enough for me to get the data as a simple JSON.
Or just call the query PIRI and then get the response. 
e.g. "BBB, CCCC, DDD, EEE, FFFF, GGGG, HHH, III, JJJ, KKK, LLL, MMM, N, OO, PPP, Q, R, S, T, U, V, W, Z, a"
Then I could write that into the InfluxDB.

jblance, 2020-05-22:
OK I have added some PI18 commands and processing (hopefully as I cant test)
The PIRI command provides the settings (or ratings) for the device (so not actual data), so havent done that one yet
I have implemented PI, ET, GS, and MOD commands
Usage is now:
```sudo mpp-solar -d /dev/hidraw0 -c GS -M PI18```
change GS to ET or MOD as needed

Add ```-D``` for lots of extra info if it doesnt work
Once we verify / correct you should be able to use the mpp-service to post results to MQTT in a way that telegraf will feed them to influx so that grafana can read them (at least thats what the few that i know of that are  using grafana are doing)
There is a couple of readmes that cover some of this

preussal, 2020-05-22:
cool, but I think a few values ​​are wrong

sudo mpp-solar -d /dev/hidraw0 -c GS -M PI18

> ac_output_active_power          011             W
> ac_output_apparent_power        0523            VA
> ac_output_frequency             0598            0.1Hz
> ac_output_voltage               500             0.1V
> battery_capacity                037             %
> battery_charging_current        000             A
> battery_discharge_current       000             A
> battery_power_direction         donothing
> battery_voltage                 000             0.1V
> battery_voltage_from_scc        000             0.1V
> battery_voltage_from_scc2       000             0.1V
> dc/ac_power_direction           AC-DC
> grid_frequency                  2217            0.1Hz
> grid_voltage                    500             0.1V
> inverter_heat_sink_temperature  000             oC
> line_power_direction            donothing
> load_connection                 disconnect
> mppt1_charger_status            abnormal
> mppt1_charger_temperature       000             oC
> mppt2_charger_status            normal but not charged
> mppt2_charger_temperature       0001            oC
> output_load_percent             007             %
> pv1_input_power                 0000            W
> pv1_input_voltage               0000            0.1V
> pv2_input_power                 0000            W
> pv2_input_voltage               0               0.1V
> setting_value_configuration_state       Nothing changed
> total_generated_energy          2217            KWh

> ac_output_active_power          011             W --> output_load_percent
> ac_output_apparent_power        0523            VA  --> ac_output_active_power 
> ac_output_frequency             0598            0.1Hz  --> ac_output_apparent_power
> ac_output_voltage               500             0.1V -->  ac_output_frequency 
> battery_capacity                037             %  --> no Idear
> battery_charging_current        000             A   -> ( batteries are coming next week )
> battery_discharge_current       000             A   -> ( batteries are coming next week )
> battery_power_direction         donothing    -> ( batteries are coming next week )
> battery_voltage                 000             0.1V   -> ( batteries are coming next week )
> battery_voltage_from_scc        000             0.1V   -> ( batteries are coming next week )
> battery_voltage_from_scc2       000             0.1V   -> ( batteries are coming next week )
> dc/ac_power_direction           AC-DC
> grid_frequency                  2217            0.1Hz   --> grid_voltage 
> grid_voltage                    500             0.1V  --> grid_frequency
> inverter_heat_sink_temperature  000             oC
> line_power_direction            donothing
> load_connection                 disconnect    --> no i have load on it
> mppt1_charger_status            abnormal   --> can be is night
> mppt1_charger_temperature       000             oC 
> mppt2_charger_status            normal but not charged
> mppt2_charger_temperature       0001            oC
> output_load_percent             007             %  --> battery_voltage 
> pv1_input_power                 0000            W  --> can be is night
> pv1_input_voltage               0000            0.1V --> can be is night
> pv2_input_power                 0000            W --> have only one mppt
> pv2_input_voltage               0               0.1V --> have only one mppt
> setting_value_configuration_state       Nothing changed
> total_generated_energy          2217            KWh  --> ac_output_voltage

preussal, 2020-05-22:
i found the Problem, this line  ["int", "Total generated energy", "KWh"],

diff --git a/mppsolar/commands/_pi18_gs.json b/mppsolar/commands/_pi18_gs.json
index 84eec92..42c6703 100644
--- a/mppsolar/commands/_pi18_gs.json
+++ b/mppsolar/commands/_pi18_gs.json
@@ -6,7 +6,6 @@
     "type": "QUERY",
     "supports": ["PI18"],
     "response": [
-            ["int", "Total generated energy", "KWh"],
             ["int", "Grid voltage", "0.1V"],
             ["int", "Grid frequency", "0.1Hz"],
             ["int", "AC output voltage", "0.1V"],

```
ac_output_active_power          0708            W
ac_output_apparent_power        0795            VA
ac_output_frequency             499             0.1Hz
ac_output_voltage               2209            0.1V
battery_capacity                000             %
battery_charging_current        000             A
battery_discharge_current       000             A
battery_power_direction         donothing
battery_voltage                 007             0.1V
battery_voltage_from_scc        000             0.1V
battery_voltage_from_scc2       000             0.1V
dc/ac_power_direction           donothing
grid_frequency                  499             0.1Hz
grid_voltage                    2209            0.1V
inverter_heat_sink_temperature  037             oC
line_power_direction            input
load_connection                 connect
local_parallel_id               0
mppt1_charger_status            abnormal
mppt1_charger_temperature       000             oC
mppt2_charger_status            abnormal
mppt2_charger_temperature       000             oC
output_load_percent             015             %
pv1_input_power                 0001            W
pv1_input_voltage               0000            0.1V
pv2_input_power                 0000            W
pv2_input_voltage               0000            0.1V
setting_value_configuration_state       Nothing changed
```
[image]

preussal, 2020-05-22:
and here the other commands

sudo mpp-solar -d /dev/hidraw0 -c ET -M PI18
```
total_generated_energy          00023870        KWh
```
sudo mpp-solar -d /dev/hidraw0 -c GS -M PI18
```
ac_output_active_power          1530            W
ac_output_apparent_power        1583            VA
ac_output_frequency             500             0.1Hz
ac_output_voltage               2199            0.1V
battery_capacity                000             %
battery_charging_current        000             A
battery_discharge_current       000             A
battery_power_direction         donothing
battery_voltage                 007             0.1V
battery_voltage_from_scc        000             0.1V
battery_voltage_from_scc2       000             0.1V
dc/ac_power_direction           donothing
grid_frequency                  500             0.1Hz
grid_voltage                    2199            0.1V
inverter_heat_sink_temperature  037             oC
line_power_direction            input
load_connection                 connect
local_parallel_id               0
mppt1_charger_status            abnormal
mppt1_charger_temperature       000             oC
mppt2_charger_status            abnormal
mppt2_charger_temperature       000             oC
output_load_percent             031             %
pv1_input_power                 0001            W
pv1_input_voltage               0000            0.1V
pv2_input_power                 0000            W
pv2_input_voltage               0000            0.1V
setting_value_configuration_state       Nothing changed
```
sudo mpp-solar -d /dev/hidraw0 -c MOD -M PI18
```
working_mode                    Bypass mode
```
sudo mpp-solar -d /dev/hidraw0 -c PI -M PI18
```
protocol_version                18
```

jblance, 2020-05-23:
oh, oops - seems like i made a copy paste error.

I have fixed that and added a result type to deal with the 0.1V type of results. So now the results should be correct without having to divide some of them by 10

preussal, 2020-05-23:
looks great

sudo mpp-solar -d /dev/hidraw0 -c GS -M PI18
```
ac_output_active_power          1754            W
ac_output_apparent_power        1801            VA
ac_output_frequency             50.0            Hz
ac_output_voltage               214.5           V
battery_capacity                000             %
battery_charging_current        000             A
battery_discharge_current       000             A
battery_power_direction         donothing
battery_voltage                 0.7             V
battery_voltage_from_scc        0.0             V
battery_voltage_from_scc2       0.0             V
dc/ac_power_direction           donothing
grid_frequency                  50.0            Hz
grid_voltage                    214.5           V
inverter_heat_sink_temperature  037             oC
line_power_direction            input
load_connection                 connect
local_parallel_id               0
mppt1_charger_status            abnormal
mppt1_charger_temperature       000             oC
mppt2_charger_status            abnormal
mppt2_charger_temperature       000             oC
output_load_percent             036             %
pv1_input_power                 0001            W
pv1_input_voltage               0.0             V
pv2_input_power                 0000            W
pv2_input_voltage               0.0             V
setting_value_configuration_state       Nothing changed
```

preussal, 2020-05-24:
no sun today

```
ac_output_active_power          0461            W
ac_output_apparent_power        0496            VA
ac_output_frequency             50.0            Hz
ac_output_voltage               225.9           V
battery_capacity                000             %
battery_charging_current        000             A
battery_discharge_current       000             A
battery_power_direction         donothing
battery_voltage                 0.7             V
battery_voltage_from_scc        0.0             V
battery_voltage_from_scc2       0.0             V
dc/ac_power_direction           DC-AC
grid_frequency                  50.0            Hz
grid_voltage                    225.9           V
inverter_heat_sink_temperature  039             oC
line_power_direction            input
load_connection                 connect
local_parallel_id               0
mppt1_charger_status            charging
mppt1_charger_temperature       000             oC
mppt2_charger_status            abnormal
mppt2_charger_temperature       000             oC
output_load_percent             009             %
pv1_input_power                 0441            W
pv1_input_voltage               203.3           V
pv2_input_power                 0000            W
pv2_input_voltage               0.0             V
setting_value_configuration_state       Nothing changed
```

preussal, 2020-05-24:
I tried to write this into the influx using the MQTT.
I got this issue.

 sudo mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i
```
Traceback (most recent call last):
  File "/usr/local/bin/mpp-info-pub", line 11, in <module>
    load_entry_point('mpp-solar==0.4.1', 'console_scripts', 'mpp-info-pub')()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mpp_info_pub.py", line 57, in main
    _data = mp.getInfluxLineProtocol(_command)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mpputils.py", line 46, in getInfluxLineProtocol
    return self.inverter.getInfluxLineProtocol(cmd)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppinverter.py", line 205, in getInfluxLineProtocol
    return result.getInfluxLineProtocol()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppcommand.py", line 355, in getInfluxLineProtocol
    msgs.append('setting={} nvalue={},unit="{}"'.format(key, float(result)/10, resp_format[2]))
ValueError: could not convert string to float: b'D1062232,499,2232,499,0971,0710,019,008,000,000,000,000,000,044,000,000,0520,0000,1941,0000,0,2,0,1,0,2,1,0'

```

preussal, 2020-05-24:
hi,
I wrote a little bash that writes the data into a JSON, I can then load the data into the Influx with telegraph.

```
#!/bin/bash

MPPSOLAR=`sudo mpp-solar -d /dev/hidraw0 -c GS -M PI18 -R |sed "s/'//g; s/\[//g; s/\]//g; s/,//g"`
echo ${MPPSOLAR} | awk '{printf \
"{ \"Grid voltage\":" ($1/10) \
", \"Grid frequency\":" ($2/10) \
", \"AC output voltage\":" ($3/10) \
", \"AC output frequency\":" ($4/10) \
", \"AC output apparent power\":" int($5) \
", \"AC output active power\":" int($6) \
", \"AC output current\":" int($5)/($3/10) \
", \"Output load percent\":" int($7) \
", \"Battery voltage\":" int($8) \
", \"Battery voltage from SCC\":" int($9) \
", \"Battery voltage from SCC2\":" int($10) \
", \"Battery discharge current\":" int($11) \
", \"Battery charging current\":" int($12) \
", \"Battery capacity percent\":" int($13) \
", \"Inverter heat sink temperature\":" int($14) \
", \"PV1 Input power\":" int($17) \
", \"PV1 Input voltage\":" $19/10 \
", \"PV1 Input current\":" int($17)/($19/10) \
}
{ printf ", \"MPPT1 charger status\":" }
{ if ( $22==0 ) printf "abnormal"; else if ( $22==1 ) printf "normal but not charged"; else if ( $22==2 ) printf "charging" }
{ printf ", \"Load connection\":" }
{ if ( $24==0 ) printf "disconnect"; else if ( $24==1 ) printf "connect" }
{ printf ", \"Battery power direction\":" }
{ if ( $25==0 ) printf "donothing"; else if ( $25==1 ) printf "charge"; else if ( $25==2 ) printf "discharge" }
{ printf ", \"DC/AC power direction\":" }
{ if ( $26==0 ) printf "donothing"; else if ( $26==1 ) printf "AC-DC"; else if ( $26==2 ) printf "DC-AC" }
{ printf ", \"Line power direction\":" }
{ if ( $27==0 ) printf "donothing"; else if ( $27==1 ) printf "input"; else if ( $27==2 ) printf "output" }
{ printf " }" }'
```
Output
`{ "Grid voltage":227.5, "Grid frequency":50, "AC output voltage":227.5, "AC output frequency":50, "AC output apparent power":568, "AC output active power":520, "AC output current":2.4967, "Output load percent":11, "Battery voltage":8, "Battery voltage from SCC":0, "Battery voltage from SCC2":0, "Battery discharge current":0, "Battery charging current":0, "Battery capacity percent":0, "Inverter heat sink temperature":42, "PV1 Input power":131, "PV1 Input voltage":195.2, "PV1 Input current":0.671107, "MPPT1 charger status":charging, "Load connection":connect, "Battery power direction":donothing, "DC/AC power direction":DC-AC, "Line power direction":input }`

jblance, 2020-05-24:
oops, missed a couple of places to change the split of the responses
i have updated the code, so hopefully you wont need to awk script
or if you want to keep using it, dont update....

preussal, 2020-05-25:
you are absolutely right. The awk script is crap :)

I already used Telegraf and therefore I would like to have the data in another DB.
That's why I added this.
Main [[outputs.influxdb]]
```
[[outputs.influxdb]]
.
.
.
  # Drop all measurements that start with "GSenergy"
  namedrop = ["GSenergy"]
```

/etc/telegraf/telegraf.d/mppsolar.conf
```
[[inputs.mqtt_consumer]]
  servers = ["tcp://127.0.0.1:1883"]
  topics = [
    "#",
  ]
  data_format = "influx"

[[outputs.influxdb]]
  urls = ["http://127.0.0.1:8086"]
  database = "db_meters"
  skip_database_creation = true
  username = "db_meters"
  password = ""
  # Only accept measurements that start with "GSenergy"
  namepass = ["GSenergy"]
```

This means that the data is not written to the Telegraf DB but only to the db_meters.

And here is the first Grafana design
[image]

preussal, 2020-05-25:
Since I read out the data from my ModBus Energy Meter in seconds. 
I created the cron for MPP like this.

```
* * * * *       root    /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i > /home/pi/cron.out 2>&1
* * * * *       root    sleep 10; /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i > /home/pi/cron.out 2>&1
* * * * *       root    sleep 15; /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i > /home/pi/cron.out 2>&1
* * * * *       root    sleep 20; /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i > /home/pi/cron.out 2>&1
* * * * *       root    sleep 25; /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i > /home/pi/cron.out 2>&1
* * * * *       root    sleep 30; /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i > /home/pi/cron.out 2>&1
* * * * *       root    sleep 35; /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i > /home/pi/cron.out 2>&1
* * * * *       root    sleep 40; /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i > /home/pi/cron.out 2>&1
* * * * *       root    sleep 45; /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i > /home/pi/cron.out 2>&1
* * * * *       root    sleep 50; /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i > /home/pi/cron.out 2>&1
* * * * *       root    sleep 55; /usr/local/bin/mpp-info-pub -q localhost -d /dev/hidraw0 -c GS -M PI18 -i > /home/pi/cron.out 2>&1
```

your script takes about 4.3 seconds to complete.
So I can read the inverter data every 5 seconds.

jblance, 2020-05-25:
That's what the daemon is intended for. It runs the commands continuously (or with a configurable pause) and send to mqtt broker.

But as always, whatever works is best :-)

jblance, 2020-05-28:
closing as seems resolved - reopen if this is not the case

Regards

closed 2020-05-28 by jblance