jblance/mpp-solar #20: this isn't an issue...

Tolmino, 2020-05-27:
Hi John Blance and thanks for your project.
i'm thinking about to write directly in InfluxDB via InfluxDBClient python module in daemon mode.
i gave a look here: https://influxdb-python.readthedocs.io/en/latest/examples.html
and it seems a python module very simple to use.

why you choose to use mosquitto and telegraf for read from mosquitto and write to InfluxDB ?

jblance, 2020-05-27:
Hi

There are always many ways of doing things :-)
I chose MQTT as it is a good way to add flexibility. It can go to influx via telegraf with no real changes. It can also be used to feed Home Assistant, it can also interface with Node-red for more manipulation.

Writing directly to Influx wouldnt be hard to add to the service - just define a new 'format' and implement the logic 

I recently added a different format, just to make Home Assistant integration a bit easier....

Tolmino, 2020-05-28:
yes, i'have seen new commits !!

now i'm using it on PIP-5048MK and works, giving last value as unknown_value_in_byte_response 010

sometime also i have an error:
mpp-solar -d /dev/hidraw0 -c QPIGS
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.4.1', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python3.5/dist-packages/mpp_solar-0.4.1-py3.5.egg/mppsolar/__init__.py", line 83, in main
    results = mp.getResponseDict(args.command)
  File "/usr/local/lib/python3.5/dist-packages/mpp_solar-0.4.1-py3.5.egg/mppsolar/mpputils.py", line 40, in getResponseDict
    return self.inverter.getResponseDict(cmd)
  File "/usr/local/lib/python3.5/dist-packages/mpp_solar-0.4.1-py3.5.egg/mppsolar/mppinverter.py", line 215, in getResponseDict
    result = self.execute(cmd)
  File "/usr/local/lib/python3.5/dist-packages/mpp_solar-0.4.1-py3.5.egg/mppsolar/mppinverter.py", line 351, in execute
    return self._doDirectUsbCommand(command)
  File "/usr/local/lib/python3.5/dist-packages/mpp_solar-0.4.1-py3.5.egg/mppsolar/mppinverter.py", line 335, in _doDirectUsbCommand
    command.setByteResponse(response_line)
  File "/usr/local/lib/python3.5/dist-packages/mpp_solar-0.4.1-py3.5.egg/mppsolar/mppcommand.py", line 139, in setByteResponse
    self.response_dict = self.getResponseDict()
  File "/usr/local/lib/python3.5/dist-packages/mpp_solar-0.4.1-py3.5.egg/mppsolar/mppcommand.py", line 434, in getResponseDict
    msgs[key] = [float(result), resp_format[2]]
ValueError: could not convert string to float: 'NAK'

it is running on a raspberry pi 3
maybe an error depending by python 3.5? do i need to upgrade python to 3.7/3.8 ?

jblance, 2020-05-28:
No that's a code issue - the inverter responded 'NAK' which is a negative acknowledgement
It could be for a variety of reasons, often you just need to try again
The code for checking invalid responses is currently disabled until i fix it (then you'd get a 'invalid response: NAK' message

jjsilva33, 2020-05-29:
Hello friend, First congratulations on your project. and second I am experiencing this same problem after you made these updates. last month I did some tests and it was working well in python 2. but when I installed recent error this error appeared when executing the command mpp-solar -s [image]

Tolmino, 2020-05-30:
i gave a look to my git logs.... and with 4may code that error wasn't appear so much

Tolmino, 2020-05-30:
diff mpp-solar-tolm/mppsolar/mppcommand.py mpp-solar/mppsolar/mppcommand.py 
433,437c433
<             msgs[key] = [result, resp_format[2]]
<             # Look for an error
<             if ( msgs[key] == ['NAK','V']):
<                 return msgs
<             elif (resp_format[0] == 'float'):
---
>             if (resp_format[0] == 'float'):

jblance, 2020-06-10:
> Hello friend, First congratulations on your project. and second I am experiencing this same problem after you made these updates. last month I did some tests and it was working well in python 2. but when I installed recent error this error appeared when executing the command mpp-solar -s [image]

Hi

You'd need to supply a -D or -I run so i can see which response is failing

It is true that I have updated to return numbers instead of strings and it looks like I have indicated something is an int instead of the float response you are getting

jblance, 2020-06-10:
> to removing annoing error i simply have added an error check to mppcommand.py
> maybe usefull manage error in **init.py** to restart query. if John Blance like, i can do it
> 
> `diff mpp-solar-tolm/mppsolar/mppcommand.py mpp-solar/mppsolar/mppcommand.py `
> `433,437c433`
> `< msgs[key] = [result, resp_format[2]]`
> `< # Look for an error`
> `< if ( msgs[key] == ['NAK','V']):`
> `< return msgs`
> `< elif (resp_format[0] == 'float'):`
> `---`
> `> if (resp_format[0] == 'float'):`

I need to re-enable the validResponse checking which will solve this - I'll do this next

Tolmino, 2020-06-10:
Great !! i have seen it now....

Tolmino, 2020-06-10:
tested:

now, when error appears i have that :

mpp-solar -d /dev/hidraw0 -c QPIGS
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.4.1', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python3.5/dist-packages/mpp_solar-0.4.1-py3.5.egg/mppsolar/__init__.py", line 83, in main
    results = mp.getResponseDict(args.command)
  File "/usr/local/lib/python3.5/dist-packages/mpp_solar-0.4.1-py3.5.egg/mppsolar/mpputils.py", line 40, in getResponseDict
    return self.inverter.getResponseDict(cmd)
  File "/usr/local/lib/python3.5/dist-packages/mpp_solar-0.4.1-py3.5.egg/mppsolar/mppinverter.py", line 219, in getResponseDict
    return result.getResponseDict()
  File "/usr/local/lib/python3.5/dist-packages/mpp_solar-0.4.1-py3.5.egg/mppsolar/mppcommand.py", line 414, in getResponseDict
    msgs['response'] = [self.getResponse().replace('\r', ''), '']
AttributeError: 'list' object has no attribute 'replace'

jblance, 2020-06-10:
I've commited a fix (hopefully)
However, the real problem is that you are getting a failure on the command you are running

jblance, 2020-06-10:
are you running the mpp-solar service as well?

Tolmino, 2020-06-10:
no, because i must reinstall os on that raspberry, and i'm searching time to do that ;)
as soon i'll do, i write you .

jjsilva33, 2020-06-11:
Hello I restored the raspbian and installed the mpp-solar again then follow the debug below:

 mpp-solar  -D -d /dev/ttyUSB0 -b 2400 -c QPIGS
INFO:MPP-Solar:command QPIGS
INFO:MPP-Solar:Serial device used: /dev/ttyUSB0, baud rate: 2400
INFO:MPP-Solar:Loading commands for inverter model: standard
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-p                                                                                                                                                             ackages/mpp_solar-0.4.2-py3.7.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-p                                                                                                                                                             ackages/mpp_solar-0.4.2-py3.7.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-p                                                                                                                                                             ackages/mpp_solar-0.4.2-py3.7.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-p                                                                                                                                                             ackages/mpp_solar-0.4.2-py3.7.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-p                                                                                                                                                             ackages/mpp_solar-0.4.2-py3.7.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-p                                                                                                                                                             ackages/mpp_solar-0.4.2-py3.7.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-p                                                                                                                                                             ackages/mpp_solar-0.4.2-py3.7.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-p                                                                                                                                                             ackages/mpp_solar-0.4.2-py3.7.egg/mppsolar/commands/muchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MUCHGC supported all inverters
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MUCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
DEBUG:MPP-Solar:Generated CRC bf 86 bf86
DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-p                                                                                                                                                             ackages/mpp_solar-0.4.2-py3.7.egg/mppsolar/commands/pbcv.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command PBCV supported all inverters
INFO:MPP-Solar:... command PBCV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBCV
DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
DEBUG:MPP-Solar:Generated CRC 64 5a 645a
DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-p                                                                                                                                                             ackages/mpp_solar-0.4.2-py3.7.egg/mppsolar/commands/pbdv.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command PBDV supported all inverters
INFO:MPP-Solar:... command PBDV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBDV
DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
DEBUG:MPP-Solar:Generated CRC fd cd fdcd
DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
[... cut ...]                           r/mppinverter.py", line 219, in getResponseDict
    return result.getResponseDict()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.2-py3.7.egg/mppsola                                                                                                                                                             r/mppcommand.py", line 438, in getResponseDict
    msgs[key] = [int(result), resp_format[2]]
ValueError: invalid literal for int() with base 10: '01.1'**

jblance, 2020-06-14:
what inverter are you using?
I think that the standard QPIGS definition is incorrect for the response you are getting (neither does the LV5048 QPIGS definition fit).
Also what does ```mpp-solar -c QPI``` give?

closed 2021-01-10 by jblance