jblance/mpp-solar #22: DEBUG:MPP-Solar:USB read error: [Errno 11] Resource temporarily unavailable

danieltroger, 2020-06-04:
Hello, I have a PIP-4024MSP from MPPSolar and have discovered your awesome utility. First and foremost thanks.

I have a raspberry pi zero connected to the inverter, and after booting the pi it works fine, however after ~20 minutes no command can be executed any longer because of this error:

```
root@zero:/home/pi# mpp-solar -d /dev/hidraw0 -s -D
INFO:MPP-Solar:command QID
INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
DEBUG:MPP-Solar:Device matches hidraw regex
INFO:MPP-Solar:Loading commands for inverter model: standard
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/muchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MUCHGC supported all inverters
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MUCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
DEBUG:MPP-Solar:Generated CRC bf 86 bf86
DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbcv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBCV supported all inverters
INFO:MPP-Solar:... command PBCV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBCV
DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
DEBUG:MPP-Solar:Generated CRC 64 5a 645a
DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbdv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBDV supported all inverters
INFO:MPP-Solar:... command PBDV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBDV
DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
DEBUG:MPP-Solar:Generated CRC fd cd fdcd
DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbft.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBFT supported all inverters
INFO:MPP-Solar:... command PBFT loaded for inverter model: standard
[... 345 lines cut ...]
DEBUG:MPP-Solar:usb byte_response was: b''
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.4.1', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/__init__.py", line 64, in main
    fullStatus = mp.getFullStatus()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mpputils.py", line 60, in getFullStatus
    data = self.getResponseDict("QPIGS")
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mpputils.py", line 40, in getResponseDict
    return self.inverter.getResponseDict(cmd)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppinverter.py", line 215, in getResponseDict
    result = self.execute(cmd)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppinverter.py", line 351, in execute
    return self._doDirectUsbCommand(command)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppinverter.py", line 335, in _doDirectUsbCommand
    command.setByteResponse(response_line)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppcommand.py", line 139, in setByteResponse
    self.response_dict = self.getResponseDict()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppcommand.py", line 434, in getResponseDict
    msgs[key] = [float(result), resp_format[2]]
ValueError: could not convert string to float: 
root@zero:/home/pi# 
```

I would appreciate a way to fix this without rebooting. Rebooting makes it work again for ~20 minutes but it feels stupid to reboot that often.

danieltroger, 2020-06-04:
I hoped using the daemon would maybe cirumvent it, it doesn't:

```
Jun  4 21:00:28 zero systemd[409]: mpp-solar.service: Scheduled restart job, restart counter is at 76.
Jun  4 21:00:28 zero systemd[409]: Stopped MPP Solar Service.
Jun  4 21:00:28 zero systemd[409]: Starting MPP Solar Service...
Jun  4 21:00:32 zero python3[1390]: MPP-Solar-Service: Initializing ...
Jun  4 21:00:32 zero python3[1390]: MPP-Solar-Service: Config file: /etc/mpp-solar/mpp-solar.conf
Jun  4 21:00:32 zero python3[1390]: MPP-Solar-Service: Config setting - pause: 0
Jun  4 21:00:32 zero python3[1390]: MPP-Solar-Service: Config setting - mqtt_broker: 192.168.178.100
Jun  4 21:00:32 zero python3[1390]: MPP-Solar-Service: Config setting - command sections found: 2
Jun  4 21:00:32 zero systemd[409]: Started MPP Solar Service.
Jun  4 21:00:32 zero python3[1390]: MPP-Solar-Service: item {'mp': <mppsolar.mpputils.mppUtils object at 0xb5e927b0>, 'command': 'QPIGS', 'format': 'influx2', 'tag': 'Inverter1'}
Jun  4 21:00:48 zero python3[1390]: Traceback (most recent call last):
Jun  4 21:00:48 zero python3[1390]:   File "/usr/local/bin/mpp-solar-service", line 11, in <module>
Jun  4 21:00:48 zero python3[1390]:     load_entry_point('mpp-solar==0.4.1', 'console_scripts', 'mpp-solar-service')()
Jun  4 21:00:48 zero python3[1390]:   File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mpp_solar_service.py", line 65, in main
Jun  4 21:00:48 zero python3[1390]:     _data = item['mp'].getInfluxLineProtocol2(item['command'])
Jun  4 21:00:48 zero python3[1390]:   File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mpputils.py", line 43, in getInfluxLineProtocol2
Jun  4 21:00:48 zero python3[1390]:     return self.inverter.getInfluxLineProtocol2(cmd)
Jun  4 21:00:48 zero python3[1390]:   File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppinverter.py", line 195, in getInfluxLineProtocol2
Jun  4 21:00:48 zero python3[1390]:     result = self.execute(cmd)
Jun  4 21:00:48 zero python3[1390]:   File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppinverter.py", line 351, in execute
Jun  4 21:00:48 zero python3[1390]:     return self._doDirectUsbCommand(command)
Jun  4 21:00:48 zero python3[1390]:   File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppinverter.py", line 335, in _doDirectUsbCommand
Jun  4 21:00:48 zero python3[1390]:     command.setByteResponse(response_line)
Jun  4 21:00:48 zero python3[1390]:   File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppcommand.py", line 139, in setByteResponse
Jun  4 21:00:48 zero python3[1390]:     self.response_dict = self.getResponseDict()
Jun  4 21:00:48 zero python3[1390]:   File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppcommand.py", line 434, in getResponseDict
Jun  4 21:00:48 zero python3[1390]:     msgs[key] = [float(result), resp_format[2]]
Jun  4 21:00:48 zero python3[1390]: ValueError: could not convert string to float:
Jun  4 21:00:48 zero systemd[409]: mpp-solar.service: Main process exited, code=exited, status=1/FAILURE
Jun  4 21:00:48 zero systemd[409]: mpp-solar.service: Failed with result 'exit-code'.
Jun  4 21:00:49 zero systemd[409]: mpp-solar.service: Service RestartSec=100ms expired, scheduling restart.
Jun  4 21:00:49 zero systemd[409]: mpp-solar.service: Scheduled restart job, restart counter is at 77.
```

danieltroger, 2020-06-04:
Disconnecting and reconnecting USB also brings it back for some time.

danieltroger, 2020-06-08:
I added an empty USB hub between the PI Zero and the inverter and powered the PI from a 2A Apple charger instead of some chinese USB connector and this error no longer gets the whole hidraw0 stuck.

The daemon crashes or errors often but it succeeds in restarting, so it's fine.

danieltroger, 2020-06-16:
I have started to reencounter this issue and think it might have to do with electromagnetic interference from the inverter or the huge currents going from the battery to it, as I have another single board computer in close proximity randomly getting HIGH on a GPIO pin which should be LOW, but only when the inverter is on.

@jblance where do you have your PIs mounted physically, and how long USB cables do you use?

jblance, 2020-06-16:
Hi

My Pi (Pi3) is about 50cm from the inverters and I have one connection with a USB-serial adapter and one connection with a direct USB to the inverter. Cable length is about 1 meter in total

Does ```dmesg``` or the syslog give any information?
What happens if you send one command and then wait for 30 min before sending another, does it still fail?

jblance, 2020-06-16:
Note - i run the service against the USB-serial adapter connection (for no real reason)

danieltroger, 2020-06-16:
Thanks for replying!

Do you use the USB cable that came with the inverter? Do you have the pi in a case? Is it or the USB cable close to the hight current DC cables?

I have checked dmesg once and found two related lines, can't remember them however. I had the pi on and below my battery pack previously. I downloaded a magnetometer app and there's a field of around 110-180µT around the batteries, which is strangely enough oscillating and increases when I apply a load to the inverter

[image]
Screenshot is with ~100w vs ~1kw load

I tried putting the pi besides the inverter where the field is weaker (~50µT) and it has been working fine for the last 8 hours, I'll see how it holds up.

I have always restarted the pi when I noticed the issue, but there have been interruptions longer than 30minutes. I have also tried only querying it every 10 minutes, the same thing happens.

If it happens again I'll stop the daemon and retry after 30 minutes.

[image]

danieltroger, 2020-06-16:
It just got a new quirk, it disconnects /dev/hidraw0 and shows up again as /dev/hidraw1

```
[15065.750945] dwc2 20980000.usb: dwc2_hc_halt() Channel can't be halted
[15107.581826] usb 1-1: input irq status -5 received
[18582.248689] dwc2 20980000.usb: dwc2_hc_halt() Channel can't be halted
[41096.524503] dwc2 20980000.usb: dwc2_hc_halt() Channel can't be halted
[44397.355155] dwc2 20980000.usb: dwc2_hc_halt() Channel can't be halted
[47546.710637] dwc2 20980000.usb: dwc2_hc_halt() Channel can't be halted
[54276.533330] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[54276.533366] usb 1-1: USB disconnect, device number 2
[54276.953112] usb 1-1: new low-speed USB device number 3 using dwc2
[54277.229339] usb 1-1: New USB device found, idVendor=0665, idProduct=5161, bcdDevice= 0.02
[54277.229363] usb 1-1: New USB device strings: Mfr=3, Product=1, SerialNumber=0
[54277.250669] hid-generic 0003:0665:5161.0002: hiddev96,hidraw1: USB HID v1.11 Device [HID 0665:5161] on usb-20980000.usb-1/input0
```

No other issues though

danieltroger, 2020-06-16:
LOL, it even says EMI (electromagnetic interference): `54276.533330] usb usb1-port1: disabled by hub (EMI?), re-enabling...`

jblance, 2020-06-16:
wow, thats a problem
maybe some shielding?

> Do you use the USB cable that came with the inverter? 

Yes
> Do you have the pi in a case? 

Yes - just a plastic case though
> Is it or the USB cable close to the high current DC cables?

No, Pi and cables are the opposite side (and above) to the battery and high current cabling

jblance, 2020-06-16:
sorry, click close instead of comment

danieltroger, 2020-06-23:
I also tried to put the pi and USB cables to the left and above the inverter and batteries. I no longer got the EMI warning in DMESG but the resource not available today. (Which is an improvement from getting it multiple times every day). This was on a 4.8v PSU though, maybe 5v might improve it.

However because I couldn't be bothered further improving it I simply made my pi automatically reboot every time it fails. In case anyone else needs this (very dirty and in PHP, unacceptable spaghetti code blah blah might work however)

On the pi itself:
In /home/pi/autorestart.php (which you need to `chmod +xxx /home/pi/autorestart.php`)
```
#!/usr/bin/php
<?php
$a = json_decode(file_get_contents("http://192.168.178.100/lastentry.php?" . time()),1);
$b = (int) substr($a[0],0,-9);
$c = (int) $a[1];
$d = time()-$b;
if($c < 10 || $d > 120){ // reboot if reported voltage <10v or no new value since 120s
shell_exec("reboot");
}
```
In roots crontab add execution every minute:
`sudo crontab -e`
and then add `* * * * * /home/pi/autorestart.php`

On the server where I have influxdb and an apache webserver running, add a file called lastentry.php to apache document root:
```
<?php
$x = shell_exec('influx -execute \'SELECT LAST("battery_voltage") FROM "mpp-solar"\' -database mppsolar');
$y = explode("\n",$x)[3];
$z = explode(" ",$y);
echo json_encode($z);
```

closed 2020-06-23 by danieltroger