jblance/mpp-solar #23: Q1 query protocol change?

danieltroger, 2020-06-08:
this is what happens when I try q1 query on PIP-4024MSP:

Note this line `DEBUG:MPP-Solar:usb byte_response was: b'(1 0 0000 15472 00001 00 00 03 000 026 019 028 00 00 000 0040 0000 0000 00.00 11\x0e\xc8\r'` and compare with `["(00000 00000 01 01 00 059 045 053 068 00 00 000 0040 0580 0000 50.00 139", "39B9"]` from [here](https://github.com/jblance/mpp-solar/blob/master/mppsolar/commands/q1.json)

They seem to have added 3 more values

```
pi@zero:/var/log $ mpp-solar -d /dev/hidraw0 -c Q1 -D
INFO:MPP-Solar:command Q1
INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
DEBUG:MPP-Solar:Device matches hidraw regex
INFO:MPP-Solar:Loading commands for inverter model: standard
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/muchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MUCHGC supported all inverters
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MUCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
DEBUG:MPP-Solar:Generated CRC bf 86 bf86
DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbcv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBCV supported all inverters
INFO:MPP-Solar:... command PBCV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBCV
DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
DEBUG:MPP-Solar:Generated CRC 64 5a 645a
DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbdv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBDV supported all inverters
INFO:MPP-Solar:... command PBDV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBDV
DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
DEBUG:MPP-Solar:Generated CRC fd cd fdcd
DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbft.json
DEBUG:MPP-Solar:-----No supports 
[... 241 lines cut ...]
INFO:MPP-Solar:DIRECT USB connection: executing Q1
Q1 query

DEBUG:MPP-Solar:length of to_send: 5
DEBUG:MPP-Solar:1 chunk send
DEBUG:MPP-Solar:usb byte_response was: b'(1 0 0000 15472 00001 00 00 03 000 026 019 028 00 00 000 0040 0000 0000 00.00 11\x0e\xc8\r'
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.4.1', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/__init__.py", line 83, in main
    results = mp.getResponseDict(args.command)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mpputils.py", line 40, in getResponseDict
    return self.inverter.getResponseDict(cmd)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppinverter.py", line 215, in getResponseDict
    result = self.execute(cmd)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppinverter.py", line 351, in execute
    return self._doDirectUsbCommand(command)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppinverter.py", line 335, in _doDirectUsbCommand
    command.setByteResponse(response_line)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppcommand.py", line 139, in setByteResponse
    self.response_dict = self.getResponseDict()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppcommand.py", line 443, in getResponseDict
    msgs[key] = [resp_format[2][int(result)], '']
IndexError: list index out of range
```

jblance, 2020-06-10:
Yeah it definitely looks different to what I have seen - unfortuately it is also an undocumented command - so makes it very difficult to use 
I found it useful, but cant really do much without knowing what it actually is (and different models seem to use different command response definitions too)

jblance, 2020-06-10:
Do you know what each of the response items are?
Can you also run a ```mpp-solar -c QPI``` command and provide the output
Cheers

danieltroger, 2020-06-10:
Nope, but the mpp-solar dameon. That might explain it

jblance, 2020-06-11:
Yeah, can only have one thing trying to communicate at a time or you'll get strange results

danieltroger, 2020-06-16:
So idk if I miscommunicated this, but when I opened the issue I stopped the mpp-solar daemon before I tried it. Trying other things I had commands succeed at the same time, which is why I assumed it would work, but the Q1 response is unexpected to the script, even when the daemon is definitely stoppped.

If you closed it because it's impossible to know which values are which that's fine and keep it closed, I just wanted to make it clear that I didn't originally encounter the issue because I had the daemon running.

```
pi@zero:~ $ systemctl --user stop mpp-solar
pi@zero:~ $  mpp-solar -d /dev/hidraw0 -c Q1 -D
INFO:MPP-Solar:command Q1
INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
DEBUG:MPP-Solar:Device matches hidraw regex
INFO:MPP-Solar:Loading commands for inverter model: standard
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/muchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MUCHGC supported all inverters
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MUCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
DEBUG:MPP-Solar:Generated CRC bf 86 bf86
DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbcv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBCV supported all inverters
INFO:MPP-Solar:... command PBCV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBCV
DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
DEBUG:MPP-Solar:Generated CRC 64 5a 645a
DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbdv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBDV supported all inverters
INFO:MPP-Solar:... command PBDV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBDV
DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
DEBUG:MPP-Solar:Generated CRC fd cd fdcd
DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/commands/pbft.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBFT supported all inverters
[... 915 lines cut ...]
Q1 query

DEBUG:MPP-Solar:length of to_send: 5
DEBUG:MPP-Solar:1 chunk send
DEBUG:MPP-Solar:usb byte_response was: b'(1 0 0000 07280 00001 00 00 03 000 030 023 032 01 00 000 0040 0000 0000 00.00 11\xbaW\r'
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.4.1', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/__init__.py", line 83, in main
    results = mp.getResponseDict(args.command)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mpputils.py", line 40, in getResponseDict
    return self.inverter.getResponseDict(cmd)
[... cut ...]
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppinverter.py", line 335, in _doDirectUsbCommand
    command.setByteResponse(response_line)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppcommand.py", line 139, in setByteResponse
    self.response_dict = self.getResponseDict()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.1-py3.7.egg/mppsolar/mppcommand.py", line 443, in getResponseDict
    msgs[key] = [resp_format[2][int(result)], '']
IndexError: list index out of range
pi@zero:~ $ 
```

jblance, 2020-06-16:
Yeah, unfortunately Q1 is undocumented - if you can determine what the response items are it is easy enough to create a definition file (the current version is definitely wrong for your inverter)
There are only 4 or 5 responses in your debug that might be of use, so you may be able to determine what they are from Watchpower and/or trial and error.

Even the current definitions for Q1 are pretty rough (and grabbed from a forum)

Sorry I cant really be of any help

jblance, 2020-06-16:
Also check QPIGS raw output (or update to latest code from github), I have updated it based on @nigel16494's feedback - there is a 'PV Input Power' value is some responses (not sure if it is true for QPI = 87 000 though)

danieltroger, 2020-06-16:
OMFG you're awesome! I updated and it added PV input power. I'm happy! Fuck Q1, pw input power is all I want.
`pv_input_power                	429             W`

```
pi@zero:~/mpp-solar $ mpp-solar -d /dev/hidraw1 -c QPIGS -D
INFO:MPP-Solar:command QPIGS
INFO:MPP-Solar:Serial device used: /dev/hidraw1, baud rate: 2400
DEBUG:MPP-Solar:Device matches hidraw regex
INFO:MPP-Solar:Loading commands for inverter model: standard
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/commands/muchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MUCHGC supported all inverters
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MUCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
DEBUG:MPP-Solar:Generated CRC bf 86 bf86
DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/commands/pbcv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBCV supported all inverters
INFO:MPP-Solar:... command PBCV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBCV
DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
DEBUG:MPP-Solar:Generated CRC 64 5a 645a
DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/commands/pbdv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBDV supported all inverters
INFO:MPP-Solar:... command PBDV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBDV
DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
DEBUG:MPP-Solar:Generated CRC fd cd fdcd
DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/commands/pbft.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBFT supported all inverters
INFO:MPP-Solar:... command PBFT loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBFT
[... 276 lines cut ...]
battery_charging_current      	0               A
battery_discharge_current     	25              A
battery_voltage               	26.5            V
battery_voltage_from_scc      	26.5            V
bus_voltage                   	419             V
inverter_heat_sink_temperature	36              Deg_C
is_ac_charging_on             	0               True - 1/False - 0
is_battery_voltage_to_steady_while_charging	0               True - 1/False - 0
is_charging_on                	1               True - 1/False - 0
is_charging_to_float          	0               True - 1/False - 0
is_configuration_changed      	1               True - 1/False - 0
is_load_on                    	1               True - 1/False - 0
is_reserved                   	0               True - 1/False - 0
is_sbu_priority_version_added 	0               True - 1/False - 0
is_scc_charging_on            	1               True - 1/False - 0
is_scc_firmware_updated       	0               True - 1/False - 0
is_switched_on                	1               True - 1/False - 0
pv_input_current_for_battery  	16.0            A
pv_input_power                	429             W
pv_input_voltage              	76.5            V
rsv1                          	0               A
rsv2                          	0               A
pi@zero:~/mpp-solar $ 

```

closed 2020-06-16 by danieltroger