jblance/mpp-solar #24: difference in values between queries QPIGS and QPGS0

Tolmino, 2020-06-10:
hi, i have noticed that i have some difference in values returned from queries QPIGS and QPGS0
i don't have parallel controller.... and i don't know if that parameters are read wrong for that.

i write output from different queries:
mpp-solar -d /dev/hidraw0 -c QPIGS
ac_input_frequency              49.9            Hz
ac_input_voltage                225.1           V
ac_output_active_power          505             W
ac_output_apparent_power        551             VA
ac_output_frequency             49.9            Hz
ac_output_load                  11              %
ac_output_voltage               229.8           V
battery_capacity                65              %
battery_charging_current        0               A
battery_discharge_current       3               A
battery_voltage                 51.7            V
battery_voltage_from_scc        51.82           V
bus_voltage                     357             V
inverter_heat_sink_temperature  35              Deg_C
is_ac_charging_on               0               True - 1/False - 0
is_battery_voltage_to_steady_while_charging     0               True - 1/False - 0
is_charging_on                  1               True - 1/False - 0
is_configuration_changed        0               True - 1/False - 0
is_load_on                      1               True - 1/False - 0
is_sbu_priority_version_added   0               True - 1/False - 0
is_scc_charging_on              1               True - 1/False - 0
is_scc_firmware_updated         0               True - 1/False - 0
pv_input_current_for_battery    7               A
pv_input_voltage                58.9            V
unknown_value_in_byte_response  010      

mpp-solar -d /dev/hidraw0 -c QPGS0
ac_output_active_power          608             W
ac_output_apparent_power        621             VA
ac_output_frequency             50.02           Hz
ac_output_voltage               230.0           V
battery_capacity                47              %
battery_charging_current        0               A
battery_discharge_current       6               A
battery_voltage                 49.9            V
charger_source_priority         Solar first     
fault_code                      No fault        
grid_frequency                  50.01           Hz
grid_voltage                    226.8           V
is_ac_charging                  0               True - 1/False - 0
is_battery_over_voltage         0               True - 1/False - 0
is_battery_under_voltage        0               True - 1/False - 0
is_configuration_changed        0               True - 1/False - 0
is_line_lost                    0               True - 1/False - 0
is_load_on                      1               True - 1/False - 0
is_scc_charging                 1               True - 1/False - 0
is_scc_ok                       1               True - 1/False - 0
load_percentage                 12              %
max_ac_charger_current          2               A
max_charger_current             60              A
max_charger_range               140             A
output_mode                     single machine  
parallel_instance_number        valid           
pv_input_current                7               A
pv_input_voltage                65.2            V
serial_number                   92931908103931  
total_ac_output_apparent_power  621             VA
total_ac_output_percentage      12              %
total_charging_current          0               A
total_output_active_power       608             W
work_mode                       Battery Mode    

if i restart raspberry connected to inverter (PIP5048MK) the parameters returned are equal for a while... but after some time they return different.
the wrong are returned from QPGS0

i am simply using QPIGS ....

jblance, 2020-06-10:
that is weird - i dont see how the Raspberry Pi or the code could be contributing to this
the reboot would drop the usb connectivity and reinitialize it, but i can see how that would change the actual 'answer' to a command over a serial connection.
maybe parallel query doesnt work for non-parallel inverters?

if you add ```-R``` to the command you will get the raw output (just to make sure the python code isnt doing anything weird)

Tolmino, 2020-06-10:
mpp-solar -d /dev/hidraw0 -c QPGS0 -R
['1', '92931908103931', 'B', '00', '226.1', '49.97', '230.2', '50.00', '0506', '0462', '010', '54.5', '013', '093', '066.0', '013', '00506', '00462', '009', '10100010', '0', '1', '060', '140', '02', '22', '000']

Tolmino, 2020-06-10:
mpp-solar -d /dev/hidraw0 -c QPIGS -R
['223.3', '49.9', '229.9', '50.0', '0551', '0509', '011', '367', '53.10', '009', '079', '0035', '0019', '067.4', '52.97', '00000', '00010110', '00', '00', '01038', '010']

jblance, 2020-06-10:
Wow - that is different
Something weird with the inverter response then, i checked my parallel 4048s (with a pi that has been up 16 days) and they are very close
not really sure how to even start working this (other than perhaps trying with a usb-serial adapter)

closed 2020-06-28 by jblance