jblance/mpp-solar #25: Support for LV2424 model?

kevinxyao, 2020-06-14:
I got an invalid_byte_response when trying this on a LV2424 model.

jblance, 2020-06-14:
please run the command with debug enabled and post output (add -D to command)

also please run a ```mpp-solar -c QPI``` (add any device / baud / model parameters as needed)

kevinxyao, 2020-06-15:
ubuntu@mppsolar:~$ mpp-solar -D
INFO:MPP-Solar:command QID
INFO:MPP-Solar:Serial device used: /dev/ttyUSB0, baud rate: 2400
INFO:MPP-Solar:Loading commands for inverter model: standard
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/muchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MUCHGC supported all inverters
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MUCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
DEBUG:MPP-Solar:Generated CRC bf 86 bf86
DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/pbcv.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command PBCV supported all inverters
INFO:MPP-Solar:... command PBCV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBCV
DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
DEBUG:MPP-Solar:Generated CRC 64 5a 645a
DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/pbdv.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command PBDV supported all inverters
INFO:MPP-Solar:... command PBDV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBDV
DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
DEBUG:MPP-Solar:Generated CRC fd cd fdcd
DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/pbft.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command PBFT supported all inverters
INFO:MPP-Solar:... command PBFT loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBFT
DEBUG:MPP-Solar:Calculating CRC for b'PBFT'
DEBUG:MPP-Solar:Generated CRC bb ed bbed
DEBUG:MPP-Solar:Full byte command: b'PBFT\xbb\xed\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/pbt.json
DEBUG:MPP-Solar:-----No supports
[... 225 lines cut ...]
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/qvfw2.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command QVFW2 supported all inverters
INFO:MPP-Solar:... command QVFW2 loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for QVFW2
DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
DEBUG:MPP-Solar:Searching for cmd 'QID'
INFO:MPP-Solar:SERIAL connection: executing QID
Device Serial Number inquiry
 -- queries the device serial number

DEBUG:MPP-Solar:port /dev/ttyUSB0, baudrate 2400
DEBUG:MPP-Solar:Command execution attempt 1...
DEBUG:MPP-Solar:serial byte_response was: b''
DEBUG:MPP-Solar:Byte_Response length: 0
DEBUG:MPP-Solar:Byte Response invalid as too short
INFO:MPP-Solar:Invalid byte_response
error                           Invalid byte_response
response

ubuntu@mppsolar:~$ ubuntu@mppsolar:~$ mpp-solar -c QPI
error                           Invalid byte_response
response

jblance, 2020-06-15:
No response from the inverter at all
What is your setup?
What computer / os / versions is running mppsolar?
What connectivity do you have to the inverter?
What type of USB to serial adapter are you using?
Can you disconnect the USB, reconnect it and post the ```dmesg``` and syslog output

kevinxyao, 2020-06-16:
LV2424, with USB to RJ45(rs232) cable, also tried usb to usb cable
raspi 3b+, Linux version 5.4.0-1012-raspi (buildd@bos02-arm64-033) (gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2))

[image]

[    3.802872] usb 1-1.1.3: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[    3.802887] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.802894] usb 1-1.1.3: Product: FT232R USB UART
[    3.802902] usb 1-1.1.3: Manufacturer: FTDI
[    3.802907] usb 1-1.1.3: SerialNumber: AD0K6BL2
[    4.042859] usb 1-1.1.1: new high-speed USB device number 5 using dwc_otg
[    4.143464] usb 1-1.1.1: New USB device found, idVendor=0424, idProduct=7800, bcdDevice= 3.00
[    4.143478] usb 1-1.1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   14.734961] usbcore: registered new interface driver usbserial_generic
[   14.735011] usbserial: USB Serial support registered for generic
[   14.792786] usbcore: registered new interface driver ftdi_sio
[   14.792843] usbserial: USB Serial support registered for FTDI USB Serial Device
[   14.793234] usb 1-1.1.3: Detected FT232RL
[   14.795313] usb 1-1.1.3: FTDI USB Serial Device converter now attached to ttyUSB0
[   15.232107] usbcore: registered new interface driver brcmfmac

jblance, 2020-06-16:
Is it the only communications cable that is connected?
Have you tried with sudo? ie ```sudo mpp-solar -c QID -I```
I wonder if it is a different protocol, could you try ```mpp-solar -c GS -M PI18 -I```

kevinxyao, 2020-06-16:
Yes, it is the only communication cable that is connected.
Tried both commands, same result:
INFO:MPP-Solar:Invalid byte_response
error                           Invalid byte_response

nigel16494, 2020-06-16:
Can you check you have access to the USB . Go into the /dev directory and do an ls -l . On my system I set it using chmod 777 <usb> e.g. chmod 777 ttyUSB0

kevinxyao, 2020-06-17:
chmod 777 ttyUSB0 in /dev
crwxrwxrwx 1 root dialout 188,   0 Apr  1 17:23 ttyUSB0

then tried
sudo mpp-solar -c QID -I
mpp-solar -c GS -M PI18 -I

INFO:MPP-Solar:Invalid byte_response
error Invalid byte_response

last few lines of  "mpp-solar -D"
DEBUG:MPP-Solar:port /dev/ttyUSB0, baudrate 2400
DEBUG:MPP-Solar:Command execution attempt 1...
DEBUG:MPP-Solar:serial byte_response was: b''
DEBUG:MPP-Solar:Byte_Response length: 0
DEBUG:MPP-Solar:Byte Response invalid as too short
INFO:MPP-Solar:Invalid byte_response
error                           Invalid byte_response
response

jblance, 2020-06-17:
Do you have another cable and adapter that you can try?
How about the direct usb connection to the inverter?

kevinxyao, 2020-06-23:
Tried another cable.  Got a bit more info this time.

$sudo mpp-solar -c GS -M PI18 -I

INFO:MPP-Solar:Invalid byte_response
error                           Invalid byte_response
response                        0000 000 0000 000 0000 0000 000 266 266 000 000 000 070 029 032 000 0005 0000 0362 0000 0 2 0 0 0 0 0 0

jblance, 2020-06-23:
That's looking better. Try the 
```mpp-solar -c QPI -I``` command

kevinxyao, 2020-06-23:
$dmesg |grep usb
[    4.079819] usb 1-1.1.1: New USB device strings: Mfr=0, Product=0,
SerialNumber=0
[   12.329415] usbcore: registered new interface driver usbserial_generic
[   12.329452] usbserial: USB Serial support registered for generic
[   12.358994] usbcore: registered new interface driver pl2303
[   12.359050] usbserial: USB Serial support registered for pl2303
[   12.363404] usb 1-1.2: pl2303 converter now attached to ttyUSB0
[   12.811146] usbcore: registered new interface driver brcmfmac

$sudo  mpp-solar -c QPI -I

INFO:MPP-Solar:... command QVFW2 loaded for inverter model: standard
INFO:MPP-Solar:SERIAL connection: executing QPI
Protocol ID inquiry
 -- queries the device protocol ID. e.g. 30 for HS series

protocol_id                     0

jblance, 2020-06-23:
Hmm, no joy there then. ok how about
```sudo mpp-solar -c GS -M PI18 -D```
and
```sudo mpp-solar -c QPIGS -D```

kevinxyao, 2020-06-24:
$sudo mpp-solar -c QPIGS -D
<...>
packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/
qpiws.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command QPIWS supported all inverters
INFO:MPP-Solar:... command QPIWS loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for QPIWS
DEBUG:MPP-Solar:Calculating CRC for b'QPIWS'
DEBUG:MPP-Solar:Generated CRC b4 da b4da
DEBUG:MPP-Solar:Full byte command: b'QPIWS\xb4\xda\r'
DEBUG:MPP-Solar:Loading command information from
/usr/local/lib/python3.8/dist-p

 ackages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/qvfw.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command QVFW supported all inverters
INFO:MPP-Solar:... command QVFW loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for QVFW
DEBUG:MPP-Solar:Calculating CRC for b'QVFW'
DEBUG:MPP-Solar:Generated CRC 62 99 6299
DEBUG:MPP-Solar:Full byte command: b'QVFWb\x99\r'
DEBUG:MPP-Solar:Loading command information from
/usr/local/lib/python3.8/dist-p

 ackages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/qvfw2.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command QVFW2 supported all inverters
INFO:MPP-Solar:... command QVFW2 loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for QVFW2
DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
DEBUG:MPP-Solar:Searching for cmd 'QPIGS'
INFO:MPP-Solar:SERIAL connection: executing QPIGS
General Status Parameters inquiry
 -- queries the value of various metrics from the Inverter

DEBUG:MPP-Solar:port /dev/ttyUSB0, baudrate 2400
DEBUG:MPP-Solar:Command execution attempt 1...
DEBUG:MPP-Solar:serial byte_response was: b'^0\x1b\xe3\r'
DEBUG:MPP-Solar:Byte_Response length: 5
DEBUG:MPP-Solar:CRC resp         27, 227
DEBUG:MPP-Solar:Calculating CRC for b'^0'
DEBUG:MPP-Solar:Generated CRC 1b e3 1be3
DEBUG:MPP-Solar:CRC calc         27 227
DEBUG:MPP-Solar:CRCs match
DEBUG:MPP-Solar:Response invalid as insufficient number of elements in
byte_resp
                     onse. Got 1, expected as least 17
INFO:MPP-Solar:Invalid byte_response
error                           Invalid byte_response
response

$sudo mpp-solar -c GS -M PI18 -D
 <...>
DEBUG:MPP-Solar:Calculating CRC for b'QPIRI'
DEBUG:MPP-Solar:Generated CRC f8 54 f854
DEBUG:MPP-Solar:Full byte command: b'QPIRI\xf8T\r'
DEBUG:MPP-Solar:Loading command information from
/usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/qpiws.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command QPIWS supported all inverters
INFO:MPP-Solar:... command QPIWS loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for QPIWS
DEBUG:MPP-Solar:Calculating CRC for b'QPIWS'
DEBUG:MPP-Solar:Generated CRC b4 da b4da
DEBUG:MPP-Solar:Full byte command: b'QPIWS\xb4\xda\r'
DEBUG:MPP-Solar:Loading command information from
/usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/qvfw.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command QVFW supported all inverters
INFO:MPP-Solar:... command QVFW loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for QVFW
DEBUG:MPP-Solar:Calculating CRC for b'QVFW'
DEBUG:MPP-Solar:Generated CRC 62 99 6299
DEBUG:MPP-Solar:Full byte command: b'QVFWb\x99\r'
DEBUG:MPP-Solar:Loading command information from
/usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.2-py3.8.egg/mppsolar/commands/qvfw2.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command QVFW2 supported all inverters
INFO:MPP-Solar:... command QVFW2 loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for QVFW2
DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
DEBUG:MPP-Solar:Searching for cmd 'GS'
INFO:MPP-Solar:SERIAL connection: executing GS
General status query
 -- Query general status information

DEBUG:MPP-Solar:port /dev/ttyUSB0, baudrate 2400
DEBUG:MPP-Solar:Command execution attempt 1...
DEBUG:MPP-Solar:serial byte_response was:
b'^D1061191,599,1191,599,1410,1410,058,260,261,000,012,000,058,037,047,000,1247,0000,0768,0000,0,2,0,1,2,2,1,0\x9a\x05\r'
DEBUG:MPP-Solar:Byte_Response length: 111
DEBUG:MPP-Solar:CRC resp         154, 5
DEBUG:MPP-Solar:Calculating CRC for
b'^D1061191,599,1191,599,1410,1410,058,260,261,000,012,000,058,037,047,000,1247,0000,0768,0000,0,2,0,1,2,2,1,0'
DEBUG:MPP-Solar:Generated CRC 9a 5 9a05
DEBUG:MPP-Solar:CRC calc         154 5
DEBUG:MPP-Solar:CRCs match
DEBUG:MPP-Solar:Response invalid as insufficient number of elements in
byte_response. Got 1, expected as least 28
INFO:MPP-Solar:Invalid byte_response
error                           Invalid byte_response
response                        1191 599 1191 599 1410 1410 058 260 261 000
012 000 058 037 047 000 1247 0000 0768 0000 0 2 0 1 2 2 1 0

jblance, 2020-06-24:
Ok so its definitely the D type protocol, can you check which version
```sudo mpp-solar -c PI -M PI18 -D```

jblance, 2020-06-24:
Are you using the version from github, or pypi?

kevinxyao, 2020-06-24:
github

jblance, 2020-06-24:
> github

Cool, can you update (and check the protocol) - the response is of the PI17 or PI18 format, but not sure why its not parsing with -M PI18, its possible that I fixed it after the version you are running..

kevinxyao, 2020-06-26:
I re-ran
sudo python3 ./setup.py install

the mpp-solar verion is 0.4.2

$sudo mpp-solar -c GS -M PI18 -D
DEBUG:MPP-Solar:port /dev/ttyUSB0, baudrate 2400
DEBUG:MPP-Solar:Command execution attempt 1...
DEBUG:MPP-Solar:serial byte_response was:
b'^D1061207,600,1207,600,0183,0183,007,270,270,000,000,000,100,031,043,000,0540,0000,0814,0000,0,2,0,1,1,2,2,0e\xf6\r'
DEBUG:MPP-Solar:Byte_Response length: 111
DEBUG:MPP-Solar:CRC resp         101, 246
DEBUG:MPP-Solar:Calculating CRC for
b'^D1061207,600,1207,600,0183,0183,007,270,270,000,000,000,100,031,043,000,0540,0000,0814,0000,0,2,0,1,1,2,2,0'
DEBUG:MPP-Solar:Generated CRC 65 f6 65f6
DEBUG:MPP-Solar:CRC calc         101 246
DEBUG:MPP-Solar:CRCs match
DEBUG:MPP-Solar:Response invalid as insufficient number of elements in
byte_response. Got 1, expected as least 28
INFO:MPP-Solar:Invalid byte_response
error                           Invalid byte_response
response                        1207 600 1207 600 0183 0183 007 270 270 000
000 000 100 031 043 000 0540 0000 0814 0000 0 2 0 1 1 2 2 0

$sudo mpp-solar -c QPIGS -D
DEBUG:MPP-Solar:port /dev/ttyUSB0, baudrate 2400
DEBUG:MPP-Solar:Command execution attempt 1...
DEBUG:MPP-Solar:serial byte_response was: b'^0\x1b\xe3\r'
DEBUG:MPP-Solar:Byte_Response length: 5
DEBUG:MPP-Solar:CRC resp         27, 227
DEBUG:MPP-Solar:Calculating CRC for b'^0'
DEBUG:MPP-Solar:Generated CRC 1b e3 1be3
DEBUG:MPP-Solar:CRC calc         27 227
DEBUG:MPP-Solar:CRCs match
DEBUG:MPP-Solar:Response invalid as insufficient number of elements in
byte_response. Got 1, expected as least 17
INFO:MPP-Solar:Invalid byte_response
error                           Invalid byte_response
response                        0

jblance, 2020-06-26:
That's still same version, you'll need to update from got first.
How did you get the code from git originally?

kevinxyao, 2020-06-26:
git clone https://github.com/jblance/mpp-solar.git

jblance, 2020-06-26:
ok, so refresh the repo
(in the directory that you cloned to:
```git pull```
then re-run the setup
you should have version 0.4.4 if you have the latest
```mpp-solar -h``` will display the running version

kevinxyao, 2020-06-26:
The previous post got truncated.  It is on 0.4.4
the mpp-solar verion was at 0.4.2, got updated to 0.4.4

jblance, 2020-06-26:
oh, ok sorry
can you try the protocol query
```sudo mpp-solar -c PI -M PI18 -D```

jblance, 2020-06-26:
i might have found the/an issue
Can you try version 0.4.5 from git

kevinxyao, 2020-06-27:
I think you fixed it too.
I got valid output now using:
$sudo mpp-solar -c QPIGS -D

Now I need to follow the rest of your guide and put this on graphana.

closed 2020-06-28 by jblance

bkjerry82, 2020-07-11:
Hello, I looked on git but only see version 4.4 and I am having the same issues as above. How can I update go Version 4.5?

jblance, 2020-07-11:
You need to build from source at the moment (I havent made a new release)
I'll make a new release next week if you dont want to build from source

bkjerry82, 2020-07-11:
Thanks for the prompt response. If you had direction on how to build from source…. Ill give it a go. If not, ill wait for next week. In any case thank you for you effort and I appreciate your time building this. Thanks! Jeremy Sent from Mail for Windows 10 From: John BlanceSent: Saturday, July 11, 2020 5:00 PMTo: jblance/mpp-solarCc: bkjerry82; CommentSubject: Re: [jblance/mpp-solar] Support for LV2424 model? (#25) You need to build from source at the moment (I havent made a new release)I'll make a new release next week if you dont want to build from source

jblance, 2020-07-11:
It's pretty straightforward
- download zip and extract or clone the repo
- from the directory with setup.py tuned ```sudo python3 setup.py install```
Check version with ```mpp-solar-h```

bkjerry82, 2020-07-11:
Do you have the URL to the zip?

jblance, 2020-07-11:
Click on the green button on the repos main github page and there is a download zip option

bkjerry82, 2020-07-12:
I was able to build and install. Thanks. Now showing 4.6. However still receiving an error. Info below.

pi@MPP-Pi:~ $ sudo mpp-solar -d /dev/hidraw0 -c QPIGS -D
INFO:MPP-Solar:command QPIGS
INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
DEBUG:MPP-Solar:Device matches hidraw regex
INFO:MPP-Solar:Loading commands for inverter model: standard
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/muchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MUCHGC supported all inverters
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MUCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
DEBUG:MPP-Solar:Generated CRC bf 86 bf86
DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/pbcv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBCV supported all inverters
INFO:MPP-Solar:... command PBCV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBCV
DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
DEBUG:MPP-Solar:Generated CRC 64 5a 645a
DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/pbdv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBDV supported all inverters
INFO:MPP-Solar:... command PBDV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBDV
DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
DEBUG:MPP-Solar:Generated CRC fd cd fdcd
DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/pbft.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBFT supported all inverters
INFO:MPP-Solar:... command PBFT loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBFT
DEBUG:MPP-Solar:Calculating CRC for b'PBFT'
DEBUG:MPP-Solar:Generated CRC bb ed bbed
[... 247 lines cut ...]
DEBUG:MPP-Solar:Generate full byte command for QVFW2
DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
DEBUG:MPP-Solar:Searching for cmd 'QPIGS'
INFO:MPP-Solar:DIRECT USB connection: executing QPIGS
General Status Parameters inquiry
 -- queries the value of various metrics from the Inverter

DEBUG:MPP-Solar:length of to_send: 8
DEBUG:MPP-Solar:1 chunk send
DEBUG:MPP-Solar:usb byte_response was: b'^0\x1b\xe3\r'
DEBUG:MPP-Solar:Byte_Response length: 5
DEBUG:MPP-Solar:CRC resp	 27, 227
DEBUG:MPP-Solar:Calculating CRC for b'^0'
DEBUG:MPP-Solar:Generated CRC 1b e3 1be3
DEBUG:MPP-Solar:CRC calc	 27 227
DEBUG:MPP-Solar:CRCs match
DEBUG:MPP-Solar:getResponse with protocol not PI18
DEBUG:MPP-Solar:Response invalid as insufficient number of elements in byte_response. Got 1, expected as least 21
INFO:MPP-Solar:Invalid byte_response
DEBUG:MPP-Solar:getResponse with protocol not PI18
error                         	Invalid byte_response 
response                      	0               
pi@MPP-Pi:~ $

jblance, 2020-07-12:
Looks like a different protocol
Try ```sudo mpp-solar -d /dev/hidraw0 -c PI -D -M PI18```