jblance/mpp-solar #26: QPIGS update for PV power for a 2424LV

nigel16494, 2020-06-14:
I needed the PV input power from the panels so I changed the QPIGS a little at the end to add some extra fields and fix the unknown byte issue, it seems to be working OK now for several weeks.

My qpigs.json is
{
    "name": "QPIGS",
    "description": "General Status Parameters inquiry",
    "help": " -- queries the value of various metrics from the Inverter",
    "type": "QUERY",
    "nosupports": ["LV5048"],
    "response": [
            ["float", "AC Input Voltage", "V"],
            ["float", "AC Input Frequency", "Hz"],
            ["float", "AC Output Voltage", "V"],
            ["float", "AC Output Frequency", "Hz"],
            ["int", "AC Output Apparent Power", "VA"],
            ["int", "AC Output Active Power", "W"],
            ["int", "AC Output Load", "%"],
            ["int", "BUS Voltage", "V"],
            ["float", "Battery Voltage", "V"],
            ["int", "Battery Charging Current", "A"],
            ["int", "Battery Capacity", "%"],
            ["int", "Inverter Heat Sink Temperature", "Deg_C"],
            ["int", "PV Input Current for Battery", "A"],
            ["float", "PV Input Voltage", "V"],
            ["float", "Battery Voltage from SCC", "V"],
            ["int", "Battery Discharge Current", "A"],
            ["flags", "Device Status", [
		            "is_sbu_priority_version_added",
                "is_configuration_changed",
                "is_scc_firmware_updated",
                "is_load_on",
                "is_battery_voltage_to_steady_while_charging",
                "is_charging_on",
                "is_scc_charging_on",
                "is_ac_charging_on"
                ]
            ],
            ["int", "RSV1", "A"],
            ["int", "RSV2", "A"],
            ["int", "PV Input Power", "W"] ,
            ["flags", "Device Status2", [
		"is_charging_to_float",
                "is_switched_on",
                "is_reserved"
         
                ]
            ]
 
        ],
    "test_responses": [
            ["(000.0 00.0 230.0 49.9 0161 0119 003 460 57.50 012 100 0069 0014 103.8 57.45 00000 00110110 00 00 00856 010", "248C"]
        ],
	"regex": ""
}

Hope this helps

jblance, 2020-06-14:
great thanks - can you run a QPI command ```mpp-solar -c QPI``` (add any device / model parameters as needed)
This is because I think the different protocol versions have slightly different responses and I am trying to put together a better way of choosing the correct response definition

So, maybe the tool checks QPI first then selects the right response definition from there (ie my 4048 gives ```PI30```)

nigel16494, 2020-06-15:
pi@raspberrypi:~ $ mpp-solar -c QPI -d /dev/hidraw0 
protocol_id                   	PI30          

Also I was doing some fuse and rewiring work and so I could test the "unknown byte" flags.  I found that it works also.
is_switched_on                	1     ---> when my inverter is on
is_switched_on                	0     ---> when my inverter is off

Let me know how I can help further.

Thanks, Nigel

jblance, 2020-06-15:
Great thanks
Have added this change to the repo
Cheers
John

closed 2020-06-15 by jblance