jblance/mpp-solar #27: Problem to start daemon

Tolmino, 2020-06-21:
hi, i can use 
mpp-info-pub -d /dev/hidraw0 -q 127.0.0.1 -c QPIGS
and mosquito receive datas 

but if i try to start service i have:
/usr/bin/python3 /usr/local/bin/mpp-solar-service -c /etc/mpp-solar/mpp-solar.conf
MPP-Solar-Service: Initializing ...
MPP-Solar-Service: Config file: /etc/mpp-solar/mpp-solar.conf
MPP-Solar-Service: Config setting - pause: 5
MPP-Solar-Service: Config setting - mqtt_broker: 127.0.0.1
MPP-Solar-Service: Config setting - command sections found: 1
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar-service", line 11, in <module>
    load_entry_point('mpp-solar==0.4.3', 'console_scripts', 'mpp-solar-service')()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.3-py3.7.egg/mppsolar/mpp_solar_service.py", line 52, in main
    systemd.daemon.notify('READY=1')
  File "/usr/local/lib/python3.7/dist-packages/systemd/daemon.py", line 39, in notify
    raise TypeError("state must be an instance of Notification")
TypeError: state must be an instance of Notification

i attach my mpp-solar.conf
[SETUP]
# Number of seconds to pause between command execution loop
# 0 is no pause, greater than 60 will cause service restarts
pause=5
mqtt_broker=127.0.0.1

[Inverter1]
model=standard
port=/dev/hidraw0
baud=2400
command=QPIGS
tag=Inverter1
format=influx2

any suggest?

jblance, 2020-06-21:
It looks like an incompatible python systemd

How did you install the python systemd support? (pip or apt?)
can you do a ```pip show systemd``` or ```dpkg -l python*systemd``` and post results

Tolmino, 2020-06-22:
pip3 show systemd
Name: systemd
Version: 0.16.1
Summary: Python systemd wrapper
Home-page: http://github.com/mosquito/python-systemd
Author: Dmitry Orlov <me@mosquito.su>
Author-email: me@mosquito.su
License: Apache
Location: /usr/local/lib/python3.7/dist-packages
Requires: 
Required-by: 

dpkg -l python*systemd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name            Version      Architecture Description
+++-===============-============-============-=================================
ii  python3-systemd 234-2+b1     armhf        Python 3 bindings for systemd

Tolmino, 2020-06-22:
i think i have installed with pip3

jblance, 2020-06-22:
I dont think that version is compatible. I used apt-get to install as per the readme.
Can you try that way?
Otherwise there may be a way to cater for the different versions (though I currently dont know what that would be...)

Tolmino, 2020-06-22:
what version i must install ?
i can download it and compile

Tolmino, 2020-06-22:
simply made 2 tests: 
version that install pip3 is incompatible
but the one that install apt-get works

jblance, 2020-06-23:
yeah the usage of the two is different
I used the apt package python3-systemd (which is on pypi as systemd-python)

closed 2020-07-08 by jblance

gianfrdp, 2024-07-07:
Hello,
I think systemd-python pip package should be a requirement for systemd.
I lost 2h today trying to install in venv and configure service

After installing with
```
pip install mppsolar[ble,systemd]
```

pip installed "cysystemd 1.6.2", but with this package didn't work.

```
Jul 07 18:41:20 nuc python3[10970]: You are missing dependencies in order to be able to use the --daemon flag.
Jul 07 18:41:20 nuc python3[10970]: To install them, use that command:
Jul 07 18:41:20 nuc python3[10970]:     python -m pip install 'mppsolar[systemd]'
```

Finally I found this issue, and installing systemd-python worked.