jblance/mpp-solar #30: All is installed but then hwo does it work ?

HuguesD, 2020-06-25:
Hello,

Thank you for posting this mpp solar service.
I have installed all of the modules, mqtt, influx, telegraph, Grafana and of course mppsolar sucesfully according to your instructions.

But now , starting Graphana doesn't display anything. It seems I have to create and configure a data source (I suppose Influxdb) and a monitoring panel. Is this correct? If so, how do I configure the dashboards/panels to visualize the mpp solar data collected from the inverter ? 
I 'm feeling I'm missing Something or that the instructions stopped Midway ?

All help welcome

Thanks

jblance, 2020-06-25:
Yes you'll need to configure an influx data source and then add panels.
This is standard grafana tasks, so grafana forums etc are the best bet to find info (I'm no expert in grafana)
Best check is to make sure that influx is populating with data

HuguesD, 2020-06-26:
Thanks. 
Issue at the moment is that I have no screen monitor next to the inverter so I can't configure graphana/influx and plug the serial cable to the inverter at the same time.
I guess I'll have to purchase one extra screen to setup next to the inverter.

jblance, 2020-06-26:
Should need to. I have influx and grafana setup on a pi separate from the one connected to the inverter.
Even if you have everything on the pi connected to the inverter you should be able to ssh to check influx and use the a web browser to connect to grafana (and configure it)

If you need help, please describe what you have and the results of troubleshooting the steps and I'll do my best to help you along

HuguesD, 2020-06-26:
Thx a lot, I appreciate it. I'll let you know when I get the first data on a grafana graph. At the moment I connect to grafana without issue. Just need to configure a panel and hope I get the data from Influx.

HuguesD, 2020-06-26:
[image]
Ok, some feedback in my testing: the mpp-solar command does not provide expected results. I either get serial port error or sometimes (but erratically using the same command) I get an output but not as expected (see picture from screenshot)

I confirm I have the inverter on ttyUSB0, so that is not the issue.
My inverter model is a "INVERTER KING 5KW PAR-A" (It is a MK 5048 model from mpp-solar, rebranded)
Is the mode the issue as your software only supports LV5048 and not MK5048 ? (command syntax might be different ?)

I also join two other screenshots showing the influxdb where I do not seem to have the right tables? and another sreenshot showing the mpp-solar service is active.

An idea what's wrong and what to correct ?

HuguesD, 2020-06-26:
the two other screens:
[image]
[image]

HuguesD, 2020-06-27:
UPdate :the command "mpp-solar -s" sometimes returns an answer, but some other times returns an error as shown on the following screenshots:
[image]

[image]

And I still have an error running the "mpp-info-pub -q localhost -c QID" test command as show on the Following screen:
[image]

Hope it helps troubleshooting

jblance, 2020-06-27:
Is the service running? Is anything else running that could be accessing
the usb?

HuguesD, 2020-06-27:
UPdate : I disabled the serial console port on Raspberry (using the ras-config tool). I also unplugged and replugged my FTDI-USB-Serial cable and watched the output of dmesg. It confirmes that it is on ttyUSB0.
Then the mpp-solar command worked and MIRACLE, the mpp-info-sub -c QID retrieved the inverter serial number.

I checked all services (telegraph, grafana-server) and they are running. I restarted to be sure the mpp-solar service.
All these services are active.

So we progress…

Now I checked the influx tables (show measurements) and it only shows the default tables : cpu, etc but no tables related to the inverter (what should I see by the way ?)

And when I troubleshoot the telegraph service, it shows this screen:
[image]

We are progressin, slowly but surely :)

jblance, 2020-06-27:
What is the config of the mpp-solar service?

jblance, 2020-06-27:
There is a need to use the influx or influx2 outputs for the mqtt messages

jblance, 2020-06-27:
ie, use:
```mpp-info-sub -c QID -i```
or
```mpp-info-sub -c QID -I```

HuguesD, 2020-06-27:
I used the defaut (example) file you provided on GitHub for the config of the mpp-solar service. Should I modify it ?

HuguesD, 2020-06-27:
I noticed that retrying the simple mpp-solar -s command does not work anynore when it worked 30 minutes ago without touching anything! Unplugging and replugging the serial/usb cable solves this. It is like Raspberry disconnects the serial/usb port after a while (is this a power saving thing that I should deactivate?)

Trying your command mpp-info-sub -c QID -i
creates this error:
[image]

HuguesD, 2020-06-27:
I have now two tables showing in influx : QPIGS and mpp-solar. I do not know why they appear now when they did not until now. 
I now Added mpp_info_sub -q localhost -c QPIGS -i in a crontab to get regular measurements. 
So using crontab, do I still need to run the mpp-solar service ? (I guess the service never created data in influx). What's the purpose of the mp-solar service ?

HuguesD, 2020-06-27:
I succeeded to have real time data on Grafana, HUrraaayy!
[image]

I'd like to pinpoint what was wrong but I suspect:
-first issues with using the USB/Serial port. I did two things : disabling the serial console (using rasp-config) and I have to unplug /replug the serial/usb cable on the Raspberry PI4 once in a while (this I Don't understand yet what is going on)
-I thought the mpp solar running as a service would regularly launch data collection from the inverter to pupulate influx. It does not do that (I still Don't know what is the purpose of the mpp-solar service as it appears to be useless). Instead I configured crontab to run the mpp-info-pub -q localhost -c QPIGS -i every minute. This command générâtes data in influxdb

With data in Influxdb coming in regularly every minute, I then created panels in Grafana which is super easy.

I'm very glad I coud finally make it work.

For memo my config is:
-WKS EVO 5Kw 48V inverter (mpp solar MK5048 inverter)
-Raspberry PI4
-Using a USB/Serial (FTDI) cable plugged on a USB port of the Raspberry on one end, and plugged into the PC/Serial port of the inverter on the other end

DMIINC, 2020-06-27:
Purpose of mpp-solar as service is to do it faster then one minute interval (lowest crontab will go). Did you follow setup instructions? Did you setup file mpp-solar.conf? With out it service will not work.

HuguesD, 2020-06-28:
> Purpose of mpp-solar as service is to do it faster then one minute interval (lowest crontab will go). Did you follow setup instructions? Did you setup file mpp-solar.conf? With out it service will not work.

I followed the instructions to the letter and spent two days rechecking everything. 
I will check again the cofiguraton of the mpp-solar service.

Otherwise crontab justs do the job fine and I do not need more data points than every minute and I'm happy with that.

jblance, 2020-06-28:
Do not use the service unless you really need it (it makes things more complicated)
If cron is working well for you then stick with cron.

Cheers
John

closed 2020-07-08 by jblance