jblance/mpp-solar #31: Getting re-started... python issue?

cowpen, 2020-06-27:
I had mpp-solar installed briefly a couple of months ago, but didn't have time to pursue to completion.

Now I'm back to it. I pulled the latest using:

git clone https://github.com/jblance/mpp-solar.git
cd mpp-solar
sudo python ./setup.py install

That went fine. 
sudo mpp-solar -h
gives me 
[...]
`MPP Solar Command Utility, version: 0.4.5`

But now...
mpp-solar -M LV5048 -d /dev/hidraw0 -l
outputs:
```
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.4.5', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.5-py2.7.egg/mppsolar/__init__.py", line 50, in main
    mp = mppUtils(args.device, args.baud, args.model)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.5-py2.7.egg/mppsolar/mpputils.py", line 31, in __init__
    self.inverter = mppInverter(serial_device, baud_rate, inverter_model)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.5-py2.7.egg/mppsolar/mppinverter.py", line 146, in __init__
    self._commands = getCommandsFromJson(inverter_model)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.5-py2.7.egg/mppsolar/mppinverter.py", line 100, in getCommandsFromJson
    help=getDataValue(data, 'help'), crc_function=getDataValue(data, 'crc'), prefix=getDataValue(data, 'prefix'), protocol=protocol))
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.5-py2.7.egg/mppsolar/mppcommand.py", line 124, in __init__
    self.byte_command = get_byte_command(self.cmd_str, self.crc_function)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.5-py2.7.egg/mppsolar/mppcommand.py", line 78, in get_byte_command
    byte_cmd = bytes(cmd, 'utf-8')
TypeError: str() takes at most 1 argument (2 given)
```
Do I have a python problem here? Any help appreciated.

cowpen, 2020-06-27:
Just saw the new beta release is for python3. Do I maybe need to reinstall using this instead?:
sudo python3 ./setup.py install

nigel16494, 2020-06-27:
You definitely have a python problem, you're running python3 code in python 2. Bytes doesn't use str() in python 3.  I used PIP3 I think to install it .

cowpen, 2020-06-27:
Thanks. I reinstalled using python3 and no longer get that error.
Something new to chase down now:

```
pi@powerpi:~/mpp-solar $ mpp-solar -M LV5048 -d /dev/hidraw0 -c QPIGS
error                           No byte_response

```

cowpen, 2020-06-27:
Here's the debug output which does include some inverter responses:

```
sudo mpp-solar -M LV5048 -d /dev/hidraw0 -c QPIGS -D
INFO:MPP-Solar:command QPIGS
INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
DEBUG:MPP-Solar:Device matches hidraw regex
INFO:MPP-Solar:Loading commands for inverter model: LV5048
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full byte command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/commands/muchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MUCHGC supported all inverters
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full byte command for MUCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
DEBUG:MPP-Solar:Generated CRC bf 86 bf86
DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/commands/pbcv.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command PBCV supported all inverters
INFO:MPP-Solar:... command PBCV loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full byte command for PBCV
DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
DEBUG:MPP-Solar:Generated CRC 64 5a 645a
DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/commands/pbdv.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command PBDV supported all inverters
INFO:MPP-Solar:... command PBDV loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full byte command for PBDV
DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
DEBUG:MPP-Solar:Generated CRC fd cd fdcd
DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/commands/pbft.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command PBFT supported all inverters
INFO:MPP-Solar:... command PBFT loaded for inverter model: LV5048
DEBUG:MPP-Solar:Generate full byte command for PBFT
DEBUG:MPP-Solar:Calculating CRC for b'PBFT'
[... 288 lines cut ...]
battery_voltage                 56.4            V
battery_voltage_from_scc        0.0             V
bus_voltage                     224             V
inverter_heat_sink_temperature  42              Deg_C
is_battery_over                 0               True - 1/False - 0
is_battery_under                0               True - 1/False - 0
is_l1_ac_charging_on            0               True - 1/False - 0
is_l1_line_not_ok               1               True - 1/False - 0
is_l1_scc_charging_on           0               True - 1/False - 0
is_l1_scc_ok                    0               True - 1/False - 0
is_load_on                      0               True - 1/False - 0
l1_ac_input_frequency           60.0            Hz
l1_ac_input_voltage             122.6           V
l1_ac_output_active_power       93              W
l1_ac_output_apparent_power     131             VA
l1_ac_output_frequency          59.9            Hz
l1_ac_output_load               5               %
l1_ac_output_voltage            119.9           V
pv_input_current_for_battery    0.0             A
pv_input_voltage                0.0             V
unknown_value_in_byte_response  010
pi@powerpi:~ $
```
`
`

cowpen, 2020-06-27:
Seems to be working, but sort of intermittently. Sent command once and received an error. Immediately sent it again, and received a response.

[image]

jblance, 2020-06-27:
Is the mppsolar service running?
I

cowpen, 2020-06-27:
No. Was only trying from shell to start.

Rebooted and tried again - 3 times in succession:
[image]

cowpen, 2020-06-28:
I've looked for any other processes or services that might be interfering with /dev/hidraw0 and found none. But it seems like the NAK and invalid byte responses are only intermittent. Might not be a problem.

Something new that popped up while configuring MQTT... unable to run mpp-info-pub:
[image]

DMIINC, 2020-06-28:
How long is your cable?

cowpen, 2020-06-28:
About 3 feet. Standard A-B USB which came with the inverter. Have been using it without issue with another logging program (https://github.com/opengd/solar-sis) but that one only measures one side of my inverter / charge controller.

DMIINC, 2020-06-28:
What is your setup? Are you om Pi?

cowpen, 2020-06-28:
Yes. 3B+

DMIINC, 2020-06-28:
Well you are....... How busy it is?

DMIINC, 2020-06-28:
Sometimes due to load it will not go through with command

DMIINC, 2020-06-28:
I noticed that when I tried to run everything on Pi. influxdb, telegraf, grafana, mpp-solar

cowpen, 2020-06-28:
Pretty idle:
[image]

DMIINC, 2020-06-28:
Yup.......

DMIINC, 2020-06-28:
Other thing that I noticed is that MPP does not have very good serial to USB adapter inside inverter. I switched to USB to serial adapters on Pi and no issues.

cowpen, 2020-06-28:
Good idea. I have a USB to serial adapter and a cable I made for the LV5048. Just never used it before. I'll try that today instead.

If you're running the mpp-solar service instead of cron, would you be willing to post your /etc/mpp-solar/mpp-solar.conf?

DMIINC, 2020-06-28:
Sure but I have three inverters. Comment out as necessary . Just a sec.

DMIINC, 2020-06-28:
[SETUP]
# Number of seconds to pause between command execution loop
# 0 is no pause, greater than 60 will cause service restarts
pause=0
mqtt_broker=127.0.0.1

# This example would work on a single LV5048
[Inverter1_L1]
model=LV5048
port=/dev/ttyUSB2
baud=2400
command=QPIGS
tag=Inverter1
format=influx2

[Inverter1_L2]
model=LV5048
port=/dev/ttyUSB2
baud=2400
command=QPIGS2
tag=Inverter1
format=influx2

[Inverter2_L1]
model=LV5048
port=/dev/ttyUSB1
baud=2400
command=QPIGS
tag=Inverter2
format=influx2

[Inverter2_L2]
model=LV5048
port=/dev/ttyUSB1
baud=2400
command=QPIGS2
tag=Inverter2
format=influx2

[Inverter3_L1]
model=LV5048
port=/dev/ttyUSB0
baud=2400
command=QPIGS
tag=Inverter3
format=influx2

[Inverter3_L2]
model=LV5048
port=/dev/ttyUSB0
baud=2400
command=QPIGS2
tag=Inverter3
format=influx2

cowpen, 2020-06-28:
Thank you. Very helpful.
Unfortunately, I'm not able to start the service. I suspect this is related to the issue where I'm unable to run mpp-info-pub. Still not sure what's wrong with that. 
[image]

DMIINC, 2020-06-28:
Have you installed all of the prerequisites for mpp-solar service?

DMIINC, 2020-06-28:
Follow this guide from step 3 or 4

https://github.com/jblance/mpp-solar/blob/master/docs/ubuntu_install.md

cowpen, 2020-06-28:
OK. I used the apt-get method described here: https://github.com/jblance/mpp-solar/blob/master/daemon/README.md
Maybe I need to pip install instead?

cowpen, 2020-06-28:
Well that didn't work:
[image]

DMIINC, 2020-06-28:
Might have to... I know @jblance started removing support for Python2 some time ago. i am not skillful enough to know if anything needs to be changed updated for getting this to work properly or something to that effect. I am sure he will chime in...

jblance, 2020-06-28:
> I've looked for any other processes or services that might be interfering with /dev/hidraw0 and found none. But it seems like the NAK and invalid byte responses are only intermittent. Might not be a problem.
> 
> Something new that popped up while configuring MQTT... unable to run mpp-info-pub:
> [image]

You need the 'paho-mqtt' module - which is installed with ```pip install paho-mqtt``` or ```pip3 install paho-mqtt``` (if you have python2 as default

jblance, 2020-06-28:
> Well that didn't work:
> [image]

try with sudo, also it seems the pip packages are different to those in the apt repo - I use apt to install the systemd package

cowpen, 2020-06-28:
Thank you! I also installed systemd using apt and it says I'm good:
```
pi@powerpi:~/mpp-solar $ sudo apt-get install python3-systemd
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-systemd is already the newest version (234-2+b1).
0 upgraded, 0 newly installed, 0 to remove and 200 not upgraded.
```
I ran `sudo pip3 install paho-mqtt` and the error related to that is gone. mpp-info-pub apparently defaults to ttyUSB0. Can I specify hidraw0 from the command line?

```
pi@powerpi:~/mpp-solar $ mpp-info-pub -q localhost -c QID
WARNING:MPP-Solar:Serial read error: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
Traceback (most recent call last):
  File "/usr/local/bin/mpp-info-pub", line 11, in <module>
    load_entry_point('mpp-solar==0.4.5', 'console_scripts', 'mpp-info-pub')()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/mpp_info_pub.py", line 37, in main
    serial_number = mp.getSerialNumber()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/mpputils.py", line 52, in getSerialNumber
    return self.inverter.getSerialNumber()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.5-py3.7.egg/mppsolar/mppinverter.py", line 172, in getSerialNumber
    self._serial_number = response["serial_number"][0]
KeyError: 'serial_number'
```

jblance, 2020-06-28:
Yep with -d 
```
 mpp-info-pub -h
usage: mpp-info-pub [-h] [-s] [-t] [-c COMMAND] [-d DEVICE] [-M MODEL]
                    [-b BAUD] [-q BROKER] [-i] [-I] [--tag TAG]

MPP Solar Inverter Info Utility

optional arguments:
  -h, --help            show this help message and exit
  -s, --grabsettings    Also get the inverter settings
  -t, --getstatus       Use the getstatus "helper"
  -c COMMAND, --command COMMAND
                        Command to execute [comma separated]
  -d DEVICE, --device DEVICE
                        Serial device(s) to communicate with [comma separated]
  -M MODEL, --model MODEL
                        Specifies the inverter model to select commands for,
                        defaults to "standard", currently supports LV5048
  -b BAUD, --baud BAUD  Baud rate for serial communications
  -q BROKER, --broker BROKER
                        MQTT Broker hostname
  -i, --influx          Use Influx Line Protocol for the messgae format
  -I, --influx2         Use Influx Line Protocol II for the messgae format
  --tag TAG             Influx tag to use for all commands in this execution
```

cowpen, 2020-06-28:
Much appreciated! 

All errors are gone now. But running `mpp-info-pub -q localhost -d /dev/hidraw0 -c QPIG`S just drops me back to the command line. Nothing reported to STDOUT. Should I see something?

cowpen, 2020-06-28:
OK - nevermind previous. I set up the listener `mosquitto_sub -h localhost -v -t "#"` and reran the commend. Output showed up! I think I'm on my way now. Thanks again for your help.

jblance, 2020-06-28:
great news

closed 2020-06-28 by jblance

cowpen, 2020-06-28:
I was prematurely optimistic. Unable to start mpp-solar service. Here is my mpp-solar.conf:

```
pi@powerpi:~ $ more /etc/mpp-solar/mpp-solar.conf
[SETUP]                 # Required section
pause=5                 # Number of seconds to pause at the end of each loop (0 is no pause)
mqtt_broker=127.0.0.1    # Hostname / IP address of the MQTT broker

[Inverter1_L1]      # To combine 2 commands for Influx math define multiple sections with the same tag
model=LV5048
port=/dev/hidraw0
baud=2400
command=QPIGS
tag=Inverter1
format=influx2

[Inverter1_L2]      # To combine 2 commands for Influx math define multiple sections with the same tag
model=LV5048
port=/dev/hidraw0
baud=2400
command=QPIGS2
tag=Inverter1
format=influx2
```
I tried to start service `systemctl --user start mpp-solar` but it fails. Watching /var/log/syslog yields the attached.
[tail_f_syslog_mpp-solar_service.txt](https://github.com/jblance/mpp-solar/files/4843146/tail_f_syslog_mpp-solar_service.txt)

jblance, 2020-06-28:
the service doesnt like the inline comments, try
```
[SETUP]
pause=5
mqtt_broker=127.0.0.1

[Inverter1_L1]
model=LV5048
port=/dev/hidraw0
baud=2400
command=QPIGS
tag=Inverter1
format=influx2

[Inverter1_L2]
model=LV5048
port=/dev/hidraw0
baud=2400
command=QPIGS2
tag=Inverter1
format=influx2
```

cowpen, 2020-06-29:
Thanks. That did the trick. Service started.

cowpen, 2020-06-29:
Pretty sure I did something wrong. Also, it takes about 20 seconds after issuing the influx command before I get the ">" prompt.
:

```
pi@powerpi:~ $ influx
use mppsolar
Connected to http://localhost:8086 version 1.7.10
InfluxDB shell version: 1.7.10
> use mppsolar
Using database mppsolar
> show measurements
name: measurements
name
----
cpu
disk
diskio
kernel
mem
mpp-solar
processes
swap
system

jblance, 2020-06-29:
it can be a bit slow - raspberry pi is a little small for a DB server
the service also sends a fair bit of information and quite regularly, it is possible to overload the DB, especially if everything is all on one pi

cowpen, 2020-06-29:
Definitely almost there. I can see there's actual data using Chronograph (below). But I can't find the db in Grafana.

[image]

cowpen, 2020-06-29:
Added data source. Cocking with gas! 
Thanks again for your help. Very glad to be finally logging both sides of my inverter. Much appreciated!