jblance/mpp-solar #33: Voltronic InfiniSolar E 5.5KW

eliafino, 2020-07-10:
Hi, thanks for this project!

I receive only NAK response... which setting for "-m"?

Regards,
Elia

jblance, 2020-07-10:
What command are you running? 
Please supply the output after adding '-D' to the command (this provides a lot of debug messages)
Also can you run the 'QPI' command and let me know the result

eliafino, 2020-07-11:
This is Debug for QPI

[log.txt](https://github.com/jblance/mpp-solar/files/4906260/log.txt)

jblance, 2020-07-11:
Are you saying your working good gives a good response?  (As there is not crc in the command you send) what response do you get? What do you get if you send QPI instead of QPIGS?

eliafino, 2020-07-11:
With QPI response is ['PI16']

eliafino, 2020-07-11:
I found info on the PI15 / PI16 protocol [here](https://github.com/zykh/nut/blob/nutdrv_qx_voltronic-sunny_rebased%2Bcommand/drivers/nutdrv_qx_voltronic-sunny.c)

jblance, 2020-07-11:
Ok, so a new protocol.
The master node base doesnt deal with alternative protocols very well, but I have started a branch to specifically deal with different protocols (branch is called refactor)

Next week I'll try adding pi16 protocol handler
Do you know how to checkout a branch and use that code?
What do you need that the linked tool doesnt provide?

Cheers

eliafino, 2020-07-12:
I see branch refactor, ok.
The links I sent are part of a software that manages UPS, I reported it only to understand the protocol. I haven't found anything else.

jblance, 2020-07-20:
I have added the first cut at PI16 (QPIGS and QPI only so far)
Can you try it against your inverter to see if it gives the correct response

eliafino, 2020-07-20:
root@raspberrypi:~# mpp-solar -c QPIGS -P PI16 -p /dev/hidraw0
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.6.1', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.6.1-py3.7.egg/mppsolar/__init__.py", line 145, in main
    results = device.run_command(command=args.command, show_raw=args.showraw)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.6.1-py3.7.egg/mppsolar/devices/mppsolar.py", line 40, in run_command
    response = self._port.send_and_receive(command, show_raw, self._protocol)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.6.1-py3.7.egg/mppsolar/io/hidrawio.py", line 71, in send_and_receive
    decoded_response = protocol.decode(response_line)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.6.1-py3.7.egg/mppsolar/protocols/protocol.py", line 106, in decode
    result = result.decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x82 in position 10: invalid start byte

eliafino, 2020-07-20:
root@raspberrypi:~# mpp-solar -R -c QPIGS -P PI16 -p /dev/hidraw0
Parameter                       Value           Unit
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.6.1', 'console_scripts', 'mpp-solar')()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.6.1-py3.7.egg/mppsolar/__init__.py", line 154, in main
    op.output(data=results, tag=tag, mqtt_broker=args.mqttbroker)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.6.1-py3.7.egg/mppsolar/outputs/screen.py", line 18, in output
    print(f'{key:<30}\t{value:<15}\t{unit:<4}')
TypeError: unsupported format string passed to bytes.__format__

jblance, 2020-07-20:
Can you post the full debug response (add ```-D```) to the command

eliafino, 2020-07-20:
[debug.txt](https://github.com/jblance/mpp-solar/files/4947019/debug.txt)

jblance, 2020-07-20:
Weird it seems null padded at the start and there IS a crc on the return.
Can you also do a debug of QPI command so I can see what that gives

eliafino, 2020-07-20:
[debugQPIGS.txt](https://github.com/jblance/mpp-solar/files/4947073/debugQPIGS.txt)
[debugQPI.txt](https://github.com/jblance/mpp-solar/files/4947074/debugQPI.txt)

jblance, 2020-07-20:
OK they both have crc, I wonder if all responses do. Maybe set commands responses have no crc?
No nulls at the start of these 2 responses though....
I'll update to remove the crc of  the responses tomorrow and try to add some more commands

eliafino, 2020-07-20:
Ok, thank you very much!

jblance, 2020-07-21:
made some changes - so hopefully the response is correctly parsed now
also added QPIBI command
can you try with version 0.6.2

jblance, 2020-07-21:
also let me know which commands you need the most (start with queries)

eliafino, 2020-07-21:
root@raspberrypi:/usr/src/mpp-solar# mpp-solar -c QPIGS -P PI16 -p /dev/hidraw0
Parameter                       Value           Unit
grid_voltage                    221.8           V
output_power                    0               W
grid_frequency                  50.0            Hz
output_current                  1.8             A
ac_output_voltage_r             233.4           V
ac_output_power_r               349             W
ac_output_frequency             50.1            Hz
ac_output_current_r             1.4             A
output_load_percent             7               %
pbus_voltage                    409.1           V
sbus_voltage                    409.1           V
positive_battery_voltage        56.7            V
negative_battery_voltage        0.0             V
battery_capacity                93              %
pv1_input_power                 0               W
pv2_input_power                 0               W
pv3_input_power                 0               W
pv1_input_voltage               0.0             V
pv2_input_voltage               0.0             V
pv3_input_voltage               0.0             V
max_temperature                 33.0            °C
status_todo                     D---110001

Perfect my friend!

eliafino, 2020-07-21:
If you can, I need QMOD and QED

jblance, 2020-07-21:
Great, also next will be testing the output modules 
What are you going to do with the data?

eliafino, 2020-07-21:
Internet page, MySQL and grafana

jblance, 2020-07-21:
how do you plan to get it to those places? mqtt? direct?
QMOD added

eliafino, 2020-07-21:
Directly to my MySQL server. Thanks for QMOD

jblance, 2020-07-21:
attempt at QED....

jblance, 2020-07-21:
so i have the idea of output modules.
if you have python code for the mysql updates, it is possible that an output module could be created (probably with a config file somewhere for the specific details etc)
then the command becomes 
``` mpp-solar -p /dev/hidraw0 -P PI16 -c QED19900101 -o mysql```
or similar

eliafino, 2020-07-21:
Interesting... but for now QED -> "NAK", no way to have production for today: QED20200721

Not finished jet, [here](http://www.eliafino.it:3000/d/solare/impianto-fotovoltaico-ibrido-eliafino?from=now%2Fd&to=now&orgId=1&refresh=5s&kiosk=tv)

jblance, 2020-07-21:
try QED now
not sure 'checksum' is the same as the CRC though, do you have an example that works from your python script?

eliafino, 2020-07-21:
I was unable to obtain the reading with any script, not even sending the command directly to the serial port. Only "NAK"

jblance, 2020-07-21:
can you try QED with version 0.6.4

jblance, 2020-07-21:
and post a debug log if it fails again

jblance, 2020-07-21:
oh, missed padding the checksum, fixed in 0.6.5

eliafino, 2020-07-22:
Now I see "0000000", in the next days I can connect solar panels and try to read production.Thanks very much for now!

jblance, 2020-07-22:
Great

closed 2020-08-04 by jblance