jblance/mpp-solar #34: Obtain individual inverter information in a 3 phase installation

alpujarra, 2020-07-26:
Dear all:

First of all I would like to thank you for the great work done in this repo and for putting this magnificent tool within everyone's reach.

I have an installation of 3 Axpert MKS inverter configured in a 3-phased way. The inverter number 1 is the one who is receiving the solar input and it is where I have connected my raspberry pi. Currently I am obtaing the aggregated information of all three with the QPIGS command, 

I would like to know if it is possible to obtain the individual information of each inverter. I have tried out the QPGS and QP2GS commands but the answer is always "CRITICAL:MPP-Solar:Command not found".

Thank you very much for your help.

Best regards.

Francisco

jblance, 2020-07-26:
There are different protocols for different versions of inverter.
The QP2GS command is in the LV5048 command set (add ```-M L:5048``` to the command to use it) 
Better is to check the protocol with the QPI command.

Does QPGS1 command give results?

alpujarra, 2020-07-27:
Hi John, thank you for  your quick answer.

The answer to the QPI command is the following:
protocol_id                   	PI30 

I obtain also an answer to the QPGS1 command.

fran@Valor:~ $ sudo mpp-solar -d /dev/hidraw0 -c QPGS1
ac_output_active_power        	67              W
ac_output_apparent_power      	115             VA
ac_output_frequency           	49.99           Hz
ac_output_voltage             	230.1           V
battery_capacity              	50              %
battery_charging_current      	0               A
battery_discharge_current     	1               A
battery_voltage               	49.7            V
charger_source_priority       	Solar + Utility 
fault_code                    	No fault        
grid_frequency                	0.0             Hz
grid_voltage                  	0.0             V
is_ac_charging                	0               True - 1/False - 0
is_battery_over_voltage       	0               True - 1/False - 0
is_battery_under_voltage      	0               True - 1/False - 0
is_configuration_changed      	0               True - 1/False - 0
is_line_lost                  	1               True - 1/False - 0
is_load_on                    	1               True - 1/False - 0
is_scc_charging               	0               True - 1/False - 0
is_scc_ok                     	0               True - 1/False - 0
load_percentage               	2               %
max_ac_charger_current        	30              A
max_charger_current           	30              A
max_charger_range             	140             A
output_mode                   	Phase 2 of 3 phase output 
parallel_instance_number      	valid           
pv_input_current              	0               A
pv_input_voltage              	0.0             V
serial_number                 	92931803100929  
total_ac_output_apparent_power	299             VA
total_ac_output_percentage    	1               %
total_charging_current        	15              A
total_output_active_power     	188             W
work_mode                     	Battery Mode 

Thanks a lot for you help.

Francisco

jblance, 2020-07-27:
Good news, in case you haven't already realized it QPGS2 will likely give you the 3rd phase inverter details

alpujarra, 2020-07-31:
Yes, you are right. With QPGS"x" command I am able to obtain the information of each individual inverter using the appropiate value of "x":
0 - For inverter 1 
1- For inverter 2
2 - For inverter 3

However I am afraid that I have not configured in the right way the mpp-solar service to get the statistics of each inverter. The "mpp-solar.conf" file that I am using is the following one:
[SETUP]
pause=5
mqtt_broker=127.0.0.1

[Aggregated_All_Inverter_Info]
model=standard
port=/dev/hidraw0
baud=2400
command=QPIGS
tag=Aggregated
format=influx2

[Inverter1]
model=standard
port=/dev/hidraw0
baud=2400
command=QPIGS0
tag=Inverter1
format=influx2

[Inverter2]
model=standard
port=/dev/hidraw0
baud=2400
command=QPIGS1
tag=Inverter2
format=influx2

[Inverter3]
model=standard
port=/dev/hidraw0
baud=2400
command=QPIGS2
tag=Inverter3
format=influx2

In Graphana I am able to see the different tags "Inverter" but with no data.

Can you help me to identify what I am doing wrong?

Thank you very much.

Francisco

jblance, 2020-07-31:
Well first off the inverter1 - 3 commands are incorrect (you have QPIGS1 instead of QPGS1)
Other problems are difficult to see from just the conf file
- what is the service status results?
- try subscribing to the mqtt broker in a command window and see what is getting published 
- what is telegraf service status
- what is in the influxdb

alpujarra, 2020-07-31:
Again you are right. The commands were wrong. Sorry for that.

I have review the points you mention in your previous answer. Here are the results.

The mpp-solar service is ok and working without errors.
I have subscribed to the mqtt broker using the following command 
       sudo mosquitto_sub -h localhost -v -t "mpp-solar"
and I have only seen message from the "Aggregated" command but not for the other ones (Inverter1, Inverter2 and Inverter3). I suppose that this is the reason because I didn't see any data in Graphana related to Inveter commands but I don't know how to fix it.

The telegraph service is also running and there is only a warning message "Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable."

The influx db is also working.

I hope I provide you enough information to help me.

Thanks again for your help.

Francisco

jblance, 2020-08-01:
Please provide the logs from the mpp-solar service

jblance, 2020-08-01:
Also restart the service and supply the startup logs

alpujarra, 2020-08-01:
I attached the log file. 
[mppsolar.log](https://github.com/jblance/mpp-solar/files/5009613/mppsolar.log)

I have also restarted de service:
fran@Valor:~ $ systemctl --user restart mpp-solar
fran@Valor:~ $ systemctl --user status  mpp-solar
● mpp-solar.service - MPP Solar Service
   Loaded: loaded (/etc/systemd/user/mpp-solar.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-08-01 03:29:13 CEST; 13s ago
 Main PID: 902 (python3)
   CGroup: /user.slice/user-1001.slice/user@1001.service/mpp-solar.service
           └─902 /usr/bin/python3 /usr/local/bin/mpp-solar-service -c /etc/mpp-solar/mpp-solar.conf

I attach also the logs after restarting the process.
[mppsolar_restarted.log](https://github.com/jblance/mpp-solar/files/5009648/mppsolar_restarted.log)

Thanks for your help.

Francisco

jblance, 2020-08-01:
Can you re post the configuration file
Does the mppsolar command on the command line work without sudo?

alpujarra, 2020-08-01:
Hi again. Here is the configuration file:

[mpp-solar.conf.txt](https://github.com/jblance/mpp-solar/files/5010213/mpp-solar.conf.txt)

The mppsolar command does not work without sudo in the command line.

Don't heasitate to ask me any other information.

Have a nice day.

Francisco

jblance, 2020-08-01:
Config still has the wrong commands in it
You need to either get the mppsolar command working without sudo (as in the docs somewhere) or to move the service to be a system service (not in the docs but covered in the original source which is linked in the docs)

(Sorry on my phone so can't easily be more specifc)

alpujarra, 2020-08-04:
Hi John, the problem is already solved. I updated the commands in the config file. In addition I have solved also the problem with the execution with sudo. Now all is working.

Thank you so much for your help. 

Francisco

jblance, 2020-08-04:
Great news
Glad it works for you

Cheers
John

closed 2020-08-04 by jblance