jblance/mpp-solar #36: Communication with Pylontech batteries

alpujarra, 2020-08-04:
Hi everyone:

I would like to ask you if anyone can recommend me a repository like this or any other information for connecting my Raspberry to my Pylontech US2000 Plus batteries.

I would like to improve my installation obtaining a better estimation of the State of Charge of the batteries compared with the one performed by my Axpert 5K inverters. 

Thank you very much in advance. 

Francisco

jblance, 2020-08-04:
I dont have any information regarding these batteries.
Do they have a connected monitoring or management device?
How do you connect now?

(as an example, another user uses the JKBMS to monitor his batteries via an app and bluetooth - we worked together to get that functionality to work from python)

jblance, 2020-08-04:
It looks like there is a CAN interface, so you'd need:
1. a CAN module for the Pi (eg https://www.beyondlogic.org/adding-can-controller-area-network-to-the-raspberry-pi/)
2. the correct cable 
3. some software that understands whatever commands the batteries support
The mpp-solar could be extended to do point 3 if the command set was known

jblance, 2020-08-04:
or via the serial console (would still to know commands etc)

jblance, 2020-08-04:
Also see https://powerforum.co.za/topic/2408-pylontech-us2000b-how-to-read-data-from-them/
and/or https://www.photovoltaikforum.com/thread/118958-pylontech-us2000b-daten-%C3%BCber-konsole-rs232-auslesen/?t=118958&start=10

alpujarra, 2020-08-24:
Hi John:

I have succeded in communicating with my Pylontech US2000B batteries and I am able to obtain the SoC, Voltage, Current and Temperature.

I am using the console serial port. I found a github repo with a implementation of this protocol (at least part of it) in python. The repo is the following: https://github.com/AlexanderPollak/Solar-Control-Program

As you can seen the repo has no updates from last year and some weeks ago I raised an issue asking info about the status of the functionality implemented in it. I didn't get any answer. The repo includes the communication part with Pylontech batteries and also with a number of inverters.

However this weekend I have tried the part related to Pylontech batteries and it's works for me. I build myself the communication cable following the indications I've found in the Powerforum forum.

In your previous answer you mention the possibility to extend the mpp-solar with this communication protocol. Perphaps there is the possibility to do it with the repo I mentioned before. What do you think?

Now I an intererested to understand how I am able to provide the SoC obtained from the batteries to my Axpert inverters. I don't know if this is already supported in mpp-solar.

Thank you very much for your help.

Francisco

suniram, 2020-08-28:
Hi Francisco,

You can connect to the Pylontech console port via RS232. Baud rate 115200. 

I have found this great thread with most of the info you require. https://powerforum.co.za/topic/2322-youdas-off-grid-lab/page/4/

alpujarra, 2020-09-11:
Hi Surinam:

Thank you for your comment. As I mention in my last post I made the communication cable and I succeed in communicating with the Pylontech batteries using the console port via RS232 using the code available in the Alexander Pollack repo (https://github.com/AlexanderPollak/Solar-Control-Program).

@jblance , as I mentioned in my last post now I an intererested to understand how I am able to provide the SoC obtained from the batteries to my Axpert inverters. I don't know if this is already supported in mpp-solar.

In addition, I get in contact with Allexader Pollack, the owner of the repo I mentioned before. I think it could be great if we can add this Pylontech module to the mpp-solar.

What do you think.

Best.

Francisco

jblance, 2020-09-24:
@alpujarra does your inverter have a way of 'telling it' the SoC?

I can add the Pylontech command set, it will need to wait until i merge the development branch back into the master codebase though

alpujarra, 2020-09-24:
Hi John:

I really don't know. I have three Axpert MKS inverters. I suppose that it is possible to tell this inverters what is the SoC because this is what other commercial software like iccsoftware does (https://iccsoftware.co.za/) but I don't know how to do it. I hope that this could be done by mpp-solar.

Best.

Francisco

jblance, 2020-10-20:
I dont think the linked software 'tells' the inverter the SoC, but rather displays it for you.
From the features list
* Support for Victron Energy BMV battery monitor, used for true state of charge (SOC) monitoring. Monitor Cycles and temperatures.
* Support for Pylontech, Hubble , Revov and Narada, used for true state of charge (SOC) monitoring.

closed 2020-11-26 by jblance