jblance/mpp-solar #39: What I´m doing wrong while Installaion

mastadook, 2020-10-02:
Hi,

I zhink this is also no direct Issue but I can´t install it on Raspi:

`pi@homecontrolx:~ $ sudo python /home/pi/mpp-solar/setup.py install
running install
running bdist_egg
running egg_info
creating mpp_solar.egg-info
writing requirements to mpp_solar.egg-info/requires.txt
writing mpp_solar.egg-info/PKG-INFO
writing top-level names to mpp_solar.egg-info/top_level.txt
writing dependency_links to mpp_solar.egg-info/dependency_links.txt
writing entry points to mpp_solar.egg-info/entry_points.txt
writing manifest file 'mpp_solar.egg-info/SOURCES.txt'
reading manifest file 'mpp_solar.egg-info/SOURCES.txt'
writing manifest file 'mpp_solar.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
warning: install_lib: 'build/lib.linux-armv7l-2.7' does not exist -- no Python modules to install

installing package data to build/bdist.linux-armv7l/egg
running install_data
creating build
creating build/bdist.linux-armv7l
creating build/bdist.linux-armv7l/egg
error: can't copy 'daemon/mpp-solar.service': doesn't exist or not a regular file`

Can someone help here?

Regards
Clemens

jblance, 2020-10-02:
You need to change into the directory that has the setup.py file in it.

mastadook, 2020-10-02:
thank you, it´s now installed but shows Errors:

pi@homecontrolx:~/mpp-solar $ mpp-solar -h
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar==0.4.6', 'console_scripts', 'mpp-solar')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.6-py2.7.egg/mppsolar/__init__.py", line 7, in <module>
    from .mpputils import mppUtils
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.6-py2.7.egg/mppsolar/mpputils.py", line 8, in <module>
    from .mppinverter import mppInverter
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.6-py2.7.egg/mppsolar/mppinverter.py", line 18, in <module>
    from .mppcommand import mppCommand
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.6-py2.7.egg/mppsolar/mppcommand.py", line 431
    resp_format = ['string', f'Unknown value in byte_response {i}', '']
                                                                 ^
SyntaxError: invalid syntax

Does the Install looks fine?

`pi@homecontrolx:/mpp-solar $ sudo python ./setup.py install
running install
running bdist_egg
running egg_info
creating mpp_solar.egg-info
writing requirements to mpp_solar.egg-info/requires.txt
writing mpp_solar.egg-info/PKG-INFO
writing top-level names to mpp_solar.egg-info/top_level.txt
writing dependency_links to mpp_solar.egg-info/dependency_links.txt
writing entry points to mpp_solar.egg-info/entry_points.txt
writing manifest file 'mpp_solar.egg-info/SOURCES.txt'
reading manifest file 'mpp_solar.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'mpp_solar.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-2.7
creating build/lib.linux-armv7l-2.7/mppsolar
copying mppsolar/mpputils.py -> build/lib.linux-armv7l-2.7/mppsolar
copying mppsolar/mppcommand.py -> build/lib.linux-armv7l-2.7/mppsolar
copying mppsolar/mpp_solar_service.py -> build/lib.linux-armv7l-2.7/mppsolar
copying mppsolar/mpp_info_pub.py -> build/lib.linux-armv7l-2.7/mppsolar
copying mppsolar/version.py -> build/lib.linux-armv7l-2.7/mppsolar
copying mppsolar/mppinverter.py -> build/lib.linux-armv7l-2.7/mppsolar
copying mppsolar/__init__.py -> build/lib.linux-armv7l-2.7/mppsolar
creating build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/_pi18_et.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/_pi18_gs.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/_pi18_mod.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/_pi18_pi.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/f.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/mchgc.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/mnchgc.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/muchgc.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/pbcv.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/pbdv.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/pbft.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/pbt.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/pcp.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/pcvv.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
copying mppsolar/commands/pepd.json -> build/lib.linux-armv7l-2.7/mppsolar/commands
[... 122 lines cut ...]
                                                                 ^
SyntaxError: invalid syntax

Adding mpp-solar 0.4.6 to easy-install.pth file
Installing mpp-solar script to /usr/local/bin
Installing mpp-info-pub script to /usr/local/bin
Installing mpp-solar-service script to /usr/local/bin

Installed /usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.6-py2.7.egg
Processing dependencies for mpp-solar==0.4.6
Searching for pyserial
Reading https://pypi.org/simple/pyserial/
Downloading https://files.pythonhosted.org/packages/fb/44/83f81048acfc32265a1a027f2297e9c784bd7ad0dd9d5e766eb0a1ae05e8/pyserial-3.5b0-py2.py3-none-any.whl#sha256=82db8a0a384fe8d401e35be48d6fb0316b536218defd8acdfaa5ad5086da312a
Best match: pyserial 3.5b0
Processing pyserial-3.5b0-py2.py3-none-any.whl
Installing pyserial-3.5b0-py2.py3-none-any.whl to /usr/local/lib/python2.7/dist-packages
writing requirements to /usr/local/lib/python2.7/dist-packages/pyserial-3.5b0-py2.7.egg/EGG-INFO/requires.txt
Adding pyserial 3.5b0 to easy-install.pth file
Installing pyserial-miniterm script to /usr/local/bin
Installing pyserial-ports script to /usr/local/bin

Installed /usr/local/lib/python2.7/dist-packages/pyserial-3.5b0-py2.7.egg
Finished processing dependencies for mpp-solar==0.4.6`

Thank you

jblance, 2020-10-02:
Only python3 is supported by mppsolar now.
So if you are using a system that defaults to python2 you will need to use python3 in the commands (including install)

mastadook, 2020-10-07:
thank you for help, this was successfull now :)

jblance, 2020-10-20:
good to hear

closed 2020-10-20 by jblance