jblance/mpp-solar #40: trying to use with pip 2424msxe

ninlee, 2020-10-17:
can you tell me why i am getting this error and how to fix it.

after i run this command 
mpp-info-pub -q localhost -c QID

i get this output
pi@grafpi ~ $ mpp-info-pub -q localhost -c QID
Traceback (most recent call last):
  File "/usr/local/bin/mpp-info-pub", line 9, in <module>
    load_entry_point('mpp-solar==0.4.6', 'console_scripts', 'mpp-info-pub')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.6-py2.7.egg/mppsolar/__init__.py", line 7, in <module>
    from .mpputils import mppUtils
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.6-py2.7.egg/mppsolar/mpputils.py", line 8, in <module>
    from .mppinverter import mppInverter
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.6-py2.7.egg/mppsolar/mppinverter.py", line 18, in <module>
    from .mppcommand import mppCommand
  File "/usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.6-py2.7.egg/mppsolar/mppcommand.py", line 431
    resp_format = ['string', f'Unknown value in byte_response {i}', '']
                                                                 ^
SyntaxError: invalid syntax

thanks for the help in advance.

jblance, 2020-10-17:
Python2 isn't supported anymore, so you'll need to use python3 to install and run 

Also its best to start with the command line tool and making sure it works before moving the the publishing one

ninlee, 2020-10-20:
hi again,

i am new to this and a dont know how to go about using python 3

jblance, 2020-10-20:
On the pi you need to use the 'python3' command instead of 'python' 

Do this for the install

Also make sure the python2 install is removed, eg
```sudo rm -rf /usr/local/lib/python2.7/dist-packages/mpp_solar-0.4.6-py2.7.egg/```

ninlee, 2020-10-20:
i am getting this error now 

Traceback (most recent call last):
  File "/usr/local/bin/mpp-info-pub", line 11, in <module>
    load_entry_point('mpp-solar==0.4.6', 'console_scripts', 'mpp-info-pub')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/mpp_info_pub.py", line 12, in <module>
    import paho.mqtt.publish as publish
ModuleNotFoundError: No module named 'paho'

jblance, 2020-10-20:
You are now missing the paho python module (for the python3 install)
First install pip3
```sudo apt-get install python3-pip```
Then install the python3 version of paho
```sudo pip3 install paho-mqtt```

Then retry the command - again I recommend troubleshooting with the ```mpp-solar``` command first

ninlee, 2020-10-20:
looks like i am getting closer.

i am connected to the inverter via usb cable. i see u talk about adding -d /dev/hidraw0 to the commands. how am i to do this?

WARNING:MPP-Solar:Serial read error: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
Traceback (most recent call last):
  File "/usr/local/bin/mpp-info-pub", line 11, in <module>
    load_entry_point('mpp-solar==0.4.6', 'console_scripts', 'mpp-info-pub')()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/mpp_info_pub.py", line 37, in main
    serial_number = mp.getSerialNumber()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/mpputils.py", line 52, in getSerialNumber
    return self.inverter.getSerialNumber()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/mppinverter.py", line 172, in getSerialNumber
    self._serial_number = response["serial_number"][0]
KeyError: 'serial_number'

jblance, 2020-10-20:
Start with
```sudo mpp-solar -d /dev/hidraw0```

ninlee, 2020-10-20:
the serial number showed up
 what now 👍

jblance, 2020-10-20:
what are you trying to do?
A good next check is
```sudo mpp-solar -d /dev/hidraw0 -c QPI```

ninlee, 2020-10-20:
it came back with 
protocol_id                     PI30

i am trying to setup the grafana stuff to log and control my Pip

i got some stats too 
================ Status ==================
Parameter                       Value           Unit
ac_input_frequency              59.9            Hz
ac_input_voltage                239.0           V
ac_output_active_power          708             W
ac_output_apparent_power        788             VA
ac_output_frequency             59.9            Hz
ac_output_load                  29              %
ac_output_voltage               239.0           V
battery_capacity                100             %
battery_charging_current        0               A
battery_discharge_current       0               A
battery_voltage                 28.5            V
battery_voltage_from_scc        0.0             V
bus_voltage                     398             V
inverter_heat_sink_temperature  43              Deg_C
is_ac_charging_on               0               True - 1/False - 0
is_battery_voltage_to_steady_while_charging     0               True - 1/False - 0
is_charging_on                  0               True - 1/False - 0
is_charging_to_float            0               True - 1/False - 0
is_configuration_changed        0               True - 1/False - 0
is_load_on                      1               True - 1/False - 0
is_reserved                     0               True - 1/False - 0
is_sbu_priority_version_added   0               True - 1/False - 0
is_scc_charging_on              0               True - 1/False - 0
is_scc_firmware_updated         0               True - 1/False - 0
is_switched_on                  1               True - 1/False - 0
pv_input_current_for_battery    0.0             A
pv_input_power                  0               W
pv_input_voltage                0.0             V
rsv1                            0               A
rsv2                            0               A

ninlee, 2020-10-20:
how so i change the command default from using /dev/ttyUSB0 to -d /dev/hidraw0?

jblance, 2020-10-20:
> how so i change the command default from using /dev/ttyUSB0 to -d /dev/hidraw0?

you dont

if you want to publish via mqtt - the mpp-info-pub script will help, 
```sudo mpp-info-pub -h``` 
will give you the available options - it also accepts the -d option so you can run it with  -d /dev/hidraw0

eg ```sudo mpp-info-pub -d /dev/hidraw0 -q localhost -t``` will publish the status to a mqtt broker running on the localhost

ninlee, 2020-10-22:
hi again 

when i run show measurements this is my output

> use mppsolar
Using database mppsolar
> show measurements
name: measurements
name
----
cpu
disk
diskio
kernel
mem
processes
swap
system
> 

is that correct? what am i doing wrong? i am not seeing the mppsolar database in grafana also

thanks

jblance, 2020-10-22:
You'll have to give a lot more detail.
How are you getting the info from mppsolar to influx?
What troubleshooting steps have you done and what are the results?
Mppsolar just talks to the inverter and gets results. There is a helper that posts results to a mqtt broker. From there on is other software.
I'm guessing you are using telegraf to read from mqtt and write into influx. If so, have you checked the mqtt messages? What about the telegraf logs and service status?

ninlee, 2020-10-22:
hi am vrey new to this and most of what you just said there just went over my head. i followed the instructions you have on github. i did everything from installing and test MQTT, Schedule Regular Queries (and MQTT message posts) using

 * * * * * /usr/local/bin/mpp-info-pub -q localhost -c QPIGS -i > /home/pi/cron.out 2>&1
installing influx,grafana and telegraf and configering with the information  you have on github.

i am at the point where you say to check what is stored in influx.

telegraf service is started, not sure how to check the logs

thanks

ninlee, 2020-10-22:
elegraf.service - The plugin-driven server agent for reporting metrics into InfluxDB
   Loaded: loaded (/lib/systemd/system/telegraf.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-10-22 04:53:38 EDT; 19min ago
     Docs: https://github.com/influxdata/telegraf
 Main PID: 445 (telegraf)
    Tasks: 12 (limit: 2065)
   CGroup: /system.slice/telegraf.service
           └─445 /usr/bin/telegraf -config /etc/telegraf/telegraf.conf -config-directory /etc/tele

Oct 22 04:53:38 raspberrypi systemd[1]: Started The plugin-driven server agent for reporting metri
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Starting Telegraf 1.16.0
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Loaded inputs: cpu disk diskio 
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Loaded aggregators:
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Loaded processors:
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Loaded outputs: influxdb
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Tags enabled: host=raspberrypi
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! [agent] Config: Interval:10s, Q
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! [inputs.mqtt_consumer] Connecte

not sure if that is the log

ninlee, 2020-10-22:
Oct 22 04:53:38 raspberrypi systemd[1]: Started The plugin-driven server agent for reporting metrics into InfluxDB.
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Starting Telegraf 1.16.0
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Loaded inputs: cpu disk diskio kernel mem mqtt_consumer processes swap system
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Loaded aggregators:
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Loaded processors:
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Loaded outputs: influxdb
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! Tags enabled: host=raspberrypi
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! [agent] Config: Interval:10s, Quiet:false, Hostname:"raspberrypi", Flush Interval:10s
Oct 22 04:53:46 raspberrypi telegraf[445]: 2020-10-22T08:53:46Z I! [inputs.mqtt_consumer] Connected [tcp://127.0.0.1:1883]

jblance, 2020-10-23:
can you post the output of:
```egrep -v "#|^$" /etc/telegraf/telegraf.conf```
```ls /etc/telegraf/telegraf.d/```
```cat /etc/telegraf/telegraf.d/*.conf```

closed 2021-01-10 by jblance

santosh-mahadev, 2021-03-26:
I am also using MPP SOLAR PIP2424 MSE inverter.
I connected using usb cable and I am also facing similar problems
[image]

jblance, 2021-03-26:
That looks like it is getting the correct responses.

If you are having problems further along (ie trying to get info into influx or home assistant etc) then you'll need to provide more details on what you are trying to do.