jblance/mpp-solar #41: communication not working

jec0408, 2020-11-23:
Hi 

I´m having some problems with communicating with the inverter 
its connected Raspberry pi3+ direct usb inverter. I was hoping if you could help me..

 
usb 1-1.5: new low-speed USB device number 6 using dwc_otg
[  196.244185] usb 1-1.5: New USB device found, idVendor=0665, idProduct=5161, bcdDevice= 0.02
[  196.244196] usb 1-1.5: New USB device strings: Mfr=3, Product=1, SerialNumber=0
[  196.256988] hid-generic 0003:0665:5161.0002: hiddev96,hidraw0: USB HID v1.11 Device [HID 0665:5161] on usb-3f980000.usb-1.5/input0

pi@raspberrypi:~ $ lsusb
Bus 001 Device 006: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 001 Device 005: ID 058f:6387 Alcor Micro Corp. Flash Drive
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Check service status systemctl --user status mpp-solar

mpp-solar.service - MPP Solar Service
   Loaded: loaded (/etc/systemd/user/mpp-solar.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-11-23 12:17:33 CET; 1min 4s ago
 Main PID: 586 (python3)
   CGroup: /user.slice/user-1000.slice/user@1000.service/mpp-solar.service
           └─586 /usr/bin/python3 /usr/local/bin/mpp-solar-service -c /etc/mpp-solar/mpp-solar.conf

nov 23 12:17:23 raspberrypi systemd[511]: Starting MPP Solar Service...
nov 23 12:17:33 raspberrypi python3[586]: MPP-Solar-Service: Initializing ...
nov 23 12:17:33 raspberrypi python3[586]: MPP-Solar-Service: Config file: /etc/mpp-solar/mpp-solar.conf
nov 23 12:17:33 raspberrypi python3[586]: MPP-Solar-Service: Config setting - pause: 5
nov 23 12:17:33 raspberrypi python3[586]: MPP-Solar-Service: Config setting - mqtt_broker: localhost
nov 23 12:17:33 raspberrypi python3[586]: MPP-Solar-Service: Config setting - command sections found: 1
nov 23 12:17:33 raspberrypi python3[586]: MPP-Solar-Service: item {'mp': <mppsolar.mpputils.mppUtils object at 0x75c56770>, 'command': 'QPGS', 'format': '
nov 23 12:17:33 raspberrypi systemd[511]: Started MPP Solar Service.
nov 23 12:17:33 raspberrypi python3[586]: CRITICAL:MPP-Solar:Command not found

pi@raspberrypi:~ $ sudo mpp-solar -M LV5161 -d /dev/ttyUSB0 -c QPIGS
WARNING:MPP-Solar:Serial read error: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
error                         	No byte_response 
pi@raspberrypi:~ $ 

pi@raspberrypi:~ $ mpp-solar -M LV5161 -d /dev/hidraw0 -c QPIGS
error                         	Invalid byte_response 
response                      	NAK             
pi@raspberrypi:~ $ 

when connecting other stuff to USB port it shows up as AMA0
pi@raspberrypi:~ $ sudo mpp-solar -M LV5161 -d /dev/ttyAMA0 -c QPIGS
WARNING:MPP-Solar:Serial read error: Write timeout
error                         	No byte_response 

my set up

[SETUP]
# Number of seconds to pause between command execution loop
# 0 is no pause, greater than 60 will cause service restarts
pause=5
mqtt_broker=localhost

[Inverter_1_L1]
model=standard
port=/dev/mppsolar/direct/0
baud=2400
command=QPGS
tag=Inverter1
format=influx2

jblance, 2020-11-23:
There is no ```-M LV5161``` supported by the code (the LV5048 is just for different a different set of commands to the original)
However the communications seemed to work with ```-d /dev/hidraw0``` (though normally you need sudo for this to work)
try ```sudo mpp-solar -d /dev/hidraw0 -c QPI``` and post the output

jec0408, 2020-11-23:
pi@raspberrypi:~ $ sudo mpp-solar -d /dev/hidraw0 -c QPI
error                         	Invalid byte_response 
response                      	NAK

jblance, 2020-11-23:
try ``` sudo mpp-solar -d /dev/hidraw0 -c QPI -D```
(this will produce a lot of output)

jec0408, 2020-11-23:
pi@raspberrypi:~ $ sudo mpp-solar -d /dev/hidraw0 -c QPI -D
INFO:MPP-Solar:command QPI
INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
DEBUG:MPP-Solar:Device matches hidraw regex
INFO:MPP-Solar:Loading commands for inverter model: standard
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/muchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MUCHGC supported all inverters
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for MUCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
DEBUG:MPP-Solar:Generated CRC bf 86 bf86
DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/pbcv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBCV supported all inverters
INFO:MPP-Solar:... command PBCV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBCV
DEBUG:MPP-Solar:Calculating CRC for b'PBCV'
DEBUG:MPP-Solar:Generated CRC 64 5a 645a
DEBUG:MPP-Solar:Full byte command: b'PBCVdZ\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/pbdv.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBDV supported all inverters
INFO:MPP-Solar:... command PBDV loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBDV
DEBUG:MPP-Solar:Calculating CRC for b'PBDV'
DEBUG:MPP-Solar:Generated CRC fd cd fdcd
DEBUG:MPP-Solar:Full byte command: b'PBDV\xfd\xcd\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/pbft.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PBFT supported all inverters
INFO:MPP-Solar:... command PBFT loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for PBFT
DEBUG:MPP-Solar:Calculating CRC for b'PBFT'
DEBUG:MPP-Solar:Generated CRC bb ed bbed
DEBUG:MPP-Solar:Full byte command: b'PBFT\xbb\xed\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/pbt.json
[... 243 lines cut ...]
DEBUG:MPP-Solar:Command QVFW2 supported all inverters
INFO:MPP-Solar:... command QVFW2 loaded for inverter model: standard
DEBUG:MPP-Solar:Generate full byte command for QVFW2
DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
DEBUG:MPP-Solar:Searching for cmd 'QPI'
INFO:MPP-Solar:DIRECT USB connection: executing QPI
Protocol ID inquiry
 -- queries the device protocol ID. e.g. 30 for HS series

DEBUG:MPP-Solar:length of to_send: 6
DEBUG:MPP-Solar:1 chunk send
DEBUG:MPP-Solar:usb byte_response was: b'(NAKss\r'
DEBUG:MPP-Solar:Byte_Response length: 7
DEBUG:MPP-Solar:CRC resp	 115, 115
DEBUG:MPP-Solar:Calculating CRC for b'(NAK'
DEBUG:MPP-Solar:Generated CRC 73 73 7373
DEBUG:MPP-Solar:CRC calc	 115 115
DEBUG:MPP-Solar:CRCs match
DEBUG:MPP-Solar:Response invalid as query with NAK resp
INFO:MPP-Solar:Invalid byte_response
DEBUG:MPP-Solar:getResponse with protocol not PI18
error                         	Invalid byte_response 
response                      	NAK

jblance, 2020-11-23:
What inverter are you using?
Try  ``` sudo mpp-solar -d /dev/hidraw0 -c QPI -D -M PI18```

jec0408, 2020-11-23:
[image]
[image]
the only name on it is hybridpower 2030. and the only thing i can find on google with that name is from the company that installed the hole system.

but it looks like the inverter from mpp.
I have written the installation company to ask for model and producer. but still no reply.
thank you for taking the time to help me :)

pi@raspberrypi:~ $  sudo mpp-solar -d /dev/hidraw0 -c QPI -D -M PI18
INFO:MPP-Solar:command QPI
INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
DEBUG:MPP-Solar:Device matches hidraw regex
INFO:MPP-Solar:Loading commands for inverter model: PI18
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command ET supported by model PI18
INFO:MPP-Solar:... command ET loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P005ET
DEBUG:MPP-Solar:Calculating CRC for b'^P005ET'
DEBUG:MPP-Solar:Generated CRC 4e 91 4e91
DEBUG:MPP-Solar:Full byte command: b'^P005ETN\x91\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command GS supported by model PI18
INFO:MPP-Solar:... command GS loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P005GS
DEBUG:MPP-Solar:Calculating CRC for b'^P005GS'
DEBUG:MPP-Solar:Generated CRC 58 14 5814
DEBUG:MPP-Solar:Full byte command: b'^P005GSX\x14\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MOD supported by model PI18
INFO:MPP-Solar:... command MOD loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P006MOD
DEBUG:MPP-Solar:Calculating CRC for b'^P006MOD'
DEBUG:MPP-Solar:Generated CRC dd be ddbe
DEBUG:MPP-Solar:Full byte command: b'^P006MOD\xdd\xbe\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PI supported by model PI18
INFO:MPP-Solar:... command PI loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P005PI
DEBUG:MPP-Solar:Calculating CRC for b'^P005PI'
DEBUG:MPP-Solar:Generated CRC 71 8b 718b
DEBUG:MPP-Solar:Full byte command: b'^P005PIq\x8b\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/muchgc.json
[... 276 lines cut ...]
INFO:MPP-Solar:... command QVFW2 loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for QVFW2
DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
DEBUG:MPP-Solar:Searching for cmd 'QPI'
INFO:MPP-Solar:DIRECT USB connection: executing QPI
Protocol ID inquiry
 -- queries the device protocol ID. e.g. 30 for HS series

DEBUG:MPP-Solar:length of to_send: 6
DEBUG:MPP-Solar:1 chunk send
DEBUG:MPP-Solar:usb byte_response was: b'(NAKss\r'
DEBUG:MPP-Solar:Byte_Response length: 7
DEBUG:MPP-Solar:CRC resp	 115, 115
DEBUG:MPP-Solar:Calculating CRC for b'(NAK'
DEBUG:MPP-Solar:Generated CRC 73 73 7373
DEBUG:MPP-Solar:CRC calc	 115 115
DEBUG:MPP-Solar:CRCs match
DEBUG:MPP-Solar:Response invalid as query with NAK resp
INFO:MPP-Solar:Invalid byte_response
DEBUG:MPP-Solar:getResponse with protocol = PI18
error                         	Invalid byte_response 
response                      	                
pi@raspberrypi:~ $

jblance, 2020-11-23:
Hmm, so we are 'sending' ok and getting a response, but the inverter is not recognizing the command we are sending
So either
- we are sending the command incorrectly, or
- the command we are sending is wrong

try
```sudo mpp-solar -d /dev/hidraw0 -c QID```
```sudo mpp-solar -d /dev/hidraw0 -c QID -M PI18```
```sudo mpp-solar -d /dev/hidraw0 -c QID -M LV5048```

And let me know the results

Do you have software that talks to the inverter?

jec0408, 2020-11-23:
I have tried using the program SolarPowerwin1.141 windows 10 USB. and it was receiving info. 

pi@raspberrypi:~ $ sudo mpp-solar -d /dev/hidraw0 -c QID
error                         	Invalid byte_response 
response                      	NAK             
pi@raspberrypi:~ $ 

pi@raspberrypi:~ $ sudo mpp-solar -d /dev/hidraw0 -c QID -M PI18
error                         	Invalid byte_response 
response                      	                
pi@raspberrypi:~ $ 

pi@raspberrypi:~ $ sudo mpp-solar -d /dev/hidraw0 -c QID -M LV5048
error                         	Invalid byte_response 
response                      	NAK             
pi@raspberrypi:~ $

jblance, 2020-11-23:
Any chance you can do a wireshark capture while running the windows
software (this is to try to determine what commands are being used)

jec0408, 2020-11-23:
hmm... okay i haven't tried that before. you mean this program to do it?
https://www.wireshark.org/download.html

jblance, 2020-11-23:
yes
its a little tricky but you should be able to figure it out. it will create
a big log file - just make sure that it is grabbing the USB connection

jec0408, 2020-11-23:
[data file.txt](https://github.com/jblance/mpp-solar/files/5586342/data.file.txt)

I think I got it.

jblance, 2020-11-24:
it doesnt really include what is needed - normally a .pcap or similar (i need that raw hex that is send / received)

jec0408, 2020-11-24:
[20201124.pcap.zip](https://github.com/jblance/mpp-solar/files/5589125/20201124.pcap.zip)
I hope this is the correct info you need

I just got a reply from the company 
The inverter is one we have manufactured at MPP-solar, so it is very similar to the models you link to.
However, ours has slightly better insides than their standard inverters.
So I can not give you a model name besides Hybridpower 2030.

jec0408, 2020-11-24:
[20201124-2.pcap.zip](https://github.com/jblance/mpp-solar/files/5589480/20201124-2.pcap.zip)
The  first file. is just the program looking at the inverter.
file I´m sending now. I have also tried to change some inverter settings. 
battery - grid
grid - battery 
and back.

Northbadge, 2020-11-25:
Doesn't PI18 specify a start byte of '^'?

Northbadge, 2020-11-25:
If the inverter's using PI18, shouldn't you be trying to use PI18 commands? Afaik their all going to start with '^', so the old commands shouldn't work unless there's a piece of code I didn't read through.. 

Try with -c PI instead

Separate issue, but commands might need some reworking, in terms of inheritance and prefixes and byte command creation

jblance, 2020-11-25:
> If the inverter's using PI18, shouldn't you be trying to use PI18 commands? Afaik their all going to start with '^', so the old commands shouldn't work unless there's a piece of code I didn't read through..
> 
> Try with -c PI (and the other arguments) instead
> 

Good catch - yeah the command would need to be
```sudo mpp-solar -d /dev/hidraw0 -c PI -M PI18```
Which should use the ^ prefix

> Separate issue, but commands might need some reworking, in terms of inheritance and prefixes and byte command creation
>

Yes - thats what I am trying at the moment in refactor branch. Originally the code only supported one command set, then one but with a few minor changes. New code has flexibility to use completely different commands / protocols / and communications methods...

jblance, 2020-11-25:
> [20201124-2.pcap.zip](https://github.com/jblance/mpp-solar/files/5589480/20201124-2.pcap.zip)
> The first file. is just the program looking at the inverter.
> file I´m sending now. I have also tried to change some inverter settings.
> battery - grid
> grid - battery
> and back.

The capture is good
There are some Q type commands, though they look like the format is different - ie no leading ( and no CRC??
eg
```
51 49 44 0d 05 b3 d0 fe                           QID.....
Q  I  D  \r  
````

Response (across multiple packets)
```
0000   28 39 36 31 33 31 35 30                           (9613150
0000   33 31 30 30 32 35 30 b3                           3100250.
0000   f2 0d 00 00 00 00 00 00                           ........
```

This corresponds to the serial number from the photo - 96131503100250 with a leading ( and no crc

jec0408, 2020-11-25:
pi@raspberrypi:~ $ sudo mpp-solar -d /dev/hidraw0 -c PI -M PI18
error                         	Invalid byte_response 
response                      	                
pi@raspberrypi:~ $

jblance, 2020-11-25:
This https://www.photovoltaikforum.com/core/attachment/7135-protocol-pdf/ _might_ be the command definitions

jec0408, 2020-11-25:
Right now I have no ide how to continue. I was just hoping to get at solar monitoring system running on my raspberry pi that is all ready monitoring my solar heating system in Iobroker. 
and coding is not my strong side....

jblance, 2020-11-25:
lol - thats ok, wasnt suggesting for you to do the coding :-)
key bit was to figure out how to talk to the inverter (and how to understand the response) - which we mostly do now
as @Northbadge mentioned, the current codebase (master branch anyway) is a bit difficult to support different command / protocols.
I'm about to merge back the refactor branch (i.e update the main code) with some work that will allow the code to support different protocols more readily - this will allow you to do the monitoring you are after (assumming to are ok to wait a bit and perhaps deal with code fixes / releases)

jec0408, 2020-11-25:
okay.. thanks :) I don't want to give up I just have no idea how to proceed..  
Yes I can wait. and hopefully it will work at some point.  just tell me what you want me to test.
the company that installed the system could only provide the name hybridpower 2030.
would it help to try and contact mpp in taiwan?
any idea if they would give such info? ore company secrets

jblance, 2020-11-25:
@jec0408 how did you install from git? are you using venv / downloading a zip / clone repo etc
If you install from the refactor branch, and run
```mpp-solar -n 'test' -P PI00 -p /dev/hidraw0 -o screen -c QPI -D```
what do you get?

jblance, 2020-11-25:
> okay.. thanks :) I don't want to give up I just have no idea how to proceed..
> Yes I can wait. and hopefully it will work at some point. just tell me what you want me to test.
> the company that [...]

They might - a few people have managed to get protocol docs (I've been collecting them in the docs dir)

I'm working on things now - pretty sure we can get it to work, just need to check that the response mapping works as we go
Theres a task for you above (to hopefully determine the protocol ID)

jec0408, 2020-11-25:
pi@raspberrypi:~ $ mpp-solar -n 'test' -P PI00 -p /dev/hidraw0 -o screen -c QPI -D
usage: mpp-solar [-h] [-c COMMAND] [-D] [-I] [-d DEVICE] [-b BAUD] [-M MODEL]
                 [-l] [-s] [-t] [-R] [-p]
mpp-solar: error: unrecognized arguments: -n test -P PI00 /dev/hidraw0 -o screen
pi@raspberrypi:~ $ 

I also tried with sudo it was the same...

I got it from github clone link.
venv = Virtual Environments? then no

jblance, 2020-11-25:
the above only works from the branch
in the directory that you cloned into run the following (one line at a time):
```
git status
git pull
git checkout refactor
sudo python3 ./setup.py install
```
then try
```mpp-solar -h```
should show ```MPP Solar Command Utility, version: 0.7.0, First refactor version - under development``` on about the third line
If so, retry the above command

jblance, 2020-11-26:
Actually I have just merged the branch, so you just need to refresh the clone and reinstall

jec0408, 2020-11-26:
I just cloned the link and installed setup.py 
when i run mpp-solar -n 'test' -P PI00 -p /dev/hidraw0 -o screen -c QPI -D 
this comes now...

 mpp-solar -n 'test' -P PI00 -p /dev/hidraw0 -o screen -c QPI -D
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.3, Refactored version - not backwards compatible, add command info to output
INFO:MPP-Solar:Creating device "test" (type: "mppsolar") on port "/dev/hidraw0" using protocol "PI00" for command "QPI" (tag: QPI)
DEBUG:MPP-Solar:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:MPP-Solar:Device matches hidraw
INFO:MPP-Solar:Using hidrawio for communications
DEBUG:MPP-Solar:device.set_protocol with protocol PI00
DEBUG:MPP-Solar:mppsolar __init__ name test, port <mppsolar.io.hidrawio.HIDRawIO object at 0x764f3fd0>, protocol <mppsolar.protocols.pi00.pi00 object at 0x75fe96f0>
DEBUG:MPP-Solar:mppsolar __init__ args ()
DEBUG:MPP-Solar:mppsolar __init__ kwargs {'name': 'test', 'port': '/dev/hidraw0', 'protocol': 'PI00'}
INFO:MPP-Solar:Running command QPI
INFO:MPP-Solar:Using protocol b'PI00' with 2 commands
DEBUG:MPP-Solar:get_command_defn for: QPI
DEBUG:MPP-Solar:Found command QPI in protocol b'PI00'
DEBUG:MPP-Solar:full command: b'QPI\r'
INFO:MPP-Solar:full command b'QPI\r' for command QPI
DEBUG:MPP-Solar:get_command_defn for: QPI
DEBUG:MPP-Solar:Found command QPI in protocol b'PI00'
DEBUG:MPP-Solar:length of to_send: 4
DEBUG:MPP-Solar:1 chunk send
DEBUG:MPP-Solar:usb response was: b'(PI16\x9c\xaf\r'
INFO:MPP-Solar:response passed to decode: b'(PI16\x9c\xaf\r'
DEBUG:MPP-Solar:trimmed and split responses: [b'PI16']
DEBUG:MPP-Solar:Decoded response {'protocol_version': ['PI16', '']}
INFO:MPP-Solar:Decoded response {'protocol_version': ['PI16', ''], '_command': 'QPI', '_command_description': 'Device Protocol Version inquiry'}
DEBUG:MPP-Solar:Send and Receive Response {'protocol_version': ['PI16', ''], '_command': 'QPI', '_command_description': 'Device Protocol Version inquiry'}
DEBUG:MPP-Solar:results: {'protocol_version': ['PI16', ''], '_command': 'QPI', '_command_description': 'Device Protocol Version inquiry'}
INFO:MPP-Solar:attempting to create output processor: screen
DEBUG:MPP-Solar:processor.screen __init__ kwargs {}
INFO:MPP-Solar:Using output processor: screen
Command: QPI - Device Protocol Version inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
protocol_version              	PI16           	    
pi@raspberrypi:~/mpp-solar $ 

mpp-solar -h
MPP Solar Command Utility, version: 0.7.3, Refactored version - not backwards
compatible, add command info to output

jblance, 2020-11-26:
Good news, so use -P PI16 for the commands
Eg
mpp-solar -P PI16 -c QPIGS -p /dev/hidraw0

jec0408, 2020-12-08:
Hi again sorry i have been gone for work....

pi@raspberrypi:~/mpp-solar $ mpp-solar -P PI16 -c QPIGS -p /dev/hidraw0
Command: QPIGS - General status query
------------------------------------------------------------
Parameter                     	Value           Unit
grid_voltage                  	230.5          	V   
output_power                  	310            	W   
grid_frequency                	50.0           	Hz  
output_current                	0.0            	A   
ac_output_voltage_r           	0.0            	V   
ac_output_power_r             	0              	W   
ac_output_frequency           	0.0            	Hz  
ac_output_current_r           	0.0            	A   
output_load_percent           	0              	%   
pbus_voltage                  	446.9          	V   
sbus_voltage                  	446.9          	V   
positive_battery_voltage      	54.0           	V   
negative_battery_voltage      	0.0            	V   
battery_capacity              	75             	%   
pv1_input_power               	782            	W   
pv2_input_power               	310            	W   
pv3_input_power               	0              	W   
pv1_input_voltage             	396.4          	V   
pv2_input_voltage             	0.0            	V   
pv3_input_voltage             	0.0            	V   
max_temperature               	31.0           	°C  
status_todo                   	A---001010     	    
pi@raspberrypi:~/mpp-solar $

jec0408, 2020-12-08:
pi@raspberrypi:~ $ systemctl --user status mpp-solar
● mpp-solar.service - MPP Solar Service
   Loaded: loaded (/etc/systemd/user/mpp-solar.service; enabled; vendor preset: 
   Active: failed (Result: exit-code) since Tue 2020-12-08 19:27:24 CET; 43s ago
  Process: 1104 ExecStart=/usr/bin/python3 /usr/local/bin/mpp-solar-service -c /
 Main PID: 1104 (code=exited, status=1/FAILURE)

dec 08 19:27:24 raspberrypi systemd[498]: mpp-solar.service: Service RestartSec=
dec 08 19:27:24 raspberrypi systemd[498]: mpp-solar.service: Scheduled restart j
dec 08 19:27:24 raspberrypi systemd[498]: Stopped MPP Solar Service.
dec 08 19:27:24 raspberrypi systemd[498]: mpp-solar.service: Start request repea
dec 08 19:27:24 raspberrypi systemd[498]: mpp-solar.service: Failed with result 
dec 08 19:27:24 raspberrypi systemd[498]: Failed to start MPP Solar Service.

jec0408, 2020-12-08:
it stopped running to day... I tried to do at restart op pi. but did not help.

jblance, 2020-12-08:
can you try the command line ```mpp-solar``` command and with -D or -I for more information

jec0408, 2020-12-08:
pi@raspberrypi:~ $ mpp-solar -D
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.3, Refactored version - not backwards compatible, add command info to output
INFO:MPP-Solar:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0" using protocol "PI30" for command "None" (tag: None)
DEBUG:MPP-Solar:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
INFO:MPP-Solar:Using serialio for communications
DEBUG:MPP-Solar:device.set_protocol with protocol PI30
DEBUG:MPP-Solar:mppsolar __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb5feafb0>, protocol <mppsolar.protocols.pi30.pi30 object at 0xb5feae50>
DEBUG:MPP-Solar:mppsolar __init__ args ()
DEBUG:MPP-Solar:mppsolar __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30'}
INFO:MPP-Solar:Running command QPI
INFO:MPP-Solar:Using protocol b'PI30' with 35 commands
DEBUG:MPP-Solar:get_command_defn for: QPI
DEBUG:MPP-Solar:Found command QPI in protocol b'PI30'
DEBUG:MPP-Solar:Calculating CRC for b'QPI'
DEBUG:MPP-Solar:Generated CRC 0xbe 0xac 0xbeac
DEBUG:MPP-Solar:full command: b'QPI\xbe\xac\r'
INFO:MPP-Solar:full command b'QPI\xbe\xac\r' for command QPI
DEBUG:MPP-Solar:get_command_defn for: QPI
DEBUG:MPP-Solar:Found command QPI in protocol b'PI30'
DEBUG:MPP-Solar:port /dev/ttyUSB0, baudrate 2400
WARNING:MPP-Solar:Serial read error: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
INFO:MPP-Solar:Command execution failed
DEBUG:MPP-Solar:Send and Receive Response {'ERROR': ['Serial command execution failed', '']}
DEBUG:MPP-Solar:results: {'ERROR': ['Serial command execution failed', '']}
INFO:MPP-Solar:attempting to create output processor: screen
DEBUG:MPP-Solar:processor.screen __init__ kwargs {}
INFO:MPP-Solar:Using output processor: screen
Parameter                     	Value           Unit
ERROR                         	Serial command execution failed	

    
pi@raspberrypi:~ $ mpp-solar -I
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.3, Refactored version - not backwards compatible, add command info to output
INFO:MPP-Solar:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0" using protocol "PI30" for command "None" (tag: None)
INFO:MPP-Solar:Using serialio for communications
INFO:MPP-Solar:Running command QPI
INFO:MPP-Solar:Using protocol b'PI30' with 35 commands
INFO:MPP-Solar:full command b'QPI\xbe\xac\r' for command QPI
WARNING:MPP-Solar:Serial read error: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
INFO:MPP-Solar:Command execution failed
INFO:MPP-Solar:attempting to create output processor: screen
INFO:MPP-Solar:Using output processor: screen
Parameter                     	Value           Unit
ERROR                         	Serial command execution failed

jblance, 2020-12-08:
it needs to be the full command, something like ```mpp-solar -P PI16 -c QPIGS -p /dev/hidraw0 -I```

jec0408, 2020-12-08:
pi@raspberrypi:~ $ mpp-solar -P PI16 -c QPIGS -p /dev/hidraw0 -I
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.3, Refactored version - not backwards compatible, add command info to output
INFO:MPP-Solar:Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0" using protocol "PI16" for command "QPIGS" (tag: QPIGS)
INFO:MPP-Solar:Using hidrawio for communications
INFO:MPP-Solar:Running command QPIGS
INFO:MPP-Solar:Using protocol b'PI16' with 6 commands
INFO:MPP-Solar:full command b'QPIGS\r' for command QPIGS
INFO:MPP-Solar:response passed to decode: b'(231.5 100043 49.9 0000.0 000.0 00000 00.0 000.0 000 406.3 406.3 048.9 ---.- 049 00000 00000 ----- 010.5 ---.- ---.- 036.0 D---000200\xb0+\r'
INFO:MPP-Solar:Decoded response {'grid_voltage': [231.5, 'V'], 'output_power': [100043, 'W'], 'grid_frequency': [49.9, 'Hz'], 'output_current': [0.0, 'A'], 'ac_output_voltage_r': [0.0, 'V'], 'ac_output_power_r': [0, 'W'], 'ac_output_frequency': [0.0, 'Hz'], 'ac_output_current_r': [0.0, 'A'], 'output_load_percent': [0, '%'], 'pbus_voltage': [406.3, 'V'], 'sbus_voltage': [406.3, 'V'], 'positive_battery_voltage': [48.9, 'V'], 'negative_battery_voltage': [0.0, 'V'], 'battery_capacity': [49, '%'], 'pv1_input_power': [0, 'W'], 'pv2_input_power': [0, 'W'], 'pv3_input_power': [0, 'W'], 'pv1_input_voltage': [10.5, 'V'], 'pv2_input_voltage': [0.0, 'V'], 'pv3_input_voltage': [0.0, 'V'], 'max_temperature': [36.0, '°C'], 'status_todo': ['D---000200', ''], '_command': 'QPIGS', '_command_description': 'General status query'}
INFO:MPP-Solar:attempting to create output processor: screen
INFO:MPP-Solar:Using output processor: screen
Command: QPIGS - General status query
------------------------------------------------------------
Parameter                     	Value           Unit
grid_voltage                  	231.5          	V   
output_power                  	100043         	W   
grid_frequency                	49.9           	Hz  
output_current                	0.0            	A   
ac_output_voltage_r           	0.0            	V   
ac_output_power_r             	0              	W   
ac_output_frequency           	0.0            	Hz  
ac_output_current_r           	0.0            	A   
output_load_percent           	0              	%   
pbus_voltage                  	406.3          	V   
sbus_voltage                  	406.3          	V   
positive_battery_voltage      	48.9           	V   
negative_battery_voltage      	0.0            	V   
battery_capacity              	49             	%   
pv1_input_power               	0              	W   
pv2_input_power               	0              	W   
pv3_input_power               	0              	W   
pv1_input_voltage             	10.5           	V   
pv2_input_voltage             	0.0            	V   
pv3_input_voltage             	0.0            	V   
max_temperature               	36.0           	°C  
status_todo                   	D---000200     	    

pi@raspberrypi:~ $ mpp-solar -P PI16 -c QPIGS -p /dev/hidraw0 -D
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.3, Refactored version - not backwards compatible, add command info to output
INFO:MPP-Solar:Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0" using protocol "PI16" for command "QPIGS" (tag: QPIGS)
DEBUG:MPP-Solar:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:MPP-Solar:Device matches hidraw
INFO:MPP-Solar:Using hidrawio for communications
DEBUG:MPP-Solar:device.set_protocol with protocol PI16
DEBUG:MPP-Solar:mppsolar __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0xb5f69fd0>, protocol <mppsolar.protocols.pi16.pi16 object at 0xb5f75530>
DEBUG:MPP-Solar:mppsolar __init__ args ()
DEBUG:MPP-Solar:mppsolar __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI16'}
INFO:MPP-Solar:Running command QPIGS
INFO:MPP-Solar:Using protocol b'PI16' with 6 commands
DEBUG:MPP-Solar:get_command_defn for: QPIGS
DEBUG:MPP-Solar:Found command QPIGS in protocol b'PI16'
DEBUG:MPP-Solar:full command: b'QPIGS\r'
INFO:MPP-Solar:full command b'QPIGS\r' for command QPIGS
DEBUG:MPP-Solar:get_command_defn for: QPIGS
DEBUG:MPP-Solar:Found command QPIGS in protocol b'PI16'
DEBUG:MPP-Solar:length of to_send: 6
DEBUG:MPP-Solar:1 chunk send
DEBUG:MPP-Solar:usb response was: b'(233.3 100045 49.9 0000.0 000.0 00000 00.0 000.0 000 406.0 406.0 048.9 ---.- 049 00000 00000 ----- 010.6 ---.- ---.- 036.0 D---000200.?\r'
INFO:MPP-Solar:response passed to decode: b'(233.3 100045 49.9 0000.0 000.0 00000 00.0 000.0 000 406.0 406.0 048.9 ---.- 049 00000 00000 ----- 010.6 ---.- ---.- 036.0 D---000200.?\r'
DEBUG:MPP-Solar:trimmed and split responses: [b'233.3', b'100045', b'49.9', b'0000.0', b'000.0', b'00000', b'00.0', b'000.0', b'000', b'406.0', b'406.0', b'048.9', b'---.-', b'049', b'00000', b'00000', b'-----', b'010.6', b'---.-', b'---.-', b'036.0', b'D---000200']
DEBUG:MPP-Solar:Decoded response {'grid_voltage': [233.3, 'V'], 'output_power': [100045, 'W'], 'grid_frequency': [49.9, 'Hz'], 'output_current': [0.0, 'A'], 'ac_output_voltage_r': [0.0, 'V'], 'ac_output_power_r': [0, 'W'], 'ac_output_frequency': [0.0, 'Hz'], 'ac_output_current_r': [0.0, 'A'], 'output_load_percent': [0, '%'], 'pbus_voltage': [406.0, 'V'], 'sbus_voltage': [406.0, 'V'], 'positive_battery_voltage': [48.9, 'V'], 'negative_battery_voltage': [0.0, 'V'], 'battery_capacity': [49, '%'], 'pv1_input_power': [0, 'W'], 'pv2_input_power': [0, 'W'], 'pv3_input_power': [0, 'W'], 'pv1_input_voltage': [10. [... cut ...]
negative_battery_voltage      	0.0            	V   
battery_capacity              	49             	%   
pv1_input_power               	0              	W   
pv2_input_power               	0              	W   
pv3_input_power               	0              	W   
pv1_input_voltage             	10.6           	V   
pv2_input_voltage             	0.0            	V   
pv3_input_voltage             	0.0            	V   
max_temperature               	36.0           	°C  
status_todo                   	D---000200     	    
pi@raspberrypi:~ $

jblance, 2020-12-08:
ok the command still works ok
so you were using the service?
i hadnt realized that it was working in the refactored version
can you try running the service on the command line (with -D) and see what that gives

jec0408, 2020-12-08:
In previous text i did both commands -I and -D
service hmm. not really  I´m not really sure what im doing. so i have trying to check that everything in the guide is working....
and hopefully at some point it works....

jblance, 2020-12-08:
the service is ```/usr/local/bin/mpp-solar-service```
it should have a config file (normally /etc/mpp-solar/mpp-solar.conf)
you should be able to run it on the command line
```/usr/local/bin/mpp-solar-service -c /etc/mpp-solar/mpp-solar.conf```
But please note that the refactored code hasnt been tested for the service at the moment

jec0408, 2020-12-08:
pi@raspberrypi:~ $ /usr/local/bin/mpp-solar-service -c /etc/mpp-solar/mpp-solar.conf
MPP-Solar-Service: Initializing ...
MPP-Solar-Service: Config file: /etc/mpp-solar/mpp-solar.conf
MPP-Solar-Service: Config setting - pause: 5
MPP-Solar-Service: Config setting - mqtt_broker: localhost
MPP-Solar-Service: Config setting - command sections found: 1
MPP-Solar-Service: item {'mp': 'mppUtils(/dev/mppsolar/direct/0, 2400, standard)', 'command': 'I', 'format': 'influx2', 'tag': 'Inverter1'}
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar-service", line 11, in <module>
    load_entry_point('mpp-solar==0.7.3', 'console_scripts', 'mpp-solar-service')()
  File "/usr/local/lib/python3.7/dist-packages/mpp_solar-0.7.3-py3.7.egg/mppsolar/__init__.py", line 317, in mpp_solar_service
AttributeError: 'str' object has no attribute 'getInfluxLineProtocol2'

[SETUP]
# Number of seconds to pause between command execution loop
# 0 is no pause, greater than 60 will cause service restarts
pause=5
mqtt_broker=localhost

[Inverter_1_L1]
model=standard
port=/dev/mppsolar/direct/0
baud=2400
command=I
tag=Inverter1
format=influx2

jblance, 2020-12-09:
Yup that looks like a code issue
 I'll have a look when I get back

jblance, 2020-12-13:
though looking at your mpp-solar.conf the command you are using is invalid

jblance, 2020-12-14:
have updated the service code in the latest verion 0.7.4 - it is now using the same code as the command line, for easier debuging and can run the contents of the config file via the command line 

the service definition has changed (now uses -C instead of -c)
the config file has changes as well

can you update and try running on command line with info messages on, i.e.
```mpp-solar -C /etc/mpp-solar/mpp-solar.conf -I```

jec0408, 2020-12-17:
Hi yes i Will try when i get home from work 
On Friday 

Sendt fra min iPhone

jec0408, 2020-12-19:
pi@raspberrypi:~/mpp-solar $ mpp-solar -C /etc/mpp-solar/mpp-solar.conf -I
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.8, Refactored version - not backwards compatible, add jkbms (02 and 04) support
INFO:MPP-Solar:Model standard specified, setting protocol to PI30
INFO:MPP-Solar:Using hidrawio for communications
INFO:MPP-Solar:Command line using config file: /etc/mpp-solar/mpp-solar.conf
INFO:MPP-Solar:Running command I
INFO:MPP-Solar:Using protocol b'PI30' with 35 commands
INFO:MPP-Solar:No command_defn found for I
INFO:MPP-Solar:full command b'I\xd9\xed\r' for command I
INFO:MPP-Solar:No command_defn found for I
INFO:MPP-Solar:response passed to decode: b'(NAKss\r'
INFO:MPP-Solar:Decoded response {'ERROR': ["No definition for command I in protocol b'PI30'", ''], 'unknown_value_in_response_0': ['NAK', ''], '_command': 'I'}
INFO:MPP-Solar:attempting to create output processor: screen
INFO:MPP-Solar:Using output processor: screen
Command: I - No description found
------------------------------------------------------------
Parameter                     	Value           Unit
ERROR                         	No definition for command I in protocol b'PI30'	    
unknown_value_in_response_0   	NAK            	    
pi@raspberrypi:~/mpp-solar $

jblance, 2020-12-19:
Looks like you still have to update your conf file

Can you post the contents of it

jec0408, 2020-12-19:
[SETUP]
# Number of seconds to pause between command execution loop
# 0 is no pause, greater than 60 will cause service restarts
pause=5
mqtt_broker=localhost

[Inverter_1_L1]
model=standard
port=/dev/mppsolar/direct/0
baud=2400
command=I
tag=Inverter1
format=influx2

jblance, 2020-12-19:
Command=I needs to be something like:
command=QPIGS

Format needs to change, try
output=screen
For a start

jblance, 2020-12-19:
Sorry should be
```outputs=screen```

jec0408, 2020-12-19:
so delete format=influx2...

pause=5
mqtt_broker=localhost

[Inverter_1_L1]
model=standard
protocol=PI16
port=/dev/mppsolar/direct/0
baud=2400
command=QPIGS
tag=Inverter1
outputs=screen

jblance, 2020-12-19:
Yeah for testing

jec0408, 2020-12-19:
pi@raspberrypi:~/mpp-solar $ mpp-solar -C /etc/mpp-solar/mpp-solar.conf -I
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.8, Refactored version - not backwards compatible, add jkbms (02 and 04) support
INFO:MPP-Solar:Using hidrawio for communications
INFO:MPP-Solar:Command line using config file: /etc/mpp-solar/mpp-solar.conf
INFO:MPP-Solar:Running command QPIGS
INFO:MPP-Solar:Using protocol b'PI16' with 6 commands
INFO:MPP-Solar:full command b'QPIGS\r' for command QPIGS
INFO:MPP-Solar:response passed to decode: b'(230.9 100047 50.0 0000.0 000.0 00000 00.0 000.0 000 398.1 398.1 047.9 ---.- 043 00000 00000 ----- 018.0 ---.- ---.- 036.0 D---000200\x1a\xa0\r'
INFO:MPP-Solar:Decoded response {'grid_voltage': [230.9, 'V'], 'output_power': [100047, 'W'], 'grid_frequency': [50.0, 'Hz'], 'output_current': [0.0, 'A'], 'ac_output_voltage_r': [0.0, 'V'], 'ac_output_power_r': [0, 'W'], 'ac_output_frequency': [0.0, 'Hz'], 'ac_output_current_r': [0.0, 'A'], 'output_load_percent': [0, '%'], 'pbus_voltage': [398.1, 'V'], 'sbus_voltage': [398.1, 'V'], 'positive_battery_voltage': [47.9, 'V'], 'negative_battery_voltage': [0.0, 'V'], 'battery_capacity': [43, '%'], 'pv1_input_power': [0, 'W'], 'pv2_input_power': [0, 'W'], 'pv3_input_power': [0, 'W'], 'pv1_input_voltage': [18.0, 'V'], 'pv2_input_voltage': [0.0, 'V'], 'pv3_input_voltage': [0.0, 'V'], 'max_temperature': [36.0, '°C'], 'status_todo': ['D---000200', ''], '_command': 'QPIGS', '_command_description': 'General status query'}
INFO:MPP-Solar:attempting to create output processor: screen
INFO:MPP-Solar:Using output processor: screen
Command: QPIGS - General status query
------------------------------------------------------------
Parameter                     	Value           Unit
grid_voltage                  	230.9          	V   
output_power                  	100047         	W   
grid_frequency                	50.0           	Hz  
output_current                	0.0            	A   
ac_output_voltage_r           	0.0            	V   
ac_output_power_r             	0              	W   
ac_output_frequency           	0.0            	Hz  
ac_output_current_r           	0.0            	A   
output_load_percent           	0              	%   
pbus_voltage                  	398.1          	V   
sbus_voltage                  	398.1          	V   
positive_battery_voltage      	47.9           	V   
negative_battery_voltage      	0.0            	V   
battery_capacity              	43             	%   
pv1_input_power               	0              	W   
pv2_input_power               	0              	W   
pv3_input_power               	0              	W   
pv1_input_voltage             	18.0           	V   
pv2_input_voltage             	0.0            	V   
pv3_input_voltage             	0.0            	V   
max_temperature               	36.0           	°C  
status_todo                   	D---000200     	    
pi@raspberrypi:~/mpp-solar $

jblance, 2020-12-19:
Once we see this working will use
```outputs=influx2_mqtt```

jblance, 2020-12-19:
Looks good, try with above and check the mqtt broker to make sure you are getting the messages you expect

jec0408, 2020-12-19:
wow now I can se data coming in to database mppsolar. and also in grafana....
wow thanks...
now i just need to make a set up for view this info. :) 

Is there anything you want me to test?
ore you want me to test in the future for devilment just say so :)

jblance, 2020-12-19:
Great news
Nothing else at the moment, let me know if you find anything wrong or if
you have suggestions for improvements

closed 2021-01-10 by jblance