jblance/mpp-solar #43: MPP 3048 support

BarkinSpider, 2020-11-26:
The PC software works ok, whats involved in getting it working with your software?

jblance, 2020-11-26:
What do you mean?
Getting the PC software to work with mpp-solar software?
or 
Getting mpp-solar to work with your inverter?

BarkinSpider, 2020-11-26:
Getting your software to work on a Pi4 -> RS232 -> MPP3048
I made a wireshark capture of USB1 while running the PC software but didnt see anything obvious
The 3048 is new, looks like they may have changed the protocol.

jblance, 2020-11-26:
protocols do seem to vary a lot
do you have mpp-solar on the pi?
what version?

BarkinSpider, 2020-11-26:
Hi, Im looking for an install document for your software. I couldn't see one.
Whats the install proceedure?

I have my pi4 ready with Python 3 and USB cable to RS232 connected to my MPP3048.
I can see a few of the Q commands in the wireshark packets. Hope its compatible.

jblance, 2020-11-26:
safest is probably a venv install (is documented in the latest readme)
## venv Install
for when you want to keep the install and dependencies separate from the
rest of the environment
* create venv `python3 -m venv ~/venv/mppsolar`
    * might need python3-venv installed
* activate venv `source venv/mppsolar/bin/activate` (needed each time
before using)
* pip install from git `pip install -e "git+
https://github.com/jblance/mpp-solar.git#egg=mpp-solar"` (only needed if
the code is updated)

If you get errors, post them and the output of
python3 --version

jblance, 2020-11-26:
Posting the wireshark pcap will help check what will work and what wont

jblance, 2020-11-26:
safest is probably a venv install (is documented in the latest readme)
## venv Install ##
for when you want to keep the install and dependencies separate from the rest of the environment
* create venv ```python3 -m venv ~/venv/mppsolar```
    * note: might need python3-venv installed
* activate venv ```source venv/mppsolar/bin/activate``` (needed each time before using)
* pip install from git ```pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mpp-solar"``` (only needed ifthe code is updated)

If you get errors, post them and the output of
`python3 --version`

BarkinSpider, 2020-11-26:
[3048usb1.zip](https://github.com/jblance/mpp-solar/files/5600254/3048usb1.zip)

jblance, 2020-11-26:
Well it claims to supprt PI30 - which is the default protocol
So it should work

BarkinSpider, 2020-11-26:
I have a split phase 2 unit system, whats the default name to use for each?

[root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar/bin) > ./mpp-solar
WARNING:MPP-Solar:Serial read error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Parameter                       Value           Unit
ERROR                           Serial command execution failed

BarkinSpider, 2020-11-26:
[root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar/bin) > ./mpp-solar -I -D
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.1, Refactored version - not backwards compatible, no mpp_info_pub
INFO:MPP-Solar:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0" using protocol "PI30" for command "None" (tag: None)
DEBUG:MPP-Solar:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
INFO:MPP-Solar:Using serialio for communications
DEBUG:MPP-Solar:device.set_protocol with protocol PI30
DEBUG:MPP-Solar:mppsolar __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb5e4a110>, protocol <mppsolar.protocols.pi30.pi30 object at 0xb5e50b90>
DEBUG:MPP-Solar:mppsolar __init__ args ()
DEBUG:MPP-Solar:mppsolar __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30'}
INFO:MPP-Solar:Running command QDI
INFO:MPP-Solar:Using protocol b'PI30' with 35 commands
DEBUG:MPP-Solar:get_command_defn for: QDI
DEBUG:MPP-Solar:Found command QDI in protocol b'PI30'
DEBUG:MPP-Solar:Calculating CRC for b'QDI'
DEBUG:MPP-Solar:Generated CRC 0x71 0x1b 0x711b
DEBUG:MPP-Solar:full command: b'QDIq\x1b\r'
INFO:MPP-Solar:full command b'QDIq\x1b\r' for command QDI
DEBUG:MPP-Solar:port /dev/ttyUSB0, baudrate 2400
DEBUG:MPP-Solar:Command execution attempt 1...
WARNING:MPP-Solar:Serial read error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
INFO:MPP-Solar:Command execution failed
DEBUG:MPP-Solar:Send and Receive Response {'ERROR': ['Serial command execution failed', '']}
DEBUG:MPP-Solar:results: {'ERROR': ['Serial command execution failed', '']}
INFO:MPP-Solar:attempting to create output processor: screen
DEBUG:MPP-Solar:processor.screen __init__ kwargs {}
INFO:MPP-Solar:Using output processor: screen
Parameter                       Value           Unit
ERROR                           Serial command execution failed

jblance, 2020-11-26:
more details on venv https://github.com/jblance/mpp-solar/blob/master/docs/venv.md 
shouldnt need to be in bin directory, as long as you have activated and done the pip install

device depends on what is connected to the pi, dmesg and lsusb should help
eg for me:
```
(mppsolar) pi@batteryshed:~ $ lsusb
Bus 001 Device 005: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

(mppsolar) pi@batteryshed:~ $ dmesg |grep USB
[    0.783839] dwc_otg 3f980000.usb: new USB bus registered, assigned bus number 1
[    0.793143] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
[    0.797665] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.807834] hub 1-0:1.0: USB hub found
[    0.847419] usbhid: USB HID core driver
[    1.236895] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    1.477144] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
[    1.481988] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    1.485289] hub 1-1:1.0: USB hub found
[    1.806893] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    1.947154] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
[    1.952680] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.050423] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:21:fb:d0
[    2.146893] usb 1-1.4: new full-speed USB device number 4 using dwc_otg
[    2.305655] usb 1-1.4: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[    2.311487] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.314473] usb 1-1.4: Product: FT232R USB UART
[    2.426897] usb 1-1.5: new low-speed USB device number 5 using dwc_otg
[    2.595523] usb 1-1.5: New USB device found, idVendor=0665, idProduct=5161, bcdDevice= 0.02
[    2.601281] usb 1-1.5: New USB device strings: Mfr=3, Product=1, SerialNumber=0
[    2.614866] hid-generic 0003:0665:5161.0001: hiddev96,hidraw0: USB HID v1.11 Device [HID 0665:5161] on usb-3f980000.usb-1.5/input0
[    6.821574] usbserial: USB Serial support registered for generic
[    6.911965] usbserial: USB Serial support registered for FTDI USB Serial Device
[    6.912336] ftdi_sio 1-1.4:1.0: FTDI USB Serial Device converter detected
[    6.915035] usb 1-1.4: FTDI USB Serial Device converter now attached to ttyUSB0
```

jblance, 2020-11-26:
my output shows a ttyUSB0 and a hidraw0 and not much else - which is because that is all that is connected (2 inverters, one direct USB and one via a SB to serial adapter)

BarkinSpider, 2020-11-26:
seems to be happy with /dev/ttyUSB1, but no data so far. Only the output I posted above. Is it because I have two inverters on the same port?

jblance, 2020-11-26:
how happy?
post output of `lsusb` and `dmesg|grep USB`
also `mpp-solar -p /dev/ttyUSB1 -c QPI -I`

BarkinSpider, 2020-11-26:
[root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar/bin) > lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 001 Device 014: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

# dmesg | grep USB
(standard input):94:[110559.745571] usb 1-1.1: new full-speed USB device number 14 using xhci_hcd
(standard input):95:[110559.880011] usb 1-1.1: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 4.00
(standard input):96:[110559.880028] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
(standard input):97:[110559.880041] usb 1-1.1: Product: USB-Serial Controller D
(standard input):100:[110559.894104] usb 1-1.1: pl2303 converter now attached to ttyUSB1

This is the same USB connection that works with windows and WatchPower

BarkinSpider, 2020-11-26:
[root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar/bin) > ./mpp-solar -p /dev/ttyUSB1 -c QPI -I
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.1, Refactored version - not backwards compatible, no mpp_info_pub
INFO:MPP-Solar:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1" using protocol "PI30" for command "QPI" (tag: QPI)
INFO:MPP-Solar:Using serialio for communications
INFO:MPP-Solar:Running command QPI
INFO:MPP-Solar:Using protocol b'PI30' with 35 commands
INFO:MPP-Solar:full command b'QPI\xbe\xac\r' for command QPI
INFO:MPP-Solar:response passed to decode: b'(PI30\x9a\x0b\r'
INFO:MPP-Solar:attempting to create output processor: screen
INFO:MPP-Solar:Using output processor: screen
Parameter                       Value           Unit
protocol_id                     PI30
(mpp-solar)

jblance, 2020-11-26:
so that worked
output was PI30
now you probably want some status info, so try
`mpp-solar -p /dev/ttyUSB1 -c QPIGS`

BarkinSpider, 2020-11-26:
Bingo!

[root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar/bin) > ./mpp-solar -p /dev/ttyUSB1 -c QPIGS
Parameter                       Value           Unit
ac_input_voltage                0.0             V
ac_input_frequency              0.0             Hz
ac_output_voltage               120.7           V
ac_output_frequency             60.0            Hz
ac_output_apparent_power        0               VA
ac_output_active_power          0               W
ac_output_load                  0               %
bus_voltage                     207             V
battery_voltage                 52.9            V
battery_charging_current        0               A
battery_capacity                87              %
inverter_heat_sink_temperature  18              °C
pv_input_current_for_battery    0.0             A
pv_input_voltage                0.0             V
battery_voltage_from_scc        0.0             V
battery_discharge_current       0               A
is_sbu_priority_version_added   0               True - 1/False - 0
is_configuration_changed        0               True - 1/False - 0
is_scc_firmware_updated         0               True - 1/False - 0
is_load_on                      1               True - 1/False - 0
is_battery_voltage_to_steady_while_charging     0               True - 1/False - 0
is_charging_on                  0               True - 1/False - 0
is_scc_charging_on              0               True - 1/False - 0
is_ac_charging_on               0               True - 1/False - 0
rsv1                            0               A
rsv2                            0               A
pv_input_power                  0               W
is_charging_to_float            0               True - 1/False - 0
is_switched_on                  1               True - 1/False - 0
is_reserved                     0               True - 1/False - 0

BarkinSpider, 2020-11-26:
How do I query individual inverters (i have 2) I want to know the PV watts of each system.

BarkinSpider, 2020-11-26:
I will be mapping all the data with Grafana, Ill send you a screenshot when its done. Ive done all my other solar equipment, see my github page.

jblance, 2020-11-26:
2 inverters that are separate boxes, or 2 in the same chassis?

BarkinSpider, 2020-11-26:
2 individual MPP3048 (seperate boxes) one for each 120v phase

jblance, 2020-11-26:
if they are connected with the 'parallel comms card', try
`mpp-solar -p /dev/ttyUSB1 -c QPGS0`
`mpp-solar -p /dev/ttyUSB1 -c QPGS1`

BarkinSpider, 2020-11-26:
Thanks for all the help! Is there a cheat sheet of useful commands and values I can set?

BarkinSpider, 2020-11-26:
[root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar/bin) > ./mpp-solar -p /dev/ttyUSB1 -c QPGS0
WARNING:MPP-Solar:Serial read error: list index out of range
Parameter                       Value           Unit
ERROR                           Serial command execution failed
(mpp-solar) [root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar/bin) > ./mpp-solar -p /dev/ttyUSB1 -c QPGS1
WARNING:MPP-Solar:Serial read error: list index out of range
Parameter                       Value           Unit
ERROR                           Serial command execution failed

jblance, 2020-11-26:
That might be an error in the code
Just to check can you add a -I (uppercase i) on the end of the command and
post the result

I'm out at the moment, so can't fix it if its an error, but will do when I
get back

BarkinSpider, 2020-11-26:
[root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar/bin) > ./mpp-solar -p /dev/ttyUSB1 -c QPGS0 -I
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.1, Refactored version - not backwards compatible, no mpp_info_pub
INFO:MPP-Solar:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1" using protocol "PI30" for command "QPGS0" (tag: QPGS0)
INFO:MPP-Solar:Using serialio for communications
INFO:MPP-Solar:Running command QPGS0
INFO:MPP-Solar:Using protocol b'PI30' with 35 commands
INFO:MPP-Solar:full command b'QPGS0?\xda\r' for command QPGS0
INFO:MPP-Solar:response passed to decode: b'(1 92812005100038 B 00 000.0 00.00 120.8 59.99 0000 0000 000 52.8 000 086 000.0 000 00012 00000 000 00000110 6 0 060 140 30 00 000\x8c\xf9\r'
WARNING:MPP-Solar:Serial read error: list index out of range
INFO:MPP-Solar:Command execution failed
INFO:MPP-Solar:attempting to create output processor: screen
INFO:MPP-Solar:Using output processor: screen
Parameter                       Value           Unit
ERROR                           Serial command execution failed
(mpp-solar) [root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar/bin) > ./mpp-solar -p /dev/ttyUSB1 -c QPGS1 -I
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.1, Refactored version - not backwards compatible, no mpp_info_pub
INFO:MPP-Solar:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1" using protocol "PI30" for command "QPGS1" (tag: QPGS1)
INFO:MPP-Solar:Using serialio for communications
INFO:MPP-Solar:Running command QPGS1
INFO:MPP-Solar:Using protocol b'PI30' with 35 commands
INFO:MPP-Solar:full command b'QPGS1/\xfb\r' for command QPGS1
INFO:MPP-Solar:response passed to decode: b'(1 92812005100035 B 00 000.0 00.00 119.2 60.00 0011 0000 000 52.9 000 087 000.0 000 00011 00000 000 00000110 5 0 060 140 30 00 000\xa6<\r'
WARNING:MPP-Solar:Serial read error: list index out of range
INFO:MPP-Solar:Command execution failed
INFO:MPP-Solar:attempting to create output processor: screen
INFO:MPP-Solar:Using output processor: screen
Parameter                       Value           Unit
ERROR                           Serial command execution failed

jblance, 2020-11-26:
Ok, it's getting responses ok, but the decode is failing

When I get back I'll add your responses as test cases and fix whatever is
the problem.

What does watchpower describe the 2 inverters as? Eg '1st phase of 3 phase'
etc

BarkinSpider, 2020-11-26:
[image]
[image]

BarkinSpider, 2020-11-26:
Im seeing so values are wrong. The battery type I have is USR (LifePO4) and the charge/float voltages are wrong.
(mpp-solar) [root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar) > bin/mpp-solar -p /dev/ttyUSB1
Parameter                       Value           Unit
ac_output_voltage               120.0           V
ac_output_frequency             60.0            Hz
max_ac_charging_current         30              A
battery_under_voltage           42.0            V
battery_float_charge_voltage    54.0            V
battery_bulk_charge_voltage     56.4            V
battery_recharge_voltage        46.0            V
max_charging_current            60              A
input_voltage_range             UPS
output_source_priority          Utility first
charger_source_priority         Solar + Utility
battery_type                    AGM
buzzer                          enabled
power_saving                    disabled
overload_restart                disabled
over_temperature_restart        disabled
lcd_backlight                   enabled
primary_source_interrupt_alarm  enabled
record_fault_code               disabled
overload_bypass                 disabled
lcd_reset_to_default            enabled
output_mode                     single machine output
battery_redischarge_voltage     54.0            V
pv_ok_condition                 As long as one unit of inverters has connect PV, parallel system will consider PV OK
pv_power_balance                PV input max power will be the sum of the max charged power and loads power
unknown_value_in_response_25    000
unknown_value_in_response_26    0

BarkinSpider, 2020-11-26:
Phase 1 of 2 phase output
Phase 2 of 2 phase output

Battery type is not AGM, its USR
charge voltages are wrong

jblance, 2020-11-26:
That looks like QDI which is default values from memory

Try -c QPIRI

BarkinSpider, 2020-11-26:
[root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar) > bin/mpp-solar -p /dev/ttyUSB1 -c QPIRI -I
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.1, Refactored version - not backwards compatible, no mpp_info_pub
INFO:MPP-Solar:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1" using protocol "PI30" for command "QPIRI" (tag: QPIRI)
INFO:MPP-Solar:Using serialio for communications
INFO:MPP-Solar:Running command QPIRI
INFO:MPP-Solar:Using protocol b'PI30' with 35 commands
INFO:MPP-Solar:full command b'QPIRI\xf8T\r' for command QPIRI
INFO:MPP-Solar:response passed to decode: b'(120.0 25.0 120.0 60.0 25.0 3000 3000 48.0 46.0 44.0 58.4 54.4 2 30 060 1 2 0 9 01 0 6 54.0 0 1 000 0\x8f\xed\r'
WARNING:MPP-Solar:Serial read error: list index out of range
INFO:MPP-Solar:Command execution failed
INFO:MPP-Solar:attempting to create output processor: screen
INFO:MPP-Solar:Using output processor: screen
Parameter                       Value           Unit
ERROR                           Serial command execution failed

jblance, 2020-11-26:
None of the responses or settings are from mppsolar they are all read from
the inverter

BarkinSpider, 2020-11-26:
you can see my settings in the images I posted from WatchPower, they are different to the mpp-solar output

jblance, 2020-11-26:
Bummer

Well once fixed that will tell you what the actual settings are

BarkinSpider, 2020-11-26:
Are changes to mppsolar required?

jblance, 2020-11-26:
Yes it needs the command definitions expanded to cater for 1st and 2nd of 2
phase as options

jblance, 2020-11-26:
ok update now and try 
`mpp-solar -p test -c QPIRI -I`

BarkinSpider, 2020-11-26:
v 0.7.3 working well on my MPP3048 parallel setup. Thanks!

[root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar) > bin/mpp-solar -c QPGS0
Command: QPGS0 - Parallel Information inquiry
------------------------------------------------------------
Parameter                       Value           Unit
parallel_instance_number        valid
serial_number                   92812005100038
work_mode                       Battery Mode
fault_code                      No fault
grid_voltage                    119.4           V
grid_frequency                  60.0            Hz
ac_output_voltage               120.7           V
ac_output_frequency             59.98           Hz
ac_output_apparent_power        1               VA
ac_output_active_power          1               W
load_percentage                 0               %
battery_voltage                 54.2            V
battery_charging_current        2               A
battery_capacity                100             %
pv_input_voltage                68.3            V
total_charging_current          9               A
total_ac_output_apparent_power  36              VA
total_output_active_power       35              W
total_ac_output_percentage      0               %
is_scc_ok                       1               True - 1/False - 0
is_ac_charging                  0               True - 1/False - 0
is_scc_charging                 1               True - 1/False - 0
is_battery_over_voltage         0               True - 1/False - 0
is_battery_under_voltage        0               True - 1/False - 0
is_line_lost                    0               True - 1/False - 0
is_load_on                      1               True - 1/False - 0
is_configuration_changed        0               True - 1/False - 0
output_mode                     Phase 2 of 2 phase output
charger_source_priority         Utility first
max_charger_current             60              A
max_charger_range               140             A
max_ac_charger_current          30              A
pv_input_current                2               A
battery_discharge_current       0               A
[root@pi4]:(/home/joe/Dev/MPP3048/mpp-solar) > bin/mpp-solar -c QPGS1
Command: QPGS1 - Parallel Information inquiry
------------------------------------------------------------
Parameter                       Value           Unit
parallel_instance_number        valid
serial_number                   92812005100035
work_mode                       Battery Mode
fault_code                      No fault
grid_voltage                    122.5           V
grid_frequency                  60.03           Hz
ac_output_voltage               121.0           V
ac_output_frequency             60.06           Hz
ac_output_apparent_power        36              VA
ac_output_active_power          34              W
load_percentage                 1               %
battery_voltage                 54.3            V
battery_charging_current        7               A
battery_capacity                100             %
pv_input_voltage                74.8            V
total_charging_current          9               A
total_ac_output_apparent_power  36              VA
total_output_active_power       34              W
total_ac_output_percentage      0               %
is_scc_ok                       1               True - 1/False - 0
is_ac_charging                  0               True - 1/False - 0
is_scc_charging                 1               True - 1/False - 0
is_battery_over_voltage         0               True - 1/False - 0
is_battery_under_voltage        0               True - 1/False - 0
is_line_lost                    0               True - 1/False - 0
is_load_on                      1               True - 1/False - 0
is_configuration_changed        0               True - 1/False - 0
output_mode                     Phase 1 of 2 phase output
charger_source_priority         Utility first
max_charger_current             60              A
max_charger_range               140             A
max_ac_charger_current          30              A
pv_input_current                7               A
battery_discharge_current       0               A

jblance, 2020-11-26:
Good news
Note there are a few output processors including a mqtt one that may help

closed 2020-11-26 by jblance