jblance/mpp-solar #44: pv_input_current appear to be an integer, needs to be a float

BarkinSpider, 2020-11-26:
pv_input_current appear to be an integer, needs to be a float. Watchpower shows 40W of power at 70v = 0.57a.
mpp-solar shows 0 for pv_input_current if it less than 1amp
version: 0.7.3 on pi4 
Linux pi4 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

BarkinSpider, 2020-11-27:
[image]

jblance, 2020-11-27:
> pv_input_current appear to be an integer, needs to be a float. Watchpower shows 40W of power at 70v = 0.57a.
> mpp-solar shows 0 for pv_input_current if it less than 1amp
> version: 0.7.3 on pi4
> Linux pi4 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

Can you run the command with -I, as the results i have seen are ints (not floats) - perhaps watchpower is calculating the value...
Happy to make it a float, but this wont help if the info is an int

jblance, 2020-11-27:
Nice dashboard!!

BarkinSpider, 2020-11-27:
Yes, I see its an integer. Watchpower must use another command to get the fractional amps to get the accurate wattage. Any idea how it does this?

Also, Is there a way to see if the charger is in Bulk or Float mode?

jblance, 2020-11-27:
Try QPIGS for pv input power
Q1 is they only command i see in PI30 for charge mode, but not all inverters support it.
Does watchpower give charge mode? Might be worth capturing a full run and seeing if there is a command that I haven't implemented.
There are protocol references in the docs directory

closed 2021-01-10 by jblance