jblance/mpp-solar #47: Can't install on OrangePi Zero

irmajavi, 2020-12-20:
I tried to install it on an OrangePi Zero but I get the following error.

root@SG-INVERTER:~/mpp-solar# sudo python3 ./setup.py install
/usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'long_description_content_type'
  warnings.warn(msg)
running install
Checking .pth file support in /usr/local/lib/python3.4/dist-packages/
/usr/bin/python3 -E -c pass
TEST PASSED: /usr/local/lib/python3.4/dist-packages/ appears to support .pth files
running bdist_egg
running egg_info
writing mpp_solar.egg-info/PKG-INFO
writing dependency_links to mpp_solar.egg-info/dependency_links.txt
writing top-level names to mpp_solar.egg-info/top_level.txt
writing entry points to mpp_solar.egg-info/entry_points.txt
writing requirements to mpp_solar.egg-info/requires.txt
reading manifest file 'mpp_solar.egg-info/SOURCES.txt'
writing manifest file 'mpp_solar.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/mppsolar
copying build/lib/mppsolar/version.py -> build/bdist.linux-armv7l/egg/mppsolar
copying build/lib/mppsolar/__init__.py -> build/bdist.linux-armv7l/egg/mppsolar
creating build/bdist.linux-armv7l/egg/mppsolar/io
copying build/lib/mppsolar/io/hidrawio.py -> build/bdist.linux-armv7l/egg/mppsolar/io
copying build/lib/mppsolar/io/testio.py -> build/bdist.linux-armv7l/egg/mppsolar/io
copying build/lib/mppsolar/io/esp32io.py -> build/bdist.linux-armv7l/egg/mppsolar/io
copying build/lib/mppsolar/io/__init__.py -> build/bdist.linux-armv7l/egg/mppsolar/io
copying build/lib/mppsolar/io/baseio.py -> build/bdist.linux-armv7l/egg/mppsolar/io
copying build/lib/mppsolar/io/serialio.py -> build/bdist.linux-armv7l/egg/mppsolar/io
copying build/lib/mppsolar/io/jkbleio.py -> build/bdist.linux-armv7l/egg/mppsolar/io
copying build/lib/mppsolar/io/jkbledelegate.py -> build/bdist.linux-armv7l/egg/mppsolar/io
creating build/bdist.linux-armv7l/egg/mppsolar/protocols
copying build/lib/mppsolar/protocols/jk04.py -> build/bdist.linux-armv7l/egg/mppsolar/protocols
copying build/lib/mppsolar/protocols/protocol.py -> build/bdist.linux-armv7l/egg/mppsolar/protocols
copying build/lib/mppsolar/protocols/pi30.py -> build/bdist.linux-armv7l/egg/mppsolar/protocols
copying build/lib/mppsolar/protocols/__init__.py -> build/bdist.linux-armv7l/egg/mppsolar/protocols
copying build/lib/mppsolar/protocols/jkabstractprotocol.py -> build/bdist.linux-armv7l/egg/mppsolar/protocols
copying build/lib/mppsolar/protocols/protocol_helpers.py -> build/bdist.linux-armv7l/egg/mppsolar/protocols
copying build/lib/mppsolar/protocols/pi16.py -> build/bdist.linux-armv7l/egg/mppsolar/protocols
copying build/lib/mppsolar/protocols/pi18.py -> build/bdist.linux-armv7l/egg/mppsolar/protocols
copying build/lib/mppsolar/protocols/pi41.py -> build/bdist.linux-armv7l/egg/mppsolar/protocols
copying build/lib/mppsolar/protocols/pi00.py -> build/bdist.linux-armv7l/egg/mppsolar/protocols
copying build/lib/mppsolar/protocols/jk02.py -> build/bdist.linux-armv7l/egg/mppsolar/protocols
creating build/bdist.linux-armv7l/egg/mppsolar/devices
copying build/lib/mppsolar/devices/__init__.py -> build/bdist.linux-armv7l/egg/mppsolar/devices
copying build/lib/mppsolar/devices/device.py -> build/bdist.linux-armv7l/egg/mppsolar/devices
copying build/lib/mppsolar/devices/jk24s.py -> build/bdist.linux-armv7l/egg/mppsolar/devices
copying build/lib/mppsolar/devices/mppsolar.py -> build/bdist.linux-armv7l/egg/mppsolar/devices
creating build/bdist.linux-armv7l/egg/mppsolar/outputs
copying build/lib/mppsolar/outputs/influx2_mqtt.py -> build/bdist.linux-armv7l/egg/mppsolar/outputs
copying build/lib/mppsolar/outputs/hass_mqtt.py -> build/bdist.linux-armv7l/egg/mppsolar/outputs
copying build/lib/mppsolar/outputs/screen.py -> build/bdist.linux-armv7l/egg/mppsolar/outputs
copying build/lib/mppsolar/outputs/hass_config.py -> build/bdist.linux-armv7l/egg/mppsolar/outputs
copying build/lib/mppsolar/outputs/__init__.py -> build/bdist.linux-armv7l/egg/mppsolar/outputs
copying build/lib/mppsolar/outputs/influx_mqtt.py -> build/bdist.linux-armv7l/egg/mppsolar/outputs
copying build/lib/mppsolar/outputs/mqtt.py -> build/bdist.linux-armv7l/egg/mppsolar/outputs
byte-compiling build/bdist.linux-armv7l/egg/mppsolar/version.py to version.cpython-34.pyc
byte-compiling build/bdist.linux-armv7l/egg/mppsolar/__init__.py to __init__.cpython-34.pyc
  File "build/bdist.linux-armv7l/egg/mppsolar/__init__.py", line 33
    log.info(f"attempting to create output processor: {output}")
                                                              ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-armv7l/egg/mppsolar/io/hidrawio.py to hidrawio.cpython-34.pyc
  File "build/bdist.linux-armv7l/egg/mppsolar/io/hidrawio.py", line 18
    log.info(f"full command {full_command} for command {command}")
                                                                ^
[... 168 lines cut ...]

root@SG-INVERTER:~/mpp-solar# mpp-solar -h
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 9, in <module>
    load_entry_point('mpp-solar==0.7.8', 'console_scripts', 'mpp-solar')()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2222, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 2164, in _find_spec
  File "<frozen importlib._bootstrap>", line 1940, in find_spec
  File "<frozen importlib._bootstrap>", line 1916, in _get_spec
  File "<frozen importlib._bootstrap>", line 1897, in _legacy_get_spec
  File "<frozen importlib._bootstrap>", line 863, in spec_from_loader
  File "<frozen importlib._bootstrap>", line 904, in spec_from_file_location
  File "/usr/local/lib/python3.4/dist-packages/mpp_solar-0.7.8-py3.4.egg/mppsolar/__init__.py", line 33
    log.info(f"attempting to create output processor: {output}")
                                                              ^
SyntaxError: invalid syntax

Any help?

jblance, 2020-12-20:
> File "", line 2237, in _find_and_load
> File "", line 2222, in _find_and_load_unlocked
> File "", line 2164, in _find_spec
> File "", line 1940, in find_spec
> File "", line 1916, in _get_spec
> File "", line 1897, in _legacy_get_spec
> File "", line 863, in spec_from_loader
> File "", line 904, in spec_from_file_location

I'm pretty sure the f strings are python3.6 or later only

Can the orangepi update to python3.6?

irmajavi, 2020-12-20:
Yes. I update to python 3.6.5. But same error happened. I do not know how to install it using python 3.6. or uninstall previous versions. Any suggestions?

jblance, 2020-12-20:
what does ```python3 --version``` give?

irmajavi, 2020-12-20:
Python 3.4.2

jblance, 2020-12-20:
try ```ls -l `which python3`* ```
e.g.
```
$ ls -l `which python3`*
lrwxrwxrwx 1 root root       9 Nov 26 05:54 /usr/bin/python3 -> python3.8
-rwxr-xr-x 1 root root 5266168 Sep 25 21:36 /usr/bin/python3.8
lrwxrwxrwx 1 root root      33 Sep 25 21:36 /usr/bin/python3.8-config -> x86_64-linux-gnu-python3.8-config
lrwxrwxrwx 1 root root      16 Oct  6 23:28 /usr/bin/python3-config -> python3.8-config
-rwxr-xr-x 1 root root     384 Aug  6 03:56 /usr/bin/python3-futurize
-rwxr-xr-x 1 root root     388 Aug  6 03:56 /usr/bin/python3-pasteurize
```

(if you have installed python 3.6, python3 might not be pointing to it - so you can either change the link or perhaps there is another executable, like python3.6)

jblance, 2020-12-20:
easier might be to try
```python3.6 --version```

irmajavi, 2020-12-20:
It says python 3.6.5

jblance, 2020-12-20:
so where the instructions say python3 use python3.6

first remove the existing install(s)
```
rm -rf /usr/local/lib/python3.4/dist-packages/mpp_solar*
```

then install using python3.6
```sudo python3.6 ./setup.py install```
and try running something simple
```mpp-solar -h```

irmajavi, 2020-12-20:
Yeah it works. Thanks for your time and help and for making this program.
Also thanks to your team.

jblance, 2020-12-20:
No problems.
Hope it is useful to you.
Just a note that there are a bunch of different output processors so if you want to feed the data somewhere they are the bits to look at (by default it just outputs to the screen)

closed 2020-12-20 by jblance