jblance/mpp-solar #50: mppsolar service not starting

irmajavi, 2020-12-27:
Hi I'm getting this error when trying to start mpp-solar service

root@SG-INVERTER:~# systemctl --user status mpp-solar
● mpp-solar.service - MPP Solar Service
   Loaded: loaded (/etc/systemd/user/mpp-solar.service; enabled)
   Active: activating (start) since Sun 2020-12-27 17:16:18 AST; 2s ago
 Main PID: 22997 (python3.6)
   CGroup: /user.slice/user-0.slice/user@0.service/mpp-solar.service
           └─22997 /usr/local/bin/python3.6 /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf

Dec 27 17:16:18 SG-INVERTER systemd[475]: Stopping MPP Solar Service...
Dec 27 17:16:18 SG-INVERTER systemd[475]: Starting MPP Solar Service...
root@SG-INVERTER:~# systemctl --user daemon-reload
root@SG-INVERTER:~# systemctl --user restart mpp-solar
Job for mpp-solar.service failed. See 'systemctl status mpp-solar.service' and 'journalctl -xn' for details.
root@SG-INVERTER:~# journalctl -xn
-- Logs begin at Sun 2020-12-27 16:33:51 AST, end at Sun 2020-12-27 17:19:04 AST. --
Dec 27 17:19:03 SG-INVERTER python3.6[23250]: op.output(data=results, tag=_tag, mqtt_broker=mqtt_broker)
Dec 27 17:19:03 SG-INVERTER python3.6[23250]: File "/usr/local/lib/python3.6/site-packages/mppsolar/outputs/hass_mqtt.py", line 21, in output
Dec 27 17:19:03 SG-INVERTER python3.6[23250]: data.pop("_command", None)
Dec 27 17:19:03 SG-INVERTER python3.6[23250]: AttributeError: 'str' object has no attribute 'pop'
Dec 27 17:19:04 SG-INVERTER systemd[475]: mpp-solar.service: main process exited, code=exited, status=1/FAILURE
Dec 27 17:19:04 SG-INVERTER systemd[475]: Failed to start MPP Solar Service.
-- Subject: Unit UNIT has failed
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit UNIT has failed.
--
-- The result is failed.
Dec 27 17:19:04 SG-INVERTER systemd[475]: Unit mpp-solar.service entered failed state.
Dec 27 17:19:04 SG-INVERTER systemd[475]: mpp-solar.service holdoff time over, scheduling restart.
Dec 27 17:19:04 SG-INVERTER systemd[475]: Stopping MPP Solar Service...
-- Subject: Unit UNIT has begun shutting down
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit UNIT has begun shutting down.
Dec 27 17:19:04 SG-INVERTER systemd[475]: Starting MPP Solar Service...
-- Subject: Unit UNIT has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit UNIT has begun starting up.

jblance, 2020-12-27:
What version are you running?
Can you post your config file
Can you try running it on the command line with outputs=screen and debug on

irmajavi, 2020-12-27:
# Required section
[SETUP]
# Number of seconds to pause at the end of each loop (0 is no pause)
pause=5
# Hostname / IP address of the MQTT broker
mqtt_broker=192.168.2.36:1883

[Inverter_1]
model=standard
protocol=PI30
port=/dev/hidraw0
baud=2400
command=QPGS1
outputs=hass_mqtt,influx2_mqtt

I'm using latest version 0.4.6

in command line works fine, I just want to make everything automatic.

root@SG-INVERTER:~#
mpp-solar -P PI30 -q 192.168.2.36 -p test -o screen,hass_mqtt,influx2_mqtt -c QPIGS
Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                       Value           Unit
ac_input_voltage                0.0             V
ac_input_frequency              0.0             Hz
ac_output_voltage               230.0           V
ac_output_frequency             49.9            Hz
ac_output_apparent_power        161             VA
ac_output_active_power          119             W
ac_output_load                  3               %
bus_voltage                     460             V
battery_voltage                 57.5            V
battery_charging_current        12              A
battery_capacity                100             %
inverter_heat_sink_temperature  69              °C
pv_input_current_for_battery    14.0            A
pv_input_voltage                103.8           V
battery_voltage_from_scc        57.45           V
battery_discharge_current       0               A
is_sbu_priority_version_added   0               True - 1/False - 0
is_configuration_changed        0               True - 1/False - 0
is_scc_firmware_updated         1               True - 1/False - 0
is_load_on                      1               True - 1/False - 0
is_battery_voltage_to_steady_while_charging     0               True - 1/False - 0
is_charging_on                  1               True - 1/False - 0
is_scc_charging_on              1               True - 1/False - 0
is_ac_charging_on               0               True - 1/False - 0
rsv1                            0               A
rsv2                            0               A
pv_input_power                  856             W
is_charging_to_float            0               True - 1/False - 0
is_switched_on                  1               True - 1/False - 0
is_reserved                     0               True - 1/False - 0
root@SG-INVERTER:~#

jblance, 2020-12-27:
You can run the full series command on the command line and add debug to get more info, eg
```mpp-solar -C /etc/mpp-solar/mpp-solar.conf -D```

jblance, 2020-12-27:
I'm not sure the :1883 syntax is supported by the mqtt libraries, try removing that from the config file

irmajavi, 2020-12-27:
login as: root
root@192.168.2.36's password:

system info:
  Distro......: Debian GNU/Linux 8 (jessie)
  Kernel......: Linux 3.4.39
  Uptime......: up 5 hours, 14 minutes
  Load........: 4.03 (1m), 4.13 (5m), 4.24 (15m)
  Processes...: 135 (root), 11 (user), 146 (total)
  CPU.........:  (5 vCPU)
  CPU Temp....: 0°C
  Memory......: 436M used, 10M free, 447M total

service status:
  influx: ● active telegraf: ● active
  mosquitto: ● active

disk usage:
  /                               8% used out of  31G
  [==================================================]
  /media/boot                     8% used out of  67M
  [==================================================]
Last login: Sun Dec 27 18:18:23 2020 from laptop-306eplq6
root@SG-INVERTER:~# mpp-solar -C /etc/mpp-solar/mpp-solar.conf -D
INFO:MPP-Solar:MPP Solar Command Utility, version: 0.7.8, Refactored version - not backwards compatible, add jkbms (02 and 04) support
DEBUG:MPP-Solar:args.configfile is true: /etc/mpp-solar/mpp-solar.conf
DEBUG:MPP-Solar:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
INFO:MPP-Solar:Using testio for communications
DEBUG:MPP-Solar:device.set_protocol with protocol PI30
DEBUG:MPP-Solar:mppsolar __init__ name Inverter_1, port <mppsolar.io.testio.TestIO object at 0xb6248410>, protocol <mppsolar.protocols.pi30.pi30 object at 0xb6248ed0>
DEBUG:MPP-Solar:mppsolar __init__ args ()
DEBUG:MPP-Solar:mppsolar __init__ kwargs {'name': 'Inverter_1', 'port': 'test', 'protocol': 'PI30', 'outputs': 'hass_mqtt,influx2_mqtt'}
INFO:MPP-Solar:Command line using config file: /etc/mpp-solar/mpp-solar.conf
DEBUG:MPP-Solar:getting results from device: mppsolar device - name: Inverter_1, port: <mppsolar.io.testio.TestIO object at 0xb6248410>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb6248ed0> for command: QPIGS, tag: None, outputs: hass_mqtt,influx2_mqtt
INFO:MPP-Solar:Running command QPIGS
INFO:MPP-Solar:Using protocol b'PI30' with 35 commands
DEBUG:MPP-Solar:get_command_defn for: QPIGS
DEBUG:MPP-Solar:Found command QPIGS in protocol b'PI30'
DEBUG:MPP-Solar:Calculating CRC for b'QPIGS'
DEBUG:MPP-Solar:Generated CRC 0xb7 0xa9 0xb7a9
DEBUG:MPP-Solar:full command: b'QPIGS\xb7\xa9\r'
INFO:MPP-Solar:full command b'QPIGS\xb7\xa9\r' for command QPIGS
DEBUG:MPP-Solar:get_command_defn for: QPIGS
DEBUG:MPP-Solar:Found command QPIGS in protocol b'PI30'
DEBUG:MPP-Solar:Raw response b'(000.0 00.0 230.0 49.9 0161 0119 003 460 57.50 012 100 0069 0014 103.8 57.45 00000 00110110 00 00 00856 010$\x8c\r'
INFO:MPP-Solar:response passed to decode: b'(000.0 00.0 230.0 49.9 0161 0119 003 460 57.50 012 100 0069 0014 103.8 57.45 00000 00110110 00 00 00856 010$\x8c\r'
DEBUG:MPP-Solar:trimmed and split responses: [b'000.0', b'00.0', b'230.0', b'49.9', b'0161', b'0119', b'003', b'460', b'57.50', b'012', b'100', b'0069', b'0014', b'103.8', b'57.45', b'00000', b'00110110', b'00', b'00', b'00856', b'010']
DEBUG:MPP-Solar:Decoded response {'ac_input_voltage': [0.0, 'V'], 'ac_input_frequency': [0.0, 'Hz'], 'ac_output_voltage': [230.0, 'V'], 'ac_output_frequency': [49.9, 'Hz'], 'ac_output_apparent_power': [161, 'VA'], 'ac_output_active_power': [119, 'W'], 'ac_output_load': [3, '%'], 'bus_voltage': [460, 'V'], 'battery_voltage': [57.5, 'V'], 'battery_charging_current': [12, 'A'], 'battery_capacity': [100, '%'], 'inverter_heat_sink_temperature': [69, '°C'], 'pv_input_current_for_battery': [14.0, 'A'], 'pv_input_voltage': [103.8, 'V'], 'battery_voltage_from_scc': [57.45, 'V'], 'battery_discharge_current': [0, 'A'], 'is_sbu_priority_version_added': [0, 'True - 1/False - 0'], 'is_configuration_changed': [0, 'True - 1/False - 0'], 'is_scc_firmware_updated': [1, 'True - 1/False - 0'], 'is_load_on': [1, 'True - 1/False - 0'], 'is_battery_voltage_to_steady_while_charging': [0, 'True - 1/False - 0'], 'is_charging_on': [1, 'True - 1/False - 0'], 'is_scc_charging_on': [1, 'True - 1/False - 0'], 'is_ac_charging_on': [0, 'True - 1/False - 0'], 'rsv1': [0, 'A'], 'rsv2': [0, 'A'], 'pv_input_power': [856, 'W'], 'is_charging_to_float': [0, 'True - 1/False - 0'], 'is_switched_on': [1, 'True - 1/False - 0'], 'is_reserved': [0, 'True - 1/False - 0']}
INFO:MPP-Solar:Decoded response {'ac_input_voltage': [0.0, 'V'], 'ac_input_frequency': [0.0, 'Hz'], 'ac_output_voltage': [230.0, 'V'], 'ac_output_frequency': [49.9, 'Hz'], 'ac_output_apparent_power': [161, 'VA'], 'ac_output_active_power': [119, 'W'], 'ac_output_load': [3, '%'], 'bus_voltage': [460, 'V'], 'battery_voltage': [57.5, 'V'], 'battery_charging_current': [12, 'A'], 'battery_capacity': [100, '%'], 'inverter_heat_sink_temperature': [69, '°C'], 'pv_input_current_for_battery': [14.0, 'A'], 'pv_input_voltage': [103.8, 'V'], 'battery_voltage_from_scc': [57.45, 'V'], 'battery_discharge_current': [0, 'A'], 'is_sbu_priority_version_added': [0, 'True - 1/False - 0'], 'is_configuration_changed': [0, 'True - 1/False - 0'], 'is_scc_firmware_updated': [1, 'True - 1/False - 0'], 'is_load_on': [1, 'True - 1/False - 0'], 'is_battery_voltage_to_steady_while_charging': [0, 'True - 1/False - 0'], 'is_charging_on': [1, 'True - 1/False - 0'], 'is_scc_charging_on': [1, 'True - 1/False - 0'], 'is_ac_charging_on': [0, 'True - 1/False - 0'], 'rsv1': [0, 'A'], 'rsv2': [0, 'A'], 'pv_input_power': [856, 'W'], 'is_charging_to_float': [0, 'True - 1/False - 0'], 'is_switched_on': [1, 'True - 1/False - 0'], 'is_reserved': [0, 'True - 1/False - 0'], '_command': 'QPIGS', '_command_description': 'General Status Parameters inquiry'}
[... cut ...] alse - 0'], 'is_reserved': [0, 'True - 1/False - 0'], '_command': 'QPIGS', '_command_description': 'General Status Parameters inquiry'}
INFO:MPP-Solar:attempting to create output processor: hass_mqtt
DEBUG:MPP-Solar:processor.hass_mqtt __init__ kwargs {}
INFO:MPP-Solar:attempting to create output processor: influx2_mqtt
DEBUG:MPP-Solar:processor.influx2_mqtt __init__ kwargs {}
INFO:MPP-Solar:Using output processor: hass_mqtt
INFO:MPP-Solar:Using output processor: influx2_mqtt
root@SG-INVERTER:~#

jblance, 2020-12-27:
That worked, did you change the config file?
You might need to install the service as a system service if you have to run the commands as root

irmajavi, 2020-12-27:
yes is working now. yes i change the config. thanks again.

closed 2020-12-28 by jblance