jblance/mpp-solar #53: Not able to connect to MPP Hybrid LV2424 Inverter

danb35, 2021-01-06:
This looks similar to #25, but that issue is closed, and I'm not sure it's actually dealing with the same model (as the model numbering is still a little confusing to me).  The "model name" is marked (both on the box and on the inverter itself) as "HV LV 2424."  Using release 0.4.6, I'm not able to connect to--or at least retrieve useful information from--the inverter.  Using some of the troubleshooting commands in the other issue, it looks like it's using protocol 18:
```
pi@solpiplog:~ $ sudo mpp-solar -c PI -M PI18 -d /dev/hidraw0
protocol_version              	18              
```
But it won't even retrieve the device's serial number:
```
pi@solpiplog:~ $ sudo mpp-solar -M PI18 -d /dev/hidraw0
serial_number                 	                
```
And an information query returns an error:
```
pi@solpiplog:~ $ sudo mpp-solar -c QPIRI -M PI18 -d /dev/hidraw0
error                         	Invalid byte_response 
response                      	                
```
Debug information below:
```
pi@solpiplog:~ $ sudo mpp-solar -c PI -D -M PI18 -d /dev/hidraw0
INFO:MPP-Solar:command PI
INFO:MPP-Solar:Serial device used: /dev/hidraw0, baud rate: 2400
DEBUG:MPP-Solar:Device matches hidraw regex
INFO:MPP-Solar:Loading commands for inverter model: PI18
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command ET supported by model PI18
INFO:MPP-Solar:... command ET loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P005ET
DEBUG:MPP-Solar:Calculating CRC for b'^P005ET'
DEBUG:MPP-Solar:Generated CRC 4e 91 4e91
DEBUG:MPP-Solar:Full byte command: b'^P005ETN\x91\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command GS supported by model PI18
INFO:MPP-Solar:... command GS loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P005GS
DEBUG:MPP-Solar:Calculating CRC for b'^P005GS'
DEBUG:MPP-Solar:Generated CRC 58 14 5814
DEBUG:MPP-Solar:Full byte command: b'^P005GSX\x14\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MOD supported by model PI18
INFO:MPP-Solar:... command MOD loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P006MOD
DEBUG:MPP-Solar:Calculating CRC for b'^P006MOD'
DEBUG:MPP-Solar:Generated CRC dd be ddbe
DEBUG:MPP-Solar:Full byte command: b'^P006MOD\xdd\xbe\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command PI supported by model PI18
INFO:MPP-Solar:... command PI loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P005PI
DEBUG:MPP-Solar:Calculating CRC for b'^P005PI'
DEBUG:MPP-Solar:Generated CRC 71 8b 718b
DEBUG:MPP-Solar:Full byte command: b'^P005PIq\x8b\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.7/dist-packages/mpp_solar-0.4.6-py3.7.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports 
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
[... 287 lines cut ...]
DEBUG:MPP-Solar:Generate full byte command for QVFW2
DEBUG:MPP-Solar:Calculating CRC for b'QVFW2'
DEBUG:MPP-Solar:Generated CRC c3 f5 c3f5
DEBUG:MPP-Solar:Full byte command: b'QVFW2\xc3\xf5\r'
DEBUG:MPP-Solar:Searching for cmd 'PI'
INFO:MPP-Solar:DIRECT USB connection: executing PI
Device Protocol Version inquiry
 -- queries the device protocol version

DEBUG:MPP-Solar:length of to_send: 10
DEBUG:MPP-Solar:2 chunk send
DEBUG:MPP-Solar:usb byte_response was: b'^D00518;\x03\r'
DEBUG:MPP-Solar:Byte_Response length: 10
DEBUG:MPP-Solar:CRC resp	 59, 3
DEBUG:MPP-Solar:Calculating CRC for b'^D00518'
DEBUG:MPP-Solar:Generated CRC 3b 3 3b03
DEBUG:MPP-Solar:CRC calc	 59 3
DEBUG:MPP-Solar:CRCs match
DEBUG:MPP-Solar:getResponse with protocol = PI18
DEBUG:MPP-Solar:Response valid as no invalid situations found
DEBUG:MPP-Solar:getResponse with protocol = PI18
DEBUG:MPP-Solar:getResponse with protocol = PI18
protocol_version              	18              
pi@solpiplog:~ $ 
```

danb35, 2021-01-06:
When passing `-c QPIRI` instead, the last section of the DEBUG output reads as follows:
```
DEBUG:MPP-Solar:Searching for cmd 'QPIRI'
INFO:MPP-Solar:DIRECT USB connection: executing QPIRI
Current Settings inquiry
 -- queries the current settings from the Inverter

DEBUG:MPP-Solar:length of to_send: 8
DEBUG:MPP-Solar:1 chunk send
DEBUG:MPP-Solar:usb byte_response was: b'^0\x1b\xe3\r'
DEBUG:MPP-Solar:Byte_Response length: 5
DEBUG:MPP-Solar:CRC resp	 27, 227
DEBUG:MPP-Solar:Calculating CRC for b'^0'
DEBUG:MPP-Solar:Generated CRC 1b e3 1be3
DEBUG:MPP-Solar:CRC calc	 27 227
DEBUG:MPP-Solar:CRCs match
DEBUG:MPP-Solar:getResponse with protocol = PI18
DEBUG:MPP-Solar:Response invalid as insufficient number of elements in byte_response. Got 1, expected as least 25
INFO:MPP-Solar:Invalid byte_response
DEBUG:MPP-Solar:getResponse with protocol = PI18
error                         	Invalid byte_response 
response                      	                
pi@solpiplog:~ $ 
```

jblance, 2021-01-06:
Pi18 has different commands
Try one of the below instead of PI
GS
ET
MOD

jblance, 2021-01-06:
Probably better to use the latest code for your model of inverter to be honest (even though it is marked unstable)

danb35, 2021-01-06:
Ah, that looks better:
```
pi@solpiplog:~ $ mpp-solar -c GS -M PI18 -d /dev/hidraw0
ac_output_active_power        	1014            W
ac_output_apparent_power      	1097            VA
ac_output_frequency           	60.0            Hz
ac_output_voltage             	120.6           V
battery_capacity              	100             %
battery_charging_current      	1               A
battery_discharge_current     	0               A
battery_power_direction       	charge          
battery_voltage               	27.1            V
battery_voltage_from_scc      	27.1            V
battery_voltage_from_scc2     	0.0             V
dc/ac_power_direction         	AC-DC           
grid_frequency                	60.0            Hz
grid_voltage                  	120.6           V
inverter_heat_sink_temperature	33              oC
line_power_direction          	input           
load_connection               	connect         
local_parallel_id             	0               
mppt1_charger_status          	charging        
mppt1_charger_temperature     	38              oC
mppt2_charger_status          	abnormal        
mppt2_charger_temperature     	0               oC
output_load_percent           	42              %
pv1_input_power               	35              W
pv1_input_voltage             	65.6            V
pv2_input_power               	0               W
pv2_input_voltage             	0.0             V
setting_value_configuration_state	Something changed 
pi@solpiplog:~ $ mpp-solar -c ET -M PI18 -d /dev/hidraw0
total_generated_energy        	317             KWh
pi@solpiplog:~ $ mpp-solar -c MOD -M PI18 -d /dev/hidraw0
working_mode                  	Hybrid mode(Line mode, Grid mode) 
```
I'll give the new code a shot.  Is there a reference for Pi18?

jblance, 2021-01-06:
There is a doc in the docs directory and the new code should have an option to display avaliable commands (pi18 hasn't had many implemented as yet)

closed 2021-01-10 by jblance