jblance/mpp-solar #56: Support password-secured MQTT brokers

danb35, 2021-01-07:
According to the [docs page](https://github.com/jblance/mpp-solar/blob/master/docs/MQTT_Influx_Grafana.md) for MQTT/Influx/Grafana, "Password secured brokers are not currently supported."  I'm wanting to log to a [Kotori](https://getkotori.org/docs/) instance, which does implement a password for the MQTT broker, and I don't see that there's a way to bypass it.  Would appreciated adding this capability.

jblance, 2021-01-07:
I can try when I'm back from holiday.
Through the kotori example are all without password?
What is the format of the publish command you are using?

jblance, 2021-01-11:
added some options for username / password for mqtt messages - 0.7.10 or later
give it a go and let me know how it goes for you

closed 2021-02-16 by jblance