jblance/mpp-solar #61: After upgrading to version 0.7.22 hass_mqtt broken

nigel914, 2021-01-22:
` jkbms -n 520585004 -T 520585004 -p XX:XX:XX:XX:XX:XX -P JK02 -o  hass_mqtt -q IP ADDRESS --mqttuser username --mqttpass password
Traceback (most recent call last):
  File "/home/pi/.local/bin/jkbms", line 11, in <module>
    load_entry_point('mpp-solar', 'console_scripts', 'jkbms')()
  File "/home/pi/src/mpp-solar/mppsolar/__init__.py", line 390, in main
    topic=parser.prog,
  File "/home/pi/src/mpp-solar/mppsolar/outputs/mqtt.py", line 62, in output
    msgs = self.build_msgs(data=data, tag=tag, topic=topic)
  File "/home/pi/src/mpp-solar/mppsolar/outputs/hass_mqtt.py", line 37, in build_msgs
    payload = f'{{"name": "pm_{tag}_{key}", "state_topic": "homeassistant/sensor/pm_{tag}_{key}/state", "unit_of_measurement": "{unit}"}}'
NameError: name 'unit' is not defined
`

jblance, 2021-01-22:
oops
didnt actually know anyone was using that (is kinda a WIP)
do you use autoconfig for hass mqtt?
fixed the above error is 0.7.23 - but note that the autoconfig has been added to this output and it hasnt been tested

nigel914, 2021-01-22:
Hi @jblance thanks for the quick response. I don't use the autoconfig I configured all of the sensors manually in my hass config and works well I have multiple jkbms running a node red automation

jblance, 2021-01-22:
nice - i started doing that and realized that it was a PITA so started on autoconfig
unfortunately i killed my hass with an update and havent rebuilt as yet

did the fix work for you?

nigel914, 2021-01-22:
I have tested and this is now resolved with 0.7.23

jblance, 2021-01-22:
great

closed 2021-01-22 by jblance