jblance/mpp-solar #62: MPP-SOLAR  mpp-info-pub

Meganetpr, 2021-01-23:
Hi all, 

  Just starting and learning. I have an LV5048 with a Raspberry Pi 3 connected via serial port. Having difficult getting mpp-solar running as a service and the troubleshooting have a reference to mpp-info-hub and can't find it anywhere. What I'm missing?

/bin/sh: 1: /usr/local/bin/mpp-info-pub: not found

my conf file:
# Required section
[SETUP]
# Number of seconds to pause at the end of each loop (0 is no pause)
pause=5
#Hostname / IP address of the MQTT broker
#mqtt_broker=mqtthost
mqtt_broker=localhsot

# This example would work on a single LV5048
[Inverter1_L1]
model=LV5048
port=/dev/ttyUSB0
baud=2400
command=QPIGS
tag=Inverter1
#format=influx2

[Inverter1_L2]
model=standard
port=/dev/ttyUSB0
baud=2400
command=QPIGS2
tag=Inverter1
#format=influx2

  Thanks

DMIINC, 2021-01-23:
Hello. I am doing this from my phone. So bear with me. I will provide you with my conf file once I get back home. I have LV5048 just like you. But you have some lines wrong in your file.

Meganetpr, 2021-01-23:
Thanks, look forward to see yours conf file. I have commented and edited a few lines to troubleshoot it. I read the PI could connect direct to the USB port on the LV5048 , but don't want to get into it until sort the mpp-solar service to run.

DMIINC, 2021-01-23:
[SETUP]
# Number of seconds to pause between command execution loop
# 0 is no pause, greater than 60 will cause service restarts
pause=0
mqtt_broker=127.0.0.1

# This example would work on a single LV5048
[Inverter1_L1]
model=LV5048
protocol=PI41
port=/dev/ttyUSB1
baud=2400 
command=QPIGS
tag=Inverter1
outputs=influx2_mqtt

[Inverter1_L2]
model=LV5048
protocol=PI41
port=/dev/ttyUSB1 
baud=2400
command=QPIGS1
tag=Inverter1
outputs=influx2_mqtt

I prefer serial connection. USB is flaky on mpp prducts

Meganetpr, 2021-01-23:
Thanks a lot, will the missing mpp-info-pub be an issue?

jblance, 2021-01-23:
What version are you using?
mpp-info-pub has been removed, maybe there is a doc that needs updating

jblance, 2021-01-23:
Testing of the service in current versions can be done on the command line with the mpp-solar -C /path/to/conf.file

Meganetpr, 2021-01-25:
The version is current, and service is running OK. Working on graphics now. 
There is a few references to the mpp-info-pub on the troubleshoot documents
Thank you all

closed 2021-01-25 by Meganetpr