jblance/mpp-solar #69: Esp32

rdaguerre, 2021-02-09:
Anyone know if can i work in micropython esp32? Thanks

DMIINC, 2021-02-09:
jblance has started work here https://github.com/jblance/micropython-esp_mqtt_slave. It is possible.

jblance, 2021-02-09:
There was an initial attempt to run the whole thing on ESP - but I havent tested or tried it
I dont think it will work as it is on micropython, and would probably be best to implement a micropython specific version.

My current thinking is that the whole thing might be too big to run on ESP, and I've started the above to have a lightweight slave on ESP that listens for commands via MQTT

What is your usage scenario?

rdaguerre, 2021-02-09:
Hi, great job. I am just starting to use py. I have been using rpi with my mpp solar 5048gk solar inverter and I want to make it work on an esp32 wroom32 to give it stability and simplicity. I believe that it is possible with the base of your work. What you have in the mqtt slave, it seems to me that what is missing is something that generates the command so as not to have to make it more complex. Then, with the response of the command in the mqtt, you could work easier in HA or nodered for example. I also think that it would not be too heavy to make it return the most common digested parameters. I would very much like to help more but I'm just starting with py and esp32. I remain at your disposal in what I can help.
Ronnie (using Google translate.. 😉)

jblance, 2021-02-09:
The idea behind the slave is simplicity - it waits for a command, runs it and then returns the raw result
The main mpp-solar code would 'generate' the command (send it to mqtt) and wiat for the result, before sending the result to the relevant output(s)
The above is started but incomplete currently.

It would be achievable to have esp 'do it all' but probably with a focused and slimmed version of the full code (which is growing in complexity to allow for flexibility for different types of IO and Inverter protocols)

closed 2021-02-14 by jblance