jblance/mpp-solar #70: After upgrade jkbms comman now fails

nigel914, 2021-02-14:
`sudo  jkbms -n 520585004 -T 520585004 -p 3C:A5:4A:DE:F7:3C -P JK02 -o  hass_mqtt -D
INFO:MPP-Solar:Creating device "520585004" (type: "jkbms") on port "3C:A5:4A:DE:F7:3C (porttype=None)" using protocol "JK02" for command "None" (tag: 520585004)
DEBUG:MPP-Solar:device_class <class 'mppsolar.devices.jkbms.jkbms'>
DEBUG:MPP-Solar:jkbms __init__ args ()
DEBUG:MPP-Solar:jkbms __init__ kwargs {'name': '520585004', 'port': '3C:A5:4A:DE:F7:3C', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
DEBUG:MPP-Solar:device:get_port_type: port matches jkble ':'
INFO:MPP-Solar:Using jkbleio for communications
DEBUG:MPP-Solar:device.set_protocol with protocol JK02
DEBUG:MPP-Solar:jkbms __init__ name 520585004, port <mppsolar.io.jkbleio.JkBleIO object at 0x75ed5a50>, protocol <mppsolar.protocols.jk02.jk02 object at 0x75ed5a90>
DEBUG:MPP-Solar:Commands [(<mppsolar.devices.jkbms.jkbms object at 0x75ed59d0>, '', '520585004', 'hass_mqtt', None, None)]
INFO:MPP-Solar:Looping 1 commands
INFO:MPP-Solar:Getting results from device: jkbms device - name: 520585004, port: <mppsolar.io.jkbleio.JkBleIO object at 0x75ed5a50>, protocol: <mppsolar.protocols.jk02.jk02 object at 0x75ed5a90> for command: , tag: 520585004, outputs: hass_mqtt
INFO:MPP-Solar:JKBMS Running command
INFO:MPP-Solar:Using protocol b'JK02' with 2 commands
DEBUG:MPP-Solar:JkBLE get_command_defn for:
DEBUG:MPP-Solar:get_command_defn for:
INFO:MPP-Solar:No command_defn found for
DEBUG:MPP-Solar:self._command = , self._command_defn = None
DEBUG:MPP-Solar:No command_defn found
INFO:MPP-Solar:full command None for command
INFO:MPP-Solar:Using protocol b'JK02' with 2 commands
DEBUG:MPP-Solar:JkBLE get_command_defn for: None
DEBUG:MPP-Solar:get_command_defn: command is None
DEBUG:MPP-Solar:self._command = None, self._command_defn = None
DEBUG:MPP-Solar:No command_defn found
INFO:MPP-Solar:full command None for command None
DEBUG:MPP-Solar:JkBLE get_command_defn for: None
DEBUG:MPP-Solar:get_command_defn: command is None
Traceback (most recent call last):
  File "/usr/local/bin/jkbms", line 33, in <module>
    sys.exit(load_entry_point('mpp-solar', 'console_scripts', 'jkbms')())
  File "/home/pi/mpp-solar/src/mpp-solar/mppsolar/__init__.py", line 405, in main
    results = _device.run_command(command=_command)
  File "/home/pi/mpp-solar/src/mpp-solar/mppsolar/devices/jkbms.py", line 39, in run_command
    raw_response = self._port.send_and_receive(command=command, protocol=self._protocol)
  File "/home/pi/mpp-solar/src/mpp-solar/mppsolar/io/jkbleio.py", line 29, in send_and_receive
    record_type = command_defn["record_type"]
TypeError: 'NoneType' object is not subscriptable
`

nigel914, 2021-02-14:
The command changed  I need to add  -c getCellData

jblance, 2021-02-14:
that looks like a bug - it is supposed to default to a useful command
re-opening to remind me to fix it

jblance, 2021-02-15:
should be fixed in 0.7.34

closed 2021-02-15 by jblance