jblance/mpp-solar #71: New Protocol for new JK-BMS JK-B1A24S15P (fw 3.1.0 = Cryptograpĥic protocol)

nonoceb, 2021-02-16:
Hi,

just for warning everybody about new firmware revision that add crypto on serial bl protocol, so i was unable to collect data over RS485, CAN or even directly tapped on bl module RX TX.

I would like to connect an esp32 or esp8266 to this great piece of cell balancer but the team choose to doing that forbidden, wtf!?
If you really need to extract data from bms, just choose another brand ;)

Here bless boot log at startup not bl connected, tapped on bl module : 
rx bl module side : 
0A 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 0D 0A 3D 20 20 20 20 28 43 29 20 43 4F 50 59 52 49 47 48 54 20 32 30 32 30 20 43 68 65 6E 67 44 75 20 4A 69 4B 6F 6E 67 20 54 65 63 68 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 3D 0D 0A 3D 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 3D 0D 0A 3D 20 20 20 20 20 53 54 4D 33 32 46 31 30 33 78 43 38 20 62 6F 6F 74 6C 6F 61 64 65 72 20 28 56 65 72 73 69 6F 6E 20 31 2E 30 2E 33 29 20 20 20 20 20 20 20 20 20 20 20 20 20 20 3D 0D 0A 3D 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 3D 0D 0A 3D 20 20 20 42 75 69 6C 64 20 44 65 63 20 31 35 20 32 30 32 30 20 20 31 34 3A 32 35 3A 31 30 20 20 20 20 20 20 20 20 20 20 20 20 20 20 42 79 20 5A 68 61 6E 67 50 65 6E 67 20 20 3D 0D 0A 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 3D 0D 0A 0D 0A 54 0D 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 41 54 0D 0A 41 54 2B 42 41 55 44 38 0D 0A 41 54 0D 0A 41 54 2B 44 49 53 43 0D 0A 41 54 2B 4E 41 4D 45 4A 4B 2D 42 31 41 32 34 53 31 35 50 0D 0A

tx bl module side :
4F 4B 0D 0A 2B 42 41 55 44 3D 38 0D 0A 4F 4B 0D 0A 4F 4B 0D 0A 4F 4B 0D 0A

Then i connecting app BL to bms
rx bl module side : 
55 AA EB 90 03 00 4A 4B 2D 42 31 41 32 34 53 31 35 50 00 00 00 00 36 2E 58 00 00 00 00 00 36 2E 31 2E 30 53 00 00 79 69 00 00 37 00 00 00 4A 4B 2D 42 31 41 32 34 53 31 35 50 00 00 00 00 30 30 30 30 00 00 00 00 00 00 00 00 00 00 00 00 32 31 30 31 30 38 00 00 32 30 31 32 32 38 31 33 38 39 00 00 00 00 00 00 49 6E 70 75 74 20 55 73 65 72 64 61 74 61 00 00 30 30 30 30 30 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 DB AA 55 90 EB C8 01 01 00 00 00 00 00 00 00 00 00 00 00 00 44 55 AA EB 90 01 00 58 02 00 00 22 0B 00 00 80 0C 00 00 36 10 00 00 04 10 00 00 32 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 F0 0A 00 00 F0 49 02 00 1E 00 00 00 3C 00 00 00 48 E8 01 00 2C 01 00 00 3C 00 00 00 3C 00 00 00 2C 01 00 00 BC 02 00 00 58 02 00 00 BC 02 00 00 58 02 00 00 38 FF FF FF 9C FF FF FF 84 03 00 00 BC 02 00 00 03 00 00 00 01 00 00 00 01 00 00 00 01 00 00 00 70 17 00 00 DC 05 00 00 DC 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3E AA 55 90 EB C8 01 01 00 00 00 00 00 00 00 00 00 00 00 00 44 55 AA EB 90 02 00 AD 0E 52 0E 53 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 07 00 00 00 70 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 51 2B 00 00 00 00 00 00 00 00 00 00 30 F8 30 F8 52 01 00 00 2A 01 00 00 C2 14 00 00 70 17 00 00 00 00 00 00 8F 01 00 00 00 00 60 07 A8 69 00 00 00 00 CB 06 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 11 04 00 00 00 00 92 4A 3B 40 00 00 00 00 14 09 00 00 00 00 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 55 AA EB 90 02 01 AD 0E 52 0E 53 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 07 00 00 00 70 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 51 2B 00 00 00 00 00 00 00 00 00 00 30 F8 30 F8 52 01 00 00 28 01 00 00 C2 14 00 00 70 17 00 00 00 00 00 00 8F 01 00 00 00 00 5F 07 A9 69 00 00 00 00 CB 06 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 11 04 00 00 00 00 92 4A 3B 40 00 00 00 00 46 08 00 00 00 00 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 B9 55 AA EB 90 02 02 AD 0E 52 0E 53 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 07 00 00 00 70 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 51 2B 00 00 00 00 00 00 00 00 00 00 30 F8 30 F8 52 01 00 00 48 01 00 00 C2 14 00 00 70 17 00 00 00 00 00 00 8F 01 00 00 00 00 6F 07 A9 69 00 00 00 00 CB 06 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 11 04 00 00 00 00 92 4A 3B 40 00 00 00 00 DF 07 00 00 00 00 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 82 55 AA EB 90 02 03 AD 0E 52 0E 53 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 07 00 00 00 70 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 51 2B 00 00 00 00 00 00 00 00 00 00 30 F8 30 F8 52 01 00 00 26 01 00 00 C2 14 00 00 70 17 00 00 00 00 00 00 8F 01 00 00 00 00 5E 07 AA 69 00 00 00 00 CB 06 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 11 04 00 00 00 00 92 4A 3B 40 00 00 00 00 AD 08 00 00 00 00 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 55 AA EB 90 02 04 AD 0E 52 0E 53 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 07 00 00 00 70 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 51 2B 00 00 00 00 00 00 00 00 00 00 30 F8 30 F8 52 01 00 00 48 01 00 00 C2 14 00 00 70 17 00 00 00 00 00 00 8F 01 00 00 00 00 6F 07 AA 69 00 00 00 00 CB 06 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 11 04 00 00 00 00 92 4A 3B 40 00 00 00 00 46 08 00 00 00 00 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0 [... cut ...] 0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0C

tx bl module side :
2B 43 4F 4E 4E 45 43 54 45 44 0D 0A AA 55 90 EB 97 00 03 A3 2B C9 94 0E 5A 91 E8 44 14 05 1C 99 AA 55 90 EB 96 00 9D 20 DB 6E B5 72 A1 A8 E0 44 56 69 83 EC 2B 44 49 53 43 4F 4E 4E 45 43 54 45 44 0D 0A

Some data are different at each connection (roling code?, chalenge code?) so i think the better is to stop using this crapware.
Sorry for this bad news, hope it's help to choose better bms

jblance, 2021-02-16:
Does the app still give results?

nonoceb, 2021-02-16:
yes the app give perfect data but it's QT5 android app, so hard to decompile to understand.
it's easyier for me to understand balancer hardware and make my own gpl :)

jblance, 2021-02-16:
bummer
is the hardware new, or just new firmware?
does the jkbms code give any data or just fail? as the above look to be the same overall data, but not encoded the same

nonoceb, 2021-02-16:
Yes it's near from the same data structure but i was unable to find my 3S cell voltage in the hex response.
I can record more data if you want but the simple fact the design team put "cryptography change" clearly in the versioning log at 3.1.0, adding the fact the support don't answer to this problem, the fact they don't want to sell a submodule to decode their encoding say to me i was against stupid people.

I was looking for a good bms for professional use, i was very happy with the old balancer device and the new bms looklike awsome but ... to good to be true :D

I see on board they use stm32 chip with bootloader for programming it and a litle memory chip to investigate a litle bit more.

jblance, 2021-02-16:
Hmm, what cells are you using?
Is the below sensible?
```
Command: getCellData - BLE Cell Data inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
header                        	55aaeb90       	    
record_type                   	02             	    
record_counter                	0              	    
voltage_cell01                	3.757          	V   
voltage_cell02                	3.666          	V   
voltage_cell03                	3.6670000000000003	V   
voltage_cell04                	0.0            	V   
voltage_cell05                	0.0            	V   
voltage_cell06                	0.0            	V   
voltage_cell07                	0.0            	V   
```

nonoceb, 2021-02-16:
lol you got it!

nonoceb, 2021-02-16:
ok i need a better understanding of your code ;) thanks a lot to give me some hope :D

jblance, 2021-02-16:
It's the jk02 format, ask if you have any questions

nonoceb, 2021-02-16:
ok thank you so much for your help and your code ;) I need now to understand my mistake on my side

jblance, 2021-02-16:
Picking one of the first cell data records in your above dump (starts 55 AA EB 90 02 00 AD 0E 52 0E 53 0E 00 00 )
The records start with `55 AA EB 90`
The there is a record type `02` is the type needed for cell voltages
Then there is a counter `00` 
Then there are 24 2byte data points for cell voltage 
`AD 0E` `52 0E` `53 0E` `00 00` etc

These are decoded by `decode2ByteHex` in protocol_helpers.py

jblance, 2021-02-16:
or as it turns out can be decoded with struct.unpack
```
t = bytes.fromhex('AD0E')
struct.unpack('<h', t)
(3757,)
```

well blow me down - the struct.unpack is kinda magic

nonoceb, 2021-02-17:
> Picking one of the first cell data records in your above dump (starts 55 AA EB 90 02 00 AD 0E 52 0E 53 0E 00 00 )
> The records start with `55 AA EB 90`
> Then there is a record type, `02` is the type needed for cell voltages
> Then there is a counter eg `00` in this instance
> Then there are 24 2byte data points for cell voltage
> `AD 0E` `52 0E` `53 0E` `00 00` etc
> 
> These are decoded by `decode2ByteHex` in protocol_helpers.py

That 's why i didn't find my cell voltage because i tried to use jk04 format instead jk02.
I think they comeback to jk02 to feed low speed rs485 at 9600bd and 1mv step voltage is enough.
The strange thing is RS485 and CAN  interface is totally silent, an helper at startup can do a lot without documentation...maybe it's not ready or implemented at this time or waiting for magic bytes input.

If i choose to replace bl module by esp8266, i think esp8266 draw a lot more current on internal isolated supply (300mA vs 10mA)

nonoceb, 2021-02-17:
> or as it turns out can be decoded with struct.unpack
> 
> ```
> t = bytes.fromhex('AD0E')
> struct.unpack('<h', t)
> (3757,)
> ```
> 
> well blow me down - the struct.unpack is kinda magic

i did understand the struct.unpack but from my point of view the real value voltage is 0E 52 = 3.666V
I dont use AD

jblance, 2021-02-24:
Is this what the app's results are?

closed 2021-03-07 by jblance

syssi, 2021-08-11:
@nonoceb Could you explain how did you capture the data? Did you use this (labeled) RX/TX pins of the BK3432 board: https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/images/jk-bd6a17s6p/005-labeled.jpg

Which baudrate should I expect?

nonoceb, 2021-08-12:
Hi it's the serial pins near antenna, i don't remember baud rate but it's a standard speed maybe 115200