jblance/mpp-solar #73: PIP2424MSE1, 220-240vac, 24v MPP solar inverter setup

santosh-mahadev, 2021-03-27:
Hi, I am trying to use this but unable to get data in influxdb
I install mpp-solar using command `sudo pip install mpp-solar`
and install MQTT / Influx / Grafana Setup

tested `mpp-solar -p /dev/hidraw0 -c QPI` it is showing PI30 as output
 I tested these two lines 
In CMD1: start (verbosely) listening to all MQTT topics on the local broker mosquitto_sub -h localhost -v -t "#"
In CMD2: send a test message mosquitto_pub -h localhost -t "basetopic/subtopic" -m "this is the message"
and mqtt is working fine
but when i try to run this `mpp-solar -q localhost -c QID` this give error as i connected with usb cable so i changed command to 
`mpp-solar -q localhost -c QID -p /dev/hidraw0` this give output to screen not on CMD2 window

How to get on mqtt window outout for this command 

I added this line to  `* * * * * /usr/local/bin/mpp-solar -q localhost -c QPIGS -i > /home/pi/cron.out 2>&1`
but it give error on cron.out file that -i arrgument is not correct so i change this line to  `* * * * * /usr/local/bin/mpp-solar -p /dev/hidraw0 -q localhost -c QPIGS  > /home/pi/cron.out 2>&1`
now output is showing on cron.out file
```

Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
ac_input_voltage              	269.0          	V   
ac_input_frequency            	50.0           	Hz  
ac_output_voltage             	228.0          	V   
ac_output_frequency           	50.0           	Hz  
ac_output_apparent_power      	88             	VA  
ac_output_active_power        	88             	W   
ac_output_load                	3              	%   
bus_voltage                   	342            	V   
battery_voltage               	24.38          	V   
battery_charging_current      	0              	A   
battery_capacity              	34             	%   
inverter_heat_sink_temperature	37             	°C  
pv_input_current_for_battery  	2.0            	A   
pv_input_voltage              	45.6           	V   
battery_voltage_from_scc      	24.53          	V   
battery_discharge_current     	3              	A   
is_sbu_priority_version_added 	1              	True - 1/False - 0
is_configuration_changed      	0              	True - 1/False - 0
is_scc_firmware_updated       	0              	True - 1/False - 0
is_load_on                    	1              	True - 1/False - 0
is_battery_voltage_to_steady_while_charging	0              	True - 1/False - 0
is_charging_on                	1              	True - 1/False - 0
is_scc_charging_on            	1              	True - 1/False - 0
is_ac_charging_on             	1              	True - 1/False - 0
rsv1                          	0              	A   
rsv2                          	3              	A   
pv_input_power                	64             	W   
is_charging_to_float          	1              	True - 1/False - 0
is_switched_on                	0              	True - 1/False - 0
is_reserved                   	0              	True - 1/False - 0

```

i try to Installing as a Service

but there was no folder /etc/mpp-solar in my setup for saving mpp-solar.conf file 
so i created one but  this command  `systemctl --user list-unit-files|grep mpp-solar`
give this error Failed to connect to bus: No such file or directory

i am unable to run these command 
mpp-info-sub -c QID
mpp-info-pub -q localhost -c QPIGS -i
bash: mpp-info-sub: command not found

Please let suggest solution so that i can use this for my inverter

jblance, 2021-03-27:
Ok, pip install doesn't install the service files.
What version do you have?
You'll need to get them manually from gut hub.
However best to get it working first.
Where are you reading about mpp_info_pub? This is very old?
You need to add ```-o influx2_mqtt ``` to your command to get telegraf compatible mqtt output.

santosh-mahadev, 2021-03-28:
after adding  `-o influx2_mqtt ` influx have new table mpp-solar and i can see data in it.

Where are you reading about mpp_info_pub? This is very old?
I go through other issues where someone was using this command 

How to start service in place of cron job.

Do I have to create new folder mpp-solar  and file  mpp-solar.conf  in /etc/  as my setup don't have it. Also I don't find mpp-solar.service file

jblance, 2021-03-28:
[mpp-solar.service](https://github.com/jblance/mpp-solar/blob/master/daemon/mpp-solar.service) 
An [example conf](https://github.com/jblance/mpp-solar/blob/master/daemon/mpp-solar.conf.example)  
Note the conf needs editing to suit you and placing in ```/etc/mpp-solar/mpp-solar.conf```
The service file needs to go to ```/etc/systemd/user/mpp-solar.service```

Some [documentation](https://github.com/jblance/mpp-solar/blob/master/daemon/README.md) on running as a service

santosh-mahadev, 2021-03-28:
i created mpp-solar.service file but unable to run this command systemctl --user list-unit-files|grep mpp-solar
Failed to connect to bus: No such file or directory
get this error

jblance, 2021-03-28:
What OS are you using?
What files are in the ```/etc/systemd/user/``` directory
What happens if you remove the file you edited?

santosh-mahadev, 2021-03-29:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

there was no file in /etc/systemd/user/ i added mpp-solar.service file

jblance, 2021-04-05:
And if you remove the file in  /etc/systemd/user/ and try
``` systemctl --user list-unit-files``` what happens?

joerghohn, 2021-04-16:
I also have it running on Debian Buster and got the same error. 
Just move the "mpp-solar.service" file to "/etc/systemd/system/" and run it as a system service instead of a user service.
Description ist here: [](https://github.com/jblance/mpp-solar/blob/master/docs/service-README.md#moving-the-unit-file)https://github.com/jblance/mpp-solar/blob/master/docs/service-README.md#moving-the-unit-file

closed 2021-04-26 by jblance