jblance/mpp-solar #74: Broken pipe when running mpp-solar

danb35, 2021-03-27:
I'd taken a bit of a break from this, as I was running into too many problems and my batteries hadn't arrived in any event, but now I'm back to it.  I'm running `mpp-solar` on a Ubuntu 20.04 VM with my MPP HV LV 2424 inverter connected by USB, with that passed through to the VM and appearing to the OS as `/dev/hidraw0`.  But whenever I run it, I get a "broken pipe" error:
```
(mppsolar) root@solar:~# mpp-solar -v
Solar Device Command Utility, version: 0.7.37, recent changes: update ved work, add a get command to ved
(mppsolar) root@solar:~# mpp-solar -p /dev/hidraw0 -P PI18 -D
INFO:root:Solar Device Command Utility, version: 0.7.37, recent changes: update ved work, add a get command to ved
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI18"
DEBUG:root:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:device:mppsolar __init__ args ()
DEBUG:device:mppsolar __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI18', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
DEBUG:device:get_port_type: port matches hidraw
INFO:device:set_port: Using hidrawio for communications
DEBUG:device:set_protocol: Protocol PI18
DEBUG:device:mppsolar __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0x7f4e795f9eb0>, protocol <mppsolar.protocols.pi18.pi18 object at 0x7f4e79607040>
DEBUG:root:Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f4e795f9f40>, '', '', 'screen', None, None)]
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x7f4e795f9eb0>, protocol: <mppsolar.protocols.pi18.pi18 object at 0x7f4e79607040> for command: , tag: , outputs: screen
INFO:device:run_command: Running command 
INFO:pi18:get_full_command: Using protocol b'PI18' with 4 commands
DEBUG:AbstractProtocol:get_command_defn: Processing command 'PI'
DEBUG:AbstractProtocol:get_command_defn: Found command PI in protocol b'PI18'
DEBUG:protocol_helpers:Calculating CRC for b'PI'
DEBUG:protocol_helpers:Generated CRC 0xd7 0x52 0xd752
DEBUG:pi18:get_full_command: full command: b'^P005PI\xd7R\r'
INFO:device:run_command: full command b'^P005PI\xd7R\r' for command PI
DEBUG:AbstractProtocol:get_command_defn: Processing command 'PI'
DEBUG:AbstractProtocol:get_command_defn: Found command PI in protocol b'PI18'
DEBUG:HIDRawIO:send_and_receive: length of to_send: 10
DEBUG:HIDRawIO:send_and_receive: 2 chunk send
Traceback (most recent call last):
  File "/root/venv/mppsolar/bin/mpp-solar", line 33, in <module>
    sys.exit(load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')())
  File "/root/venv/mppsolar/src/mpp-solar/mppsolar/__init__.py", line 378, in main
    results = _device.run_command(command=_command)
  File "/root/venv/mppsolar/src/mpp-solar/mppsolar/devices/device.py", line 295, in run_command
    raw_response = self._port.send_and_receive(
  File "/root/venv/mppsolar/src/mpp-solar/mppsolar/io/hidrawio.py", line 42, in send_and_receive
    os.write(usb0, to_send[:5])
BrokenPipeError: [Errno 32] Broken pipe
(mppsolar) root@solar:~# 
```
My first thought was that there was a problem with the USB connection, but it seems to be good--the problem persists after unplugging and reconnecting the cable as well.  Where should I be looking to troubleshoot this?

jblance, 2021-03-27:
I've not seen that error before
It's getting that as it tries to write to the USB. Are there any other things using the USB? How about in the host system? Are there any settings in the VM space that influence the access? Noting that hidraw is a pretty low level device, so something might be trying to block it

Otherwise I'll try to post a simple write python script and we can use that to test outside of the complicated mpp-solar code

jblance, 2021-03-27:
One Google hit suggested a vm memory issue - how much memory is allocated to the vm?

danb35, 2021-03-27:
The VM has 2 GB RAM allocated, and is currently reporting 640 MB used.  It shouldn't have anything else using the USB; the VM is dedicated to monitoring the inverter (and BMS, hopefully, but I haven't gotten to working on that, as I don't yet have all the batteries).  I don't see that any other VM on that host has any USB devices assigned (which is as it should be), and nothing on the host should be doing anything with USB.

Though at this point, I'm wondering if it'd be worthwhile to spin up a VM on a different host and see if I'm still seeing the issue...

danb35, 2021-03-27:
> I'm wondering if it'd be worthwhile to spin up a VM on a different host and see if I'm still seeing the issue...

Tried this, no change.  Fresh Ubuntu 20.04 Server VM, with all updates, and I've never connected the inverter to this host.  Passed through the device to the VM, and it's shown when I run `lsusb`.  Ran `apt install python3-pip python3-venv`.  Then `sudo pip3 install mpp-solar`.  Still seeing the same error, and it looks like it's happening in the same place:
```
dan@solar:~$ sudo mpp-solar -p /dev/hidraw0 -D
INFO:MPP-Solar:Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30" for command "None" (tag: None)
DEBUG:MPP-Solar:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:MPP-Solar:mppsolar __init__ args ()
DEBUG:MPP-Solar:mppsolar __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
DEBUG:MPP-Solar:device:get_port_type: port matches hidraw
INFO:MPP-Solar:Using hidrawio for communications
DEBUG:MPP-Solar:device.set_protocol with protocol PI30
DEBUG:MPP-Solar:mppsolar __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0x7f7601f9d3d0>, protocol <mppsolar.protocols.pi30.pi30 object at 0x7f7601f98be0>
DEBUG:MPP-Solar:Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f7601f9d520>, '', None, 'screen', None, None)]
INFO:MPP-Solar:Looping 1 commands
INFO:MPP-Solar:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x7f7601f9d3d0>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x7f7601f98be0> for command: , tag: None, outputs: screen
INFO:MPP-Solar:Running command 
INFO:MPP-Solar:Using protocol b'PI30' with 35 commands
DEBUG:MPP-Solar:Calculating CRC for b'QPI'
DEBUG:MPP-Solar:Generated CRC 0xbe 0xac 0xbeac
DEBUG:MPP-Solar:full command: b'QPI\xbe\xac\r'
INFO:MPP-Solar:full command b'QPI\xbe\xac\r' for command QPI
DEBUG:MPP-Solar:get_command_defn for: QPI
DEBUG:MPP-Solar:Found command QPI in protocol b'PI30'
DEBUG:MPP-Solar:length of to_send: 6
DEBUG:MPP-Solar:1 chunk send
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/__init__.py", line 405, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/devices/device.py", line 281, in run_command
    raw_response = self._port.send_and_receive(
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/io/hidrawio.py", line 38, in send_and_receive
    os.write(usb0, to_send)
BrokenPipeError: [Errno 32] Broken pipe
```

jblance, 2021-03-27:
[hidrawtest.py.txt](https://github.com/jblance/mpp-solar/files/6216473/hidrawtest.py.txt)
Try running the attached (after naming to hidrawtest.py)

danb35, 2021-03-27:
```
dan@solar:~$ python3 hidrawtest.py 
send_and_receive: sending: b'^P005PI\xd7R\r'
send_and_receive: length of to_send: 10
send_and_receive: error: [Errno 32] Broken pipe
> /home/dan/hidrawtest.py(29)<module>()
-> exit(1)
(Pdb) 
```

jblance, 2021-03-28:
Try editing the file to send ```b"PI\r"``` and if that fails also try ```"PI\r" ```

jblance, 2021-03-28:
Also what is your virtualisation setup?
Also try the below at the ```(pdb)``` prompt
```os.isatty(usb0)```

danb35, 2021-03-28:
```
dan@solar:~$ python3 hidrawtest.py 
send_and_receive: sending: b'PI\r'
send_and_receive: length of to_send: 3
send_and_receive: error: [Errno 32] Broken pipe
> /home/dan/hidrawtest.py(30)<module>()
-> exit(1)
(Pdb) os.isatty(usb0)
False
```

```
dan@solar:~$ python3 hidrawtest.py 
send_and_receive: sending: PI
send_and_receive: length of to_send: 3
send_and_receive: error: a bytes-like object is required, not 'str'
> /home/dan/hidrawtest.py(30)<module>()
-> exit(1)
(Pdb) 
```

I'm using Proxmox VE 6.3-6 as a virtualization host on a Dell PowerEdge C6100.  The USB device is passed through to the VM, which has worked with your software in the past, I think it was somewhere in the 0.4 versions.

jblance, 2021-03-28:
And its definitley the right device number?

jblance, 2021-03-28:
Can you try this one (using python open instead of lower level  os.open()
[hidrawtest.py.txt](https://github.com/jblance/mpp-solar/files/6218061/hidrawtest.py.txt)

dustinzinnj, 2021-03-28:
Not sure if it helps, but. 
Had this same issue when setting mine up, seems to be some commands aren't returning the right code (maybe LV2424 has two verisons?). 

Using LV2424, I had to roll back to version 0.4.6 to get anything from USB or TTY adapter. I also have to use the -M PI18 in addition. I have never got the USB directly working for my LV. 

My full command is "sudo mpp-solar -d /dev/ttyUSB0 -M PI18 -c GS" The only other one written in 0.4.6 is ET, for my LV that works. Using a generic USB to RS232 converter works like a charm so far. No newer versions worked for me, not sure why. Never got around to figuring out how to add more commands yet.

jblance, 2021-03-28:
@dustinzinnj - what was your error again? You are correct in that I dont have a LV2424, but I'm keen to solve any errors that I can
0.4.6 is pretty old now, adding commands and supporting different inverters is much easier in the newer code - just need to rely on people with the inverters to test / feedback

dustinzinnj, 2021-03-28:
I'd have to update it, it's super similar with broken pipes, and cryptic messages that I -think- are the code getting the wrong response from the inverter (so it couldn't format it). If I copy the folder of my current setup, is that enough that I could roll it back after updating? I'll test it if so and report back my errors. 

Also, the command changed at some point didn't it? I know my command didn't let me do -d I think. Also, I'm not using a VM for my setup, it's just a small HP elitedesk 800 micro pc and I still had super similar errors.

jblance, 2021-03-28:
Re updating - if you use ```venv``` you get an isolated environment that wont impact your production install (see [readme link](https://github.com/jblance/mpp-solar#venv-install---recommended-if-testing-new-features--release)  )
With the venv approach, you need to install any dependancies again in the venv
The command options have changed and venv doesnt work well with sudo, but your command would be 
```mpp-solar -p /dev/ttyUSB0 -P PI18 -c GS``` in the newer code (shouldnt need sudo for /dev/ttyUSB0) 

The setter type commands work with my inverter model, but i havent used them much and they kinda suck as you need to know the actual command - eg ```mpp-solar -c PBFT54.0``` or similar to set Battery Float Charging Voltage. At some point I'll be changing them to more user friendly commands.

I've also moved from Grafana to Home Assistant for display (and so I can incorporate 'actions' etc), but there is likely similarities to 'send commands from buttons in grafana to the device'

As always if you have requests / suggestions open a (new) issue with the details and I'll see what can be achieved (though you'll need to get the new code working and be prepared for testing and debug :-)

dustinzinnj, 2021-03-28:
Here is my current error after using venv and running the command you sent:

"root@dustinz-solar:/venv/mppsolar/bin# source /venv/mppsolar/bin/activate
(mppsolar) root@dustinz-solar:/venv/mppsolar/bin# mpp-solar -p /dev/ttyUSB0 -P PI18 -c GS
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 33, in <module>
    sys.exit(load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')())
  File "/root/src/mpp-solar/mppsolar/__init__.py", line 378, in main
    results = _device.run_command(command=_command)
  File "/root/src/mpp-solar/mppsolar/devices/device.py", line 309, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/root/src/mpp-solar/mppsolar/protocols/protocol.py", line 254, in decode
    msgs[key] = [float(result) / 10, resp_format[2]]
ValueError: could not convert string to float: ''"

With -D, let me know if I missed something. 

"(mppsolar) root@dustinz-solar:/venv/mppsolar/bin# mpp-solar -p /dev/ttyUSB0 -P PI18 -c GS -D
INFO:root:Solar Device Command Utility, version: 0.7.37, recent changes: update ved work, add a get command to ved
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI18"
DEBUG:root:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:device:mppsolar __init__ args ()
DEBUG:device:mppsolar __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI18', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
DEBUG:device:get_port_type: port matches ttyusb
INFO:device:set_port: Using serialio for communications
DEBUG:device:set_protocol: Protocol PI18
DEBUG:device:mppsolar __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0x7fbb8e46e520>, protocol <mppsolar.protocols.pi18.pi18 object at 0x7fbb8e46e4c0>
DEBUG:root:Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7fbb8e46e610>, 'GS', 'GS', 'screen', None, None)]
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x7fbb8e46e520>, protocol: <mppsolar.protocols.pi18.pi18 object at 0x7fbb8e46e4c0> for command: GS, tag: GS, outputs: screen
INFO:device:run_command: Running command GS
INFO:pi18:get_full_command: Using protocol b'PI18' with 4 commands
DEBUG:AbstractProtocol:get_command_defn: Processing command 'GS'
DEBUG:AbstractProtocol:get_command_defn: Found command GS in protocol b'PI18'
DEBUG:protocol_helpers:Calculating CRC for b'GS'
DEBUG:protocol_helpers:Generated CRC 0xfe 0xcd 0xfecd
DEBUG:pi18:get_full_command: full command: b'^P005GS\xfe\xcd\r'
INFO:device:run_command: full command b'^P005GS\xfe\xcd\r' for command GS
DEBUG:AbstractProtocol:get_command_defn: Processing command 'GS'
DEBUG:AbstractProtocol:get_command_defn: Found command GS in protocol b'PI18'
DEBUG:SerialIO:send_and_receive: port /dev/ttyUSB0, baudrate 2400
DEBUG:SerialIO:send_and_receive: Executing command via serialio...
DEBUG:SerialIO:send_and_receive: serial response was: b'^0\x1b\xe3\r'
DEBUG:device:run_command: Send and Receive Response b'^0\x1b\xe3\r'
INFO:AbstractProtocol:decode: response passed to decode: b'^0\x1b\xe3\r'
DEBUG:AbstractProtocol:get_command_defn: Processing command 'GS'
DEBUG:AbstractProtocol:get_command_defn: Found command GS in protocol b'PI18'
DEBUG:AbstractProtocol:decode: trimmed and split responses: [b'']
INFO:AbstractProtocol:decode: Processing response of type default
INFO:AbstractProtocol:decode: Processing SEQUENCE type responses
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 33, in <module>
    sys.exit(load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')())
  File "/root/src/mpp-solar/mppsolar/__init__.py", line 378, in main
    results = _device.run_command(command=_command)
  File "/root/src/mpp-solar/mppsolar/devices/device.py", line 309, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/root/src/mpp-solar/mppsolar/protocols/protocol.py", line 254, in decode
    msgs[key] = [float(result) / 10, resp_format[2]]
ValueError: could not convert string to float: ''"

Hopefully I didn't miss anything, but seems to be the same error I had before. I assumed "could not convert string to float" was it was unable to understand the response and properly send it to influxdb.

jblance, 2021-03-28:
So the inverter is refusing the command (and telling us that it refused) - wonder why?
Can you run against the old command with debug enabled as well (add the ```-D``` flag) outside of the venv environment

dustinzinnj, 2021-03-29:
So I actually had to reinstall after trying to use venv, but it's back working again.  On 0.4.6 I get responses that seem in order with the device (when I check the LCD)

"dustinz@dustinz-solar:~sudo mpp-solar -M PI18 -d /dev/ttyUSB0 -c GS -D
INFO:MPP-Solar:command GS
INFO:MPP-Solar:Serial device used: /dev/ttyUSB0, baud rate: 2400
INFO:MPP-Solar:Loading commands for inverter model: PI18
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.6-py3.8.egg/mppsolar/commands/_pi18_et.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command ET supported by model PI18
INFO:MPP-Solar:... command ET loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P005ET
DEBUG:MPP-Solar:Calculating CRC for b'^P005ET'
DEBUG:MPP-Solar:Generated CRC 4e 91 4e91
DEBUG:MPP-Solar:Full byte command: b'^P005ETN\x91\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.6-py3.8.egg/mppsolar/commands/_pi18_gs.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command GS supported by model PI18
INFO:MPP-Solar:... command GS loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P005GS
DEBUG:MPP-Solar:Calculating CRC for b'^P005GS'
DEBUG:MPP-Solar:Generated CRC 58 14 5814
DEBUG:MPP-Solar:Full byte command: b'^P005GSX\x14\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.6-py3.8.egg/mppsolar/commands/_pi18_mod.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MOD supported by model PI18
INFO:MPP-Solar:... command MOD loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P006MOD
DEBUG:MPP-Solar:Calculating CRC for b'^P006MOD'
DEBUG:MPP-Solar:Generated CRC dd be ddbe
DEBUG:MPP-Solar:Full byte command: b'^P006MOD\xdd\xbe\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.6-py3.8.egg/mppsolar/commands/_pi18_pi.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command PI supported by model PI18
INFO:MPP-Solar:... command PI loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for ^P005PI
DEBUG:MPP-Solar:Calculating CRC for b'^P005PI'
DEBUG:MPP-Solar:Generated CRC 71 8b 718b
DEBUG:MPP-Solar:Full byte command: b'^P005PIq\x8b\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.6-py3.8.egg/mppsolar/commands/f.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command F supported all inverters
INFO:MPP-Solar:... command F loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for F
DEBUG:MPP-Solar:Calculating CRC for b'F'
DEBUG:MPP-Solar:Generated CRC 29 2 2902
DEBUG:MPP-Solar:Full byte command: b'F)\x02\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.6-py3.8.egg/mppsolar/commands/mchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MCHGC supported all inverters
INFO:MPP-Solar:... command MCHGC loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for MCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MCHGC'
DEBUG:MPP-Solar:Generated CRC 9d a2 9da2
DEBUG:MPP-Solar:Full byte command: b'MCHGC\x9d\xa2\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.6-py3.8.egg/mppsolar/commands/mnchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MNCHGC supported all inverters
INFO:MPP-Solar:... command MNCHGC loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for MNCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MNCHGC'
DEBUG:MPP-Solar:Generated CRC 57 23 5723
DEBUG:MPP-Solar:Full byte command: b'MNCHGCW#\r'
DEBUG:MPP-Solar:Loading command information from /usr/local/lib/python3.8/dist-packages/mpp_solar-0.4.6-py3.8.egg/mppsolar/commands/muchgc.json
DEBUG:MPP-Solar:-----No supports
DEBUG:MPP-Solar:Command MUCHGC supported all inverters
INFO:MPP-Solar:... command MUCHGC loaded for inverter model: PI18
DEBUG:MPP-Solar:Generate full byte command for MUCHGC
DEBUG:MPP-Solar:Calculating CRC for b'MUCHGC'
DEBUG:MPP-Solar:Generated CRC bf 86 bf86
DEBUG:MPP-Solar:Full byte command: b'MUCHGC\xbf\x86\r'
[... 295 lines cut ...]
ac_output_voltage               118.5           V
battery_capacity                84              %
battery_charging_current        0               A
battery_discharge_current       0               A
battery_power_direction         charge
battery_voltage                 28.3            V
battery_voltage_from_scc        0.0             V
battery_voltage_from_scc2       0.0             V
dc/ac_power_direction           DC-AC
grid_frequency                  59.9            Hz
grid_voltage                    118.5           V
inverter_heat_sink_temperature  20              oC
line_power_direction            input
load_connection                 connect
local_parallel_id               0
mppt1_charger_status            abnormal
mppt1_charger_temperature       0               oC
mppt2_charger_status            abnormal
mppt2_charger_temperature       0               oC
output_load_percent             2               %
pv1_input_power                 0               W
pv1_input_voltage               0.0             V
pv2_input_power                 0               W
pv2_input_voltage               0.0             V
setting_value_configuration_state       Nothing changed"

jblance, 2021-03-29:
So the CRC calculation is wrong in the new code...

danb35, 2021-03-29:
> And its definitley the right device number? (ie is there just one hidraw device)

Good question.  It had been `hidraw0` (under the same guest OS, same hypervisor, same host hardware), but now I see there's also a `hidraw1` device, and it gives different output:

```
dan@solar:~$ mpp-solar -p /dev/hidraw1 -P PI18 -c GS
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/__init__.py", line 405, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/devices/device.py", line 295, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/protocols/protocol.py", line 145, in decode
    msgs[key] = [float(result) / 10, resp_format[2]]
ValueError: could not convert string to float: ''
```

Using your new test script, I get:
```
dan@solar:~$ python3 hidrawtest2.py 
send_and_receive: sending: b'^P005PI\xd7R\r'
send_and_receive: length of to_send: 10
send_and_receive: USB read error: [Errno 32] Broken pipe
send_and_receive: USB read error: [Errno 32] Broken pipe
send_and_receive: USB read error: [Errno 32] Broken pipe
. . .
send_and_receive: USB read error: [Errno 32] Broken pipe
send_and_receive: usb response was:  b''
Traceback (most recent call last):
  File "hidrawtest2.py", line 52, in <module>
    usb0.close()
BrokenPipeError: [Errno 32] Broken pipe
```

If I edit that script to try `hidraw1`, I instead get:
```
. . .
send_and_receive: USB read error: [Errno 110] Connection timed out
send_and_receive: USB read error: [Errno 110] Connection timed out
send_and_receive: USB read error: [Errno 110] Connection timed out
send_and_receive: usb response was:  b''
Traceback (most recent call last):
  File "hidrawtest2.py", line 52, in <module>
    usb0.close()
TimeoutError: [Errno 110] Connection timed out
```

jblance, 2021-03-29:
@danb35 hidraw1 looked better
Turns out the crc calc was wrong, can you try hidraw1 with debug on (and/or update to latest version from github first)

danb35, 2021-03-29:
Well, progress.  Following the instructions in the README, I installed the latest github version in a venv, and ran it against `hidraw1`.  Still not right, but better:
```
(mppsolar) root@solar:~# mpp-solar -v
Solar Device Command Utility, version: 0.7.38, recent changes: change pi18 crc
(mppsolar) root@solar:~# mpp-solar -p /dev/hidraw1 -P PI18 -c GS
Command: GS - General status query
------------------------------------------------------------
Parameter                     	Value           Unit
warning0                      	Command GS was rejected	    
```

Debug output:

```
(mppsolar) root@solar:~# mpp-solar -p /dev/hidraw1 -P PI18 -c GS -D
INFO:root:Solar Device Command Utility, version: 0.7.38, recent changes: change pi18 crc
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw1 (porttype=None)" using protocol "PI18"
DEBUG:root:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:device:mppsolar __init__ args ()
DEBUG:device:mppsolar __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw1', 'protocol': 'PI18', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
DEBUG:device:get_port_type: port matches hidraw
INFO:device:set_port: Using hidrawio for communications
DEBUG:device:set_protocol: Protocol PI18
DEBUG:device:mppsolar __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0x7fe49f39de80>, protocol <mppsolar.protocols.pi18.pi18 object at 0x7fe49f3ab0a0>
DEBUG:root:Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7fe49f39df10>, 'GS', 'GS', 'screen', None, None)]
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x7fe49f39de80>, protocol: <mppsolar.protocols.pi18.pi18 object at 0x7fe49f3ab0a0> for command: GS, tag: GS, outputs: screen
INFO:device:run_command: Running command GS
INFO:pi18:get_full_command: Using protocol b'PI18' with 4 commands
DEBUG:AbstractProtocol:get_command_defn: Processing command 'GS'
DEBUG:AbstractProtocol:get_command_defn: Found command GS in protocol b'PI18'
DEBUG:pi18:get_full_command: _pre_cmd: b'^P005GS'
DEBUG:protocol_helpers:Calculating CRC for b'^P005GS'
DEBUG:protocol_helpers:Generated CRC 0x58 0x14 0x5814
DEBUG:pi18:get_full_command: full command: b'^P005X\x14\r'
INFO:device:run_command: full command b'^P005X\x14\r' for command GS
DEBUG:AbstractProtocol:get_command_defn: Processing command 'GS'
DEBUG:AbstractProtocol:get_command_defn: Found command GS in protocol b'PI18'
DEBUG:HIDRawIO:send_and_receive: length of to_send: 8
DEBUG:HIDRawIO:send_and_receive: 1 chunk send
DEBUG:HIDRawIO:send_and_receive: usb response was: b'^0\x1b\xe3\r'
DEBUG:device:run_command: Send and Receive Response b'^0\x1b\xe3\r'
INFO:AbstractProtocol:decode: response passed to decode: b'^0\x1b\xe3\r'
DEBUG:AbstractProtocol:get_command_defn: Processing command 'GS'
DEBUG:AbstractProtocol:get_command_defn: Found command GS in protocol b'PI18'
DEBUG:AbstractProtocol:decode: trimmed and split responses: ['NAK']
INFO:AbstractProtocol:decode: Processing response of type default
INFO:AbstractProtocol:decode: Processing SEQUENCE type responses
INFO:device:run_command: Decoded response {'raw_response': ['^0\x1bã\r', ''], '_command': 'GS', '_command_description': 'General status query', 'WARNING0': ['Command GS was rejected', '']}
DEBUG:root:results: {'raw_response': ['^0\x1bã\r', ''], '_command': 'GS', '_command_description': 'General status query', 'WARNING0': ['Command GS was rejected', '']}
INFO:root:attempting to create output processor: screen
DEBUG:screen:processor.screen __init__ kwargs {}
DEBUG:root:Using output filter: None
INFO:screen:output: Using output processor: screen
DEBUG:screen:output: kwargs {'data': {'raw_response': ['^0\x1bã\r', ''], '_command': 'GS', '_command_description': 'General status query', 'WARNING0': ['Command GS was rejected', '']}, 'tag': 'GS', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Command: GS - General status query
------------------------------------------------------------
Parameter                     	Value           Unit
DEBUG:helpers:key_wanted: No filter and key warning0 not excluded by excl_filter None so key wanted
warning0                      	Command GS was rejected	    
DEBUG:root:Not daemon, so not looping
```

jblance, 2021-03-29:
oops, my bad
try git pull now and try again

danb35, 2021-03-29:
Ah ha!
```
(mppsolar) root@solar:~# mpp-solar -v
Solar Device Command Utility, version: 0.7.38, recent changes: change pi18 crc
(mppsolar) root@solar:~# mpp-solar -p /dev/hidraw1 -P PI18 -c GS
Command: GS - General status query
------------------------------------------------------------
Parameter                     	Value           Unit
grid_voltage                  	6120.8         	V   
grid_frequency                	60.0           	Hz  
ac_output_voltage             	120.8          	V   
ac_output_frequency           	60.0           	Hz  
ac_output_apparent_power      	1365           	VA  
ac_output_active_power        	1284           	W   
output_load_percent           	53             	%   
battery_voltage               	27.0           	V   
battery_voltage_from_scc      	0.0            	V   
battery_voltage_from_scc2     	0.0            	V   
battery_discharge_current     	0              	A   
battery_charging_current      	1              	A   
battery_capacity              	100            	%   
inverter_heat_sink_temperature	30             	°C  
mppt1_charger_temperature     	0              	°C  
mppt2_charger_temperature     	0              	°C  
pv1_input_power               	0              	W   
pv2_input_power               	0              	W   
pv1_input_voltage             	0.0            	V   
pv2_input_voltage             	0.0            	V   
setting_value_configuration_state	Something changed	    
mppt1_charger_status          	abnormal       	    
mppt2_charger_status          	abnormal       	    
load_connection               	connect        	    
battery_power_direction       	charge         	    
dc/ac_power_direction         	AC-DC          	    
line_power_direction          	input          	    
local_parallel_id             	0              	    
(mppsolar) root@solar:~# mpp-solar -p /dev/hidraw1 -P PI18 -c ET
Command: ET - Total Generated Energy query
------------------------------------------------------------
Parameter                     	Value           Unit
total_generated_energy        	100128158      	KWh 
```
Though I think something's wrong with the math; I know my grid voltage isn't 6kV, and there's no way I've generated 100 GWh in the month or so I've had the panels up.

So I can wonder why, in exactly the same environment, the inverter got a different device name, and feel kind of silly for missing that, but definite improvement going on.

jblance, 2021-03-29:
Can you run GS with -D or -R

danb35, 2021-03-29:
```
(mppsolar) root@solar:~# mpp-solar -p /dev/hidraw1 -P PI18 -c GS -D
INFO:root:Solar Device Command Utility, version: 0.7.38, recent changes: change pi18 crc
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw1 (porttype=None)" using protocol "PI18"
DEBUG:root:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:device:mppsolar __init__ args ()
DEBUG:device:mppsolar __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw1', 'protocol': 'PI18', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
DEBUG:device:get_port_type: port matches hidraw
INFO:device:set_port: Using hidrawio for communications
DEBUG:device:set_protocol: Protocol PI18
DEBUG:device:mppsolar __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0x7fab93055e80>, protocol <mppsolar.protocols.pi18.pi18 object at 0x7fab930630a0>
DEBUG:root:Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7fab93055f10>, 'GS', 'GS', 'screen', None, None)]
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x7fab93055e80>, protocol: <mppsolar.protocols.pi18.pi18 object at 0x7fab930630a0> for command: GS, tag: GS, outputs: screen
INFO:device:run_command: Running command GS
INFO:pi18:get_full_command: Using protocol b'PI18' with 4 commands
DEBUG:AbstractProtocol:get_command_defn: Processing command 'GS'
DEBUG:AbstractProtocol:get_command_defn: Found command GS in protocol b'PI18'
DEBUG:pi18:get_full_command: _pre_cmd: b'^P005GS'
DEBUG:protocol_helpers:Calculating CRC for b'^P005GS'
DEBUG:protocol_helpers:Generated CRC 0x58 0x14 0x5814
DEBUG:pi18:get_full_command: full command: b'^P005GSX\x14\r'
INFO:device:run_command: full command b'^P005GSX\x14\r' for command GS
DEBUG:AbstractProtocol:get_command_defn: Processing command 'GS'
DEBUG:AbstractProtocol:get_command_defn: Found command GS in protocol b'PI18'
DEBUG:HIDRawIO:send_and_receive: length of to_send: 10
DEBUG:HIDRawIO:send_and_receive: 2 chunk send
DEBUG:HIDRawIO:send_and_receive: usb response was: b'^D1061210,600,1210,600,1331,1254,052,270,000,000,000,001,100,030,000,000,0000,0000,0000,0000,1,0,0,1,1,1,1,0WW\r'
DEBUG:device:run_command: Send and Receive Response b'^D1061210,600,1210,600,1331,1254,052,270,000,000,000,001,100,030,000,000,0000,0000,0000,0000,1,0,0,1,1,1,1,0WW\r'
INFO:AbstractProtocol:decode: response passed to decode: b'^D1061210,600,1210,600,1331,1254,052,270,000,000,000,001,100,030,000,000,0000,0000,0000,0000,1,0,0,1,1,1,1,0WW\r'
DEBUG:AbstractProtocol:get_command_defn: Processing command 'GS'
DEBUG:AbstractProtocol:get_command_defn: Found command GS in protocol b'PI18'
DEBUG:AbstractProtocol:decode: trimmed and split responses: [b'61210', b'600', b'1210', b'600', b'1331', b'1254', b'052', b'270', b'000', b'000', b'000', b'001', b'100', b'030', b'000', b'000', b'0000', b'0000', b'0000', b'0000', b'1', b'0', b'0', b'1', b'1', b'1', b'1', b'0']
INFO:AbstractProtocol:decode: Processing response of type default
INFO:AbstractProtocol:decode: Processing SEQUENCE type responses
INFO:device:run_command: Decoded response {'raw_response': ['^D1061210,600,1210,600,1331,1254,052,270,000,000,000,001,100,030,000,000,0000,0000,0000,0000,1,0,0,1,1,1,1,0WW\r', ''], '_command': 'GS', '_command_description': 'General status query', 'Grid voltage': [6121.0, 'V'], 'Grid frequency': [60.0, 'Hz'], 'AC output voltage': [121.0, 'V'], 'AC output frequency': [60.0, 'Hz'], 'AC output apparent power': [1331, 'VA'], 'AC output active power': [1254, 'W'], 'Output load percent': [52, '%'], 'Battery voltage': [27.0, 'V'], 'Battery voltage from SCC': [0.0, 'V'], 'Battery voltage from SCC2': [0.0, 'V'], 'Battery discharge current': [0, 'A'], 'Battery charging current': [1, 'A'], 'Battery capacity': [100, '%'], 'Inverter heat sink temperature': [30, '°C'], 'MPPT1 charger temperature': [0, '°C'], 'MPPT2 charger temperature': [0, '°C'], 'PV1 Input power': [0, 'W'], 'PV2 Input power': [0, 'W'], 'PV1 Input voltage': [0.0, 'V'], 'PV2 Input voltage': [0.0, 'V'], 'Setting value configuration state': ['Something changed', ''], 'MPPT1 charger status': ['abnormal', ''], 'MPPT2 charger status': ['abnormal', ''], 'Load connection': ['connect', ''], 'Battery power direction': ['charge', ''], 'DC/AC power direction': ['AC-DC', ''], 'Line power direction': ['input', ''], 'Local parallel ID': [0, '']}
DEBUG:root:results: {'raw_response': ['^D1061210,600,1210,600,1331,1254,052,270,000,000,000,001,100,030,000,000,0000,0000,0000,0000,1,0,0,1,1,1,1,0WW\r', ''], '_command': 'GS', '_command_description': 'General status query', 'Grid voltage': [6121.0, 'V'], 'Grid frequency': [60.0, 'Hz'], 'AC output voltage': [121.0, 'V'], 'AC output frequency': [60.0, 'Hz'], 'AC output apparent power': [1331, 'VA'], 'AC output active power': [1254, 'W'], 'Output load percent': [52, '%'], 'Battery voltage': [27.0, 'V'], 'Battery voltage from SCC': [0.0, 'V'], 'Battery voltage from SCC2': [0.0, 'V'], 'Battery discharge current': [0, 'A'], 'Battery charging current': [1, 'A'], 'Battery capacity': [100, '%'], 'Inverter heat sink temperature': [30, '°C'], 'MPPT1 charger temperature': [0, '°C'], 'MPPT2 charger temperature': [0, '°C'], 'PV1 Input power': [0, 'W'], 'PV2 Input power': [0, 'W'], 'PV1 Input voltage': [0.0, 'V'], 'PV2 Input voltage': [0.0, 'V'], 'Setting value configuration state': ['Something changed', ''], 'MPPT1 charger status [... cut ...]
(mppsolar) root@solar:~# mpp-solar -p /dev/hidraw1 -P PI18 -c GS -R
usage: mpp-solar [-h] [-n NAME] [-p PORT] [--porttype PORTTYPE] [-P {PI00,PI16,PI18,PI30,PI41,VED}] [-T TAG] [-b BAUD]
                 [-o [OUTPUT]] [--keepcase] [--filter FILTER] [--exclfilter EXCLFILTER] [-q MQTTBROKER] [--mqttport MQTTPORT]
                 [--mqttuser MQTTUSER] [--mqttpass MQTTPASS] [-c [COMMAND]] [-C [CONFIGFILE]] [--daemon] [--getstatus]
                 [--getsettings] [-v] [-D] [-I]
mpp-solar: error: unrecognized arguments: -R
(mppsolar) root@solar:~# 
```

jblance, 2021-03-29:
Fixed a mistake, try again after updating

danb35, 2021-03-29:
```
(mppsolar) root@solar:~# mpp-solar -p /dev/hidraw1 -P PI18 -c GS
Command: GS - General status query
------------------------------------------------------------
Parameter                     	Value           Unit
grid_voltage                  	120.9          	V   
grid_frequency                	60.0           	Hz  
ac_output_voltage             	120.9          	V   
ac_output_frequency           	60.0           	Hz  
ac_output_apparent_power      	1355           	VA  
ac_output_active_power        	1273           	W   
output_load_percent           	53             	%   
battery_voltage               	27.0           	V   
battery_voltage_from_scc      	0.0            	V   
battery_voltage_from_scc2     	0.0            	V   
battery_discharge_current     	0              	A   
battery_charging_current      	1              	A   
battery_capacity              	100            	%   
inverter_heat_sink_temperature	30             	°C  
mppt1_charger_temperature     	0              	°C  
mppt2_charger_temperature     	0              	°C  
pv1_input_power               	0              	W   
pv2_input_power               	0              	W   
pv1_input_voltage             	0.0            	V   
pv2_input_voltage             	0.0            	V   
setting_value_configuration_state	Something changed	    
mppt1_charger_status          	abnormal       	    
mppt2_charger_status          	abnormal       	    
load_connection               	connect        	    
battery_power_direction       	charge         	    
dc/ac_power_direction         	AC-DC          	    
line_power_direction          	input          	    
local_parallel_id             	0              	    
(mppsolar) root@solar:~# mpp-solar -p /dev/hidraw1 -P PI18 -c ET
Command: ET - Total Generated Energy query
------------------------------------------------------------
Parameter                     	Value           Unit
total_generated_energy        	128158         	KWh 
(mppsolar) root@solar:~# mpp-solar -p /dev/hidraw1 -P PI18 -c MOD
Command: MOD - Working mode query
------------------------------------------------------------
Parameter                     	Value           Unit
working_mode                  	Hybrid mode(Line mode, Grid mode)	    
(mppsolar) root@solar:~# 
```

jblance, 2021-03-29:
Does 128158 KWh sound right?

jblance, 2021-03-29:
[image]
Cos thats what it is based on

danb35, 2021-03-29:
> Does 128158 KWh sound right?

Not off the top of my head (it's had 2 kW of panels feeding it for about a month).  I can probably check in a couple of days to see what the inverter console says--I'm not there at the moment.  I'd have guessed it was something with the units, but checking the doc it indicates that the output is whole numbers of kWh.  My back-of-the-envelope math would suggest the result is in Wh instead.  But I can probably get a better answer Tuesday.

jblance, 2021-03-29:
Yeah Wh would seem more likely
Let me know - pretty easy to change the command definition

dustinzinnj, 2021-03-29:
Well, went and updated, but it seems some parts didn't update. I was running a cronjob but it's somehow stuck to the old version. New version and command works for my LV too, though. 

"dustinz@dustinz-solar:/home$ mpp-info-pub -q localhost -p /dev/ttyUSB0 -P PI18
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 584, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 901, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 792, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (mpp-solar 0.7.38 (/home/src/mpp-solar), Requirement.parse('mpp-solar==0.4.6'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/mpp-info-pub", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3254, in <module>
    def _initialize_master_working_set():
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3237, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3266, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 586, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 599, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 787, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'mpp-solar==0.4.6' distribution was not found and is required by the application"

jblance, 2021-03-29:
The info_pub command has gone in the new version, so you should remove /usr/local/bin/mpp-info-pub

The new command uses output modules eg ```-o mqtt ``` just use -o to get a list of commands 

The old install approach did not cleanup, the pip install will clean up old versions, but lags behind a little (and you need to clean up the other versions first)

jblance, 2021-03-29:
I generally do ```sudo rm -rf /usr/local/lib/python*/dist-packages/mpp_solar*``` to clean-up when I'm developing
You could ```sudo rm -f /usr/local/bin/mpp*``` as well 
then do a fresh install

dustinzinnj, 2021-03-29:
Alright so back to conversion errors it seems, unless I have the syntax wrong. 

dustinz@dustinz-solar:~$ sudo mpp-solar -p /dev/ttyUSB0 -P PI18 -c GS -T GS -o influx_mqtt -q localhost
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 33, in <module>
    sys.exit(load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')())
  File "/home/dustinz/src/mpp-solar/mppsolar/__init__.py", line 378, in main
    results = _device.run_command(command=_command)
  File "/home/dustinz/src/mpp-solar/mppsolar/devices/device.py", line 309, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/home/dustinz/src/mpp-solar/mppsolar/protocols/protocol.py", line 259, in decode
    msgs[key] = [float(result) / 10, resp_format[2]]
ValueError: could not convert string to float: ''

jblance, 2021-03-29:
Add -D for debug so I can see where the error is

dustinzinnj, 2021-03-29:
After rebooting it works, but it isn't sending it to mqtt/influx, and the commands in the setup from long ago don't seem to work for me for troubleshooting now

jblance, 2021-03-29:
Trace mqtt messages with mosquito_sub 
Check next step (is it telegraf? Check service logs)...

Is influx_mqtt the right format? Maybe it's influx2_mqtt

dustinzinnj, 2021-03-29:
So I've tried both formats and it's getting -something- as the graphs update correctly but there isnt any values. I'll have to test it more tomorrow after work. Could definitely be telegraf, as i've only used mqtt/telegraf in this scenario so it could be broken from something I've done.  Will report back here or a new issue, whichever is easier.

jblance, 2021-03-29:
Suggest checking the messages that are being sent by using mosquito_sub, post an example and we can see if there's a problem with the messages 
The telegraf service log normally logs any errors.

danb35, 2021-03-30:
> Yeah Wh would seem more likely

That does seem to be the case.  This morning, here's the result from the software:
```
(mppsolar) root@solar:~# mpp-solar -p /dev/hidraw1 -P PI18 -c ET
Command: ET - Total Generated Energy query
------------------------------------------------------------
Parameter                     	Value           Unit
total_generated_energy        	134899         	KWh 
```

According to the inverter console, total is 134 kWh.

closed 2021-04-05 by jblance