jblance/mpp-solar #75: Possible to send JSON via MQTT?

danb35, 2021-04-02:
I'm trying to log data from mpp-solar to a Kotori instance (see https://github.com/daq-tools/kotori/issues/50), and it seems they'd prefer to have the data in a JSON block.  I see that mpp-solar supports that output (`-o json`), but it looks like it just writes to the screen rather than to the specified MQTT broker.  Is there a way to do this that I'm missing?  Or if not, could it be added?

jblance, 2021-04-03:
It shouldn't be too hard. I'll try to add this in the next day or two

jblance, 2021-04-05:
added in version 0.7.40 e.g. below (make sure to change -p, -q and --mqtttopic options to suit)
```$ mpp-solar -p test -o json_mqtt -q screen --mqtttopic kotori -c QDI```
```
{'topic': 'kotori', 'payload': '{"ac_output_voltage": 230.0, "ac_output_frequency": 50.0, "max_ac_charging_current": 30, 
"battery_under_voltage": 42.0, "battery_float_charge_voltage": 54.0, "battery_bulk_charge_voltage": 56.4, 
"battery_recharge_voltage": 46.0, "max_charging_current": 60, "input_voltage_range": "Appliance", "output_source_priority": 
"Utility first", "charger_source_priority": "Solar + Utility", "battery_type": "AGM", "buzzer": "enabled", "power_saving": "disabled", 
"overload_restart": "disabled", "over_temperature_restart": "disabled", "lcd_backlight": "enabled", 
"primary_source_interrupt_alarm": "enabled", "record_fault_code": "disabled", "overload_bypass": "disabled", 
"lcd_reset_to_default": "enabled", "output_mode": "single machine output", "battery_redischarge_voltage": 54.0, 
"pv_ok_condition": "As long as one unit of inverters has connect PV, parallel system will consider PV OK", "pv_power_balance": 
"PV input max power will be the sum of the max charged power and loads power", "unknown_value_in_response_25": "000"}'}
```

closed 2021-04-05 by jblance

danb35, 2021-04-06:
Excellent, this now seems to be working:
[image]

jblance, 2021-04-06:
Great news

amotl, 2021-04-06:
Thank you all!

As Kotori's gallery [1] does not include a corresponding energy monitoring use-case yet [2], I would love to finally create such a walkthrough and I believe using `mpp-solar` there would make absolute sense.

So, if you would like to volunteer on such a document, may I humbly ask you if you can share some images of your setup, @danb35? I really love to exhibit more real world scenarios where Kotori is used, optimally similar to https://getkotori.org/docs/gallery/hydro2motion.html and https://getkotori.org/docs/gallery/weewx.html.

[1] https://getkotori.org/docs/gallery/
[2] Actually it has, but I haven't been able to find time to create a document from this:
[image]

danb35, 2021-04-06:
I'll have to give that some thought.  I'm very much a n00b when it comes to Grafana, so I'm sure the dashboard isn't yet anywhere near where it could be.  Here's what I put together this morning:
[image]

Would you be interested in the solar hardware setup, or more in the software side of things?  I'd think the latter would be more relevant, but the image you shared suggests the former is also of interest.

amotl, 2021-04-06:
> Would you be interested in the solar hardware setup, or more in the software side of things?

Actually, as I don't have any such hardware around, I would indeed be interested in having nice pictures of the real hardware in the field.