jblance/mpp-solar #76: Strip down requested information from JKBMS

alexanderfitu, 2021-04-05:
Hi all

Firstly, thanks for the excellent jkbms tool, it really allows some powerful stuff. 

I am using the BMS's as part of an electric vehicle. I am using mpp-solar to pull information from 2 BMS's every 30 seconds or so to obtain hi/lo cell data to control charging etc. 

Specifically, I am using getCellData. I wonder if pulling only the data I need will speed up the data retrieval from the BMS unit? Specifically,  only need cell voltages. 

The only reason I mention it, is the app can poll the data much quicker than this script, so I am wondering if the script is pulling more data than the app does.

alexanderfitu, 2021-04-05:
Looking at jk02.py it looks like it sends a query, and then all the data is supplied by the BMS, so I suspect this will not be possible unless we/I know the request codes for more specific data.

jblance, 2021-04-05:
It's more a problem with the way the code connects and disconnects each time.
The device spams a bunch of responses that could just be read continuously until it stops and then another request could be sent. This would speed things up for the jkbms usage, just wouldn't be generic for other situations. [Though not disconnecting each time would work and is on my "must get to it one day" list] there are likely a couple of other speed improvements possible for the jkbms logic if it was optimized.

alexanderfitu, 2021-04-05:
That makes sense, especially considering the way the android app works (connects, then updates all values continuously until disconnected.). It would certainly make my error handling if the system cannot connect easier, as at the moment I am using locks and presence of files to confirm whether the info is accurate. 

I do appreciate that its not the primary use for this package though and that its use is "out of scope"

jblance, 2021-04-05:
There's an early code base on my git "jkbms" that might be a better starting point as it is only jkbms focused (tho the decodes are better in mpp-solar)
Depending on what exactly you are trying to do, you might be better to start from the earlier code and modify it to keep outputting the results without disconnecting. 

There is an uptime result returned, you could use that to validate the age of the results

Just a guess on your intent though

alexanderfitu, 2021-04-05:
Thanks! That's a great idea, I will take a look. The main issue I have is that the jkbms pulls the data, then outputs it in JSON format to a text file, then my stuff goes through it all. If the data never changes the reset of the script never knows etc, but I think I have something useable now. I may fork yours off and post up my code, in case its helpful for somebody else.

jblance, 2021-04-05:
what is your script written in? Might be easier to grab the relevant bits of jkbms and include them directly
If you figure out any other commands or content in the results from jkbms, I'd appreciate knowing so i can add it to my work

alexanderfitu, 2021-04-08:
It's all in Bash (I know, but I don't know python and I did not want to learn a new language to figure out what I want to do). It is attached 
[checkbattery.txt](https://github.com/jblance/mpp-solar/files/6278661/checkbattery.txt)

jblance, 2021-04-12:
wow - you are brave using bash
the way you are using would make it hard to use mpp-solar and 'keep the connection alive'
the python code kinda does:
1. connect via bluetooth 
2. send init command to bms
3. ignore result
4. send actual command to bms
5. get result 
6. close connection

It does this each time you call it
To make it faster / more like the app the approach would need to loop at 5 and redo 4 when needed, but this would need to be in an environment that was persistant (ie loops)  and your bash script is a run many type of approach

alexanderfitu, 2021-04-13:
I agree. It is not ideal, but I have 0 experience in python and at least with bash, I can make it do what I want, with some limitations. 
At the moment, I have it working reliably, at the expense of a 30 second delay if values go past a certain point. There are some situations where the BMS returns gibberish, or the connection times out, so handling that is important and the default output of the charger needs to be off. 

I may invest some time into learning python if I could obtain a continuous data stream from both BMS's without looping as you say.

jblance, 2021-04-14:
Worth learning python! Python can directly interact with the Pi GPIO as well
Note your code will need to loop and maintain state during the looping
It is an interesting problem 

I'm no expert, but feel free to ask if you have any questions (discussions area probably better for this)

closed 2021-04-14 by jblance