jblance/mpp-solar #83: Any chance of supporting Daly "Smart" BMSs?

danb35, 2021-04-11:
Daly makes a fairly-popular line of BMSs with communications ability via both serial and CAN.  They have software available, but of course it's Windows and GUI-only, and doesn't support logging the data.  https://github.com/njfaria/dalylog can apparently do the job, but it's Raspberry Pi-only, GUI-only, and apparently closed-source (as its author doesn't post the source, only a compiled binary for the rPi)--but at least it does appear to support data logging via MQTT.

Since this project includes jkbms, I was hoping you'd be willing to support other BMSs like the Daly.  I've found documentation of their protocol; it's attached to this issue.  Since I have one of these units, I'd be happy to test as it goes along.  Thanks for considering.

[Part 4_ Daly RS485+UART Protocol.pdf](https://github.com/jblance/mpp-solar/files/6292886/Part.4_.Daly.RS485%2BUART.Protocol.pdf)

jblance, 2021-04-11:
I'm definitely willing to add support to other devices
From the protocol doc you sent it doesnt look too complicated (so long as we can get responses)
What is your setup? Do you have a Linux type device connected to the Daly?

danb35, 2021-04-11:
> Do you have a Linux type device connected to the Daly?

I will shortly--I need to get a USB hub so I can connect both it and the inverter to the Ubuntu 20.04 VM I'm currently using for mpp-solar, but that hub should be here in a couple of days.  I should also connect a Windows laptop just to make sure the communications are working OK (they ship a USB-RS485 adapter with screw terminals--I'm pretty sure I have it wired correctly, but really should rule it out as a possible source of problems before getting into other things).

jblance, 2021-04-12:
[dalytest.py.txt](https://github.com/jblance/mpp-solar/files/6293141/dalytest.py.txt)
When you get the daly connected can you run this test (you'll need to edit the port if the daly isnt on /dev/ttyUSB0)

The protocol doc has a number of vague statements, so this is my first guess and what a command might look like...
You'll see in the file a section that contains the command components
```
## Command components ???
startFlag = bytes.fromhex("A5")
moduleAddress = bytes.fromhex("80")
commandID = bytes.fromhex("90")
dataLength = bytes.fromhex("08")
data = bytes.fromhex("00" * 8) 
```
If it doesnt work you can try changing these to see if you can get a response
It is also possible the checksum calculation is wrong....

Let me know what you get back

jblance, 2021-04-12:
Actually will likely need a \n termination on the command, try this file
[dalytest.py.txt](https://github.com/jblance/mpp-solar/files/6293403/dalytest.py.txt)

danb35, 2021-04-16:
Thanks for the test script.  I'm having some comms problems even with the Windows software, so trying to sort those out first.

firemen027, 2021-04-19:
Just want to say : You make my day

Presently use John Blance Mpp-solar on raspberry PI 4b linked to  (LV-MK) 3000w, it work flawless.
I also use a Daly BMS, and I'm on the same boat as danb35, présent solution are not working well for me.
Daly log have strange output for me and is not really for my needs

I'm willing to do lot of test on a rpi when i'm at my cabin. (where I run my off grid setup)
I have uart to usb adapter from Daly
Also have uart to Bluetooth adapter from Daly: I can only use one at a time.

I can assist you and danb35 as you need ( next presence at the cabin will be april 23 to 25)

My best regards

jblance, 2021-04-19:
Thanks for the offer of help
Key is to understand the protocol a bit better - as always the docs are vague in all the important parts
Above is a script (most likely the second one) to try and talk to the BMS and see what response is given - if you get a chance running the script against your bms would help me understand the protocol and implement it

firemen027, 2021-04-24:
Hi,  Here the first response for the script

pi@rpichalet:~/jblanc $ python3 dalytest.py
checksum: b'\xbd'
sending command b'\xa5\x80\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00\xbd\n'
wrote 14 bytes
Got response: b'\xa5\x01\x90\x08\x02\x10\x00\x00uo\x03\xbc\xf3'
Got response: b''
Got response: b''
Got response: b''
Got response: b''
^CTraceback (most recent call last):
  File "dalytest.py", line 48, in <module>
    time.sleep(10)
KeyboardInterrupt

firemen027, 2021-04-24:
Hi again, I modified the script and change the command for 91, will do for some command

pi@rpichalet:~/jblanc $ python3 dalytest91.py
checksum: b'\xbe'
sending command b'\xa5\x80\x91\x08\x00\x00\x00\x00\x00\x00\x00\x00\xbe\n'
wrote 14 bytes
Got response: b'\xa5\x01\x91\x08\x0c\xeb\x05\x0c\xdd\x0b\x03\xb9\xeb'
Got response: b''

For command 92

pi@rpichalet:~/jblanc $ python3 dalytest92.py
checksum: b'\xbf'
sending command b'\xa5\x80\x92\x08\x00\x00\x00\x00\x00\x00\x00\x00\xbf\n'
wrote 14 bytes
Got response: b'\xa5\x01\x92\x08,\x01,\x01\xdd\x0b\x03\xb9>'
Got response: b''

for command 93

pi@rpichalet:~/jblanc $ python3 dalytest93.py
checksum: b'\xc0'
sending command b'\xa5\x80\x93\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc0\n'
wrote 14 bytes
Got response: b'\xa5\x01\x93\x08\x02\x00\x01|\x00\x04\x0f\x10\xe3'
Got response: b''
^CTraceback (most recent call last):

Command 94

pi@rpichalet:~/jblanc $ python3 dalytest94.py
checksum: b'\xc1'
sending command b'\xa5\x80\x94\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc1\n'
wrote 14 bytes
Got response: b'\xa5\x01\x94\x08\x10\x01\x00\x00\x00\x00\x02\x10e'
Got response: b''
^CTraceback (most recent call last):

command 95   (I feel the request is good because this command is for cell voltage, in my case I got 16 cell presently around 3.300 volts 

pi@rpichalet:~/jblanc $ python3 dalytest95.py
checksum: b'\xc2'
sending command b'\xa5\x80\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc2\n'
wrote 14 bytes
Got response: b'\xa5\x01\x95\x08\x01\x0c\xea\x0c\xec\x0c\xea\x108\xa5\x01\x95\x0                                                            8\x02\x0c\xea\x0c\xeb\x0c\xea\x108\xa5\x01\x95\x08\x03\x0c\xeb\x0c\xe9\x0c\xdf\x                                                            10-\xa5\x01\x95\x08\x04\x0c\xe1\x0c\xdd\x0c\xe9\x10"\xa5\x01\x95\x08\x05\x0c\xe9                                                            \x0c\xea\x0c\xe9\x108\xa5\x01\x95\x08\x06\x0c\xe2\x00\x00\x00\x00\x10G\xa5\x01\x                                                            95\x08\x07\x00\x00\x00\x00\x00\x00\x10Z\xa5\x01\x95\x08\x08\x00\x00\x00\x00\x00\                                                            x00\x10[\xa5\x01\x95\x08\t\x00\x00\x00\x00\x00\x00\x10\\\xa5\x01\x95\x08\n'
Got response: b'\x00\x00\x00\x00\x00\x00\x10]\xa5\x01\x95\x08\x0b\x00\x00\x00\x0                                                            0\x00\x00\x10^\xa5\x01\x95\x08\x0c\x00\x00\x00\x00\x00\x00\x10_\xa5\x01\x95\x08\                                                            r\x00\x00\x00\x00\x00\x00\x10`\xa5\x01\x95\x08\x0e\x00\x00\x00\x00\x00\x00\x10a\                                                            xa5\x01\x95\x08\x0f\x00\x00\x00\x00\x00\x00\x10b\xa5\x01\x95\x08\x10\x00\x00\x00                                                            \x00\x00\x00\x10c'
Got response: b''
^CTraceback (most recent call last):

firemen027, 2021-04-24:
Part 2

command 96

pi@rpichalet:~/jblanc $ python3 dalytest96.py
checksum: b'\xc3'
sending command b'\xa5\x80\x96\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc3\n'
wrote 14 bytes
Got response: b'\xa5\x01\x96\x08\x01.\x00\x00\x00\x00\x00\x00s\xa5\x01\x96\x08\x02\x00\x00\x00\x00\x00\x00\x00F'
Got response: b''
^CTraceback (most recent call last):

command 97 (again i feel the command is ok, this is the balancing status and we are in the night, so no balancing is active .. just 0000 everywhere)

pi@rpichalet:~/jblanc $ python3 dalytest97.py
checksum: b'\xc4'
sending command b'\xa5\x80\x97\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc4\n'
wrote 14 bytes
Got response: b'\xa5\x01\x97\x08\x00\x00\x00\x00\x00\x00\x00\x00E'
Got response: b''
^CTraceback (most recent call last):

Command 98

pi@rpichalet:~/jblanc $ python3 dalytest98.py
checksum: b'\xc5'
sending command b'\xa5\x80\x98\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc5\n'
wrote 14 bytes
Got response: b'\xa5\x01\x98\x08\x00\x08\x00\x00\x00\x00\x00\x00N'
Got response: b''
^CTraceback (most recent call last):

I dunno if it's the good way to show the response from BMS, But i paste from terminal.
Also I always paste the part where I put the ctrl c, because a loop seem to begin in the script.

And the last part: sorry for my english: I,m a french canadian ;)

thanks!!

jblance, 2021-04-24:
Thanks, that looks positive (getting data returned at least)
Can you give the approx values that you'd be expecting (at least for some of the commands)
I'll have a go at implementing the protocol decode soon

(Your English is great, so no worries there)
Regards

firemen027, 2021-04-24:
Thanks again, 
here the capture from interface (bluetooth connexion) few minutes ago

[image]

firemen027, 2021-04-24:
Also here fresh result from command: third hex for command number ( few minutes difference from screen capture)

Got response: b'\xa5\x01\x90\x08\x02\x14\x00\x00uE\x03x\x89'
Got response: b'\xa5\x01\x91\x08\r\x00\x0f\x0c\xfe\x01\x03x\xe1'
Got response: b'\xa5\x01\x92\x087\x017\x01\xfe\x01\x03x*'
Got response: b'\xa5\x01\x93\x08\x02\x01\x01x\x00\x03\xcb@\xcb'
Got response: b'\xa5\x01\x94\x08\x10\x01\x00\x00\x00\x00\x03@\x96'

next one : two response

Got response: b'\xa5\x01\x95\x08\x01\x0c\xfd\x0c\xfe\x0c\xfe@\xa1\xa5\x01\x95\x08\x02\x0c\xfe\x0c\xfe\x0c\xfe@\xa3\xa5\x01\x95\x08\x03\x0c\xfe\x0c\xfe\x0c\xfe@\xa4\xa5\x01\x95\x08\x04\x0c\xfe\x0c\xfc\x0c\xfe@\xa3\xa5\x01\x95\x08\x05\x0c\xfe\x0c\xff\x0c\xfe@\xa7\xa5\x01\x95\x08\x06\x0c\xfc\x00\x00\x00\x00@\x91\xa5\x01\x95\x08\x07\x00\x00\x00\x00\x00\x00@\x8a\xa5\x01\x95\x08\x08\x00\x00\x00\x00\x00\x00@\x8b\xa5\x01\x95\x08\t\x00\x00\x00\x00\x00\x00@\x8c\xa5\x01\x95\x08\n'
Got response: b'\x00\x00\x00\x00\x00\x00@\x8d\xa5\x01\x95\x08\x0b\x00\x00\x00\x00\x00\x00@\x8e\xa5\x01\x95\x08\x0c\x00\x00\x00\x00\x00\x00@\x8f\xa5\x01\x95\x08\r\x00\x00\x00\x00\x00\x00@\x90\xa5\x01\x95\x08\x0e\x00\x00\x00\x00\x00\x00@\x91\xa5\x01\x95\x08\x0f\x00\x00\x00\x00\x00\x00@\x92\xa5\x01\x95\x08\x10\x00\x00\x00\x00\x00\x00@\x93'

Got response: b'\xa5\x01\x96\x08\x017\x00\x00\x00\x00\x00\x00|\xa5\x01\x96\x08\x02\x00\x00\x00\x00\x00\x00\x00F'
Got response: b'\xa5\x01\x97\x08\x00\x00\x00\x00\x00\x00\x00\x00E'
Got response: b'\xa5\x01\x98\x08\x00\x00\x00\x00\x00\x00\x00\x00F'

Hope this help

My best regards

spilegi, 2021-04-26:
I have one of these Smart BMS's as well and would be more than happy to help in any way possible.

jblance, 2021-04-26:
Latest commit has a few DALY protocol commands implemented - be good if you could test to see what we get, command will be something like
```
mpp-solar -p /dev/ttyUSB0 -b 9600 -P DALY -c SOC -I
```
Currently have the first few commands implemented:
```
Command: command help - List available commands for protocol DALY
------------------------------------------------------------
Parameter                     	Value           Unit
SOC                           	State of Charge	    
cellMinMax                    	Cell Minimum and Maximum Voltages	    
cellTemperatures              	Cell Minimum and Maximum Temperatures	    
mosStatus                     	mosStatus      	    
status                        	Status Information	
```

Let me know what responses you get (and if it works what name corrections are needed)
Note to self, need to implement localization / translate ability....

spilegi, 2021-04-26:
Here's the output that I recieved

(mppsolar) root@raspberrypi:~# mpp-solar -p /dev/ttyUSB7  -b 9600 -P DALY -c SOC -I
INFO:root:Solar Device Command Utility, version: 0.7.48, recent changes: add DALY v1
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB7 (porttype=None)" using protocol "DALY"
INFO:device:set_port: Using serialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x75eb4f70>, protocol: <mppsolar.protocols.daly.daly object at 0x75eb4ed0> for command: SOC, tag: SOC, outputs: screen
INFO:device:run_command: Running command SOC
INFO:daly:get_full_command: Using protocol b'DALY' with 5 commands
INFO:device:run_command: full command b'\xa5\x80\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00\xbd\n' for command SOC
INFO:AbstractProtocol:decode: response passed to decode: b'\xa5\x01\x90\x08\x01\t\x00\x00u\xcf\x03\n\x99'
INFO:AbstractProtocol:decode: Processing response of type POSITIONAL
INFO:device:run_command: Decoded response {'raw_response': ['¥\x01\x90\x08\x01\t\x00\x00uÏ\x03\n\x99', ''], '_command': 'SOC', '_command_description': 'State of Charge', 'pressure': [26.5, 'V'], 'acquistion': [0.0, 'V'], 'current': [15.9, 'A'], 'SOC': [92.1, '%']}
INFO:root:attempting to create output processor: screen
INFO:screen:output: Using output processor: screen
Command: SOC - State of Charge
------------------------------------------------------------
Parameter                     	Value           Unit
pressure                      	26.5           	V   
acquistion                    	0.0            	V   
current                       	15.9           	A   
soc                           	92.1           	%   
(mppsolar) root@raspberrypi:~#

jblance, 2021-04-27:
good news
not sure on the terms either (they're from a translated chinese doc)
pressure = is that total voltage? or is pressure correct

firemen027, 2021-04-27:
wow!!
I return back to the cabin next week end, I can confirm lot of information, Spilegi, thanks for the help.
Pressure is the battery bank voltage I think!

spilegi, 2021-04-27:
All the other commands seem to be working great.....  Min/max, status etc...

I have one request, which I think lots of others will find useful. 

The listing of each cell and it's corresponding voltage.

jblance, 2021-04-27:
Thanks for the confirmation
Yes, I'm working on the cell voltages command - it is more complicated as it returns multiple frames (the number depends on the number of cells)
Watch this space....

spilegi, 2021-04-27:
jblance,  awesome  thanks for all your hard work.

jblance, 2021-04-28:
ok added cellVoltages command and changed cellTemperatures to cellMinMaxTemps and added a different cellTemperatures one

The new cellTemperatures is a little wierd as the temps are 40C offset, so no temp result gives -40 (as would -40 - looking at you Canadians... @firemen027 ), result example I got from @firemen027 only had 1 temp result and the doc talks about "monomer temperature 1 -16, but with 21 possibe results...)

spilegi, 2021-04-28:
Cell Voltages looks good with some warnings, I run 24v 8s, so output looks great

(mppsolar) root@raspberrypi:~# mpp-solar -p /dev/ttyUSB7 -b 9600 -P DALY -c cellVoltages -I
INFO:root:Solar Device Command Utility, version: 0.7.49, recent changes: add DALY cellVoltages
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB7 (porttype=None)" using protocol "DALY"
INFO:device:set_port: Using serialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x764ca7f0>, protocol: <mppsolar.protocols.daly.daly object at 0x75ecbfb0> for command: cellVoltages, tag: cellVoltages, outputs: screen
INFO:device:run_command: Running command cellVoltages
INFO:daly:get_full_command: Using protocol b'DALY' with 7 commands
INFO:device:run_command: full command b'\xa5\x80\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc2\n' for command cellVoltages
INFO:AbstractProtocol:decode: response passed to decode: b'\xa5\x01\x95\x08\x01\x0c\xcd\x0c\xd6\x0c\xd6\x8eo\xa5\x01\x95\x08\x02\x0c\xd7\x0c\xd2\x0c\xd6\x8ev\xa5\x01\x95\x08\x03\x0c\xd4\x0c\xd5\x00\x00\x8e\x95\xa5\x01\x95\x08\x04\x00\x00\x00\x00\x00\x00\x8e\xd5\xa5\x01\x95\x08\x05\x00\x00\x00\x00\x00\x00\x8e\xd6\xa5\x01\x95\x08\x06\x00\x00\x00\x00\x00\x00\x8e\xd7\xa5\x01\x95\x08\x07\x00\x00\x00\x00\x00\x00\x8e\xd8\xa5\x01\x95\x08\x08\x00\x00\x00\x00\x00\x00\x8e\xd9\xa5\x01\x95\x08\t\x00\x00\x00\x00\x00\x00\x8e\xda\xa5\x01\x95\x08\n\x00\x00\x00\x00\x00\x00\x8e\xdb\xa5'
INFO:daly:is multiframe response - assuming ok for now
INFO:daly:Multi frame response with 11 frames
INFO:AbstractProtocol:decode: Processing response of type MULTIFRAME-POSITIONAL
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b'\x0c\xcd'
INFO:protocol_helpers:Hex b'\x0c\xcd' 2 byte decoded to 3277
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b'\x0c\xd6'
INFO:protocol_helpers:Hex b'\x0c\xd6' 2 byte decoded to 3286
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+3:02d} Voltage' b'\x0c\xd6'
INFO:protocol_helpers:Hex b'\x0c\xd6' 2 byte decoded to 3286
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b'\x0c\xd7'
INFO:protocol_helpers:Hex b'\x0c\xd7' 2 byte decoded to 3287
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b'\x0c\xd2'
INFO:protocol_helpers:Hex b'\x0c\xd2' 2 byte decoded to 3282
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+3:02d} Voltage' b'\x0c\xd6'
INFO:protocol_helpers:Hex b'\x0c\xd6' 2 byte decoded to 3286
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b'\x0c\xd4'
INFO:protocol_helpers:Hex b'\x0c\xd4' 2 byte decoded to 3284
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b'\x0c\xd5'
INFO:protocol_helpers:Hex b'\x0c\xd5' 2 byte decoded to 3285
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+3:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+3:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+3:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+3:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+3:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+3:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
[... cut ...]
cell_05_temperature           	-40            	°C  
cell_06_temperature           	-40            	°C  
cell_07_temperature           	-40            	°C  
cell_08_temperature           	-40            	°C  
cell_09_temperature           	-40            	°C  
cell_10_temperature           	-40            	°C  
cell_11_temperature           	-40            	°C  
cell_12_temperature           	-40            	°C  
cell_13_temperature           	-40            	°C  
cell_14_temperature           	-40            	°C

danb35, 2021-04-29:
Just ran across this thread which might add some other information here:
https://diysolarforum.com/threads/decoding-the-daly-smartbms-protocol.21898/

I'm still having comms problems with mine--sorry I haven't been as helpful here as I'd meant to be.

spilegi, 2021-04-29:
I have a handful of these BMS's. I'm receiving checksum errors on some of them

INFO:root:Solar Device Command Utility, version: 0.7.49, recent changes: add DALY cellVoltages
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB5 (porttype=None)" using protocol "DALY"
DEBUG:root:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:device:mppsolar __init__ args ()
DEBUG:device:mppsolar __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB5', 'protocol': 'DALY', 'baud': 9600, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
DEBUG:device:get_port_type: port matches ttyusb
INFO:device:set_port: Using serialio for communications
DEBUG:device:set_protocol: Protocol DALY
DEBUG:device:mppsolar __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0x76543570>, protocol <mppsolar.protocols.daly.daly object at 0x76543630>
DEBUG:root:Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x76543550>, 'SOC', 'SOC', 'screen', None, None)]
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x76543570>, protocol: <mppsolar.protocols.daly.daly object at 0x76543630> for command: SOC, tag: SOC, outputs: screen
INFO:device:run_command: Running command SOC
INFO:daly:get_full_command: Using protocol b'DALY' with 7 commands
DEBUG:AbstractProtocol:get_command_defn: Processing command 'SOC'
DEBUG:AbstractProtocol:get_command_defn: Found command SOC in protocol b'DALY'
DEBUG:daly:get_full_command: full command: b'\xa5\x80\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00\xbd\n'
INFO:device:run_command: full command b'\xa5\x80\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00\xbd\n' for command SOC
DEBUG:AbstractProtocol:get_command_defn: Processing command 'SOC'
DEBUG:AbstractProtocol:get_command_defn: Found command SOC in protocol b'DALY'
DEBUG:SerialIO:send_and_receive: port /dev/ttyUSB5, baudrate 9600
DEBUG:SerialIO:send_and_receive: Executing command via serialio...
DEBUG:SerialIO:send_and_receive: serial response was: b'\xa5\xd2\x03R\x05\xdc\x0c\x80\x00\x01\x00\x08\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x01,\x01,\x0eB\x0eB\n\xf0\x01$\x01$\x00\xe0\x00\xe0r\x13r\x13yWyW\x00i\x00i\x00\x00\x00\x00\x00n\x00n\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\xff\x0c\xf8\x002\x00\x01\x00\x01\x03\xe8\x00(YU'
DEBUG:device:run_command: Send and Receive Response b'\xa5\xd2\x03R\x05\xdc\x0c\x80\x00\x01\x00\x08\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x01,\x01,\x0eB\x0eB\n\xf0\x01$\x01$\x00\xe0\x00\xe0r\x13r\x13yWyW\x00i\x00i\x00\x00\x00\x00\x00n\x00n\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\xff\x0c\xf8\x002\x00\x01\x00\x01\x03\xe8\x00(YU'
INFO:AbstractProtocol:decode: response passed to decode: b'\xa5\xd2\x03R\x05\xdc\x0c\x80\x00\x01\x00\x08\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x01,\x01,\x0eB\x0eB\n\xf0\x01$\x01$\x00\xe0\x00\xe0r\x13r\x13yWyW\x00i\x00i\x00\x00\x00\x00\x00n\x00n\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\xff\x0c\xf8\x002\x00\x01\x00\x01\x03\xe8\x00(YU'
DEBUG:daly:checking validity of b'\xa5\xd2\x03R\x05\xdc\x0c\x80\x00\x01\x00\x08\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x01,\x01,\x0eB\x0eB\n\xf0\x01$\x01$\x00\xe0\x00\xe0r\x13r\x13yWyW\x00i\x00i\x00\x00\x00\x00\x00n\x00n\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\xff\x0c\xf8\x002\x00\x01\x00\x01\x03\xe8\x00(YU'
INFO:AbstractProtocol:DALY checksum did not match for response b'\xa5\xd2\x03R\x05\xdc\x0c\x80\x00\x01\x00\x08\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x01,\x01,\x0eB\x0eB\n\xf0\x01$\x01$\x00\xe0\x00\xe0r\x13r\x13yWyW\x00i\x00i\x00\x00\x00\x00\x00n\x00n\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\xff\x0c\xf8\x002\x00\x01\x00\x01\x03\xe8\x00(YU'
INFO:device:run_command: Decoded response {'ERROR': ["DALY checksum did not match for response b'\\xa5\\xd2\\x03R\\x05\\xdc\\x0c\\x80\\x00\\x01\\x00\\x08\\x00\\x00\\x00\\x00\\x00\\x01\\x00\\x00\\x00\\x00\\x00\\x00\\x01,\\x01,\\x0eB\\x0eB\\n\\xf0\\x01$\\x01$\\x00\\xe0\\x00\\xe0r\\x13r\\x13yWyW\\x00i\\x00i\\x00\\x00\\x00\\x00\\x00n\\x00n\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\xff\\x00\\xff\\x0c\\xf8\\x002\\x00\\x01\\x00\\x01\\x03\\xe8\\x00(YU'", '']}
DEBUG:root:results: {'ERROR': ["DALY checksum did not match for response b'\\xa5\\xd2\\x03R\\x05\\xdc\\x0c\\x80\\x00\\x01\\x00\\x08\\x00\\x00\\x00\\x00\\x00\\x01\\x00\\x00\\x00\\x00\\x00\\x00\\x01,\\x01,\\x0eB\\x0eB\\n\\xf0\\x01$\\x01$\\x00\\xe0\\x00\\xe0r\\x13r\\x13yWyW\\x00i\\x00i\\x00\\x00\\x00\\x00\\x00n\\x00n\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\xff\\x00\\xff\\x0c\\xf8\\x002\\x00\\x01\\x00\\x01\\x03\\xe8\\x00(YU'", '']}
INFO:root:attempting to create output processor: screen
DEBUG:screen:processor.screen __init__ kwargs {}
DEBUG:root:Using output filter: None
INFO:screen:output: Using output processor: screen
DEBUG:screen:output: kwargs {'data': {'ERROR': ["DALY checksum did not match for response b'\\xa5\\xd2\\x03R\\x05\\xdc\\x0c\\x80\\x00\\x01\\x00\\x08\\x00\\x00\\x00\\x00\\x00\\x01\\x00\\x00\\x00\\x00\\x00\\x00\\x01,\\x01,\\x0eB\\x0eB\\n\\xf0\\x01$\\x01$\\x00\\xe0\\x00\\xe0r\\x13r\\x13yWyW\\x00i\\x00i\\x00\\x00\\x00\\x00\\x00n\\x00n\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\xff\\x00\\xff\\x0c\\xf8\\x002\\x00\\x01\\x00\\x01\\x03\\xe8\\x00(YU'", '']}, 'tag': 'SOC', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': Fal [... cut ...]
DEBUG:helpers:key_wanted: No filter and key error not excluded by excl_filter None so key wanted
error                         	DALY checksum did not match for response b'\xa5\xd2\x03R\x05\xdc\x0c\x80\x00\x01\x00\x08\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x01,\x01,\x0eB\x0eB\n\xf0\x01$\x01$\x00\xe0\x00\xe0r\x13r\x13yWyW\x00i\x00i\x00\x00\x00\x00\x00n\x00n\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\xff\x0c\xf8\x002\x00\x01\x00\x01\x03\xe8\x00(YU'	    
DEBUG:root:Not daemon, so not looping
root@raspberrypi:/dev/serial/by-path#

jblance, 2021-04-29:
Well that response doesnt meet the format of the documented protocol i have, so not surprising that the crc check fails
It almost looks like the baud rate is wrong or you are gettting multiple results mixed up.
Are the ones that fail consistently the same BMS and are there any version dor model differences?
Is anything else trying to communicate to the BMSs (including wirelessly)?
Can you try other common baud rates, e.g 2400, 19200 etc

spilegi, 2021-04-30:
I'll give that a shot.  The initial one I had was a 250a model. The ones that aren't working are 60a versions.  I'm trying  a PC instead of a pi.  

It's odd.  The errors appear to be only with Cell Voltages.  I've attached the output.  SOC seems to be working just fine.  but cellVoltages, will often only error out or only display 3 or 4 cells

root@solar:/etc/solarmonitor/bms# mpp-solar -p /dev/ttyUSB0 -b 9600 -P DALY -c SOC -I
INFO:root:Solar Device Command Utility, version: 0.7.49, recent changes: add DALY cellVoltages
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "DALY"
INFO:device:set_port: Using serialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x7f4b7f97b910>, protocol: <mppsolar.protocols.daly.daly object at 0x7f4b7f97b850> for command: SOC, tag: SOC, outputs: screen
INFO:device:run_command: Running command SOC
INFO:daly:get_full_command: Using protocol b'DALY' with 7 commands
INFO:device:run_command: full command b'\xa5\x80\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00\xbd\n' for command SOC
INFO:AbstractProtocol:decode: response passed to decode: b'\xa5\x01\x90\x08\x01\x03\x00\x00uu\x00\xf9%'
INFO:AbstractProtocol:decode: Processing response of type POSITIONAL
INFO:AbstractProtocol:Got template r/10 for Battery Bank Voltage b'\x01\x03'
INFO:protocol_helpers:Hex b'\x01\x03' 2 byte decoded to 259
INFO:AbstractProtocol:Got template r/10 for acquistion b'\x00\x00'
INFO:protocol_helpers:Hex b'\x00\x00' 2 byte decoded to 0
INFO:AbstractProtocol:Got template (r-30000)/10 for Current b'uu'
INFO:protocol_helpers:Hex b'uu' 2 byte decoded to 30069
INFO:AbstractProtocol:Got template r/10 for SOC b'\x00\xf9'
INFO:protocol_helpers:Hex b'\x00\xf9' 2 byte decoded to 249
INFO:device:run_command: Decoded response {'raw_response': ['¥\x01\x90\x08\x01\x03\x00\x00uu\x00ù%', ''], '_command': 'SOC', '_command_description': 'State of Charge', 'Battery Bank Voltage': [25.9, 'V'], 'acquistion': [0.0, 'V'], 'Current': [6.9, 'A'], 'SOC': [24.9, '%']}
INFO:root:attempting to create output processor: screen
INFO:screen:output: Using output processor: screen
Command: SOC - State of Charge
------------------------------------------------------------
Parameter                     	Value           Unit
battery_bank_voltage          	25.9           	V   
acquistion                    	0.0            	V   
current                       	6.9            	A   
soc                           	24.9           	%   
root@solar:/etc/solarmonitor/bms# mpp-solar -p /dev/ttyUSB0 -b 9600 -P DALY -c cellVoltages -I
INFO:root:Solar Device Command Utility, version: 0.7.49, recent changes: add DALY cellVoltages
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "DALY"
INFO:device:set_port: Using serialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x7f5069e30910>, protocol: <mppsolar.protocols.daly.daly object at 0x7f5069e30850> for command: cellVoltages, tag: cellVoltages, outputs: screen
INFO:device:run_command: Running command cellVoltages
INFO:daly:get_full_command: Using protocol b'DALY' with 7 commands
INFO:device:run_command: full command b'\xa5\x80\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc2\n' for command cellVoltages
INFO:AbstractProtocol:decode: response passed to decode: b'\xa5\x01\x95\x08\x01\x0c\xa3\x0c\xa7\x0c\xa6\xf9Q\xa5\x01\x95\x08\x02\x0c\xa8\x0c\xa5\x0c\xa4\xf9S\xa5\x01\x95\x08\x03\x0c\xa5\x0c\xa6\x00\x00\xf9\xa2\xa5\x01\x95\x08\x04\x00\x00\x00\x00\x00\x00\xf9@\xa5\x01\x95\x08\x05\x00\x00\x00\x00\x00\x00\xf9A\xa5\x01\x95\x08\x06\x00\x00\x00\x00\x00\x00\xf9B\xa5\x01\x95\x08\x07\x00\x00\x00\x00\x00\x00\xf9C\xa5\x01\x95\x08\x08\x00\x00\x00\x00\x00\x00\xf9D\xa5\x01\x95\x08\t\x00\x00\x00\x00\x00\x00\xf9E\xa5\x01\x95\x08\n\x00\x00\x00\x00\x00\x00\xf9F\xa5\x01\x95\x08\x0b'
INFO:daly:is multiframe response - assuming ok for now
INFO:daly:Multi frame response with 13 frames
INFO:AbstractProtocol:decode: Processing response of type MULTIFRAME-POSITIONAL
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b'\x0c\xa3'
INFO:protocol_helpers:Hex b'\x0c\xa3' 2 byte decoded to 3235
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b'\x0c\xa7'
INFO:protocol_helpers:Hex b'\x0c\xa7' 2 byte decoded to 3239
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+3:02d} Voltage' b'\x0c\xa6'
INFO:protocol_helpers:Hex b'\x0c\xa6' 2 byte decoded to 3238
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b'\x0c\xa8'
INFO:protocol_helpers:Hex b'\x0c\xa8' 2 byte decoded to 3240
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b'\x0c'
WARNING:protocol_helpers:Hex encoded value must be 2 bytes long. Was 1 length
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+3:02d} Voltage' b''
WARNING:protocol_helpers:Hex encoded value must be 2 bytes long. Was 0 length
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+1:02d} Voltage' b''
WARNING:protocol_helpers:Hex encoded value must be 2 bytes long. Was 0 length
INFO:AbstractProtocol:Got template r/1000 for f'Cell {3*f+2:02d} Voltage' b''
WARNING:protocol_helpers:Hex encoded value must be 2 bytes long. Was 0 length
[... cut ...]
cell_30_voltage               	0.0            	V   
cell_31_voltage               	0.0            	V   
cell_32_voltage               	0.0            	V   
cell_33_voltage               	0.0            	V   
cell_34_voltage               	0.0            	V   
cell_35_voltage               	0.0            	V   
cell_36_voltage               	0.249          	V   
cell_37_voltage               	0.0            	V   
cell_38_voltage               	0.0            	V   
cell_39_voltage               	0.0            	V

jblance, 2021-04-30:
ok i fixed a couple of mistakes in the frame split logic in 0.7.50 or later
How many cells should there be? (i still only get 8 with your data)
```
Command: cellVoltages - Cell Voltages Information
------------------------------------------------------------
Parameter                     	Value           Unit
cell_01_voltage               	3.235          	V   
cell_02_voltage               	3.239          	V   
cell_03_voltage               	3.238          	V   
cell_04_voltage               	3.24           	V   
cell_05_voltage               	3.237          	V   
cell_06_voltage               	3.236          	V   
cell_07_voltage               	3.237          	V   
cell_08_voltage               	3.238          	V   
cell_09_voltage               	0.0            	V   
cell_10_voltage               	0.0            	V   
cell_11_voltage               	0.0            	V   
...
```
I think the reading from the serial port might be truncating some data - might need a different approach to the read

spilegi, 2021-04-30:
there are 8 cells

I'll attempt it tonight

firemen027, 2021-05-01:
Hi everybody,

I'm back to the cabin for one night, so I'm in a rush to test.

Found one issue for the command 94, status 
The string convertion is wrong, the hex is 10, the display show 10, but my string is 16 (sem like there's no conversion)
Here a capture

[image]

For the second Byte (temperature) the value is around 13 to 15 celcius presently. I try to understand if it's the good field

More to come 
Thank you very Much JBlance

firemen027, 2021-05-01:
Again in my testing: 

In the command 91 cellMinMaxVoltages, the doc describ 4 return, soc is not supposed to come with this command

[image]

firemen027, 2021-05-01:
Hi again

Question: how I can put all command in the config file?

Or I must made a section for each command?

Thanks again

firemen027, 2021-05-02:
It's me again

I add this to the config file.

[DALY]
#model=standard
protocol=DALY
port=/dev/ttyUSB0
baud=9600
command=SOC
tag=DALYSOC
outputs=mqtt

restart de the deamon, but I received many error from serial port, but also received some good value.
Almost 10 error for one good reading, 

The error are Mainly : no response

Sometime: Serial command execution failed
And sometime: DALY checksum did not match for response b'\xa5'

[image]

I change the loop pause to 10, same effect.

I'm still searching, but I understand a lot more,

Thanks again

firemen027, 2021-05-02:
Yes me again,

Mostly everything is now running and send to mqtt, Node Red is working to, 
Have some random error in the checksum for the SOC command

Here the error display in mqtt

DALY checksum did not match for response b'\xa5\x01\x90\x08\x02\r'

35 error in the last 20 minutes, presently cloudy here, seem to happen when sun appear between cloud. (high input from panel)

Again: thank you JBlance

jblance, 2021-05-02:
Oh I see, looks like the standard SerialIO approach wont work for the DALY protocol (it takes the '\r' as end of the data)
I'll look at a new IO approach - this will probably address the errors as well.

Re multiple commands in the config file - just add them to the command line separated by comments, e.g.:
```
# This example would work on a single LV5048
[LV5048]
protocol=PI41
port=/dev/hidraw0
command=QPGS0,QP2GS0
tag=Inverter1
outputs=influx2_mqtt
```

jblance, 2021-05-02:
I have added a new serial handler for the DALY BMS
Can you try it and see if it works / addresses any of the errors
to use it (after updating to 0.7.51 or later add `--porttype DALY` to the command line

jblance, 2021-05-02:
> Hi everybody,
> 
> I'm back to the cabin for one night, so I'm in a rush to test.
> 
> Found one issue for the command 94, status
> The string convertion is wrong, the hex is 10, the display show 10, but my string is 16 (sem like there's no conversion)
>
> In the command 91 cellMinMaxVoltages, the doc describ 4 return, soc is not supposed to come with this command
> 

OK, I've removed SOC from all commands except SOC and change 'Battery String' to an int
0.7.51 or later has these changes

firemen027, 2021-05-03:
For you J Blance.

Will be back at the cabin in 3 weeks, so no more test until this, So all change you make will be test later.

Here what I can display with your work!! More tuning to do but I'm very satisfied.

[image]

jblance, 2021-05-03:
ok - looks good
what version of influxdb are you using?

firemen027, 2021-05-03:
> ok - looks good
> what version of influxdb are you using?

Présently InfluxDB v1.8.4

danb35, 2021-05-03:
After three USB-RS485 adapters failed to work, a USB-UART adapter finally got me talking to the BMS.  So thanks, all but @jblance, for testing (which I'd intended to be able to do), and thanks, @jblance, for the work so far in implementing this.  Most of the commands are working as expected for me:
```
root@solar:~# mpp-solar -p /dev/ttyUSB0 -b 9600 -P DALY -c SOC
Command: SOC - State of Charge
------------------------------------------------------------
Parameter                     	Value           Unit
battery_bank_voltage          	27.2           	V
acquistion                    	0.0            	V
current                       	-1.7           	A
soc                           	96.0           	%
root@solar:~# mpp-solar -p /dev/ttyUSB0 -b 9600 -P DALY -c status
Command: status - Status Information
------------------------------------------------------------
Parameter                     	Value           Unit
battery_string                	8
temperature                   	02
charger_status                	disconnected
load_status                   	disconnected
flags_(todo)                  	00
charge/discharge_cycles       	322            	cycles
reserved                      	00
root@solar:~# mpp-solar -p /dev/ttyUSB0 -b 9600 -P DALY -c cellTemperatures
Command: cellTemperatures - Cell Temperature Information
------------------------------------------------------------
Parameter                     	Value           Unit
cell_01_temperature           	24             	°C
cell_02_temperature           	26             	°C
cell_03_temperature           	-40            	°C
cell_04_temperature           	-40            	°C
cell_05_temperature           	-40            	°C
cell_06_temperature           	-40            	°C
cell_07_temperature           	-40            	°C
cell_08_temperature           	-40            	°C
cell_09_temperature           	-40            	°C
cell_10_temperature           	-40            	°C
cell_11_temperature           	-40            	°C
cell_12_temperature           	-40            	°C
cell_13_temperature           	-40            	°C
cell_14_temperature           	-40            	°C
```
But `cellVoltages` isn't:
```
root@solar:~#  mpp-solar -p /dev/ttyUSB0 -b 9600 -P DALY -c cellVoltages -I
INFO:root:Solar Device Command Utility, version: 0.7.52, recent changes: corrections to DALY protocol
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "DALY"
INFO:device:set_port: Using serialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x7fb080e211c0>, protocol: <mppsolar.protocols.daly.daly object at 0x7fb080e21220> for command: cellVoltages, tag: cellVoltages, outputs: screen
INFO:device:run_command: Running command cellVoltages
INFO:daly:get_full_command: Using protocol b'DALY' with 7 commands
INFO:device:run_command: full command b'\xa5\x80\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc2\n' for command cellVoltages
INFO:AbstractProtocol:decode: response passed to decode: b'\xa5\x01\x95\x08\x01\r'
INFO:AbstractProtocol:DALY checksum did not match for response b'\xa5\x01\x95\x08\x01\r'
INFO:device:run_command: Decoded response {'ERROR': ["DALY checksum did not match for response b'\\xa5\\x01\\x95\\x08\\x01\\r'", '']}
INFO:root:attempting to create output processor: screen
INFO:screen:output: Using output processor: screen
Parameter                     	Value           Unit
error                         	DALY checksum did not match for response b'\xa5\x01\x95\x08\x01\r'
root@solar:~# mpp-solar -p /dev/ttyUSB0 --porttype DALY -b 9600 -P DALY -c cellVoltages -I
INFO:root:Solar Device Command Utility, version: 0.7.52, recent changes: corrections to DALY protocol
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=DALY)" using protocol "DALY"
INFO:device:set_port: Port overide - using port 'DALY'
INFO:device:set_port: Using dalyserialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0x7f73657451f0>, protocol: <mppsolar.protocols.daly.daly object at 0x7f73657451c0> for command: cellVoltages, tag: cellVoltages, outputs: screen
INFO:device:run_command: Running command cellVoltages
INFO:daly:get_full_command: Using protocol b'DALY' with 7 commands
INFO:device:run_command: full command b'\xa5\x80\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc2\n' for command cellVoltages
WARNING:DalySerialIO:send_and_receive: Serial read error: unsupported operand type(s) for +=: 'NoneType' and 'bytes'
INFO:DalySerialIO:Command execution failed
INFO:root:attempting to create output processor: screen
INFO:screen:output: Using output processor: screen
Parameter                     	Value           Unit
error                         	Serial command execution failed
```

jblance, 2021-05-03:
Try adding `--porttype daly` the the command

danb35, 2021-05-03:
```
root@solar:~# mpp-solar -p /dev/ttyUSB0 --porttype daly -b 9600 -P DALY -c cellVoltages -I
INFO:root:Solar Device Command Utility, version: 0.7.52, recent changes: corrections to DALY protocol
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=daly)" using protocol "DALY"
INFO:device:set_port: Port overide - using port 'daly'
INFO:device:set_port: Using dalyserialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0x7fc69e5bc1f0>, protocol: <mppsolar.protocols.daly.daly object at 0x7fc69e5bc1c0> for command: cellVoltages, tag: cellVoltages, outputs: screen
INFO:device:run_command: Running command cellVoltages
INFO:daly:get_full_command: Using protocol b'DALY' with 7 commands
INFO:device:run_command: full command b'\xa5\x80\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc2\n' for command cellVoltages
WARNING:DalySerialIO:send_and_receive: Serial read error: unsupported operand type(s) for +=: 'NoneType' and 'bytes'
INFO:DalySerialIO:Command execution failed
INFO:root:attempting to create output processor: screen
INFO:screen:output: Using output processor: screen
Parameter                     	Value           Unit
error                         	Serial command execution failed
```

jblance, 2021-05-03:
it looks like i havent got it quite right yet
can you change `-I` to `-D` and post the output

jblance, 2021-05-03:
actually, found a bug - yes pushed an update

danb35, 2021-05-03:
Man, you're quick:
```
root@solar:~# mpp-solar -p /dev/ttyUSB0 --porttype daly -b 9600 -P DALY -c cellVoltages -I
INFO:root:Solar Device Command Utility, version: 0.7.53, recent changes: corrections to DALY protocol
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=daly)" using protocol "DALY"
INFO:device:set_port: Port overide - using port 'daly'
INFO:device:set_port: Using dalyserialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0x7fdf23480610>, protocol: <mppsolar.protocols.daly.daly object at 0x7fdf234805b0> for command: cellVoltages, tag: cellVoltages, outputs: screen
INFO:device:run_command: Running command cellVoltages
INFO:daly:get_full_command: Using protocol b'DALY' with 7 commands
INFO:device:run_command: full command b'\xa5\x80\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc2\n' for command cellVoltages
INFO:AbstractProtocol:decode: response passed to decode: b'\xa5\x01\x95\x08\x01\rM\r[\r8\x00K\xa5\x01\x95\x08\x02\rF\rV\r\\\x00d\xa5\x01\x95\x08\x03\ra\r1\x00\x00\x00\xf2\xa5\x01\x95\x08\x04\x00\x00\x00\x00\x00\x00\x00G\xa5\x01\x95\x08\x05\x00\x00\x00\x00\x00\x00\x00H\xa5\x01\x95\x08\x06\x00\x00\x00\x00\x00\x00\x00I\xa5\x01\x95\x08\x07\x00\x00\x00\x00\x00\x00\x00J\xa5\x01\x95\x08\x08\x00\x00\x00\x00\x00\x00\x00K\xa5\x01\x95\x08\t\x00\x00\x00\x00\x00\x00\x00L\xa5\x01\x95\x08\n\x00\x00\x00\x00\x00\x00\x00M\xa5\x01\x95\x08\x0b\x00\x00\x00\x00\x00\x00\x00N\xa5\x01\x95\x08\x0c\x00\x00\x00\x00\x00\x00\x00O\xa5\x01\x95\x08\r\x00\x00\x00\x00\x00\x00\x00P\xa5\x01\x95\x08\x0e\x00\x00\x00\x00\x00\x00\x00Q\xa5\x01\x95\x08\x0f\x00\x00\x00\x00\x00\x00\x00R\xa5\x01\x95\x08\x10\x00\x00\x00\x00\x00\x00\x00S'
INFO:daly:is multiframe response - assuming ok for now
INFO:daly:Multi frame response with 16 frames
INFO:AbstractProtocol:decode: Processing response of type MULTIFRAME-POSITIONAL
INFO:device:run_command: Decoded response {'raw_response': ['¥\x01\x95\x08\x01\rM\r[\r8\x00K¥\x01\x95\x08\x02\rF\rV\r\\\x00d¥\x01\x95\x08\x03\ra\r1\x00\x00\x00ò¥\x01\x95\x08\x04\x00\x00\x00\x00\x00\x00\x00G¥\x01\x95\x08\x05\x00\x00\x00\x00\x00\x00\x00H¥\x01\x95\x08\x06\x00\x00\x00\x00\x00\x00\x00I¥\x01\x95\x08\x07\x00\x00\x00\x00\x00\x00\x00J¥\x01\x95\x08\x08\x00\x00\x00\x00\x00\x00\x00K¥\x01\x95\x08\t\x00\x00\x00\x00\x00\x00\x00L¥\x01\x95\x08\n\x00\x00\x00\x00\x00\x00\x00M¥\x01\x95\x08\x0b\x00\x00\x00\x00\x00\x00\x00N¥\x01\x95\x08\x0c\x00\x00\x00\x00\x00\x00\x00O¥\x01\x95\x08\r\x00\x00\x00\x00\x00\x00\x00P¥\x01\x95\x08\x0e\x00\x00\x00\x00\x00\x00\x00Q¥\x01\x95\x08\x0f\x00\x00\x00\x00\x00\x00\x00R¥\x01\x95\x08\x10\x00\x00\x00\x00\x00\x00\x00S', ''], '_command': 'cellVoltages', '_command_description': 'Cell Voltages Information', 'Cell 01 Voltage': [3.405, 'V'], 'Cell 02 Voltage': [3.419, 'V'], 'Cell 03 Voltage': [3.384, 'V'], 'Cell 04 Voltage': [3.398, 'V'], 'Cell 05 Voltage': [3.414, 'V'], 'Cell 06 Voltage': [3.42, 'V'], 'Cell 07 Voltage': [3.425, 'V'], 'Cell 08 Voltage': [3.377, 'V'], 'Cell 09 Voltage': [0.0, 'V'], 'Cell 10 Voltage': [0.0, 'V'], 'Cell 11 Voltage': [0.0, 'V'], 'Cell 12 Voltage': [0.0, 'V'], 'Cell 13 Voltage': [0.0, 'V'], 'Cell 14 Voltage': [0.0, 'V'], 'Cell 15 Voltage': [0.0, 'V'], 'Cell 16 Voltage': [0.0, 'V'], 'Cell 17 Voltage': [0.0, 'V'], 'Cell 18 Voltage': [0.0, 'V'], 'Cell 19 Voltage': [0.0, 'V'], 'Cell 20 Voltage': [0.0, 'V'], 'Cell 21 Voltage': [0.0, 'V'], 'Cell 22 Voltage': [0.0, 'V'], 'Cell 23 Voltage': [0.0, 'V'], 'Cell 24 Voltage': [0.0, 'V'], 'Cell 25 Voltage': [0.0, 'V'], 'Cell 26 Voltage': [0.0, 'V'], 'Cell 27 Voltage': [0.0, 'V'], 'Cell 28 Voltage': [0.0, 'V'], 'Cell 29 Voltage': [0.0, 'V'], 'Cell 30 Voltage': [0.0, 'V'], 'Cell 31 Voltage': [0.0, 'V'], 'Cell 32 Voltage': [0.0, 'V'], 'Cell 33 Voltage': [0.0, 'V'], 'Cell 34 Voltage': [0.0, 'V'], 'Cell 35 Voltage': [0.0, 'V'], 'Cell 36 Voltage': [0.0, 'V'], 'Cell 37 Voltage': [0.0, 'V'], 'Cell 38 Voltage': [0.0, 'V'], 'Cell 39 Voltage': [0.0, 'V'], 'Cell 40 Voltage': [0.0, 'V'], 'Cell 41 Voltage': [0.0, 'V'], 'Cell 42 Voltage': [0.0, 'V'], 'Cell 43 Voltage': [0.0, 'V'], 'Cell 44 Voltage': [0.0, 'V'], 'Cell 45 Voltage': [0.0, 'V'], 'Cell 46 Voltage': [0.0, 'V'], 'Cell 47 Voltage': [0.0, 'V'], 'Cell 48 Voltage': [0.0, 'V']}
INFO:root:attempting to create output processor: screen
INFO:screen:output: Using output processor: screen
Command: cellVoltages - Cell Voltages Information
------------------------------------------------------------
Parameter                     	Value           Unit
cell_01_voltage               	3.405          	V
cell_02_voltage               	3.419          	V
cell_03_voltage               	3.384          	V
cell_04_voltage               	3.398          	V
cell_05_voltage               	3.414          	V
cell_06_voltage               	3.42           	V
cell_07_voltage               	3.425          	V
cell_08_voltage               	3.377          	V
cell_09_voltage               	0.0            	V
cell_10_voltage               	0.0            	V
cell_11_voltage               	0.0            	V
cell_12_voltage               	0.0            	V
cell_13_voltage               	0.0            	V
cell_14_voltage               	0.0            	V
cell_15_voltage               	0.0            	V
cell_16_voltage               	0.0            	V
cell_17_voltage               	0.0            	V
cell_18_voltage               	0.0            	V
[... cut ...]
cell_38_voltage               	0.0            	V
cell_39_voltage               	0.0            	V
cell_40_voltage               	0.0            	V
cell_41_voltage               	0.0            	V
cell_42_voltage               	0.0            	V
cell_43_voltage               	0.0            	V
cell_44_voltage               	0.0            	V
cell_45_voltage               	0.0            	V
cell_46_voltage               	0.0            	V
cell_47_voltage               	0.0            	V
cell_48_voltage               	0.0            	V
```

jblance, 2021-05-03:
that looks better
i should really default to using the daly porttype for protocol daly...

firemen027, 2021-05-12:
Hi everybody,

I'm back from my cabin for a short half day. I Update de MPP-solar from jblance, change my conf file. After understanding how to add the port type in the conf file: everything work perfectly. Else I miss something, I didn't find the command for the remaining AH in the cell bank and I can live with IT. Seriously you make a very good job on that in a very short time.
I really appréciate. 

Again my best regards

Gaetan

jblance, 2021-05-13:
Glad it is working, the mosStatus command gives a remaining capacity result, but I dont know the decode yet. Ideally someone would give the output of the mosStatus command and the 'answer' from elsewhere so i can implement and check the result

closed 2021-05-21 by jblance