jblance/mpp-solar #84: error when running some inquiries

joerghohn, 2021-04-13:
Hi, 
first thank you very much for this software.
I have tis inverter: https://www.amazon.de/dp/B07YL3VVQ2/ref=pe_3044161_185740101_TE_item and running mpp-solar version: 0.7.41
Atm I only use -c QPIGS and send the output via MQTT to ioBroker, so this is nothing I need here, just wanted to let you know.

I have a problem when running command different Qxxx commands, resulting in a traceback.
This happens for:
QBOOT, QOPM, QPGS, QPIRI, 
The outputs with the first 3 commands are (replace <command> by the command without Q:
```
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')()
  File "/root/src/mpp-solar/mppsolar/__init__.py", line 387, in main
    results = _device.run_command(command=_command)
  File "/root/src/mpp-solar/mppsolar/devices/device.py", line 309, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/root/src/mpp-solar/mppsolar/protocols/abstractprotocol.py", line 263, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
ValueError: invalid literal for int() with base 10: '<command>'

```
The output for QPIRI is:
```
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')()
  File "/root/src/mpp-solar/mppsolar/__init__.py", line 387, in main
    results = _device.run_command(command=_command)
  File "/root/src/mpp-solar/mppsolar/devices/device.py", line 309, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/root/src/mpp-solar/mppsolar/protocols/abstractprotocol.py", line 254, in decode
    msgs[key] = [int(result), resp_format[2]]
ValueError: invalid literal for int() with base 10: '-'

```

Greetings from Rhein-Main-Area, Germany
Jörg

jblance, 2021-04-13:
Thanks for the comments
Can you run the commands with ```-I``` added and post the output
Also can you run the QPI command and post the result 

Cheers

joerghohn, 2021-04-13:
QPI:
```
Command: QPI - Protocol ID inquiry
------------------------------------------------------------
Parameter                       Value           Unit
protocol_id                     PI30                
```
QPIRI with -I:
```
INFO:root:Solar Device Command Utility, version: 0.7.41, recent changes: add daly (rs485) protocol
INFO:root:Creating device "FirstInverter" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
INFO:device:set_port: Using serialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: FirstInverter, port: <mppsolar.io.serialio.SerialIO object at 0xb713652c>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb71367ec> for command: QPIRI, tag: QPIRI, outputs: screen
INFO:device:run_command: Running command QPIRI
INFO:AbstractProtocol:get_full_command: Using protocol b'PI30' with 35 commands
INFO:device:run_command: full command b'QPIRI\xf8T\r' for command QPIRI
INFO:AbstractProtocol:decode: response passed to decode: b'(230.0 13.0 230.0 50.0 13.0 3000 2400 24.0 23.0 21.0 28.2 27.0 0 30 50 0 2 1 - 01 1 0 26.0 0 0\xb9\xbd\r'
INFO:AbstractProtocol:decode: Processing response of type default
INFO:AbstractProtocol:decode: Processing SEQUENCE type responses
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')()
  File "/root/src/mpp-solar/mppsolar/__init__.py", line 387, in main
    results = _device.run_command(command=_command)
  File "/root/src/mpp-solar/mppsolar/devices/device.py", line 309, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/root/src/mpp-solar/mppsolar/protocols/abstractprotocol.py", line 254, in decode
    msgs[key] = [int(result), resp_format[2]]
ValueError: invalid literal for int() with base 10: '-'
```

QOPM with -I:
```
INFO:root:Solar Device Command Utility, version: 0.7.41, recent changes: add daly (rs485) protocol
INFO:root:Creating device "FirstInverter" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
INFO:device:set_port: Using serialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: FirstInverter, port: <mppsolar.io.serialio.SerialIO object at 0xb719212c>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb70f57ec> for command: QOPM, tag: QOPM, outputs: screen
INFO:device:run_command: Running command QOPM
INFO:AbstractProtocol:get_full_command: Using protocol b'PI30' with 35 commands
INFO:device:run_command: full command b'QOPM\xa5\xc5\r' for command QOPM
INFO:AbstractProtocol:decode: response passed to decode: b'QOPM\xa5\xc5\r'
INFO:AbstractProtocol:decode: Processing response of type default
INFO:AbstractProtocol:decode: Processing SEQUENCE type responses
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')()
  File "/root/src/mpp-solar/mppsolar/__init__.py", line 387, in main
    results = _device.run_command(command=_command)
  File "/root/src/mpp-solar/mppsolar/devices/device.py", line 309, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/root/src/mpp-solar/mppsolar/protocols/abstractprotocol.py", line 263, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
ValueError: invalid literal for int() with base 10: 'OPM'
```

If you need something else, let me know.
Jörg

jblance, 2021-04-13:
QOPM isnt working for you (it also doesnt return a valid result on my system), but the validity check only looks at the crc and the result is the same as the command for some reason, so it tests as valid
QPIRI is failng as it was expecting a number and is getting a ```-```
I have made some changes that should fix this bug in version 0.7.42
The QPIRI command does give an answer to what QOPM should return - it looks like QOPM isnt supported on all models (mine included)

QPIRI result from your debug:
```
Command: QPIRI - Current Settings inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
ac_input_voltage              	230.0          	V   
ac_input_current              	13.0           	A   
ac_output_voltage             	230.0          	V   
ac_output_frequency           	50.0           	Hz  
ac_output_current             	13.0           	A   
ac_output_apparent_power      	3000           	VA  
ac_output_active_power        	2400           	W   
battery_voltage               	24.0           	V   
battery_recharge_voltage      	23.0           	V   
battery_under_voltage         	21.0           	V   
battery_bulk_charge_voltage   	28.2           	V   
battery_float_charge_voltage  	27.0           	V   
battery_type                  	AGM            	    
max_ac_charging_current       	30             	A   
max_charging_current          	50             	A   
input_voltage_range           	Appliance      	    
output_source_priority        	SBU first      	    
charger_source_priority       	Solar first    	    
max_parallel_units            	-              	units
machine_type                  	Off Grid       	    
topology                      	transformer    	    
output_mode                   	single machine output	    
battery_redischarge_voltage   	26.0           	V   
pv_ok_condition               	As long as one unit of inverters has connect PV, parallel system will consider PV OK	    
pv_power_balance              	PV input max current will be the max charged current	    

```

Thanks again for the feedback and diagnosis

joerghohn, 2021-04-14:
So I assume with QBOOT, and QPGS it might be the same as with QOPM. For me it does not matter because I do not need the commands. I will only monitor some values with ioBroker and QPIGS gives me all values I need.
Thank you very much.
Jörg

jblance, 2021-04-14:
Not sure about QBOOT, but QPGSn is only for parallel setups, so it not working for you makes sense.
I'll check QBOOT on my system to see if it actually works (I've implemented all the commands in the docs, but not run them all against a live system) 
If you paste a -I run of any commands that fail on your system I can try to improve the code so it either works or fails cleanly

joerghohn, 2021-04-14:
QPIRI is working with 0.7.42

joerghohn, 2021-04-14:
Other Commands throwing a Traceback (QBOOT, QOPM, QPGS):

QBOOT -I:
```
INFO:root:Solar Device Command Utility, version: 0.7.42, recent changes: deal with non-numerical result better 
INFO:root:Creating device "FirstInverter" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
INFO:device:set_port: Using serialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: FirstInverter, port: <mppsolar.io.serialio.SerialIO object at 0xb709452c>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb70947ec> for command: QBOOT, tag: QBOOT, outputs: screen
INFO:device:run_command: Running command QBOOT
INFO:AbstractProtocol:get_full_command: Using protocol b'PI30' with 35 commands
INFO:device:run_command: full command b'QBOOT\x0b\x88\r' for command QBOOT
INFO:AbstractProtocol:decode: response passed to decode: b'QBOOT\x0b\x88\r'
INFO:AbstractProtocol:decode: Processing response of type default
INFO:AbstractProtocol:decode: Processing SEQUENCE type responses
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')()
  File "/root/src/mpp-solar/mppsolar/__init__.py", line 387, in main
    results = _device.run_command(command=_command)
  File "/root/src/mpp-solar/mppsolar/devices/device.py", line 309, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/root/src/mpp-solar/mppsolar/protocols/abstractprotocol.py", line 267, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
ValueError: invalid literal for int() with base 10: 'BOOT'
```

QOPM -I:
```
INFO:root:Solar Device Command Utility, version: 0.7.42, recent changes: deal with non-numerical result better 
INFO:root:Creating device "FirstInverter" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
INFO:device:set_port: Using serialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: FirstInverter, port: <mppsolar.io.serialio.SerialIO object at 0xb714d54c>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb714d80c> for command: QOPM, tag: QOPM, outputs: screen
INFO:device:run_command: Running command QOPM
INFO:AbstractProtocol:get_full_command: Using protocol b'PI30' with 35 commands
INFO:device:run_command: full command b'QOPM\xa5\xc5\r' for command QOPM
INFO:AbstractProtocol:decode: response passed to decode: b'QOPM\xa5\xc5\r'
INFO:AbstractProtocol:decode: Processing response of type default
INFO:AbstractProtocol:decode: Processing SEQUENCE type responses
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')()
  File "/root/src/mpp-solar/mppsolar/__init__.py", line 387, in main
    results = _device.run_command(command=_command)
  File "/root/src/mpp-solar/mppsolar/devices/device.py", line 309, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/root/src/mpp-solar/mppsolar/protocols/abstractprotocol.py", line 267, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
ValueError: invalid literal for int() with base 10: 'OPM'
```

QPGS -I:
```
INFO:root:Solar Device Command Utility, version: 0.7.42, recent changes: deal with non-numerical result better 
INFO:root:Creating device "FirstInverter" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
INFO:device:set_port: Using serialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: FirstInverter, port: <mppsolar.io.serialio.SerialIO object at 0xb707d52c>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb707d7ec> for command: QPGS, tag: QPGS, outputs: screen
INFO:device:run_command: Running command QPGS
INFO:AbstractProtocol:get_full_command: Using protocol b'PI30' with 35 commands
INFO:device:run_command: full command b'QPGS\xa3\x8c\r' for command QPGS
INFO:AbstractProtocol:decode: response passed to decode: b'QPGS\xa3\x8c\r'
INFO:AbstractProtocol:decode: Processing response of type default
INFO:AbstractProtocol:decode: Processing SEQUENCE type responses
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')()
  File "/root/src/mpp-solar/mppsolar/__init__.py", line 387, in main
    results = _device.run_command(command=_command)
  File "/root/src/mpp-solar/mppsolar/devices/device.py", line 309, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/root/src/mpp-solar/mppsolar/protocols/abstractprotocol.py", line 267, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
ValueError: invalid literal for int() with base 10: 'PGS'
```

jblance, 2021-04-14:
It's weird how the commands that don't work get the command returned by your system. I should be able to add a test for that.
Note QPGS command needs a number ie QPGS0, ill check about adding a check for that too

joerghohn, 2021-04-14:
The Inverter is a rather cheap China Variant without even a manufacturer name on the system or in the handbook
QPGS0 shows:
```
INFO:root:Solar Device Command Utility, version: 0.7.42, recent changes: deal with non-numerical result better 
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
INFO:device:set_port: Using serialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xb709456c>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb709482c> for command: QPGS0, tag: QPGS0, outputs: screen
INFO:device:run_command: Running command QPGS0
INFO:AbstractProtocol:get_full_command: Using protocol b'PI30' with 35 commands
INFO:device:run_command: full command b'QPGS0?\xda\r' for command QPGS0
INFO:AbstractProtocol:decode: response passed to decode: b'QPGS0?\xda\r'
INFO:AbstractProtocol:decode: Processing response of type default
INFO:AbstractProtocol:decode: Processing SEQUENCE type responses
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 11, in <module>
    load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')()
  File "/root/src/mpp-solar/mppsolar/__init__.py", line 387, in main
    results = _device.run_command(command=_command)
  File "/root/src/mpp-solar/mppsolar/devices/device.py", line 309, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/root/src/mpp-solar/mppsolar/protocols/abstractprotocol.py", line 267, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
ValueError: invalid literal for int() with base 10: 'PGS0'
```

jblance, 2021-04-14:
So this command is not supported by your system either

jblance, 2021-04-14:
I have created new issues for the problems above, will close this and track changes in #88 and #89

closed 2021-04-14 by jblance