jblance/mpp-solar #98: BMS JK-B2A24S15P: Hardware Version 6.X Software Version:6.1.35 - Cell data output are not correct.

CRaccani, 2021-04-18:
JK-B2A24S15P Voltages appear to be read/decoded incorrectly.
processInfoRecord(self, record) returns correct info.
processCellDataRecord(self, record) returns incorrect info.

Please see the DEBUG output below.
Hope you can shed some light. 
I can do what ever tests you suggest to fix this.

Thank you.

Using your jkbms python code as follows:

$ jkbms -D -p
Query BMS via BLE
Reading config file: /etc/jkbms/jkbms.conf
Logging level: 10
DEBUG:JKBMS-BT:Config data - name: Power Wall 1, model: JK-B2A24S, mac: 3c:a5:51:86:01:8c, command: command, tag: Power_Wall_1, format: influx2
DEBUG:JKBMS-BT:Additional config - records: 1, maxConnectionAttempts: 3, mqttBroker: None
jkBMS Logging level: 10
DEBUG:JKBMS-BT:JKBMS instance --- name: Power Wall 1, model: JK-B2A24S, mac: 3c:a5:51:86:01:8c, command: command, tag: Power_Wall_1, format: influx2, records: 1, maxConnectionAttempts: 3, mqttBroker: None
Delegate JKBMS instance --- name: Power Wall 1, model: JK-B2A24S, mac: 3c:a5:51:86:01:8c, command: command, tag: Power_Wall_1, format: influx2, records: 1, maxConnectionAttempts: 3, mqttBroker: None
INFO:JKBMS-BT:Attempting to connect to Power Wall 1
INFO:JKBMS-BT:Connected to b'JK-B2A24S15P\x00'
INFO:JKBMS-BT:Read characteristic: Characteristic <ffe3>, handle 10
INFO:JKBMS-BT:Enable 0x0b handle
INFO:JKBMS-BT:Enable read handle
INFO:JKBMS-BT:Write getInfo to read handle
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
INFO:JKBMS-BT:Processing info record
INFO:JKBMS-BT:Record number: 18
INFO:JKBMS-BT:VendorID: JK-B2A24S15P
{'VendorID': 'JK-B2A24S15P'}:influx2:None:Power_Wall_1
INFO:JKBMS-BT:Device Name: JK-B2A24S15P
{'DeviceName': 'JK-B2A24S15P'}:influx2:None:Power_Wall_1
DEBUG:JKBMS-BT:Pass Code: 1234
INFO:JKBMS-BT:Hardware Version: 6.X
{'HardwareVersion': '6.X'}:influx2:None:Power_Wall_1
INFO:JKBMS-BT:Software Version: 6.1.3S
{'SoftwareVersion': '6.1.3S'}:influx2:None:Power_Wall_1
INFO:JKBMS-BT:Uptime: 2D23H29M59S
{'Uptime': '2D23H29M59S'}:influx2:None:Power_Wall_1
INFO:JKBMS-BT:Power Up Times: 5
{'Power Up Times: 5'}:influx2:None:Power_Wall_1
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
INFO:JKBMS-BT:notificationData has ACK
INFO:JKBMS-BT:Write getCellInfo to read handle
INFO:JKBMS-BT:Grabbing 1 records (after inital response)
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
DEBUG:JKBMS-BT:From handle: 0x0a Got 20 bytes of data
[... 497 lines cut ...]
INFO:JKBMS-BT:After position 0: 3.002498149871826
DEBUG:JKBMS-BT:hexString: bytearray(b'\xee(@@')
DEBUG:JKBMS-BT:hex(byte1): 0x40
DEBUG:JKBMS-BT:byte1Low: 0
DEBUG:JKBMS-BT:hex(byte2): 0x40
DEBUG:JKBMS-BT:byte2High: 4
DEBUG:JKBMS-BT:byte2Low: 0
DEBUG:JKBMS-BT:hex(byte3): 0x28
DEBUG:JKBMS-BT:byte3High: 2
DEBUG:JKBMS-BT:byte3Low: 8
DEBUG:JKBMS-BT:hex(byte4): 0xee
DEBUG:JKBMS-BT:byte4High: 14
DEBUG:JKBMS-BT:byte4Low: 14
DEBUG:JKBMS-BT:step1: 0.25
DEBUG:JKBMS-BT:step2: 0.015625
DEBUG:JKBMS-BT:step3: 0.0009765625
DEBUG:JKBMS-BT:step4: 6.103515625e-05
DEBUG:JKBMS-BT:step5: 3.814697265625e-06
DEBUG:JKBMS-BT:step6: 2.384185791015625e-07
DEBUG:JKBMS-BT:Hex bytearray(b'\xee(@@') decoded to 3.002498149871826
{'ResistanceCell23': 3.002498149871826}:influx2:None:Power_Wall_1
INFO:JKBMS-BT:Cell: 24, Resistance: 0.0000
{'ResistanceCell24': 0.0}:influx2:None:Power_Wall_1
Got 1 records
INFO:JKBMS-BT:Disconnecting...

jblance, 2021-04-18:
What version of the code are you using? (as this looks old)
```jkbms -v```

CRaccani, 2021-04-19:
Wow, thanks for that speedy reply.
Of course you are right, old version taken from your other place. Just noticed newer code at MPP-Solar, just installed and used protocol JK-02, 04 did not give correct cell info.

Marvellous, looks good.
Did notice that battery voltage says: -12.305 V, when this is a 14S battery?

See output below.

Thank again.

pi@piThinkPad:~ $ jkbms -P JK02 -p 3c:a5:51:86:01:8c  -c getCellData
Command: getCellData - BLE Cell Data inquiry
------------------------------------------------------------
Parameter                       Value           Unit
header                          55aaeb90
record_type                     02
record_counter                  156
voltage_cell01                  3.802           V
voltage_cell02                  3.803           V
voltage_cell03                  3.8             V
voltage_cell04                  3.802           V
voltage_cell05                  3.803           V
voltage_cell06                  3.804           V
voltage_cell07                  3.8             V
voltage_cell08                  3.804           V
voltage_cell09                  3.8             V
voltage_cell10                  3.804           V
voltage_cell11                  3.803           V
voltage_cell12                  3.803           V
voltage_cell13                  3.8             V
voltage_cell14                  3.803           V
voltage_cell15                  0.0             V
voltage_cell16                  0.0             V
voltage_cell17                  0.0             V
voltage_cell18                  0.0             V
voltage_cell19                  0.0             V
voltage_cell20                  0.0             V
voltage_cell21                  0.0             V
voltage_cell22                  0.0             V
voltage_cell23                  0.0             V
voltage_cell24                  0.0             V
average_cell_voltage            3.802           V
delta_cell_voltage              0.004           V
current_balancer                1.541
resistance_cell01               0.109           Ohm
resistance_cell02               0.11            Ohm
resistance_cell03               0.111           Ohm
resistance_cell04               0.11            Ohm
resistance_cell05               0.111           Ohm
resistance_cell06               0.109           Ohm
resistance_cell07               0.112           Ohm
resistance_cell08               0.108           Ohm
resistance_cell09               0.108           Ohm
resistance_cell10               0.108           Ohm
resistance_cell11               0.108           Ohm
resistance_cell12               0.108           Ohm
resistance_cell13               0.111           Ohm
resistance_cell14               0.108           Ohm
resistance_cell15               0.0             Ohm
resistance_cell16               0.0             Ohm
resistance_cell17               0.0             Ohm
resistance_cell18               0.0             Ohm
resistance_cell19               0.0             Ohm
resistance_cell20               0.0             Ohm
resistance_cell21               0.0             Ohm
resistance_cell22               0.0             Ohm
resistance_cell23               0.0             Ohm
resistance_cell24               0.0             Ohm
resistance_cell25               0.0             Ohm
battery_voltage                 -12.305         V
battery_t1                      26.900000000000002      °C
battery_t2                      22.1            °C
mos_temp                        28.599999999999998      °C
percent_remain                  69
capacity_remain                 8.544
nominal_capacity                12.231
capacity_cycle                  37.236000000000004
time                            3D0H46M42S
current_charge                  0.004
current_discharge               0.0
pi@piThinkPad:~ $ jkbms -P JK02 -p 3c:a5:51:86:01:8c  -c getInfo
Command: getInfo - BLE Device Information inquiry
------------------------------------------------------------
Parameter                       Value           Unit
header                          55aaeb90
record_type                     03
record_counter                  146
device_model                    JK-B2A24S1
hardware_version                5P6.X
software_version                6.1.3S
device_name                     JK-B2A24S15P
device_passcode                 1234
unknown1                        210121
unknown2                        2012282119
user_data                       Input Userdata
settings_passcode?              123456

CRaccani, 2021-04-19:
getInfo:
unknown1 = Manufacturing Date
unknown2 = Serial Number
Gleaned from the Mobile App/About box.

jblance, 2021-04-19:
Thanks for the unknowns info - have added that to the definition
re the voltage - can you send a full debug and I'll try to figure out what is wrong there

CRaccani, 2021-04-19:
Full debug file attached.
Actual battery Voltage should be: 53.19 V

Thanks.

File: [JKBMS Debug Log.txt](https://github.com/jblance/mpp-solar/files/6332764/JKBMS.Debug.Log.txt)

CRaccani, 2021-04-19:
``
["loop", 25, "Resistance_Cell", "Ohm", "2ByteHex"],
["discard", 4, "discard2", ""], ["2ByteHex", 1, "Battery_Voltage", "V"],
``

CRaccani, 2021-04-19:
Possible fix:
1. set 24 for loop of Resistance_Cell
2. Change 2ByteHex to 4ByteHex1000 for Battery_Voltage

Evidence:
Debug output: Info: `protocol_helpers:Hex bytearray(b'\xc1\xcf') 2 byte decoded to -12.351`

WORD(CFC1) = -12351
DWORD(0000CFC1) = 53185

Not far off, do you think?

jblance, 2021-04-19:
very close
the resistance loop isnt the one that needs changing, there are discards before and after (discards are bytes that i dont know what they are for or they are padding - ie:
```
["discard", 4, "discard2", ""],
["2ByteHex", 1, "Battery_Voltage", "V"],
["discard", 10, "discard3", ""],
```

If i change it to 
```
["discard", 4, "discard2", ""],
["4ByteHex1000", 1, "Battery_Voltage", "V"],
["discard", 8, "discard3", ""],
```

I get the below with your data (and it seems better for the other test data too)
```
...
esistance_cell24             	0.0            	Ohm 
resistance_cell25             	0.0            	Ohm 
battery_voltage               	53.185         	V   
battery_t1                    	24.5           	°C  
battery_t2                    	21.8           	°C  
mos_temp                      	26.1           	°C  
percent_remain                	69             	    
...

```

Fixed in v 0.7.46

CRaccani, 2021-04-19:
Very true, but in reality, there are at most 24 resistance values. So logically that should be the case.
Then you would just discard 6 before Battery_voltage. The Unknowns hopefully would present themselves at some point.
But, both work, so..

Many thanks for your excellent work.
I'm going to test the RS485 next, hoping it will be faster than the BLE.

closed 2021-04-20 by jblance

CRaccani, 2021-04-20:
Just a niggly question: Since the max number resistance values are 24 to match the max number of cells, why do you want to make it 25? The Battery_Voltage is now 4bytes instead of 2 so this would make the discard of 4 bytes after the resistance values unchanged? (I edited my previous comment, to show this, since I calculated incorrectly before)

jblance, 2021-04-21:
The extra bytes are from after the value as the bytes order is little endian. This fits with other 4 byte values and matches the test responses i have so far
Re 25 vs 24 this is what I saw on the app, so I maintained that in the code
[image]
As you see in this (very bad) image, the voltages are 01-24 while the resistances are 00-24, not sure why but that is why the code is as it is now

CRaccani, 2021-04-21:
Volts: 1..24 = 24 values
Resistance: 0..24 = 25 values
Got it!
That is super eagle eyed! Much respect.

CRaccani, 2021-04-21:
OK, so the saga continues.
Take a look at this screenshot, this is my test setup, 14S.
Now, look at the numbering!
Volts: 1..24
Resistance: 1..24
The inscrutability of the Chinese..

[image]

jblance, 2021-04-21:
Oh, maybe the jk04 and jk02 are different
I can update to 24 resistance if you prefer, it will just be another discarded value anyway 

Thoughts?

CRaccani, 2021-04-26:
I would update to 24 resistance values.
Values are discarded until they're discovered. Hope to help fill in some of these gaps.
Thanks for your input.