jblance/mpp-solar #99: Serial command execution failed

nigel914, 2021-04-19:
Recently picked up a Victron 500a Shunt and was trying to test the dev branch seeing you added support this is my output, the error may be on my part. This is my test system so on the dev branch any ideas.

`
pi@monitorpi:~ $ mpp-solar --version
Solar Device Command Utility, version: 0.7.45, recent changes: remove mpptest
`

`pi@monitorpi:~ $ python --version
Python 3.7.3
`

```
pi@monitorpi:~ $ mpp-solar -n shunt --port /dev/ttyUSB0  -b 19200 -P VED -D
INFO:root:Solar Device Command Utility, version: 0.7.45, recent changes: remove mpptest
INFO:root:Creating device "shunt" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "VED"
DEBUG:root:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:device:mppsolar __init__ args ()
DEBUG:device:mppsolar __init__ kwargs {'name': 'shunt', 'port': '/dev/ttyUSB0', 'protocol': 'VED', 'baud': 19200, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
DEBUG:device:get_port_type: port matches ttyusb
INFO:device:set_port: Using serialio for communications
DEBUG:device:set_protocol: Protocol VED
DEBUG:device:mppsolar __init__ name shunt, port <mppsolar.io.serialio.SerialIO object at 0x75eb8690>, protocol <mppsolar.protocols.ved.ved object at 0x75eb88d0>
DEBUG:root:Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x75eafdd0>, '', '', 'screen', None, None)]
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: shunt, port: <mppsolar.io.serialio.SerialIO object at 0x75eb8690>, protocol: <mppsolar.protocols.ved.ved object at 0x75eb88d0> for command: , tag: , outputs: screen
INFO:device:run_command: Running command
INFO:ved:get_full_command: Using protocol b'VED' with 2 commands
DEBUG:AbstractProtocol:get_command_defn: Processing command 'vedtext'
DEBUG:AbstractProtocol:get_command_defn: Found command vedtext in protocol b'VED'
DEBUG:ved:get_full_command: command is VEDTEXT type so returning VEDTEXT
INFO:device:run_command: full command VEDTEXT for command vedtext
DEBUG:AbstractProtocol:get_command_defn: Processing command 'vedtext'
DEBUG:AbstractProtocol:get_command_defn: Found command vedtext in protocol b'VED'
DEBUG:SerialIO:send_and_receive: port /dev/ttyUSB0, baudrate 19200
DEBUG:SerialIO:send_and_receive: Executing command via serialio...
WARNING:SerialIO:send_and_receive: Serial read error: unicode strings are not supported, please encode to bytes: 'VEDTEXT'
INFO:SerialIO:Command execution failed
DEBUG:device:run_command: Send and Receive Response {'ERROR': ['Serial command execution failed', '']}
DEBUG:root:results: {'ERROR': ['Serial command execution failed', '']}
INFO:root:attempting to create output processor: screen
DEBUG:screen:processor.screen __init__ kwargs {}
DEBUG:root:Using output filter: None
INFO:screen:output: Using output processor: screen
DEBUG:screen:output: kwargs {'data': {'ERROR': ['Serial command execution failed', '']}, 'tag': '', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
DEBUG:helpers:key_wanted: No filter and key error not excluded by excl_filter None so key wanted
error                           Serial command execution failed
DEBUG:root:Not daemon, so not looping

```

jblance, 2021-04-19:
try adding ```--porttype vserial``` to your command and see how it goes (I havent done extensive testing on this device type yet)

nigel914, 2021-04-19:
That seem to work here is the output and this is not a production system so I don't mind testing 

```

pi@monitorpi:~ $ mpp-solar -n shunt --port /dev/ttyUSB0  -b 19200 -D -P VED --porttype vserial
INFO:root:Solar Device Command Utility, version: 0.7.45, recent changes: remove mpptest
INFO:root:Creating device "shunt" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=vserial)" using protocol "VED"
DEBUG:root:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:device:mppsolar __init__ args ()
DEBUG:device:mppsolar __init__ kwargs {'name': 'shunt', 'port': '/dev/ttyUSB0', 'protocol': 'VED', 'baud': 19200, 'porttype': 'vserial', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
INFO:device:set_port: Port overide - using port 'vserial'
DEBUG:device:get_port_type: port matches vserial
INFO:device:set_port: Using vserialio for communications
DEBUG:device:set_protocol: Protocol VED
DEBUG:device:mppsolar __init__ name shunt, port <mppsolar.io.vserialio.VSerialIO object at 0x75f0adf0>, protocol <mppsolar.protocols.ved.ved object at 0x75f0af30>
DEBUG:root:Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x75f0add0>, '', '', 'screen', None, None)]
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: shunt, port: <mppsolar.io.vserialio.VSerialIO object at 0x75f0adf0>, protocol: <mppsolar.protocols.ved.ved object at 0x75f0af30> for command: , tag: , outputs: screen
INFO:device:run_command: Running command
INFO:ved:get_full_command: Using protocol b'VED' with 2 commands
DEBUG:AbstractProtocol:get_command_defn: Processing command 'vedtext'
DEBUG:AbstractProtocol:get_command_defn: Found command vedtext in protocol b'VED'
DEBUG:ved:get_full_command: command is VEDTEXT type so returning VEDTEXT
INFO:device:run_command: full command VEDTEXT for command vedtext
DEBUG:AbstractProtocol:get_command_defn: Processing command 'vedtext'
DEBUG:AbstractProtocol:get_command_defn: Found command vedtext in protocol b'VED'
DEBUG:VSerialIO:port /dev/ttyUSB0, baudrate 19200, records 30
DEBUG:VSerialIO:vserial response was: b'\x00z\r\n'
DEBUG:VSerialIO:vserial response was: b'PID\t0xA381\r\n'
DEBUG:VSerialIO:vserial response was: b'V\t52553\r\n'
DEBUG:VSerialIO:vserial response was: b'I\t25919\r\n'
DEBUG:VSerialIO:vserial response was: b'P\t1362\r\n'
DEBUG:VSerialIO:vserial response was: b'CE\t-170638\r\n'
DEBUG:VSerialIO:vserial response was: b'SOC\t668\r\n'
DEBUG:VSerialIO:vserial response was: b'TTG\t-1\r\n'
DEBUG:VSerialIO:vserial response was: b'Alarm\tOFF\r\n'
DEBUG:VSerialIO:vserial response was: b'Relay\tOFF\r\n'
DEBUG:VSerialIO:vserial response was: b'AR\t0\r\n'
DEBUG:VSerialIO:vserial response was: b'BMV\t712 Smart\r\n'
DEBUG:VSerialIO:vserial response was: b'FW\t0405\r\n'
DEBUG:VSerialIO:vserial response was: b'Checksum\t\xcf\r\n'
DEBUG:VSerialIO:vserial response was: b'H1\t-287914\r\n'
DEBUG:VSerialIO:vserial response was: b'H2\t-186703\r\n'
DEBUG:VSerialIO:vserial response was: b'H3\t-237007\r\n'
DEBUG:VSerialIO:vserial response was: b'H4\t7\r\n'
DEBUG:VSerialIO:vserial response was: b'H5\t0\r\n'
DEBUG:VSerialIO:vserial response was: b'H6\t-2294022\r\n'
DEBUG:VSerialIO:vserial response was: b'H7\t26\r\n'
DEBUG:VSerialIO:vserial response was: b'H8\t55489\r\n'
DEBUG:VSerialIO:vserial response was: b'H9\t53961\r\n'
DEBUG:VSerialIO:vserial response was: b'H10\t0\r\n'
DEBUG:VSerialIO:vserial response was: b'H11\t0\r\n'
DEBUG:VSerialIO:vserial response was: b'H12\t0\r\n'
DEBUG:VSerialIO:vserial response was: b'H15\t28\r\n'
DEBUG:VSerialIO:vserial response was: b'H16\t28\r\n'
DEBUG:VSerialIO:vserial response was: b'H17\t11975\r\n'
DEBUG:VSerialIO:vserial response was: b'H18\t13118\r\n'
DEBUG:device:run_command: Send and Receive Response b'\x00z\r\nPID\t0xA381\r\nV\t52553\r\nI\t25919\r\nP\t1362\r\nCE\t-170638\r\nSOC\t668\r\nTTG\t-1\r\nAlarm\tOFF\r\nRelay\tOFF\r\nAR\t0\r\nBMV\t712 Smart\r\nFW\t0405\r\nChecksum\t\xcf\r\nH1\t-287914\r\nH2\t-186703\r\nH3\t-237007\r\nH4\t7\r\nH5\t0\r\nH6\t-2294022\r\nH7\t26\r\nH8\t55489\r\nH9\t53961\r\nH10\t0\r\nH11\t0\r\nH12\t0\r\nH15\t28\r\nH16\t28\r\nH17\t11975\r\nH18\t13118\r\n'
INFO:AbstractProtocol:decode: response passed to decode: b'\x00z\r\nPID\t0xA381\r\nV\t52553\r\nI\t25919\r\nP\t1362\r\nCE\t-170638\r\nSOC\t668\r\nTTG\t-1\r\nAlarm\tOFF\r\nRelay\tOFF\r\nAR\t0\r\nBMV\t712 Smart\r\nFW\t0405\r\nChecksum\t\xcf\r\nH1\t-287914\r\nH2\t-186703\r\nH3\t-237007\r\nH4\t7\r\nH5\t0\r\nH6\t-2294022\r\nH7\t26\r\nH8\t55489\r\nH9\t53961\r\nH10\t0\r\nH11\t0\r\nH12\t0\r\nH15\t28\r\nH16\t28\r\nH17\t11975\r\nH18\t13118\r\n'
DEBUG:AbstractProtocol:get_command_defn: Processing command 'vedtext'
DEBUG:AbstractProtocol:get_command_defn: Found command vedtext in protocol b'VED'
DEBUG:AbstractProtocol:decode: trimmed and split responses: [[b'\x00z'], [b'PID', b'0xA381'], [b'V', b'52553'], [b'I', b'25919'], [b'P', b'1362'], [b'CE', b'-170638'], [b'SOC', b'668'], [b'TTG', b'-1'], [b'Alarm', b'OFF'], [b'Relay', b'OFF'], [b'AR', b'0'], [b'BMV', b'712 Smart'], [b'FW', b'0405'], [b'Checksum', b'\xcf'], [b'H1', b'-287914'], [b'H2', b'-186703'], [b'H3', b'-237007'], [b'H4', b'7'], [b'H5', b'0'], [b'H6', b'-2294022'], [b'H7', b'26'], [b'H8', b'55489'], [b'H9', b'53961'], [b'H10', b'0'], [b'H11', b'0'], [b'H12', b'0'], [b'H15', b'28'], [b'H16', b'28'], [b'H17', b'11975'], [b'H18', b'13118'], [b'']]
INFO:AbstractProtocol:decode: Processing response of type KEYED
DEBUG:AbstractProtocol:decode: Processing KEYED type responses
[... cut ...]
DEBUG:helpers:key_wanted: No filter and key maximum_auxiliary_(battery)_voltage not excluded by excl_filter None so key wanted
maximum_auxiliary_(battery)_voltage     0.028           V
DEBUG:helpers:key_wanted: No filter and key amount_of_discharged_energy not excluded by excl_filter None so key wanted
amount_of_discharged_energy     119.75          kWh
DEBUG:helpers:key_wanted: No filter and key amount_of_charged_energy not excluded by excl_filter None so key wanted
amount_of_charged_energy        131.18          kWh
DEBUG:root:Not daemon, so not looping
```

jblance, 2021-04-19:
Do any values look wrong?

jblance, 2021-04-19:
great news

closed 2021-04-19 by jblance