jblance/mpp-solar #112: JK BMS RS485 protocol

CRaccani, 2021-05-08:
Please find attached a different RS485 protocol document, sent to me by IC GOGOGO Store for the BMS I bought from them.
I have also just received their RS485 Dongle.
The Dongle plugs into the BMS but the dongle ends in 3 bare wires at the other end. I presume these are RS485- GND, A+, B+.
I have connected these to an RS485-USB dongle.
Any experience with this setup.

[image]

I tried your RS485 protocol but did not receive any bytes back. 
So looks like this BMS may use a different protocol.

Any chance of adding this protocol:
[RS485-english.pdf](https://github.com/jblance/mpp-solar/files/6446495/RS485-english.pdf)

jblance, 2021-05-09:
I've had a go at adding the first attempt at this protocol (called it jk232), likely still got work to do
can you update to version 0.7.54 or later and try with
`jkbms -p /dev/ttyUSB0 -b 9600 -P JK232 -I`
(change the port to whereever the adapter shows up), can you post the results

CRaccani, 2021-05-09:
Thank you. 
Here is the output requested:

$ jkbms -p /dev/ttyUSB0 -b 9600 -P JK232 -D
INFO:root:Solar Device Command Utility, version: 0.7.54, recent changes: add jk rs232/485 protocol
INFO:root:Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JK232"
DEBUG:root:device_class <class 'mppsolar.devices.jkbms.jkbms'>
DEBUG:device:jkbms __init__ args ()
DEBUG:device:jkbms __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JK232', 'baud': 9600, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
DEBUG:device:get_port_type: port matches ttyusb
INFO:device:set_port: Using serialio for communications
DEBUG:device:set_protocol: Protocol JK232
DEBUG:device:jkbms __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb6d609cc>, protocol <mppsolar.protocols.jk232.jk232 object at 0xb6d60a8c>
DEBUG:root:Commands [(<mppsolar.devices.jkbms.jkbms object at 0xb6d60a0c>, '', '', 'screen', None, None)]
INFO:root:Looping 1 commands
INFO:root:Getting results from device: jkbms device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xb6d609cc>, protocol: <mppsolar.protocols.jk232.jk232 object at 0xb6d60a8c> for command: , tag: , outputs: screen
INFO:device:run_command: Running command
INFO:jk232:get_full_command: Using protocol b'JK232' with 1 commands
DEBUG:AbstractProtocol:get_command_defn: Processing command 'getBalancerData'
DEBUG:AbstractProtocol:get_command_defn: Found command getBalancerData in protocol b'JK232'
DEBUG:jk232:get_full_command: cmd with start bit: bytearray(b'\xdd\x00\x00\x00\x00\x00\x00')
DEBUG:jk232:get_full_command: cmd with crc: bytearray(b'\xdd\xa5\x03\x00\xff\xfdw')
INFO:device:run_command: full command bytearray(b'\xdd\xa5\x03\x00\xff\xfdw') for command getBalancerData
DEBUG:AbstractProtocol:get_command_defn: Processing command 'getBalancerData'
DEBUG:AbstractProtocol:get_command_defn: Found command getBalancerData in protocol b'JK232'
DEBUG:SerialIO:send_and_receive: port /dev/ttyUSB0, baudrate 9600
DEBUG:SerialIO:send_and_receive: Executing command via serialio...
DEBUG:SerialIO:send_and_receive: serial response was: b''
DEBUG:device:run_command: Send and Receive Response b''
INFO:AbstractProtocol:decode: response passed to decode: b''
DEBUG:AbstractProtocol:get_command_defn: Processing command 'getBalancerData'
DEBUG:AbstractProtocol:get_command_defn: Found command getBalancerData in protocol b'JK232'
DEBUG:jk232:get_responses: responses [b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'']
DEBUG:AbstractProtocol:decode: trimmed and split responses: [b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'']
INFO:AbstractProtocol:decode: Processing response of type POSITIONAL
DEBUG:AbstractProtocol:decode: Processing POSITIONAL type responses
DEBUG:AbstractProtocol:Processing data_type: Hex2Str for data_name: Start Byte, raw_value b''
DEBUG:AbstractProtocol:Processing format string Hex2Str(raw_value)
DEBUG:protocol_helpers:Hex b'' decoded to
DEBUG:AbstractProtocol:decode: Processing POSITIONAL type responses
DEBUG:AbstractProtocol:Processing data_type: Hex2Str for data_name: Command Code, raw_value b''
DEBUG:AbstractProtocol:Processing format string Hex2Str(raw_value)
DEBUG:protocol_helpers:Hex b'' decoded to
DEBUG:AbstractProtocol:decode: Processing POSITIONAL type responses
DEBUG:AbstractProtocol:Processing data_type: Hex2Str for data_name: Status, raw_value b''
DEBUG:AbstractProtocol:Processing format string Hex2Str(raw_value)
DEBUG:protocol_helpers:Hex b'' decoded to
DEBUG:AbstractProtocol:decode: Processing POSITIONAL type responses
DEBUG:AbstractProtocol:Processing data_type: Hex2Int for data_name: Data Length, raw_value b''
DEBUG:AbstractProtocol:Processing format string Hex2Int(raw_value)
Traceback (most recent call last):
  File "/usr/local/bin/jkbms", line 33, in <module>
    sys.exit(load_entry_point('mpp-solar', 'console_scripts', 'jkbms')())
  File "/home/pi/src/mpp-solar/mppsolar/__init__.py", line 388, in main
    results = _device.run_command(command=_command)
  File "/home/pi/src/mpp-solar/mppsolar/devices/device.py", line 330, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/home/pi/src/mpp-solar/mppsolar/protocols/abstractprotocol.py", line 350, in decode
    frame_number=frame_number,
  File "/home/pi/src/mpp-solar/mppsolar/protocols/abstractprotocol.py", line 122, in process_response
    r = eval(format_string)
  File "<string>", line 1, in <module>
  File "/home/pi/src/mpp-solar/mppsolar/protocols/protocol_helpers.py", line 70, in Hex2Int
    answer = hexString[0]
IndexError: index out of range

jblance, 2021-05-09:
So still some bugs in the code - but more importantly there was no response from the device
Is `ttyUSB0` the correct device?
Do you have any indication of communications happening?
Does `dmesg` give any insight?
What does `lsusb` report?

CRaccani, 2021-05-10:
Am now using my Victron RS485 to USB cable. Higher quality and I know it works.
It has flashing LEDS when activity.
There is a heartbeat denoted by the GREEN LED which flashes on a regular basis- so heart beat from BMS?
When I run your program the GREEN LED blinks once.

OUTPUT:dmesg
:    :   : 
:    :   :  
[614891.532083] usb 1-1.4: new full-speed USB device number 67 using ehci-pci
[614891.650899] usb 1-1.4: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[614891.650902] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[614891.650904] usb 1-1.4: Product: FT232R USB UART
[614891.650906] usb 1-1.4: Manufacturer: FTDI
[614891.650908] usb 1-1.4: SerialNumber: AK08MAON
[614891.654312] ftdi_sio 1-1.4:1.0: FTDI USB Serial Device converter detected
[614891.654347] usb 1-1.4: Detected FT232RL
[614891.654772] usb 1-1.4: FTDI USB Serial Device converter now attached to ttyUSB0

OUTPUT: lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 004: ID 17ef:480c Lenovo Integrated Webcam
Bus 001 Device 067: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 066: ID 0409:005a NEC Corp. HighSpeed Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 003: ID 0a5c:2145 Broadcom Corp. BCM2045B (BDC-2.1) [Bluetooth Controller]
Bus 004 Device 002: ID 08ff:2810 AuthenTec, Inc. AES2810
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

OUTPUT protocol: 232 using Victron dongle:

 jkbms -p /dev/ttyUSB0 -b 9600 -P JK232 -D

INFO:root:Solar Device Command Utility, version: 0.7.54, recent changes: add jk rs232/485 protocol
INFO:root:Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JK232"
DEBUG:root:device_class <class 'mppsolar.devices.jkbms.jkbms'>
DEBUG:device:jkbms __init__ args ()
DEBUG:device:jkbms __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JK232', 'baud': 9600, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
DEBUG:device:get_port_type: port matches ttyusb
INFO:device:set_port: Using serialio for communications
DEBUG:device:set_protocol: Protocol JK232
DEBUG:device:jkbms __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb6e39f6c>, protocol <mppsolar.protocols.jk232.jk232 object at 0xb6e4196c>
DEBUG:root:Commands [(<mppsolar.devices.jkbms.jkbms object at 0xb6e39fcc>, '', '', 'screen', None, None)]
INFO:root:Looping 1 commands
INFO:root:Getting results from device: jkbms device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xb6e39f6c>, protocol: <mppsolar.protocols.jk232.jk232 object at 0xb6e4196c> for command: , tag: , outputs: screen
INFO:device:run_command: Running command
INFO:jk232:get_full_command: Using protocol b'JK232' with 1 commands
DEBUG:AbstractProtocol:get_command_defn: Processing command 'getBalancerData'
DEBUG:AbstractProtocol:get_command_defn: Found command getBalancerData in protocol b'JK232'
DEBUG:jk232:get_full_command: cmd with start bit: bytearray(b'\xdd\x00\x00\x00\x00\x00\x00')
DEBUG:jk232:get_full_command: cmd with crc: bytearray(b'\xdd\xa5\x03\x00\xff\xfdw')
INFO:device:run_command: full command bytearray(b'\xdd\xa5\x03\x00\xff\xfdw') for command getBalancerData
DEBUG:AbstractProtocol:get_command_defn: Processing command 'getBalancerData'
DEBUG:AbstractProtocol:get_command_defn: Found command getBalancerData in protocol b'JK232'
DEBUG:SerialIO:send_and_receive: port /dev/ttyUSB0, baudrate 9600
DEBUG:SerialIO:send_and_receive: Executing command via serialio...
DEBUG:SerialIO:send_and_receive: serial response was: b'\x00'
DEBUG:device:run_command: Send and Receive Response b'\x00'
INFO:AbstractProtocol:decode: response passed to decode: b'\x00'
DEBUG:AbstractProtocol:get_command_defn: Processing command 'getBalancerData'
DEBUG:AbstractProtocol:get_command_defn: Found command getBalancerData in protocol b'JK232'
DEBUG:jk232:get_responses: responses [b'\x00', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'']
DEBUG:AbstractProtocol:decode: trimmed and split responses: [b'\x00', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'']
INFO:AbstractProtocol:decode: Processing response of type POSITIONAL
DEBUG:AbstractProtocol:decode: Processing POSITIONAL type responses
DEBUG:AbstractProtocol:Processing data_type: Hex2Str for data_name: Start Byte, raw_value b'\x00'
[... 57 lines cut ...]
DEBUG:AbstractProtocol:get_command_defn: Found command getBalancerData in protocol b'JK485'
DEBUG:SerialIO:send_and_receive: port /dev/ttyUSB0, baudrate 9600
DEBUG:SerialIO:send_and_receive: Executing command via serialio...
DEBUG:SerialIO:send_and_receive: serial response was: b'\x00'
DEBUG:device:run_command: Send and Receive Response b'\x00'
INFO:jkAbstractProtocol:decode: response passed to decode: b'\x00'
DEBUG:jkAbstractProtocol:get_command_defn: get_command_defn for: getBalancerData
DEBUG:AbstractProtocol:get_command_defn: Processing command 'getBalancerData'
[... cut ...]
  File "/home/pi/src/mpp-solar/mppsolar/src/mpp-solar/mppsolar/__init__.py", line 388, in main
    results = _device.run_command(command=_command)
  File "/home/pi/src/mpp-solar/mppsolar/src/mpp-solar/mppsolar/devices/device.py", line 330, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/home/pi/src/mpp-solar/mppsolar/src/mpp-solar/mppsolar/protocols/jkabstractprotocol.py", line 198, in decode
    value += f"{responses.pop(0):02x}"
IndexError: pop from empty bytearray

I'm none the wiser.
Hope this helps.

jblance, 2021-05-10:
Well that output is slightly better (we are getting a response `\x00` from both) 
So that means the command being sent is likely wrong

jblance, 2021-05-10:
can you try running the below
``` python
import serial
import time

## Vars ################
port = "/dev/ttyUSB0"
baud = 9600
########################

# DD A5 03 00 FF FD 77
cmds = [b'\xdd\xa5\x03\x00\xff\xfdw', b'\xdd\xa5\x03\x00\xff\xfdw\n', b'\xdd\xa5\x03\x00\xff\xfdw', b'\xdd \xa5 \x03 \x00 \xff \xfd w', b'DD A5 03 00 FF FD 77', , b'DDA50300FFFD77']

with serial.serial_for_url(port, baud) as s:
    s.timeout = 1
    s.write_timeout = 1
    s.flushInput()
    s.flushOutput()
    for cmd in cmds:
        print (f"sending command {cmd}")
        bytes_written = s.write(cmd)
        print (f"wrote {bytes_written} bytes")

        for _ in range(5):
            response_line = s.readline()
            time.sleep(1)
            print (f"Got response: {response_line}")
```

jblance, 2021-05-10:
> Corrected syntax error for future testing.
>

Opps thanks - will edit the comment

> Could this be a timing issue?
> 

It could be - is anything else using the port? 
Does any application talk to the BMS?

> Also would it make any difference to use Serial.RS485?
> I'm not a Python programmer by the way.

It shouldnt - it looks to me like we are communicating ok, but not the correct commands (but I cant be sure)

CRaccani, 2021-05-10:
Could the Bluetooth of the BMS be messing with it? 
Does this need to be switched to RS485 mode? Power BMS off then on again, with USB dongle connected?

There was a flurry of green LED right before test script ended, but not as each command was sent.

jblance, 2021-05-10:
> Could the Bluetooth of the BMS be messing with it?
> Does this need to be switched to RS485 mode? Power BMS off then on again, with USB dongle connected?
> 

It is possible - I have never used a JKBMS with RS485 before - is there any options in the BLE app?

I know there is also a CAN bus as well, is that connected as well?
Are you certain the physical connections are correct?

> There was a flurry of green LED right before test script ended, but not as each command was sent.

Was the script still trying to read a response?

CRaccani, 2021-05-14:
Got response from seller:
I sent them this screenshot of Version information from Android App:
[image]

They sent back yet another protocol document in CHINESE.

[BMS Protocol V2.5 CHINESE.pdf](https://github.com/jblance/mpp-solar/files/6478719/BMS.Protocol.V2.5.CHINESE.pdf)

I then put this through GOOGLE Translate:

[BMS Protocol V2.5 ENGLISH (GOOGLE Translate).pdf](https://github.com/jblance/mpp-solar/files/6478725/BMS.Protocol.V2.5.ENGLISH.GOOGLE.Translate.pdf)

This protocol seems radically different to any other I have seen so far.

I gladly hand it over to you to take a bite at it.
Let us know what you think.
Kind regards.

CRaccani, 2021-05-14:
Can you check this for me. Does this look right to you?
The doc also implies a baudrate of: 115200.

<html>
<body>
<p class="MsoNormal" style="margin: 0cm 0cm 10pt; line-height: 16.8667px; font-size: 11pt; font-family: Calibri, sans-serif; color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Example: Activate BMS</p>

Serial number | Frame unit | length | Value
-- | -- | -- | --
1 | STX | 2 | 0x4E57
2 | LENGTH | 2 | 0x0013
3 | BMS ID | 4 | 0x00000000
4 | Command | 1 | 0x01
5 | Frame source | 1 | 0x00
6 | Transmission Type | 1 | 0x00
7 | Frame Information unit | 1 | 0x00
8 | Record number | 4 | 0x00000000
9 | END Flag | 1 | 0x68
10 | Checksum | 4 | 0x00004ED3

<p class="MsoNormal" style="margin: 0cm 0cm 10pt; line-height: 16.8667px; font-size: 11pt; font-family: Calibri, sans-serif; color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><o:p> </o:p></p>
</body>
</html>

<html>
<body>
<p class="MsoNormal" style="margin: 0cm 0cm 10pt; line-height: 16.8667px; font-size: 11pt; font-family: Calibri, sans-serif; color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Example: Read All Data<o:p></o:p></p>

Serial number | Frame unit | length | Value
-- | -- | -- | --
1 | STX | 2 | 0x4E57
2 | LENGTH | 2 | 0x0013
3 | BMS ID | 4 | 0x00000000
4 | Command | 1 | 0x06
5 | Frame source | 1 | 0x00
6 | Transmission Type | 1 | 0x00
7 | Frame Information unit | 1 | 0x00
8 | Record number | 4 | 0x00000000
9 | END Flag | 1 | 0x68
10 | Checksum | 4 | 0x00004ED8

<p class="MsoNormal" style="margin: 0cm 0cm 10pt; line-height: 16.8667px; font-size: 11pt; font-family: Calibri, sans-serif; color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><o:p> </o:p></p>
</body>
</html>

<html>
<body>
<p class="MsoNormal" style="margin: 0cm 0cm 10pt; line-height: 16.8667px; font-size: 11pt; font-family: Calibri, sans-serif; color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Example: Read Total Battery Voltage<o:p></o:p></p>

Serial number | Frame unit | length | Value
-- | -- | -- | --
1 | STX | 2 | 0x4E57
2 | LENGTH | 2 | 0x0013
3 | BMS ID | 4 | 0x00000000
4 | Command | 1 | 0x03
5 | Frame source | 1 | 0x00
6 | Transmission Type | 1 | 0x00
7 | Frame Information unit | 1 | 0x83
8 | Record number | 4 | 0x00000000
9 | END Flag | 1 | 0x68
10 | Checksum | 4 | 0x00004F58

<p class="MsoNormal" style="margin: 0cm 0cm 10pt; line-height: 16.8667px; font-size: 11pt; font-family: Calibri, sans-serif; color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><o:p> </o:p></p>
</body>
</html>

CRaccani, 2021-05-16:
I have tried this command to activate the bms, but no joy:

Send: 4E57 0013 00000000 01 00 00 00 00000000 68 00000121

and this one to get ALL data, still no joy:

Send: 4E57 0013 00000000 06 00 00 00 00000000 68 00000129

Contacting seller again. 
Fingers crossed.

jblance, 2021-05-16:
How are you sending it? Can you post the code you used

I should get a chance to look at the doc today.

CRaccani, 2021-05-17:
I've used both a program called Windmill comDebug and the following python script from your test script.
Do you have this BMS and are you able to test with RS485.
I'm no expert at all this so another tester would be good.
Also, may buy a different USB to RS485 dongle. Not sure if this one does Master/Slave automatically.

Here is the code:

```
import serial
import time

# Checksum 4bytes, bytes 1-2 0000 not used, bytes 3-4 cumulative total 
def crc(byteData):
    CRC = 0
    for b in byteData:
        print(("{0:02x}".format(b)))
        CRC += b
    crc_byte4 = CRC & 0xFF
    crc_byte3 = (CRC >> 8) & 0xFF
    print(("{0:02x}".format(crc_byte3)))
    print(("{0:02x}".format(crc_byte4)))
    return [crc_byte3, crc_byte4]

# 4.2.4 COMMAND codes
command_ACTIVATE = b'\x01'
command_WRITE = b'\x02'
command_READ = b'\x03'
command_SEND_PASSWORD = b'\x05'
command_READ_ALL_DATA = b'\x06'

# FRAME SOURCE 0 BMS Data box 1 Bluetooth 2 GPS 3 PC Host Computer
source_BMS_DATA_BOX = b'\x00'
source_BLUETOOTH = b'\x01'
source_HOST_PC = b'\x03'

# Transmission type
tx_type_READ_DATA = b'\x00'
tx_type_REPLY_FRAME = b'\x01'
tx_type_WRITE_DATA = b'\x02'

# 4.2 Frame Format
frame_STX = b'\x4E\x57'
frame_LENGTH = bytearray(2)
frame_BMS_ID = bytearray(4)
frame_Command = bytearray(1)
frame_SOURCE = bytearray(1)
frame_TX_Type = bytearray(1)
frame_INFO_READ = bytearray(1)
frame_REC_NUM = bytearray(4)
frame_END_FLAG = b'\x68'
frame_CRC_HIGH = b'\x00\x00'
frame_CRC_LOW = bytearray(2)

# Initialise read request
# For READ req. LENGTH = 19, 0x0013
frame_LENGTH = b'\x00\x13'
frame_BMS_ID = b'\x00\x00\x00\x00'

frame_Command = command_READ_ALL_DATA
#frame_Command = command_ACTIVATE

#frame_SOURCE = source_BMS_DATA_BOX
frame_SOURCE = source_HOST_PC

frame_TX_Type = tx_type_READ_DATA

frame_INFO_READ = b'\x00'

frame_REC_NUM = b'\x00\x00\x00\x00'

# Make request frame
request_FRAME = bytearray()
request_FRAME[0:2] = frame_STX
request_FRAME[2:2] = frame_LENGTH
request_FRAME[4:4] = frame_BMS_ID
request_FRAME[8:1] = frame_Command
request_FRAME[9:1] = frame_SOURCE
request_FRAME[10:1] = frame_TX_Type
request_FRAME[11:1] = frame_INFO_READ
request_FRAME[12:4] = frame_REC_NUM
request_FRAME[16:1] = frame_END_FLAG

crc_byte3, crc_byte4 = crc(request_FRAME[0:17])
frame_CRC_LOW[0] = crc_byte3
frame_CRC_LOW[1] = crc_byte4

request_FRAME[17:2] = frame_CRC_HIGH
request_FRAME[19:2] = frame_CRC_LOW

port = "COM9"
baud = 115200
########################

with serial.serial_for_url(port, baud) as s:
    s.timeout = 1
    s.write_timeout = 1
    s.flushInput()
    s.flushOutput()
    print (f"sending command: {request_FRAME.hex()}")
    bytes_written = s.write(request_FRAME)
    print (f"wrote {bytes_written} bytes")
    for _ in range(5):
        response_line = s.readline()
        time.sleep(1)
        print (f"Got response: {response_line}")
```

jblance, 2021-05-18:
What response do you get? Nothing or \x00 ?

It looks like you are running this on windows? Do you have a linux type system to test on?

CRaccani, 2021-05-18:
Response is always 0x00
I have both, Win 10 (COM9)  and Linux Debian 10 machines (/dev/ttyusb0).
Both respond the same: 0x00.

tried:
BMS_ID: 0x00000000, 0x00000001 presume this allows multiple BMS. May start from 1 or 0
REC_NUM: 0x00000000, 0x00000001 don't know how this is used. 

Tried: 
ACTIVATE first, then READ_ALL
Just READ_ALL
```
Seller says this should work:
4E 57 00 13 00 00 00 00 06 03 00 00 00 00 00 00 68 00 00 01 29

My comments:
1. STX: 4E 57
2. LENGTH: 19 bytes
3. BMS_ID: 00 00 00 00
4. Command: 06, READ_ALL
5. FRAME_SOURCE: 03 PC Host computer
6. TX_TYPE:  00 Read request
7. FRAME INFO: 00 set 00 when command : 06
8. REC_NUM: 00 00 00 00
9. END flag: 68
10. Checksum: 00 00 01 29
```
but again returns 0x00

CRaccani, 2021-05-27:
Progress,

READ ALL:
4E 57 00 13 00 00 00 00 06 03 00 00 00 00 00 00 68 00 00 01 29
RESPONSE:
4E 57 01 1B 00 00 00 00 06 00 01 
79 2A 
 01 0E ED 
 02 0E FA 
 03 0E F7 
 04 0E EC 
 05 0E F8 
 06 0E FA 
 07 0E F1 
 08 0E F8 
 09 0E E3 
 0A 0E FA 
 0B 0E F1 
 0C 0E FB 
 0D 0E FB 
 0E 0E F2 
80 00 1D 
81 00 1E 
82 00 1C 
83 14 EF 
84 80 D0 
85 0F 
86 02 
87 00 04 
89 00 00 00 00 
8A 00 0E 
8B 00 00 
8C 00 07 
8E 16 26 
8F 10 AE 
90 0F D2 
91 0F A0 
92 00 05 
93 0B EA 
94 0C 1C 
95 00 05 
96 01 2C 
97 00 07 
98 00 03 
99 00 05 
9A 00 05 
9B 0C E4 
9C 00 08 
9D 01 
9E 00 5A 
9F 00 46 
A0 00 64 
A1 00 64 
A2 00 14 
A3 00 46 
A4 00 46 
A5 FF EC 
A6 FF F6 
A7 FF EC 
A8 FF F6 
A9 0E 
AA 00 00 00 0E 
AB 01 
AC 01 
AD 04 11 
AE 01 
AF 01 
B0 00 0A 
B1 14 
B2 31 32 33 34 35 36 00 00 00 00 
B3 00 
B4 49 6E 70 75 74 20 55 73 
B5 32 31 30 31 
B6 00 00 E2 00 
B7 48 36 2E 58 5F 5F 53 36 2E 31 2E 33 53 5F 5F 
B8 00 
B9 00 00 00 00 
BA 42 54 33 30 37 32 30 32 30 31 32 30 30 30 30 32 30 30 35 32 31 30 30 31 
C0 01 
00 00 00 00 68 00 00 54 D1

jblance, 2021-05-27:
awesome news, what changed?
did you change the communications code ?

CRaccani, 2021-05-28:
Not using their TTL-RS485 dongle. Seems to be a faulty one.
Instead, connected a TTL-USB dongle directly to RX, TX, GND of BMS TTL-RS485 socket. 
Protocol has a transmission source of Bluetooth, could you do a quick python script to send this protocols READ ALL command, via Bluetooth?

CRaccani, 2021-05-28:
Machine readable list of Codes extracted from protocol document:
```
5.1 BMS Read/Write data identification code

RW
Code
Name
Bytes
Type
Info

R
0x79
Individual Cell voltage
3*n
HEX
The first byte is the cell number, the next two bytes is the voltage value MV. 
When reading all the data at the same time, 0x79 is followed by one byte length data, n as shown above, 
and then a group of three bytes represents the electricity Cell voltage.

R 
0x80
Read power tube temperature
2
HEX
0--140 (-40 to 100) The part exceeding 100 is negative temperature, such as 101 is negative 1 degree (100 Benchmark)

R 
0x81
Read the temperature in the battery box
2
HEX
0--140 (-40 to 100) The part exceeding 100 is negative temperature, the same as above (100 reference)

R 
0x82
Read battery temperature
2
HEX
0--140 (-40 to 100) The part exceeding 100 is negative temperature, the same as above (100 reference)

R 
0X83
Total battery voltage
2
HEX
0.01V 3500*0.01=35.00v minimum unit 10MV

R 
0X84
Current data
2
Hex
10000 (10000-11000)*0.01=-10.00a (discharge)
(10000-9500)*0.01=5.00a (charging) Accuracy 10MA unit: 0.01A
Note: C0:0x01 redefine 0x84 current data, the unit is 10MA, the highest bit is 0
Means discharging, 1 means charging
If discharging 20A, the data transmitted will be 2000 (0x07D0)
If charging 20A, the transmission data is 34768 (0x87D0)

R 
0X85
Battery remaining capacity
1
HEX
SOC, 0-100%, 

R 
0X86 
Number of battery temperature sensors
[... 430 lines cut ...]
0xBE
Battery low voltage turns off GPS
2
Hex 
unit: mv ( turn off the power to GPS when low voltage is detected )

W 
0xBF
Battery low voltage recovery GPS
2
Hex 
unit: mv ( turn on the power to the GPS when the recovery voltage value is detected )

R 
0xC0
Protocol version number
1
Hex
Default value: 0x00
0x01: Redefine 0x84 current data, the unit is 10MA, and the highest bit is 0 for discharging.
1 means charging
If discharging 20A, the data transmitted will be 2000 (0x07D0)
If charging 20A, the transmission data is 34768 (0x87D0)

```

Scotty-89, 2021-06-14:
Hello,

i see this issue here and have some important informations for you ;) 

i have an 

and i got the Rs485 working but this was a lot of trouble...

first the Documentation you have is an very old one ... i got an "newer" version after some discussion with the distributer ;)  they send me the protocol in Chinese but i but it into an translate ....  (i cant upload it here :(  i will try it later on again )
for Example they Change the Baudrate up to 115200 instead off 9600 ;)  and even more ... 

second the USB to RS485 dongle you use, you can but this into the trash on my side this wans working fine it received alswasy "random" byte sometime ...
my actual test bench you can see here: https://forum.drbacke.de/viewtopic.php?f=8&t=982 

but now the Problem with the RS485...

I am able to read out the available values form the BMS but you cant write something and the response from the distributer was the RS485 is only for read not for write ... 
and also it is not possible to read out the Current "Balance current" .... that sad :( 

so with this i would say the Rs485 is not an alternative to the Bluetooth app ... and there fore iam currently souring for the Bluetooth protocol ... did someone have this ?  or how the impl here was done ??? 
maybe some one can help me here ?

jblance, 2021-06-14:
I have not seen a bluetooth protocol document - the current implmentation was from reverse engineering the comms between app and unit.
The BLE code is pretty rough at the moment, someone on https://diysolarforum.com/ has done a heap of investigation of the set commands for the bluetooth connection, but i havent had a chance to look at them in any depth

Scotty-89, 2021-06-15:
Hey @jblance  
where i can find the BLE Communication which is impl here ? maybe we will find something out ..

jblance, 2021-06-15:
The jkbms command with a port that is a MAC address will default to ble comes. There's some details in the usage doc

jblance, 2021-06-15:
There is also an issue referencing jk02 which has a bunch of info that needs to be incorporated that is for the BLE comms

Scotty-89, 2021-06-15:
@jblance 

Sorry i am not so familiar with Python ;) where can i find the Code here ? 
where is the Req and resp def ?  or the BLE Connectino etc ...

thx ...

jblance, 2021-06-15:
Oh, well that's a bit more complicated 
The BLE commands are in jkabbstractprotocol and jk02 jk04 in protocols, there's a jkbleio in io that has the comms stuff.
BLE in python (that I've found) is non trivial

Scotty-89, 2021-06-23:
Hey sorry capturing the thread by maybe someone can help me ;)

i have the same HW and SW Version as on the top written...
over RS485 i am able to read out the values which are available but it is frustrating because the value for the Balance Current isn't there...

so my idea is to use the BLE now but i do not like the PI and i'am not the expert with this ... i love my ESP8226 or EPS32 ;)
so my Question is has some one dried to read something over BLE on this HW and SW version ? 
i tried something with the Handy app "nrf connect" but do not get anny respone if i send : aa 55 90 eb 97 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11

jblance, 2021-07-08:
the bluetooth is pretty hacky (it is more of a replay of what the app does, than a sensible approach)
the process used here follows the below steps (as you can see multiple handles are used)
```
"Enable 0x0b handle" self._device.writeCharacteristic(0x0B, b"\x01\x00")
"Enable read handle", self._device.writeCharacteristic(handleRead, b"\x01\x00")
"Write getInfo to read handle", self._device.writeCharacteristic(handleRead, b"\xaa\x55\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11")
```
Before sending the command
```
 "Write command to read handle", self._device.writeCharacteristic(handleRead, command)
```
The is likely a better of doing this, but I havent had a chance to do the trial and error approach to optimizing (I've not found an api document)

The BLE library on the Pi looks pretty specific, so dont think that will work on the ESP with micropython, but you might be able to do it with native libraries or ESPHome with https://esphome.io/components/sensor/ble_client.html

Scotty-89, 2021-07-10:
> the bluetooth is pretty hacky (it is more of a replay of what the app does, than a sensible approach)
> the process used here follows the below steps (as you can see multiple handles are used)
>
> ``` [...]

thx for your reply ;)
the "basic" communication is already working on my EPS32  now ;) 
but i struggle with the responses and Replys form the Balancer ...
maybe some one can help me here ? 

one thing which is working fine is reading the "MainInfos" 
sending: 0xaa, 0x55, 0x90, 0xeb, 0x97, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x11
response: 55 AA EB 90 3 0 4A 4B 2D 42 32 41 32 34 53 31 35 50 0 0 0 0 36 2E 58 0 0 0 0 0 36 2E 31 2E 33 53 0 0 E8 61 1 0 57 0 0 0 4A 4B 2D 42 32 41 32 34 53 31 35 50 0 0 0 0 31 32 33 34 0 0 0 0 0 0 0 0 0 0 0 0 32 31 30 31 32 31 0 0 32 30 31 32 32 38 32 30 38 38 0 0 0 0 0 0 49 6E 70 75 74 20 55 73 65 72 64 61 74 61 0 0 31 32 33 34 35 36 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

this is mainly clear because is only like Hex to ASCI ;) 

i struggle with the following: 
sending: 0xaa, 0x55, 0x90, 0xeb, 0x96, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10
response: 55 AA EB 90 1 0 58 2 0 0 80 C 0 0 E4 C 0 0 4 10 0 0 A0 F 0 0A 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1C C 0 0 30 75 0 0 1E 0 0 0 3C 0 0 0 60 EA 0 0 2C 1 0 0 3C 0 0 0 3C 0 0 0 D0 7 0 0 BC 2 0 0 58 2 0 0 BC 2 0 0 58 2 0 0 38 FF FF FF 9C FF FF FF 84 3 0 0 BC 2 0 0 E 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 40 9C 0 0 DC 5 0 0 E4 C 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 9 AA 55 90 EB C8 1 1 0 0 0 0 0 0 0 0 0 0 0 0 44 

normal i would expect here that "58 2"  is the first Cell or 258 2 0 02 is the first Cell but i have no Idea how to convert this... Anny idea ? 

i use HW 6.X and SW version 6.1.3.S  (JK-B2A24S15P)

jblance, 2021-07-10:
I haven't check as my office is pieces while I install a new external door and rebuild the wall, but that looks like a standard jk02 response to me. The breakdown is in the jk02 protocol file

closed 2021-08-04 by jblance

syssi, 2021-08-09:
> Not using their TTL-RS485 dongle. Seems to be a faulty one.
> Instead, connected a TTL-USB dongle directly to RX, TX, GND of BMS TTL-RS485 socket.

@CRaccani Did I understand your setup correctly: You dropped the chinese TTL-to-RS485 converter and you didn't use a RS485-to-USB converter. Instead you picked a common TTL-to-USB (pl2303, ftdi, ...) device to interact with the BMS at the TTL-RS485 socket?

bloddy, 2022-07-02:
@syssi  Yes he removed the TTL to RS485(their little box is just plain TTL UART to RS485 with a DC/DC converter, because on the VCC you get the full voltage of the bat bank).

Now I have a question .. anyone have succeed to write any info to the bms via TTL ?

syssi, 2022-07-02:
No. The UART-TTL is read-only.

logmod, 2022-09-18:
> @syssi Yes he removed the TTL to RS485(their little box is just plain TTL UART to RS485 with a DC/DC converter, because on the VCC you get the full voltage of the bat bank).

I know this issue is closed already, but is VCC=Vbatt (still) true on all JKBMS models? What is the TTL voltage level then? The JKBMS PCB must have step-down converter somewhere, otherwise how would they power the RS232 driver?

syssi, 2022-09-19:
To drive the microcontroller (3.3V) and the BLE module (3.3V) at the mainboard of the BMS there is a step-down converter on the mainboard. Please keep in mind this onboard step-down isn't able to provide much current. The UART-TTL is 3.3V. There is no RS232 driver (+12V/-12V levels)!

The Jikong RS485 module requires the VBAT voltage to be able to shift the TTL level signal to RS485 levels.

Could you explaint what do you mean exactly by VCC=VBAT? It's possible there is another version of the RS485 converter module now because the VBAT voltage of the new 4S BMS version is too low to drive the old converter circuit (probably).

logmod, 2022-09-19:
I want to build my own serial converter with galvanic isolation for a 48V JKBMS (still in delivery). That's why I was confused about the statement from @bloddy that VCC is getting the full voltage of the battery bank (48V). If the UART-TTL is 3.3V then I also need VCC to be 3.3V to drive the isolated UART on my board. I would assume that VCC is always 12V regardless of the BMS model and battery bank voltage, but I was hoping to get a confirmation on this. Then I could include a 12V to 3.3V step down converter on my own board.

syssi, 2022-09-19:
The statement of @bloddy is correct. The UART-TTL connector has 4 pin (`RX, TX, GND, VBAT`). The battery voltage (`VBAT`) depends on the number of cells attached to your BMS. In my case it's 16 * 3.0-3.65V = ~48-58.4V.

Please check the connection between the microcontroller of the BMS and the UART-TTL port. IMO there is a digital isolator called `122U31` in place.
 
[image]

logmod, 2022-09-19:
Thank you for the info. I just read the TI design guide for the 122U31, but I am still not clear where V_iso for the isolated TTL signal is coming from and why they put V_bat on the connector, but I guess they did not want users to tap into their internal step-down converter. 

I will just use an external wide input (40-60V) step-down converter to create the 3.3V for my isolated board. It never hurts to have another level of isolation and that is also what the factory converter module does to provide its own power.

syssi, 2022-09-19:
I would be super happy if you like to share your solution. This is my longterm goal: I want to remove the bluetooth module from my BMS or pull down the RST pin of the module permanently to get rid of bluetooth (it's insecure. the password isn't required to change important settings). The protocol between the microcontroller and the BLE module is well known (`JK02`). Instead of using the UART-TTL protocol (different payload -> different measurements) I would like to request the JK02 payload using the RX/TX pins of the BLE module by wire. This connection isn't isolated yet.

logmod, 2022-09-19:
I will test it when I finally receive my BMS. If it works I will share the results.

bloddy, 2022-10-18:
@logmod Yes all their bms goes with vbat on the connector, because they have their own external uart<>rs485/can box .. and they need to power it :) 

The implementation is not bad .. but they could have added a damn fast fuse on the pcb for the connector ... 

In my case I'm going on isolation from rs485 to RJ45, I'm using Industrial serial server, RS485 to RJ45 Ethernet from waveshare. 

For the comm you only need rx/tx/gnd .. and gnd if you don't want to really couple it you can add an resistor over 100K and it should be in the safe spot.

Because for my need's I'm doing dynamic, voltage control, amperage control, base on some thresholds, all implementation goes between RS485 protocol.. mqtt(raspberry)<>HASS, and I like to have an history of the bank's ... and remote access :)
[image]
[image]

logmod, 2022-10-18:
Actually a nice idea to convert the serial output to IP, but I wanted to get rid of the RS485 converter and keep the connection to my board as simpel and safe as possible. Not sure to understand the remark about the 100k resistor, usually a 1k resistor is used to protect TTL inputs. Could you elaborate on that?

logmod, 2022-10-20:
@syssi I have found a cheap ready-made isolator board that would fit your requirements to get the TTL signal directly from the Bluetooth interface. Its from Hailege based on an ADUM1201 Dual Channel Digital Magnetic Isolator (available e.g. on Amazon). You can power it from inside the BMS board with the available 3.3V suppy. On the outside you can directly connect it to your GPIO and the outside 3.3V. Adding a DIP switch in the connector cable to pull down RST you could have a dual-solution (BLE/serial) in parallel. 

[image]

MHunzalaAwan, 2022-11-24:
> Got response from seller: I sent them this screenshot of Version information from Android App: [image]
>
> They sent back yet another protocol document in CHINESE.
>
> [BMS Protocol V2.5 [...]

Hi, I just wanted to know if this android app you used, runs on Bluetooth or it can also work with USB to otg connected to the mobile? If it can work using the USB to otg with the mobile, then can you please share the app? Thanks.

abc2006, 2022-12-31:
No need to reopen, just want to mention that this issue was same for me. 
Communication didn't work, also got 0x00 as Answer. 
Got rid of this Dongle and connected directly via RS485 TTL, and everything works fine.
Matching Plug is the Molex PicoBlade 1.25,  but some JST 1.25 are also inbound, i will share the results as soon as they arrived.
And yeah, pay attention to the VCC-Pin, as there is full battery Voltage available.
Cheers, 
Stephan

abc2006, 2023-01-20:
1.25mm 1.25 Micro JST Stecker 15cm Kabel
https://www.ebay.de/itm/183283879151?var=690577026311
^^These are the right ones, fit a little better than the Molex.
Could sb drop a Link to a TTL-USB-Converter over here? I'm struggeling with a bunch of not-working ones.. 
Maybe someone can also drop the specifications ...

logmod, 2023-01-20:
Try searching for ICQUANZX ADUM4160 on Amazon, specs for the BM board inteface are in this thread.

kadzsol, 2023-02-12:
As I had no JST 1.25 connector at hand and did not want to cut the one from the RS485 converter, I have figured out how to use the converter with an ESP8266 to query the BMS.

As I use NODEMCU (LUA) for prototyping, first I had an issue with the RX serial buffer overflowing when receiving ALL the data at once at 115200 baud. So I had to figure out how to query only a certain value to receive less data and avoid the overflow:

voltage
4E57001300000000030300830000000068000001A9
current
4E57001300000000030300840000000068000001AA
capacity
4E57001300000000030300850000000068000001AB

The only stuff you have to change in the commands above is the code (83, 84, 85) and the CRC (sum of all values) to create your own query. See 5.1 BMS Read/Write data identification code in the table above to find your query code.

After this I still did not receive the data on the ESP but I was able to receive it with a PC using an RS485-to-USB converter. It turned out that the TTL-to-RS485 converter interfacing with the ESP was not able to switch back to receiving mode fast enough. So it missed the response of the BMS. I added a second TTL-to-RS485 module for separate sending and receiving and now it works just fine with the official converter.

The only drawaback of this approach is that I am also receiving my own command in front of the BMS response, but it is easy to filter it out in the software. 

Hope this helps someone.

SeByDocKy, 2023-05-10:
Hello,

Seems with new hardware V11x, there are now both the UART-TTL (on GPS port) and RS485 on CAN bus. There are new options available on the appz

[image]

Anyone tried to communicate  via RS485 (modbus RTU), especially now there is an option to select the (modbus) address.... I hope it means we can have several JKBMS v11x on the same bus...

AD-Codex, 2023-05-10:
> As I had no JST 1.25 connector at hand and did not want to cut the one from the RS485 converter, I have figured out how to use the converter with an ESP8266 to query the BMS.
>
> As I use NODEMCU (LUA) [...]

> As I had no JST 1.25 connector at hand and did not want to cut the one from the RS485 converter, I have figured out how to use the converter with an ESP8266 to query the BMS.
>
> As I use NODEMCU (LUA) [...]

what is the BMS response for this request ?
Is the response like 4E570013000000000303008317A4000068000001A9

kadzsol, 2023-05-10:
> what is the BMS response for this request ?
> Is the response like 4E570013000000000303008317A4000068000001A9

4E57001300000000030300830000000068000001A94E570015000000000300018300120000000068000001BB

AD-Codex, 2023-05-11:
4E57001300000000030300830000000068000001A9 is the command to get voltage value from the BMS. I need the an example that what kind of reply (receiving data ) from the BMS. 
Is it like the same request command but after /x83 first 2 hex numbers represent the value of voltage.

request           - 4E57001300000000030300830000000068000001A9
received data - 4E570013000000000303008317A4000068000001A9

Woodstockskipper, 2023-05-11:
I made a translation file in excel from my project:
[BMS commands and receives.xlsx](https://github.com/jblance/mpp-solar/files/11455354/BMS.commands.and.receives.xlsx)
I hope this helps

Maikel-K, 2024-01-21:
> Hello,
>
>
>
> Seems with new hardware V11x, there are now both the UART-TTL (on GPS port) and RS485 on CAN bus. There are new options available on the appz
>
>
>
> [image]
>
>
>
>
>
> Anyone tried to communicate [...]

Anybody tried this? I need comms with 9 BMSes!

lucize, 2024-02-09:
> Anybody tried this? I need comms with 9 BMSes!

it works on native Modbus but can't seems to find the registers on the internet and jk never replyed to my request, there are some here:

```
addr,meaning
0,battery voltage
1,number of cells
2, percent
3,capacity
4,charging ampere
5,?discharging?
6,temp1
7,temp2
8,tube temp
9,cell1
10,cell2
11,cell3
12,cell4
13,cell5
14,cell6
15,cell7
16,cell8
17,cell9
18,cell10
19,cell11
20,cell12
21,cell13
22,cell14
23,cell15
24,cell16
25,cell17
26,cell18
27,cell19
28,cell20
29,cell21
30,cell22
31,cell23
32,cell24
```
also found some on 1000+10 but don't know the meaning of them and still trying to scan all registers, I guess that an automation would we faster

lukolszewski, 2024-08-17:
Does anyone know if there is any software that can parse this protocol yet? They issue was closed twice, but I'm not seeing any solution. I can send byte series mentioned and receive replies, but I was wondering if there already is a parser that works?

lucize, 2024-08-18:
now that esphome has modbus server support should be very easy to implement, use a modbus sniffer and see what address the inverter is asking and then use the address to query the jk-bms inverter bms to see what actual data is sent on those addresses, maybe I'll do a map as now I have one of those bms

kadzsol, 2024-08-18:
> Does anyone know if there is any software that can parse this protocol yet? They issue was closed twice, but I'm not seeing any solution. I can send byte series mentioned and receive replies, but I was wondering if there already is a parser that works?

https://github.com/stamina84/jk-bms/blob/main/serial/read_bms.py

syssi, 2024-08-18:
Please keep in mind we are probably talking about different device models / modbus register addresses / port of the BMS here. On pretty new device models the protocol of the different UARTs is configurable now.

These are the modbus registers of the JK-BMS PB (inverter) series:

https://github.com/syssi/esphome-jk-bms/blob/main/docs/pb2a16s20p/BMS%20RS485%20Modbus%20V1.1.pdf
https://github.com/syssi/esphome-jk-bms/blob/main/esp32-jk-pb-modbus-example.yaml

This protocol is called `GPS protocol`:

```
Request >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
Response <<< 4E:57:01:18:00:00:00:00:06:00:01:79:27:01:0F:FC:02:0F:CF:03:0F...
```

[3 more replies]